dresden-elektronik/deconz-rest-plugin #5835: Device name

superMoYoX, 2022-02-27:
## Device

- Product name: Vikefon temperature & humidity sensor
- Manufacturer: _TZ3000_fllyghyj
- Model identifier: TS0201
- Device type : 
  - Sensor
Seems to be a bit different that this one : https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5613

## Screenshots
[image]

### Basic : 
[image]

Screenshot of Node Info Panel : 
[image]

### Power Configuration
[image]

### Identify
[image]

### Temperature Measurement
[image]

### Relative Humidity measurement
[image]

### OTAU
[image]

Smanar, 2022-03-01:
Hello, this device realy look standard I think it can be added using DDF https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet

Can start from scratch or just take a working sensor with temperature and humidity and change the usefull part 
Here there is one that make only temperature but I can help you to finish it https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5777/files

Need to update.
```
 "manufacturername": "OWON",
  "modelid": "THS317-ET",
  "product": "THS317-ET",
  "path": "/devices/ths317-et.json",
```

superMoYoX, 2022-03-07:
Okay thanks !

I promise I'll take some time to understand all of those things.
I'll tell you !

leo-beltrao, 2022-03-29:
I have the same issue with this device. I will test the proposed solution. Tks

leo-beltrao, 2022-04-02:
Good news! 
It took me a while to figure out why the Edit DDF was not opening (topic for another question). Anyway,  after restarting deconz,  it worked, and I could create the DDF file with the temperature and humidity using the standard settings. Voilà! It recognized the device. It automatically labels the device as Humidity 52. I could confirm the operation by opening the sensors in Phoscon. It is giving me the two measurements.  See screenshot below.
Thanks for providing the link to DDF Cheat Sheet.

leo-beltrao, 2022-04-02:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

leo-beltrao, 2022-04-02:
I'm not sure if I have a way to apply an offset on top of measurement for humidity. It looks like the values are a bit higher than other sensors. Tks

Smanar, 2022-04-02:
You have.
If you are sing defaut value, it s managed by the legacy code.
Else you can use as "expression" something like `Item.val = Attr.val + R.item('config/offset').val`

And if you can make a PR to submit your DDF for others ?

leo-beltrao, 2022-04-03:
@Smanar, I'll need your help. Could you send me instructions on how to send PR? In addition, could you send me further details on how I could create that expression for the offset? I don't know if this needs to be done in deConz or Home Assistance? Tks

Smanar, 2022-04-03:
Depend of you DDF
If you are just using

```
       {
          "name": "state/temperature"
        },
```

I will be done by legacy code, else if you have forced some value, you can add this expression using the DDF editor (just select state/temperature), you will all field to fill.

To send a PR just go on the folder you want to put your device. I know it's vikefon, but for me its tuya stuff so for exemple here 
https://github.com/dresden-elektronik/deconz-rest-plugin/tree/master/devices/tuya
Choose "add a file" choose a name according to other __TZ3000_fllyghyj_temp_hum_sensor.json for exemple.

Github will say you have not the right on this github, and will create a fork under you name, and at the end you can submit the PR
Don't worry, you can't break something, and all is deletable/editable, so you can retry many time.

If later you want to edit the file (and if no one have already valided it), just go on your fork and edit the file again, all will be synchronised.

closed 2022-05-02 by github-actions[bot]