dresden-elektronik/deconz-rest-plugin #5836: Tuya 4 button ZigBee 3.0 Wireless Portable Remote Control - Model number: SS6400ZB-V2

rork-68, 2022-02-27:
## Device

- Product name: Tuya zigbee 4 Gang Remote
- Manufacturer: _TZ3000_ee8nrt2l
- Model identifier: TS0044
- Device type : Remote

## Screenshots
[image]
[image]

### Basic
[image]

### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off
[image]

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above
[image]
[image]

hoodlumj3, 2022-03-01:
Can i say #metoo i have exact same remote and exact same issue in DECONZ 2.14.1 with CONBEE II FW: 0x26720700 - these remotes do not appear in Phoscon Gateway

rork-68, 2022-03-01:
> Can i say #metoo i have exact same remote and exact same issue in DECONZ 2.14.1 with CONBEE II FW: 0x26720700 - these remotes do not appear in Phoscon Gateway

Hope Deconz will start to support our remote soon!!!

hoodlumj3, 2022-03-17:
Just to stop this from going stale.
Prob not much help regarding this controller but more of an FYI... I got one of those sonoff zigbee 3.0 dongles and plugged it in and configured zigbee2mqtt on channel 15 and then paired my/this 4 button remote to the sonoff zigbee network and its working great. 
I haven't researched running multiple zigbee networks from the same HA server but no detrimental effects experienced so far.

rork-68, 2022-03-19:
> Just to stop this from going stale.
> Prob not much help regarding this controller but more of an FYI... I got one of those sonoff zigbee 3.0 dongles and plugged it in and configured zigbee2mqtt on [...]

Great job! Maybe the time has arrived to switch to an independent Zigbee bridge. I'll give it a try!! Thanks a lot for you time!

volkerhielscher, 2022-06-09:
can we reopen this please?

Smanar, 2022-06-09:
Hello, can you tru this DDF

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_ee8nrt2l",
  "modelid": "TS0044",
  "product": "TS0044",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_SWITCH",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0006"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/buttonevent"
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0006"
    },
    {
      "bind": "unicast",
      "src.ep": 2,
      "dst.ep": 1,
      "cl": "0x0006"
    },
    {
      "bind": "unicast",
      "src.ep": 3,
      "dst.ep": 1,
      "cl": "0x0006"
    },
    {
      "bind": "unicast",
      "src.ep": 4,
      "dst.ep": 1,
      "cl": "0x0006"
    }
  ]
}
```

And you will need too edit the button_maps.json file, (in the same folder than all DDF "devices" to add your model here
```

"Tuya3gangMap": {
            "vendor": "Tuya",
            "doc": "3-gang remote",
            "modelids": ["_TZ3000_t8hzpgnd", "_TZ3000_wkai4ga5", "_TZ3000_bi6lpsew", "_TZ3400_keyjhapk", "_TYZB02_key8kk7r", "_TZ3400_keyjqthh", "_TZ3400_key8kk7r", "_TZ3000_vp6clf9d", "_TYZB02_keyjqthh", "_TZ3000_peszejy7", "_TZ3000_qzjcsmar", "_TZ3000_owgcnkrh", "_TZ3000_adkvzooy", "_TZ3000_arfwfgoa", "_TZ3000_a7ouggvs", "_TZ3000_rrjr1q0u", "_TZ3000_abci1hiu", "_TZ3000_dfgbtub0"],
            "map": [
```

Don't take care at name, this part is used for all gangs number

Pedder007, 2022-06-09:
Hi @Smanar, great to find this here, as I just received one of these 4-button devices.

I loaded your DDF file and also stored the button_maps.json file in the same directory. Aren't there missing some closing brackets in the end? However I added a "]," and a "}" in the end and did a hot reload in DDF.

But I got the device not working in my deCONZ/ioBroker setup. The datapoints I see in ioBroker are:
[image]
The battery and the reachable status seems working, but nothing else, thus no reaction when pressing a button on the remote.

I close for today and maybe get a further hint from you.

Pedder007, 2022-06-10:
I quickly learned the remote new to deCONZ this morning, after deleting all former entries in deCONZ/ioBroker and also restarting deCONZ and the respective ioBroker deCONZ adapter.
I again used your above DDF file and the buttons map file. Last one, as described abvove incl. the two additional brackets.

Now I got the following in deCONZ:
[image]

And the following as data points in ioBroker:
[image]

But as result, again no status of the data points is changing when pressing a button on the remote.

Smanar, 2022-06-10:
There is probably a problem in the button_maps.json, you just need to add 

`"_TZ3000_ee8nrt2l",`

To have something like
```

            "modelids": ["_TZ3000_ee8nrt2l", "_TZ3000_t8hzpgnd", "_TZ3000_wkai4ga5", "_TZ3000_bi6lpsew", "_TZ3400_keyjhapk", "_TYZB02_key8kk7r", "_TZ3400_keyjqthh", "_TZ3400_key8kk7r", "_TZ3000_vp6clf9d", "_TYZB02_keyjqthh", "_TZ3000_peszejy7", "_TZ3000_qzjcsmar", "_TZ3000_owgcnkrh", "_TZ3000_adkvzooy", "_TZ3000_arfwfgoa", "_TZ3000_a7ouggvs", "_TZ3000_rrjr1q0u", "_TZ3000_abci1hiu", "_TZ3000_dfgbtub0"],
            "map": [
```

Pedder007, 2022-06-10:
Ah, now I got it! Thanks for the link, I wasn't aware, that this file contains info for that much other devices in total.

Thus I added now "_TZ3000_ee8nrt2l" in the list in the brackets.
In addition I added the lines 1211 - 1225.

Is that sufficient or do I have to add also parts form the beginning of the file? e. g. 1-136

I also found other lines, about devices which I'm already using lengthy, but without that file and they nevertheless work!?

Smanar, 2022-06-11:
```
Thus I added now "_TZ3000_ee8nrt2l" in the list in the brackets.
In addition I added the lines 1211 - 1225.
In addition I added lines 1-136.
```

IDK what you have done but, you JUST need to add "_TZ3000_ee8nrt2l"

>I also found other lines, about devices which I'm already using lengthy, but without that file and they nevertheless work!?

You let the file, you just add one device (you will have a new manufacture name for the same device every month with tuya, not possible to guess them). So no difference for device that already use this file.

>To test, I tried then to configure a scenario in Phoscon, but here the remote isn't offered to use, allthough it is listed in the switches overview.

I never use phoscon, to check if it work need to take a look in the API, in phoscon / help / API Information.
The value updated when you use the device is "buttonevent"

Pedder007, 2022-06-11:
Hi again, the reason for adding the lines is, that I started with an completely empty file.
I never used it before, resp. there wasn‘t such a file in the directory, where also the DDFs are stored.
Thus I thought to built such a file (currently) only with the content I need. I guess 
I will test the whole file then, as it is currently available here.

Regarding testing I will check via the API, but first have to read/remind how to use it  :-)

Smanar, 2022-06-11:
Can make with this contain for the button_maps.json file https://pastebin.com/C0ivZeiE

>but first have to read/remind how to use it :-)

Don't need to use the API, can just take a look in phoscon / help / api information, you can see direclty API or event using phoscon.

Pedder007, 2022-06-11:
Ah great! Will test later, as currently I’m not at home.

Pedder007, 2022-06-11:
I'm sorry, but it seems further on not working. I used the file you provided above and paired the remote new, but it looks as before. 
Checking the API outcome in Phoscon looks as following:

//192.168.188.200/api/D5563177CA/sensors/102
REST API
````
{
    "config": {
        "on": true,
        "reachable": true
    },
    "etag": "a7afb67673020e2137fa3ef144aa01ea",
    "lastannounced": null,
    "lastseen": "2022-06-11T22:00Z",
    "manufacturername": "_TZ3000_ee8nrt2l",
    "mode": 1,
    "modelid": "TS0044",
    "name": "GarageFB",
    "state": {
        "buttonevent": null,
        "lastupdated": "none"
    },
    "type": "ZHASwitch",
    "uniqueid": "a4:c1:38:9c:38:79:86:7f-01-0006"
}
````
Events (when pressing a button, but has to have a break of ~1 minute btw. pressing)
````
    "00:04:44:632": {
        "attr": {
            "id": "102",
            "lastannounced": null,
            "lastseen": "2022-06-11T22:04Z",
            "manufacturername": "_TZ3000_ee8nrt2l",
            "modelid": "TS0044",
            "name": "GarageFB",
            "swversion": null,
            "type": "ZHASwitch",
            "uniqueid": "a4:c1:38:9c:38:79:86:7f-01-0006"
        },
        "e": "changed",
        "id": "102",
        "r": "sensors",
        "t": "event",
        "uniqueid": "a4:c1:38:9c:38:79:86:7f-01-0006"
    }
}
````

Pedder007, 2022-06-13:
Just for completness, as I saw another curious behavior with another 4-button-remote (Moes).
I paired the remote now for testing with a Tuya hub and afterwards again paired it to my ConBee/deCONZ environment, but also this didn't bring along any change.

Pedder007, 2022-06-16:
Yes I have access and I checked the logging.
When enabling info continously messages are coming in.
When enabling info_l2 tonnes of messages are coming in.
What can I do with them?
One further observation: I think normaly, when pressing a button, the little light in the deCONZ gui shall blink blue, right? But in case of the 4-button-remote it blinks light green.

Smanar, 2022-06-17:
Yeah I know logs are realy talkative, but you need to see something just when you use the button.
Something like 

>[INFO] - No button map for: ..............
>[INFO] - No button handler for: ............

Can you check if you have not a light entry fo the device (it's visible on phoscon / light)

Pedder007, 2022-06-17:
Hi, in Phoscon I only find this in switches (=Schalter in German)
[image]
There's absolutely nothing in lights or switches which I do not know, resp. I'm not able to assign to another device.
In deCONZ it still looks as above, I pasted in earlier. Only a new ID as I paired it newly at some point of time.

I checked again the debug, but there are no [INFO]s as you wrote.
In 'info' nothing comes in, but in 'info_l2', when pressing button 1+2, the following comes in (reproducable):
````
18:30:08:267 Websocket 127.0.0.1:53450 send message: {"attr":{"id":"98","lastannounced":null,"lastseen":"2022-06-17T17:30Z","manufacturername":"_TZ3000_ee8nrt2l","modelid":"TS0044","name":"Switch 98","swversion":null,"type":"ZHASwitch","uniqueid":"a4:c1:38:9c:38:79:86:7f-01-0006"},"e":"changed","id":"98","r":"sensors","t":"event","uniqueid":"a4:c1:38:9c:38:79:86:7f-01-0006"} (ret = -1094385736)
18:30:08:269 Websocket 192.168.188.20:16830 send message: {"attr":{"id":"98","lastannounced":null,"lastseen":"2022-06-17T17:30Z","manufacturername":"_TZ3000_ee8nrt2l","modelid":"TS0044","name":"Switch 98","swversion":null,"type":"ZHASwitch","uniqueid":"a4:c1:38:9c:38:79:86:7f-01-0006"},"e":"changed","id":"98","r":"sensors","t":"event","uniqueid":"a4:c1:38:9c:38:79:86:7f-01-0006"} (ret = -1094385736)
18:30:08:272 Websocket 127.0.0.1:53450 send message: {"attr":{"id":"95","lastannounced":null,"lastseen":"2022-06-17T17:30Z","manufacturername":"_TZ3000_ee8nrt2l","modelid":"TS0044","name":"Battery 95","swversion":null,"type":"ZHABattery","uniqueid":"a4:c1:38:9c:38:79:86:7f-ff-0001"},"e":"changed","id":"95","r":"sensors","t":"event","uniqueid":"a4:c1:38:9c:38:79:86:7f-ff-0001"} (ret = -1094385736)
18:30:08:274 Websocket 192.168.188.20:16830 send message: {"attr":{"id":"95","lastannounced":null,"lastseen":"2022-06-17T17:30Z","manufacturername":"_TZ3000_ee8nrt2l","modelid":"TS0044","name":"Battery 95","swversion":null,"type":"ZHABattery","uniqueid":"a4:c1:38:9c:38:79:86:7f-ff-0001"},"e":"changed","id":"95","r":"sensors","t":"event","uniqueid":"a4:c1:38:9c:38:79:86:7f-ff-0001"} (ret = -1094385736)
````

I additionally checked 'error, error_l2 and DDF', but there the remote doesn't show anything.

The battery ID 95 I see also in ioBroker:
[image]

Smanar, 2022-06-17:
Ok, so you don't have a light entry but a ZHABattery one.
There is nothing in DDF or in legacy code to explain that, and ofc no state/buttonevent field ...

BTW I m seing you have a device with the bad endpoint FF

>"uniqueid":"a4:c1:38:9c:38:79:86:7f-ff-0001"

Wich one deconz version have you ?

Pedder007, 2022-06-17:
It is 2.15.03 / 21.4.2022

Smanar, 2022-06-18:
You still have a device with this uniqueid ? 

>a4:c1:38:9c:38:79:86:7f-ff-0001

You need to have only
>a4:c1:38:9c:38:79:86:7f-01-0006

Pedder007, 2022-06-18:
This is the ZHAbattery device (also in the debug above), which always apears, when pairing the remote to my setup.
````
17:22:49:810 Websocket 192.168.188.20:1168 send message: {"attr":{"id":"95","lastannounced":"2022-06-17T20:56:56Z","lastseen":"2022-06-18T16:22Z","manufacturername":"_TZ3000_ee8nrt2l","modelid":"TS0044","name":"Battery 95","swversion":null,"type":"ZHABattery","uniqueid":"a4:c1:38:9c:38:79:86:7f-ff-0001"},"e":"changed","id":"95","r":"sensors","t":"event","uniqueid":"a4:c1:38:9c:38:79:86:7f-ff-0001"} (ret = -1093496904)
````
But I cannot see it as device. Neither in Phoscon nor in the deCONZ gui (incl. node list) und thus cannot delete it there.
It only appears as data point in ioBroker, and in the debug.

So what do you mean by 'if you have updated your deconz version according the folder you have put the DDF, this one can be removed.'?
What has the update to do with the DDF folder? - sorry I'm not that deep in.

Smanar, 2022-06-18:
Lol, ok so what we can say at next user that will have the same issue ? Just pray and retry ^^  ?

About the DDF folder.

> All DDF provided directly with deCONZ typically reside in /usr/share/deCONZ/devices/ on a Linux system and are loaded first. However, files residing in the home directory of the user running deCONZ (e.g. /home/<DECONZUSER>/.local/share/dresden-elektronik/deCONZ/devices) will override the pre-packaged files to allow users to amend and keep their own files if desired.

So there is a folder where all DDF are reseted whaen re-installing deconz, so the DDF used can be deleted.

BTW, so the DDF is working (at least with some mystery) so can make the PR, battery is working too ?

Pedder007, 2022-06-18:
Yes pray loud 😅
Ok good to know about DDF I will check tomorrow.

My wife is happy now, as she likes this remote much more than the Osram Minis, which look like large rubbers 😀

Thanks a lot for your support and enjoy your Samedi evening with a good glass of vine - recognised in another chat that you seem to be French 🇫🇷 🍷

Salut from Germany 🙋🏻‍♂️

Smanar, 2022-06-18:
Lol, yep I m french and we will test german beer in some month for our holidays.

Tell me if all is working for I make the PR, or you can do it youself if you want.

Pedder007, 2022-06-18:
Thats funny, as we are going to spend our holidays this year in France, to have some good wine, near Narbonne.
But your French beer is also very good, have been in Lion ~3years ago and tasted Ninkasi there - very nice 😉

But I fear thats the wrong platform here to chat about beer and wine ... 😅

Smanar, 2022-06-18:
Lol, can speak about food after drink.

A PR is a Pull request, if the DDF is working need to put it here https://github.com/dresden-elektronik/deconz-rest-plugin/pulls for it will be included in next deconz version.

Pedder007, 2022-06-18:
My DDF directory currently contains 3 files:
[image]
It looks like before - nothing changed.

I now additonally did a hot reload for the remote and checked if there is a battery status, but there isn't one.
Also has a look at the Power Config Cluster in the deCONZ gui and pressed 'read', but the battery fields stay empty:

[image]

Smanar, 2022-06-19:
Oups, my bad, haven't set it in the DDF, and the procedure you have done is perfect (to force an attribute read).
Have set the defaut value for battery, but It don't work on lot of devices, if it don't work replace

```
        {
          "name": "config/battery",
          "awake": true
        },
```

by

```
        {
          "name": "config/battery",
          "parse": {"at": "0x0021", "cl": "0x0001", "ep": 1, "eval": "Item.val = Attr.val"},
          "read": {"fn": "zcl", "ep": 1, "cl": "0x0001", "at": "0x0021"},
          "refresh.interval": 4000,
          "awake": true
        },
```

The new DDF

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_ee8nrt2l",
  "modelid": "TS0044",
  "product": "TS0044",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_SWITCH",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0006"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/battery",
          "awake": true
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/buttonevent"
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 60,
          "max": 3600,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0006"
    },
    {
      "bind": "unicast",
      "src.ep": 2,
      "dst.ep": 1,
      "cl": "0x0006"
    },
    {
      "bind": "unicast",
      "src.ep": 3,
      "dst.ep": 1,
      "cl": "0x0006"
    },
    {
      "bind": "unicast",
      "src.ep": 4,
      "dst.ep": 1,
      "cl": "0x0006"
    }
  ]
}
```

Pedder007, 2022-06-19:
I'm sorry. Tested as you outlined, incl. using a really new file with the content(s) from above, but no battery is shown.
Neither in deCONZ gui nor as data point in ioBroker.
When pressing the READ button in deCONZ, the status light in the device flashes in red, which I guess shows that someting isn't right. The rest of the remote funtionality is still available.

But: I in addtion checked the Preview in the DDF window, which doesn't show the additional battery lines.
What is this, did the 'hot reload' not work? - I pressed it multiple times.
May I drag and drop the ZHABatery item manually into the device section?

Smanar, 2022-06-19:
>When pressing the READ button in deCONZ, the status light in the device flashes in red

It's normal, this device is a sleeper, you can only send request when it want, so you generally need to wake it in same time you send the request.
Or just wait for it decide to send the value.

>But: I in addtion checked the Preview in the DDF window, which doesn't show the additional battery lines.

This is not normal, have you tried to load the DDF direclty in the editor, deconz probably check them only at start.

>May I drag and drop the ZHABattery item manually into the device section?

No, you need to do like you have done, just add a config/battery, not a new sensor.

>The ts0044.ddf file now looks as following (previously also tested the config with only two lines variant, as above). but both do not work.

You have the config/battery, but miss the bind for battery return,

```
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 60,
          "max": 3600,
          "change": "0x00000001"
        }
      ]
    },
```

Pedder007, 2022-06-19:
Hi, had to do some barbeque today 😋

To your points:
Thanks for the additonal info regarding DDF and deCONZ, but:
What I'm wondering about is, that you wrote that deCONZ checks the DDF files only when starting. I thought, that with the 'hot reload' deCONZ somehow 'registers' the addtional info. If that only works when deCONZ is restarted, that would not be really convenient, as of the fact that restarting deCONZ often triggers some unwanted actions in my smarthome (moving shutters, switching switches, etc.). Normally I do not restart deCONZ without an urgent need.

Regarding the battery info:
Yes when testing I didn't add the ZHABattery item, I only added the 'config/battery' item and configured it manually.
What is interesting now, is that somewhen during barbeque, the device seems having decided to start sending a battery status:
1. The little battery icon charging bar (in the gui) switched from 0 (red) to somewhat full (green)
2. The cluster info shows 96% now
3. Also I got a data point in ioBroker which shows the 96%

So seems again whatever magic, I really have no clue, as I really didn't anything, besides barbeque 😅
I also checked the DDF file, but there the section you pasted above is not in.

Smanar, 2022-06-20:
> What I'm wondering about is, that you wrote that deCONZ checks the DDF files only when starting. I thought, that with the 'hot reload' deCONZ somehow 'registers' the addtional info. If that only [...]

Hot reaload make a "hot reload" of the DDF actually used/modified.
If you replace/edit a DDF file, not sure deconz will check if the file was modified, I can be wrong but for me it scan the folder at start, load them in memory and only update them if you change it in the editor (Can be wrong, not tested)

Ok, next issue I will purpose a barbeque.

But for me nothing magic, the device is lazy, it can take time before it decide to send the value.

>I also checked the DDF file, but there the section you pasted above is not in.

Strange, perhaps done on previous DDF (binding are not removed, you need to reset the device to remove them).
But if it work as it can make a PR with your actual one (not a big issue if battery is missing)

Pedder007, 2022-06-21:
> If you replace/edit a DDF file (using test or file editor), not sure deconz will check if the file was modified, I can be wrong but for me it scan the folder at start, load them in memory and only update them if you change it in the editor (Can be wrong, not tested)

I guess you are right on this, because before editing with the editior, I edited the file directly with a text editor and nothing happened, despite pressing 'hot reload' and these changes have not been shown in den DDF editor. Thus it seems really keeping the content of a once read DDF file in the memory, regardless if it is replaced.
 
> Ok, so note for later "next issue, purpose a barbeque."

Definitely 😂

> Strange, perhaps done on previous DDF (binding are not removed, you need to reset the device to remove them).
Yes, isn't it!? I also thought, that maybe there was something during the first pairing, when I got this single battery device (only visible in ioBroker), but now button device.

Maybe there is a kind of fragment somewhere, but also wondering, regarding the 'keeping in memory' strategy, as I did a restart somewhen...

Besides all this mistics the device is properly working and also the battery status has been update this afternoon.
And the former 'Only-Battery-Device' didn't come back until now.

PS.
I found the very detailled [guidance for DDF](https://dresden-elektronik.github.io/deconz-dev-doc/modules/ddf/) usage, but cannot find the time to read it ... 😥 ... maybe on vacations, we will see ... 😉

Smanar, 2022-06-21:
K, so if it work for you, wana make a PR or wana I make it ? to submit the device officially ?

Pedder007, 2022-06-21:
I guess it would be better if you do. I fear I'm not deep enough in to do this properly.

In addtion:
I have another remote here, which has a really wondering behavior: https://zigbee.blakadder.com/Moes_ZS-SR4-2169.html
After pairing it worked directly well, but drains the batteries within 2-3 days, when it is not connected to an original Tuya hub. 
I tested this and also had a lenghty conversion with the support staff from Moes (@aliexpress) and in the end luckily got back my money. In parallel I also found respective rating on AMZ, point in the same direction.

Would it make sense to try to get this tackled also via a respective DDF?

Smanar, 2022-06-22:
PR done have just added your model to another DDF, but its exactly the same code.

>Would it make sense to try to get this tackled also via a respective DDF?

I have see a similar issue, but some user have the issue, other not, this device have probably something special, (because its always the same code we are using), but haven't found the reason yet, so IDK if it can be solved using a DDF.

Pedder007, 2022-06-23:
Already saw this bad weather at the moment and wondered, as Spain is extreme hot for June.
However, the forcast says better, starting with Sunday, thus it will work. Better 28°C then 38°C 😄 
And Septembers in Germany can also have their surprises. Press thumbs and hear you here again soon 🖖

schumi2004, 2022-11-27:
Looks like i'm facing the same issue but then for a Tuya 4-gang remote _TZ3000_mh9px7cq
How can i have this added aswell?
https://www.ebay.com/itm/125444870534

Smanar, 2022-11-27:
Hello, he ? Wich one issue ?

Pedder007, 2022-11-27:
Ha! @schumi2004, this looks like the one I have also tested earlier, which drains massively the battery (2-3 days empty) without being conected to a Tuya hub. The design really looks nice, but not usable for me in the end.
I meanwhile use these ones, which can be ordered also in France: https://www.domadoo.fr/en/peripheriques/6015-loratap-zigbee-30-4-buttons-remote-control.html

Smanar, 2022-11-28:
Lol, ha yes the "Tiger Mosquitos", an horror this year, impossible to go outside, need a flame thrower.

And yes, perfect time, we was there during the "berlin marathon", but we didn't run ^^.

We have made some discovery on "battery drain" since the last time, you remember the DDF you was using ? Was this one https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/tuya/_TZ3000_ygvf9xzp_4gang_remote.json ?

schumi2004, 2022-11-28:
> Hello, he ? Wich one issue ?
> 
> Your model is a TS0044 ?

Model reports as TS0044 indeed.
So far the battery is still working after a week of lying around not being conencted to a Tuya hub.

I will try to create a DDF in my HassIO environment based on the previously submitted PR, that should work for the time being i think.

schumi2004, 2022-11-28:
Going to test and report back, thanks.

Smanar, 2022-11-28:
Yeah, sorry the file is blocked in PR list since so long time.

schumi2004, 2022-11-28:
For my understanding, shouldn't the device be added to the 4-gang remote device list in button_maps.json instead of the 3-gang it is currently added? Or doesn't this make any difference?

Smanar, 2022-11-28:
>Or doesn't this make any difference?

No differences, there is the same buttonmap for all tuya, from 1 gangs to 4 gangs, values or just not used.

schumi2004, 2022-11-28:
It's visible now in Deconz and HassIO but unfortunatly it doesn't seem to send out any button preses (checked this with debugging in Node Red using deconz_events and debug node)
I do see that the battery level is critical but maybe it was shipped this way.

Maybe something went wrong with editing button_map.json, maybe it isn't picked up correctly. If i restart addon all my changes are gone again and need to start over so need to figure out how to fix this. Atleast it's correctly shown in Deconz/Phoscon and i do see button presses in Deconz VNC GUI but not in HassIO

Smanar, 2022-11-28:
>it doesn't seem to send out any button preses

You probably have error visible in deconz with flag "info_l2" about missing buttonmap ?

> If i restart addon all my changes are gone again and need to start over

Docker rules, and long procedure, there is a way for DDf now, but bot for the buttonmap.
I can add it in blind mode in the previous PR, you will have it on next update, but untested ?

schumi2004, 2022-11-28:
We can try not sure if you need more information from my device?

Smanar, 2022-11-28:
If it is not a special device no, all are clone, he can have special feature but will be hard to test for you.
Will see on next Merge, have added your device here https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6244

Pedder007, 2022-11-28:
> We have made some discovery on "battery drain" since the last time, you remember the DDF you was using ? Was this one https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/tuya/_TZ3000_ygvf9xzp_4gang_remote.json ?

Oha, didn't follow up with this, as we are currently getting a new heating installed, with 4 months of delay .... So construction area since 6 months incl. geothermic drilling about 100m deep into the ground ...

I deleted the device from my setup and thus fear that the old DDF config has also gone, right?
When I find the time (towards Christmas)  I will learn it again to my setup and do some new tests.
Thanks for the hint!

Smanar, 2022-11-28:
>. So construction area since 6 months incl. geothermic drilling about 100m deep into the ground

Where are you from ? How much cost this kind of installation ? What is the minimum temperature in your country ?

Smanar, 2022-11-29:
@Pedder007 ok it s totaly out of subject, but I m realy curious ^^.

It's only possible on "rural area" ? You are using this kind of installation a lot in Germany ? In my mind was mostly used in canada.

I live close to you, in france, almost same temperature, we are using too lot of "heating pump" here, but they are all using air, we never use this kind of installation. It's something new ? You have a link for some information about this technic ? I have looked some video on youtube, but not sure it's the same thing than you, so intrusive works ....

On my side I have a simple "reversible air conditionning".

Pedder007, 2022-11-29:
> @Pedder007 ok it s totaly out of subject, …
 
 Yes, you are right ;-) Do you have a Discord account, to maybe continue there!?

Smanar, 2022-11-29:
>Afaik there is no possibility to chat here on GitHub, right?

Not realy, but I think Mimix "aka the boss" can tolerate 2/3 out of subject post. But you are right, I m on the deconz discord, Smanar too.

Mimiix, 2022-11-29:
> > Afaik there is no possibility to chat here on GitHub, right?
> 
> Not realy, but I think Mimix "aka the boss" can tolerate 2/3 out of subject post on a closed issue. But you are right, I m on the deconz discord, Smanar too.

Please :)

Smanar, 2022-11-29:
lool, you have an eye everywhere