dresden-elektronik/deconz-rest-plugin #5840: Device name

jasonwragg, 2022-03-03:
## Device

- Product name: XUELILI 16A UK Standards Smart Outlet
- Manufacturer: _TZ3000_5f43h46b
- Model identifier: TS011F
- Device type : Please remove all unrelated device types. 
  - Switch

## Screenshots

[image]
[image]

### Basic
[image]

### Identify
[image]

### On/Off
[image]

### Simple Metering
[image]

### Electrical Measurement 
[image]

jasonwragg, 2022-03-03:
> Hello, this device is working, but you miss the power consumption and ect ?

Yes thats correct.

Smanar, 2022-03-04:
OK, so ATM DDF are working fine for plug.
Can start from scratch https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet
Or just editing a working one https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5475

Change the first lines, take care manufacturername and modelid need to be the same than on deconz.
```
  "manufacturername": "Immax",
  "modelid": "Plug-230V-ZB3.0",
  "vendor": "Immax Neo",
  "product": "07048L",
```

ping-localhost, 2022-03-11:
This also applies for the EU version of the device. It reports the same items.

ping-localhost, 2022-04-02:
I would like to get this resolved. Anything I can do to help? :)

Smanar, 2022-04-03:
Yep https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5840#issuecomment-1059346046

closed 2022-05-02 by github-actions[bot]

proggprod, 2022-08-15:
I just got some of these. They get discovered, but I can't turn them on.
They work by pressing the button on the side if I do it before pairing them, but once paired, they wont turn on.
In the phoscon UI if I try to turn them on, they turn on for a moment then go back to off.

Any ideas?

proggprod, 2022-08-15:
Not sure what I did, but after some clicking around in deCONZ, these now turn on and off. But I'm not getting any energy readings. It's just stuck at 0W.
I'm using OpenHAB with my Raspbee II Pi HAT to control and monitor these items.

ping-localhost, 2022-08-15:
@proggprod I, personally, have switched to Zigbee2MQTT (using a SONOFF Zigbee 3.0 USB Dongle Plus). It found the device, I was able to update the firmware and now it works perfectly.

proggprod, 2022-08-15:
> @proggprod I, personally, have switched to Zigbee2MQTT (using a SONOFF Zigbee 3.0 USB Dongle Plus). It found the device, I was able to update the firmware and now it works perfectly.

So you think it needs new firmware?

proggprod, 2022-08-15:
[image]
It seems to get the readings in deCONZ, but it doesn't come out the other end (OpenHAB)

Mimiix, 2022-08-15:
Please guys, read smanars comment: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5840#issuecomment-1086803585

proggprod, 2022-08-15:
> Please guys, read smanars comment: [#5840 (comment)](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5840#issuecomment-1086803585)

Thank you. I read this but maybe I don't understand. Is this suggesting that I just populate those four fields? : "manufacturername": "_TZ3000_5f43h46b",
  "modelid": "TS011F",
  "vendor": "Tuya",
  "product": "XXXXX",

And does this actually mean XXXXX or am I supposed to change this to something?

Smanar, 2022-08-15:
Try this DDF

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_5f43h46b",
  "modelid": "TS011F",
  "product": "XUELILI 16A UK Standards Smart OutletDevice",
  "vendor": "XUELILI ",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_SMART_PLUG",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert",
          "default": "none"
        },
        {
          "name": "state/on",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_POWER_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0b04"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0051",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0B04"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/current",
          "refresh.interval": 300
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/power",
          "refresh.interval": 300
        },
        {
          "name": "state/voltage",
          "refresh.interval": 300
        }
      ]
    },
    {
      "type": "$TYPE_CONSUMPTION_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0702"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0051",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0702"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/consumption",
          "read": {"at": "0x0000", "cl": "0x0702", "ep": 0, "fn": "zcl" },
          "parse": {"at": "0x0000", "cl": "0x0702", "ep": 0, "eval": "Item.val = Attr.val * 10"},
          "refresh.interval": 300
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0006",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x10",
          "min": 1,
          "max": 300
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0702",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x25",
          "min": 1,
          "max": 300,
          "change": "0x0000000A"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0B04",
      "report": [
        {
          "at": "0x0505",
          "dt": "0x21",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        },
        {
          "at": "0x0508",
          "dt": "0x21",
          "min": 1,
          "max": 300,
          "change": "0x00000064"
        },
        {
          "at": "0x050B",
          "dt": "0x29",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        }
      ]
    }
  ]
}
```

proggprod, 2022-08-15:
Thanks for that. I managed to make this DDF file and the nodes load it up properly and it looks normal, but they have now gone back to not turning on, or turning on for 1 - 4 seconds then turning off again. They also don't respond to pressing the button on the  side.
I'm sure it's not a dud one because I have a few of them here and they all do the same.

proggprod, 2022-08-15:
I can confirm that this DDF works. I managed to get a plug to stay on for long enough to verify I was getting energy readings in OpenHAB.
They still usually turn off after a few seconds though. Even without anything plugged in. Sometimes it will stay on for a few mins with a battery charger plugged in drawing 75W. ???

Smanar, 2022-08-15:
There is a post on z2m abut this issue https://github.com/Koenkk/zigbee2mqtt/issues/11648

proggprod, 2022-08-15:
Thanks, I already saw this.
I think it's over voltage protection. My voltage varies between 238 and 241, which is why is was working sometimes and not at all sometimes. The product page says 90 - 250V, but it is very repeatable now that I'm looking for it. It would also explain why it happens whether paired to a hub or not. Then again, I've been wrong before...