dresden-elektronik/deconz-rest-plugin #5843: _TZ3000_4upl1fcj

hastime, 2022-03-03:
## Device

- Product name: Scene Switch
- Manufacturer: _TZ3000_4upl1fcj
- Model identifier: TS0041
- Device type : 
    - Switch

## Screenshots

### Basic
[image]

### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off (Not Greyed out)
[image]
[image]

### On/Off (Greyed out)
[image]

[image]

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

### OTAU
[image]
[image]

### Time
[image]

### Power Configuration
[image]

### Node/ Node Info
[image]
[image]

### Entries in Phoscon
[image]
[image]

### Information from API
[image]
[image]
[image]

hastime, 2022-03-03:
Hi, 

I have tried to include not only the required Clusters but all the information I could gather. I have a light and switch showing in Phoscon but I have two switch items in the API, not sure if this is because I have added the device a couple of times and played around with some DDF settings (which I removed before taking these screenshots and submitting).

I am aware that #5786 exists, but appears to have gone stale. Speaking to @Mimiix, it was suggested I created a new request and we closed the other one, to avoid duplicates. 

Thanks all! 

Let me know if you need any further information or testing, in the meantime I will continue trying to get a better understanding of DDF and if I get this working I will post.

Kind regards,

Mimiix, 2022-03-04:
@Smanar can you take a look?

Smanar, 2022-03-04:
Unfortunately, ATM the DDF core can't include battery switch.
So even the support have started, not possible to use it without editing code.

As you have only 1 endpoint, It can perhaps working with just editing the json file.
To check if it's possible , just enable deconz log, in deconz / help / debug view, with the flag "info" and "info_l2". And press the button, you will see errors about a missing map.

MBj1703, 2023-02-09:
How to reopen this, I have the same switch and it is not working

Mimiix, 2023-02-09:
Sure.

MBj1703, 2023-02-09:
To link to your entry use: [paste.debian.net/1270132](https://paste.debian.net/1270132)
To download your entry use: [paste.debian.net/download/1270132](https://paste.debian.net/download/1270132)

Smanar, 2023-02-09:
Can use this DDF

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": [" _TZ3000_itb0omhv", "_TZ3000_4upl1fcj"],
  "modelid": ["TS0041", "TS0041"],
  "product": "Tuya 1 gangs switch battery",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_SWITCH",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0006"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "refresh.interval": 84000,
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "refresh.interval": 84000,
          "awake": true,
          "read": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2"
          },
          "default": 0
        },        
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/buttonevent"
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 60,
          "max": 3600,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0006"
    }
  ]
}
```

But you need too edit the button_maps.json file (in the "devices" folder too.

```
        "Tuya3gangMap": {
            "vendor": "Tuya",
            "doc": "3-gang remote",
            "modelids": ["_TZ3000_ee8nrt2l", "_TZ3000_ygvf9xzp", "_TZ3000_t8hzpgnd", "_TZ3000_wkai4ga5", "_TZ3000_bi6lpsew", "_TZ3400_keyjhapk", "_TYZB02_key8kk7r", "_TZ3400_keyjqthh", "_TZ3400_key8kk7r", "_TZ3000_vp6clf9d", "_TYZB02_keyjqthh", "_TZ3000_peszejy7", "_TZ3000_qzjcsmar", "_TZ3000_owgcnkrh", "_TZ3000_adkvzooy", "_TZ3000_arfwfgoa", "_TZ3000_a7ouggvs", "_TZ3000_rrjr1q0u", "_TZ3000_abci1hiu", "_TZ3000_dfgbtub0", "_TZ3000_rco1yzb1", "_TZ3000_gbm10jnj", "_TZ3000_5e235jpa", "_TZ3000_sj7jbgks", "_TZ3000_w8jwkczz", "_TZ3000_dziaict4", "_TZ3000_famkxci2", "_TZ3000_itb0omhv", "_TZ3000_tzvbimpq", "_TZ3000_mh9px7cq", "_TZ3000_b7bxojrg", "_TZ3000_ufhtxr59", "_TZ3000_w3c7ouru", "_TZ3000_4upl1fcj"],
            "map": [
```

MBj1703, 2023-02-09:
Thank you.
I'm not an expert, where can i find this files, or better, where should I create it?

Smanar, 2023-02-09:
Depend of your OS, if you have the HA docker will be not possible to edit the button_maps.json file ?

MBj1703, 2023-02-09:
I'm on a Raspy.
Found the folder and edited the button_maps.json

The DDF file you posted, is it a new one? What is the file name?

Smanar, 2023-02-09:
Ok so perfect.

`All DDF provided directly with deCONZ typically reside in /usr/share/deCONZ/devices/ on a Linux system and are loaded first. However, files residing in the home directory of the user running deCONZ (e.g. /home/<DECONZUSER>/.local/share/dresden-elektronik/deCONZ/devices) will override the pre-packaged files to allow users to amend and keep their own files if desired.`

The buttonmap is on the first one.
Can create a new one in the user folder or edit in the first one `tuya\_TZ3000_itb0omhv_1gang_bat.json`

MBj1703, 2023-02-09:
Okay, edited both files, deleted the switch on the web console, restarted the Raspy, but no success.

I'm using FHEM as my smarthome and I got the information about the switch, but still wrong (2 devices, one ZHASwitch and one On/Off light). If i click, nothing happens, so no events truh API?

Smanar, 2023-02-10:
Your procedure is fine. sure there is no typo ?
What is your deconz version ?
I don't remember, you haven't the GUI ?

MBj1703, 2023-02-10:
No, I don't have a GUI on this RaspBerry only the web interface.
Version: 2.20.01 / 19.9.2022

I will now double check for typos.

Smanar, 2023-02-10:
>edited both files, deleted the switch on the web console, restarted the Raspy, but no success.

You have re-included it too ?
Both files was in  /usr/share/deCONZ/devices/ ?

It's possible to run deconz with command line to have debug line, perhaps you will see error on the DDF ? https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Debugging-101#step-2-gather-logs

Can you show the device uniqueid ? it's not something with "-FF-" ? Visible in phoscon/ help / API Information

MBj1703, 2023-02-10:
So double check everything again, restarted, now it is working!!!
Thanks a lot for your help!!

Smanar, 2023-02-10:
Np, ^^, PR created https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6740

closed 2023-02-20 by manup