dresden-elektronik/deconz-rest-plugin #5848: Add support for Sinopé TH1500ZB

MattL0, 2022-03-08:
## Device

- Product name:  Sinopé TH1500ZB Smart double-pole thermostat
- Manufacturer: Sinope Techonologies.
- Model identifier: TH1500ZB.
- Device type : Please remove all unrelated device types. 
  - Thermostat
 

## Screenshots
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

### Basic

### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

MattL0, 2022-03-08:
I see nothing from the api (nodered, deconz phoscon)

Smanar, 2022-03-08:
Hello, I m searching if it's possible to add thermostat using DDF, as your device look standard.
But I don't see one except this one but still in draft status https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5793

BabaIsYou, 2022-03-21:
It looks similar to others Sinopé thermostats partly supported like TH1124ZB (https://github.com/dresden-elektronik/deconz-rest-plugin/blob/5ce086600955a6bd696a7b74ae1ac1bcb2f0c767/general.xml) but perhaps it has to be specifically whitelisted somewhere to be seesn uin deConz app.

Smanar, 2022-03-21:
Ha nice, thx @BabaIsYou and this DDF is working ?

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "Sinope",
  "modelid": "TH1124ZB",
  "product": "TH1124ZB",
  "sleeper": false,
  "status": "Draft",
  "subdevices": [
    {
      "type": "$TYPE_THERMOSTAT",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0201"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0301",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0201",
          "0x0204",
          "0x0B05"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/heatsetpoint",
          "description": "Target temperature of a thermostat.",
          "default": 0
        },
        {
          "name": "config/mode",
          "description": "Mode of the device."
        },
        {
          "name": "config/offset",
          "description": "Relative offset to the main measured value.",
[... 248 lines cut ...]
          "at": "0x050B",
          "dt": "0x29",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        },
        {
          "at": "0x0505",
          "dt": "0x21",
          "min": 1,
          "max": 300,
          "change": "0x0000000A"
        },
        {
          "at": "0x0508",
          "dt": "0x21",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        }
      ]
    }
  ]
}
```

closed 2022-04-19 by github-actions[bot]

MattL0, 2022-04-22:
Hi, I am very sorry for the lack of reply. I got a busy schedule. 

I had tu return the device to amazon, It was making a weird  and loud click noise when controlling the heater. Maybe a relay. 
I am not sur if this is an issue or not, I think I saw somewhere on the web somebody having the same issue, so I didn’t take the chance to pass the return delay windows.

I guess if the TH1123/1124 don’t have this issue, maybe I can retry the th1500 this summer.

BabaIsYou, 2022-04-22:
Thanks for answering anyway. Th1500ZB is a relay based thermostat, Th112x are TRIAC based I think. Then less noisy.

MattL0, 2022-04-23:
Thanks to you that totally make sense ! Ordered a th1124zv to play with