dresden-elektronik/deconz-rest-plugin #5849: GLEDOPTO Zigbee Mini Switch Modul

bruesche, 2022-03-09:
## Device

Product name: GLEDOPTO Zigbee Mini Switch Modul
Manufacturer: GLEDOPTO
Model identifier: GL-SD-002
Device type : Switch

## Screenshots
[image]

[image]

[image]

[image]
[image]
[image]

[image]

### Basic

### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

bruesche, 2022-03-09:
Sorry that last time screenshots missing. Hope everything is now fine.

bruesche, 2022-03-31:
Only a comment to clear the stale lable.

Mimiix, 2022-03-31:
@Smanar can this be done with ddf?

bruesche, 2022-03-31:
I don‘t know

bruesche, 2022-04-06:
I will try it. After eastern is the next Test Slot. Sorry for the delay. I give a comment, if it works or if need help

bruesche, 2022-04-28:
While quarantine of Family and Friends I can‘t try. I try it next Time i can See my friend. Sorry for the delay

bruesche, 2022-05-21:
Must be tested. Hope next week. Sorry für delay.

bruesche, 2022-06-12:
To be continued

bruesche, 2022-07-04:
Worked for me. Very easy, very simple. Thanks a lot

Mimiix, 2022-07-04:
Can you perhaps create a PR for this device so others can benefit it as well?

bruesche, 2022-07-04:
How can I do this? Need advice. And it’s Not my device. It is from a friend. Holidays and so. 

If I have advice what to do, i can Open the issue to give the PR (???) to you. Think this is in some weeks. Summer and so

Smanar, 2022-07-05:
You can share the DDF using this post I will make the PR.
Else can just create a file (and a folder GLEDOPTO ) here https://github.com/dresden-elektronik/deconz-rest-plugin/tree/master/devices using "add file", it will create a fork under your name, create the file and prepare the PR.

bruesche, 2022-07-07:
Hi, ok, Hope i can do this after holidays. I held this Thread open.

Smanar, 2022-07-07:
Depend of OS, if you select the node than "edit DDF", you will see the folder and path on editor title.
Else on linux 

> 
All DDF provided directly with deCONZ typically reside in /usr/share/deCONZ/devices/ on a Linux system and are loaded first. However, files residing in the home directory of the user running deCONZ (e.g. /home/<DECONZUSER>/.local/share/dresden-elektronik/deCONZ/devices) will override the pre-packaged files to allow users to amend and keep their own files if desired.

bruesche, 2022-07-15:
[gl-sd-002.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/9121428/gl-sd-002.txt)
 
Here is the file that worked for me. Have renamed it to txt. You musst only change to json.
Hope everything is correct.

Thanks a lot for your help.

closed 2022-07-15 by bruesche