dresden-elektronik/deconz-rest-plugin #5861: 1CH Zigbee Switch Module-L

optikkore, 2022-03-12:
## Device

- Product name: 1CH Zigbee Switch Module-L
- Manufacturer: _TZ3000_ji4araar
- Model identifier: TS0011
- Device type : 
  - Switch

## Screenshots
[image]
[image]
[image]

### Basic

### Identify

### Groups

### Scenes

### On/Off

### OTAU

### UNKOWN

### UNKOWN

### TIME

Mimiix, 2022-03-12:
Not all screens are there, but there's a ddf in the pr list.

https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5839

So if you import that, you can use it :)

optikkore, 2022-03-12:
@Mimiix Pardon my ignorance do since I am in an HA addon I assume I need to add the file to the docker container?

Mimiix, 2022-03-12:
I think so, yes :)

closed 2022-04-10 by github-actions[bot]