dresden-elektronik/deconz-rest-plugin #5862: WXKG16LM  Aqara E1 Wireless Switch, Single rocker

yeongsong, 2022-03-12:
## Device

- Product name: Aqara E1 Wireless Switch
- Manufacturer: LUMI
- Model identifier: lumi.remote.acn003
- Device type : Switch

## Screenshots

  Required screenshots:
  - Endpoints and clusters of the node
[image]
  - Node Info panel
[image]

### Basic
[image]

### Identify
[image]

### Other clusters that are not mentioned above

[image]
[image]
[image]
[image]

Mimiix, 2022-05-24:
@Smanar can this be done with ddf?

Smanar, 2022-05-24:
lol, why you ask me ? it s not tuya stuff, this one have the lumi cluster, it s SwoopX hobby.

Seriously I don't know. I have some problem with battery switch and this one use the lumi cluster, I haven't found smilar device in actual DDF sample.

SwoopX, 2022-05-24:
This one should probably make it fully functionable:

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "$MF_LUMI",
  "modelid": "lumi.remote.acn003",
  "matchexpr": "R.endpoints.length === 1",
  "vendor": "Xiaomi",
  "product": "Aqara E1 single rocker wireless switch WXKG16LM",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_SWITCH",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0006"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0000",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0012",
          "0xFCC0"
        ],
        "out": [
          "0x0006"
        ]
      },
      "meta": {"group.endpoints": [1] },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid",
          "awake": true
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x00f7",
            "ep": 1,
            "fn": "xiaomi:special",
            "idx": "0x01",
            "script": "xiaomi_battery.js"
          }
        },
        {
          "name": "config/clickmode",
          "awake": true,
          "parse": {
            "at": "0x0125",
            "cl": "0xfcc0",
            "ep": 1,
            "eval": "if (Attr.val == 1) { Item.val = 'highspeed' } else if (Attr.val == 2) { Item.val = 'multiclick' } else { Item.val = 'unknown' }",
            "fn": "zcl",
            "mf": "0x115f"
          },
          "read": {
            "at": "0x0125",
            "cl": "0xfcc0",
            "ep": 1,
            "fn": "zcl",
            "mf": "0x115f"
          },
          "write": {
            "at": "0x0125",
            "cl": "0xfcc0",
            "dt": "0x20",
            "ep": 1,
            "eval": "if (Item.val == 'highspeed') { 1 } else if (Item.val == 'multiclick') { 2 }",
            "fn": "zcl",
            "mf": "0x115f"
          },
          "values": [
            ["\"highspeed\"", "Buttons emit only single click events, but fast"],
            ["\"multiclick\"", "Emits single, double and treble presses"] 
          ],
          "default": "multiclick"
        },
        {
          "name": "config/devicemode",
          "refresh.interval": 3000,
          "parse": {
            "at": "0x0009",
            "cl": "0xfcc0",
            "ep": 1,
            "eval": "if (Attr.val == 1) { Item.val = 'compatibility' } else if (Attr.val == 2) { Item.val = 'zigbee' } else { Item.val = 'unknown' }",
            "fn": "zcl",
            "mf": "0x115f"
          },
          "read": {
            "at": "0x0009",
            "cl": "0xfcc0",
            "ep": 1,
            "fn": "zcl",
            "mf": "0x115f"
          },
          "write": {
            "at": "0x0009",
            "cl": "0xfcc0",
            "dt": "0x20",
            "ep": 1,
            "eval": "if (Item.val == 'compatibility') { 1 } else if (Item.val == 'zigbee') { 2 }",
            "fn": "zcl",
            "mf": "0x115f"
          },
          "values": [
            ["\"compatibility\"", "Default mode for Xiaomi devices"],
            ["\"zigbee\"", "Closer to zigbee standard"] 
          ],
          "default": "compatibility"
        },
        {
          "name": "config/group",
          "default": "auto"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/buttonevent"
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "groupcast",
      "config.group": 0,
      "src.ep": 1,
      "cl": "0x0006"
    }
  ]
}
```

Smanar, 2022-05-25:
whoaaa.
You have made this DDF from scratch ? or you already have used this code for another device ?

SwoopX, 2022-05-25:
Looks pretty identical to the wireless H1 dual rocker I have here 🙂

closed 2022-06-23 by github-actions[bot]

mcf68k, 2022-10-15:
Hello, 
How can I make working the same switch but with 2 buttons (WXKG17LM) ? I already opened a new device request.
Can you confirm this one is working (WXKG16LM) before I order one ? 
Thanks

yeongsong, 2023-01-28:
> Hello,
> How can I make working the same switch but with 2 buttons (WXKG17LM) ? I already opened a new device request.
> Can you confirm this one is working (WXKG16LM) before I order one ?
> Thanks

nope, didn't work. somehow my switch got bricked after several attempts to edit the settings.

Idechix, 2024-03-15:
I've been struggling for several days to get the that module to work with deconz, but to no avail.
I've even tried adapting and modifying the DDF files of other similar modules like WXKG17LM and WXKG20LM without success.
Each time, the battery level doesn't work, and I don't receive any events when I press the button.

Any insights or suggestions would be greatly appreciated. Thanks in advance for your help!

Smanar, 2024-03-16:
Hello, what is your model id ?
And a Cluster picture to see wich one version you have ?

Idechix, 2024-03-16:
My model is lumi.remote.acn003 (WXKG16LM).

[image]

I didn't notice earlier that there are more attributes listed in mine than in yeongsong's.
Don't know if this has anything to do with the fact that mine doesn't work.

Smanar, 2024-03-16:
So there is nothing in deconz fo this device.
You can try with this DDF https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5862#issuecomment-1136459355

You will have a start of support, but you will need to edit too the button_maps.json file to add the device

If they are same device, and use same command can add your device in the WXKG17LM button map
```

        "xiaomiSwitchE1Acn004Map": {
            "vendor": "Xiaomi",
            "doc": "Xiaomi Aqara E1 dual rocker wireless switch WXKG17LM",
            "modelids": ["lumi.remote.acn004", "lumi.remote.acn003"],
```

Idechix, 2024-03-16:
That still doesn't work. (I already tried that a few days ago, but I tried it again today)

I've tried everything I know to make it work properly:
- copying the newly created DDF .json file into ~/.local/share/dresden-elektronik/deCONZ/devices
- copying xiaomi_battery.js and xiaomi_swversion.js into the same folder
- editing button_maps.json to include "lumi.remote.acn003"
- restarting deCONZ
- enabling Strict DDF mode
- pairing the device

As mentioned previously, the battery monitoring always reports 0, and there are no events sent when pressing the button.

Here are the only messages I receive from the websocket:
```
{"attr":{"id":"45", "lastannounced":null, "lastseen":"2024-03-16T15:25Z", "manufacturername":"LUMI", "mode":1, "modelid":"lumi.remote.acn003", "name":"Switch 45", "swversion":null, "type":"ZHASwitch", "uniqueid":"54:ef:44:10:00:7b:66:be-01-0006"}, "e":"changed", "id":"45", "r":"sensors", "t":"event", "uniqueid":"54:ef:44:10:00:7b:66:be-01-0006"}
{"config":{"battery":0, "clickmode":"multiclick", "devicemode":"unknown", "on":true, "reachable":true}, "e":"changed", "id":"45", "r":"sensors", "t":"event", "uniqueid":"54:ef:44:10:00:7b:66:be-01-0006"}
```

Idechix, 2024-03-16:
I've created a slightly modified version of the DDF mentioned earlier, which closely resembles the current DDF for the WXKG17LM.
Despite this, it still doesn't work. However, it could serve as a good starting point for experimentation.
```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "$MF_LUMI",
  "modelid": "lumi.remote.acn003",
  "matchexpr": "R.endpoints.length === 1",
  "vendor": "Xiaomi",
  "product": "Aqara E1 single rocker wireless switch WXKG16LM",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_SWITCH",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0006"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0000",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0012",
          "0xFCC0"
        ],
        "out": [
          "0x0006"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid",
          "awake": true
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {
            "at": "0x00f7",
            "ep": 1,
            "fn": "xiaomi:special",
            "idx": "0x08",
            "script": "xiaomi_swversion.js"
          },
          "read": {
            "fn": "none"
          }
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x00f7",
            "ep": 1,
            "fn": "xiaomi:special",
            "idx": "0x01",
            "script": "xiaomi_battery.js"
          }
        },
        {
          "name": "config/clickmode",
          "awake": true,
          "parse": {
            "at": "0x0125",
            "cl": "0xfcc0",
            "ep": 1,
            "eval": "if (Attr.val == 1) { Item.val = 'highspeed' } else if (Attr.val == 2) { Item.val = 'multiclick' } else { Item.val = 'unknown' }",
            "fn": "zcl:attr",
            "mf": "0x115f"
          },
          "read": {
            "at": "0x0125",
            "cl": "0xfcc0",
            "ep": 1,
            "fn": "zcl:attr",
            "mf": "0x115f"
          },
          "write": {
            "at": "0x0125",
            "cl": "0xfcc0",
            "dt": "0x20",
            "ep": 1,
            "eval": "if (Item.val == 'highspeed') { 1 } else if (Item.val == 'multiclick') { 2 }",
            "fn": "zcl:attr",
            "mf": "0x115f",
            "state.timeout": 2
          },
          "values": [
            ["\"highspeed\"", "Buttons emit only single click events, but fast"],
            ["\"multiclick\"", "Emits single, double and treble presses"] 
          ],
          "default": "multiclick"
        },
        {
          "name": "config/devicemode",
          "refresh.interval": 3000,
          "parse": {
            "at": "0x0009",
            "cl": "0xfcc0",
            "ep": 1,
            "eval": "if (Attr.val == 1) { Item.val = 'compatibility' } else if (Attr.val == 2) { Item.val = 'zigbee' } else { Item.val = 'unknown' }",
            "fn": "zcl:attr",
            "mf": "0x115f"
          },
          "read": {
            "at": "0x0009",
            "cl": "0xfcc0",
            "ep": 1,
            "fn": "zcl:attr",
            "mf": "0x115f"
          },
          "write": {
            "at": "0x0009",
            "cl": "0xfcc0",
            "dt": "0x20",
            "ep": 1,
            "eval": "if (Item.val == 'compatibility') { 1 } else if (Item.val == 'zigbee') { 2 }",
            "fn": "zcl:attr",
            "mf": "0x115f",
            "state.timeout": 2
          },
          "values": [
            ["\"compatibility\"", "Default mode for Xiaomi devices"],
            ["\"zigbee\"", "Closer to zigbee standard"] 
          ],
          "default": "compatibility"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/buttonevent"
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ]
}
```

Smanar, 2024-03-16:
```
  "manufacturername": "$MF_LUMI",
  "modelid": "lumi.remote.acn003",
  "matchexpr": "R.endpoints.length === 1",
```

The DDF need to work if the model id  is "lumi.remote.acn003", the vendor "LUMI" and if the device have only 1 endpoint.
You can remove the line
`  "matchexpr": "R.endpoints.length === 1",`
To be sure it's not deconz that have issue.

The device need to be reconised, it mean after the inclusion his name go from "0xXXXX" to something readable.

If you enable log in deconz/help debug view, with flag "info" and "info_l2" you need to see lines like this one
`INFO] - No button handler for: %s%s, endpoint: 0x%02X, cluster: %s, command: %s, payload: %s, zclSeq: %u\n",`

And we use thoses return to make the button_maps.json file

Idechix, 2024-03-16:
> ```
>   "manufacturername": "$MF_LUMI",
>   "modelid": "lumi.remote.acn003",
>   "matchexpr": "R.endpoints.length === 1",
> ```
> 
> The DDF need to work if the model id is "lumi.remote.acn003", the vendor "LUMI" and if the device have only 1 endpoint. You can remove the line ` "matchexpr": "R.endpoints.length === 1",` To be sure it's not deconz that have issue.
> 
> The device need to be reconised, it mean after the inclusion his name go from "0xXXXX" to something readable.
> 
> I think this part is ok now, as you are able to see return by websocket.

Yes, the device is recognized correctly.
> 
> If you enable log in deconz/help debug view, with flag "info" and "info_l2" you need to see lines like this one `[INFO] - No button handler for: %s%s, endpoint: 0x%02X, cluster: %s, command: %s, payload: %s, zclSeq: %u\n",` When you use a button.
> 
> And we use thoses return to make the button_maps.json file

There are no lines like that in the debug view when I press the button. (I tried with and without the button_maps.json modification you suggested earlier)

Smanar, 2024-03-16:
Ha yes, we forget something, the binds.
Can make a try with the on/off, you just need a bind from the device to the coordinator without report, something like
```

    {
      "bind":"unicast",
      "src.ep":1,
      "dst.ep":1,
      "cl":"0x0006",
}
```

Idechix, 2024-03-16:
Where should I insert that command?
(I'm not home tonight, so I'll try that tomorrow...)

Smanar, 2024-03-16:
In fact it's just the "config/clickmode", the DDF need to set the mode at "multiclick" perhaps this request have failed ?

Idechix, 2024-03-16:
Should it be set in deCONZ via the DDF file, or externally with a request?

Smanar, 2024-03-17:
Externally with a request, using the API.
The DDF need to make the request itself, perhaps a problem here.

Else an easier thing, set a bind from the device to the coordinator on the cluster 0x0012, this can be done using Deconz https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2077#issuecomment-554640916

Idechix, 2024-03-17:
Nothing worked...

I think values aren't correctly read from device: swversion, config/battery and config/devicemode are generally reported as none (or 0).
config/clickmode gives me a readable result, but it is sometimes 'highspeed', sometimes 'multiclick'...

Martinvdm, 2025-06-02:
Older issue, but seems still relevant.
tried with Deconz 2.30.2. It seems identical to 
 lumi.remote.acn004, this WXKG16LM is lumi.remote.acn003.
Did someone ever make this working or can help me to get this working? I have the same wireless switch.

Martinvdm, 2025-06-03:
> There is a DDF for the lumi.remote.acn004 but not for the lumi.remote.acn003 you have the second one ? There is a starting DDF here [#5862 (comment)](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5862#issuecomment-2002023092)
> 
> With some comments and the way to add support for state/buttonevent.

Yes I have the lumi.remote.acn003
I only need single click, no special tricks.
Is there any recent instruction/manual how add such ddf the correct way to deconz docker in a good way? It seems that some instructions around are already obsolete

Smanar, 2025-06-03:
The procedure in the video is still used, but it's to create or edit DDF using the editor. Here the DDF is already created, but you can try to reproduce the same DDF using the editor, but for me it's faster to use text editor.

There is https://deconz-community.github.io/ddf-tools/#/ but not tried with DDF, as on my side I have a real OS, so I can edit the file I want in less than 20s.

You can still use the DDF and catch the debug message https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5862#issuecomment-2002025515

 According to them I can try to make a PR in blind mode, if it seem correct.

Martinvdm, 2025-06-03:
Ok we are making progress.
I have created in the devices folder an xiaomi_wxkg16lm_e1_switch.json with [your ](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5862#issuecomment-1136459355) file in it. Reboot Deconz. 
Now the devices is paired and its giving logging at pairing proces:

20:51:37:479 0x00158D0001A6592F extract Xiaomi special attribute 0xFF01
20:52:33:782 DEV found DDF for 0x54EF4410013091E1, path: /opt/deCONZ/devices/xiaomi_wxkg16lm_e1_switch.json, result: 1

Only buttons are not working and Home Assistant is not able to listen to button events, probable becaus of missing button file or something?

Martinvdm, 2025-06-03:
addition to above; 
i enabled debug info and info_l2. following output:

21:22:27:829 [INFO] - No button map for: lumi.remote.acn003, unicast to: 0x0000, endpoint: 0x01, cluster: MULTISTATE_INPUT (0X0012), command: ATTRIBUTE_REPORT (0X0A), payload: 5500210100, zclSeq: 150

Martinvdm, 2025-06-04:
It seems those line are exactly the same. Could it be that this device only support single click and no multi click?

Smanar, 2025-06-05:
It's possible, there is sometime different versions with more features.
The lumi.remote.acn004 support double and triple clic, same endpoint, same cluster, same command, there is just a difference in the payload. Make a try with your device.

21:22:27:829 [INFO] - No button map for: lumi.remote.acn003, unicast to: 0x0000, endpoint: 0x01, cluster: MULTISTATE_INPUT (0X0012), command: ATTRIBUTE_REPORT (0X0A), **payload: 5500210100**, zclSeq: 150

Martinvdm, 2025-06-06:
I just edited the file whitin the docker container. So it is temporary, a pr is needed.

Martinvdm, 2025-06-07:
https://github.com/dresden-elektronik/deconz-rest-plugin/pull/8245