dresden-elektronik/deconz-rest-plugin #5864: Aqara smart smoke detector - JY-GZ-01AQ

roelvandermade, 2022-03-14:
## Device

- Product name: Aqara Smart Smoke Detector
- Manufacturer:  Aqara
- Model identifier: JY-GZ-01AQ
- Device type : Please remove all unrelated device types. 
  - Sensor
 

## Screenshots

### Basic

### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

Mimiix, 2022-03-14:
Missing screenshots.

Please add them all.

roelvandermade, 2022-03-14:
Some reference here also: https://zigbee.blakadder.com/Aqara_JY-GZ-01AQ.html

Mimiix, 2022-03-14:
In the original post please. Add all clusters as requested.

roelvandermade, 2022-03-14:
> Missing screenshots.
> 
> 
> 
> Please add them all. 

Added them all, please let me know if you need more information.

Smanar, 2022-03-14:
Hello, this device realy look classic. Do want to make a try with DDF ?
You can start from scratch https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet
Or edit an existning one, like this one https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/frient/smszb-120_smoke_detector.json
But you need to make so much change on it, and I don't find another one more clear.

roelvandermade, 2022-03-15:
> Hello, this device realy look classic. Do want to make a try with DDF ? You can start from scratch https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet Or edit an existning [...]

That's a new path for me, but I will give it a try :) 
Will loop back here if there has been any succes.

thanks

Smanar, 2022-03-15:
Sure, If you can at least make something for the device be reconized, just copy/paste your DDF (and save the file, to have the file path), I can finish it using text editor.

Smanar, 2022-04-13:
@roelvandermade You have give up ?

roelvandermade, 2022-04-15:
Not given up but more into a lack of time and understanding. Will dive back into it in a few weeks. Will let you know then how far I’ve come and where I got stuck.

Smanar, 2022-04-15:
Sure, I can't help you for time, but I can if you need more guidance.

wvuyk, 2022-07-24:
I received two of these smoke sensors today and am trying to follow up on the github thread that has been closed a while ago due to lack of response.

This is my first attempt to use the DDF editor to at least deliver the basic file Smanar was asking for. But I just do not get how to use the DDF editor. I select the device that I need to add and choose “edit DDF” this opens an Untitled file with prefilled modelid lumi.sensor_smoke.acn03 and Manufacturer LUMI as basic information. Then I enable the sleeper option and drag the ZHAFire to the left side of the window.

I checked the attributes and assume all I should change maybe is the settings for the fire attribute?
Here is where I get lost and do not know where to go from here. I have set the file on bronze and enabled bronze in the DDF options of deCONZ. I saved the file at this point and did a hot reload. Rescanned the devices and nothing seems to have changed.

When opening the ddf file on the device again the file I saved is opened so it has loaded it. I see the option to select the different parse options, including Xiaomi: special. But have no clue what to set in this. I have tried to read through documentation but am stuck here.

How can I get this moving forward to the REST API?

Thanks,

Wim

Smanar, 2022-07-25:
Hello have started a DDF can use this one as base

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "LUMI",
  "modelid": "lumi.sensor_smoke.acn03",
  "vendor": "Xiaomi",
  "product": "JY-GZ-01AQ smoke sensor",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_FIRE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0402",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0500"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "refresh.interval": 84000,
          "read": {
            "at": "0x0006",
            "cl": "0x0000",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0006",
            "cl": "0x0000",
            "ep": 1,
            "eval": "Item.val = Attr.val",
            "fn": "zcl"
          }
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery"
        },
        {
          "name": "config/enrolled"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/pending"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/fire"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/lowbattery"
        },
        {
          "name": "state/test"
        }
      ]
    },
    {
      "type": "Warning device",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "refresh.interval": 84000,
          "read": {
            "at": "0x0006",
            "cl": "0x0000",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0006",
            "cl": "0x0000",
            "ep": 1,
            "eval": "Item.val = Attr.val",
            "fn": "zcl"
          }
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert"
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0020",
          "dt": "0x20",
          "min": 300,
          "max": 43200,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 35,
      "cl": "0x0500"
    }
  ]
}
```

You can compare it with your "preview" tab, or can show screen capture.

- I have created 1 "ZHAFire" and a "Warning device"
- I m using classic method for IAS cluster
- not sure for sleeper, as this device have a siren too nope ?
- It seem this device ned a special magic during inclusion, or it s for advanced condig ...
-I don't see the Xiaomi specail cluster on capture, so I m using classic method, but I suspect an invisible cluster (need to use it for previous remark)

wvuyk, 2022-07-25:
I will check on this. I believe I saw a xiaomi cluster in my version (different firmware?) If I see those I will post the images here. Will be back on this

wvuyk, 2022-07-25:
I tried to use your version, but it did not change, I do not see the warning device here, not sure if it is needed. But there is a sirene sound when testing, so I guess something should be there. I do see the Lumi specific cluster. I will post all images from my device after performing read on each of them:

[image]

Basic cluster:
[image]

IAS zone:
[image]

Power configuration
[image]

Lumi specific part 1
[image]

Lumi specific part 2
[image]

Does this change things?

SwoopX, 2022-07-25:
DDF should be stripped down a bit like that:

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "$MF_LUMI",
  "modelid": "lumi.sensor_smoke.acn03",
  "vendor": "Xiaomi",
  "product": "Aqara smoke sensor JY-GZ-01AQ",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_FIRE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0402",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0500"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid",
          "awake": true
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "awake": true,
          "parse": {
            "at": "0x00f7",
            "ep": 1,
            "fn": "xiaomi:special",
            "idx": "0x08",
            "script": "xiaomi_swversion.js"
          },
          "read": {
            "fn": "none"
          }
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x00f7",
            "ep": 1,
            "fn": "xiaomi:special",
            "idx": "0x01",
            "script": "xiaomi_battery.js"
          }
        },
        {
          "name": "config/enrolled"
        },
        {
          "name": "config/ledindication",
          "refresh.interval": 300,
          "parse": {
            "at": "0x013c",
            "cl": "0xfcc0",
            "ep": 1,
            "eval": "Item.val = Attr.val == 1 ? true : false",
            "fn": "zcl",
            "mf": "0x115f"
          },
          "read": {
            "at": "0x013c",
            "cl": "0xfcc0",
            "ep": 1,
            "fn": "zcl",
            "mf": "0x115f"
          },
          "write": {
            "at": "0x013c",
            "cl": "0xfcc0",
            "dt": "0x20",
            "ep": 1,
            "eval": "Item.val ? false : true",
            "fn": "zcl",
            "mf": "0x115f"
          },
          "values": [
            [true, "Device LED will flash every 60 seconds"],
            [false, "Device LED off"] 
          ],
          "default": true
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/pending"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/fire"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/lowbattery"
        },
        {
          "name": "state/test"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 01,
      "cl": "0x0500"
    }
  ]
}
```

The device doesn't have any IAS WD cluster, so no warning device.

Smanar, 2022-07-26:
If the name stay an hexa number (like 0x770B) It mean the DDF is not used.
- So check model id / manufacture name
- Try a "hot relaod" in the menu

@SwoopX 
>The device doesn't have any IAS WD cluster, so no warning device.

For that I m reading he have, but use the Lumic specific cluster, not a basic device, not sure all will be possible using only DDF

wvuyk, 2022-07-26:
I updated the DDF using @SwoopX 's sample. The DDF could not be read, until I changed the bindings section to use  `"src.ep": "01",` instead of `"src.ep": 01,`. Here it could be read and the name changed to Fire 142. So the DDF is used resulting in REST API reply:

```
{
"config": {
"battery": 0,
"ledindication": false,
"on": true,
"pending": [],
"reachable": false
},
"ep": 1,
"etag": "f7271de847e69a4fdfff0b5827549bc8",
"lastannounced": null,
"lastseen": "2022-07-26T14:46Z",
"manufacturername": "LUMI",
"modelid": "lumi.sensor_smoke.acn03",
"name": "Fire 142 Fire",
"state": {
"fire": null,
"lastupdated": "none",
"lowbattery": null,
"test": null
},
"type": "ZHAFire",
"uniqueid": "54:ef:44:10:00:48:2d:c2-01-0500"
}
```
But the battery value is never updated. And if I check the preview, the bindings section is empty. So I guess reporting is not yet active? I have added a binding for battery as shown in the DDF Cheat sheet. I did that maybe 15 minutes ago, so it could be a bit longer before a first report would be send I guess? Would I need more bindings other then the battery for this device?

The DDF at its current state:

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "$MF_LUMI",
  "modelid": "lumi.sensor_smoke.acn03",
  "vendor": "Xiaomi",
  "product": "Aqara smoke sensor JY-GZ-01AQ",
  "sleeper": true,
  "status": "Gold",
  "path": "/devices/lumi.sensor_smoke.acn03.json",
  "subdevices": [
    {
      "type": "$TYPE_FIRE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0402",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0500"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x00f7",
            "ep": 1,
            "fn": "xiaomi:special",
            "idx": "0x01",
            "script": "xiaomi_battery.js"
          },
          "default": 0
        },
        {
          "name": "config/enrolled",
          "public": false
        },
        {
          "name": "config/ledindication",
          "refresh.interval": 300,
          "read": {
            "at": "0x013c",
            "cl": "0xfcc0",
            "ep": 1,
            "fn": "zcl",
            "mf": "0x115f"
          },
          "write": {
            "at": "0x013c",
            "cl": "0xfcc0",
            "dt": "0x20",
            "ep": 1,
            "eval": "Item.val ? false : true",
            "fn": "zcl",
            "mf": "0x115f"
          },
          "parse": {
            "at": "0x013c",
            "cl": "0xfcc0",
            "ep": 1,
            "eval": "Item.val = Attr.val == 1 ? true : false",
            "fn": "zcl",
            "mf": "0x115f"
          },
          "default": true
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/pending"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/fire"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/lowbattery"
        },
        {
          "name": "state/test"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0020",
          "dt": "0x20",
          "min": 300,
          "max": 43200,
          "change": "0x00000001"
        }
      ]
    }
  ]
}

```

Smanar, 2022-07-26:
>But the battery value is never updated

Its my bad, on your first capture, the lumi cluster was not present, but I m sure now it use this one for the battery, so need to change the DDF part for battery, need to remove the bind and use probably something like

```
        {
          "name": "config/battery",
          "parse": {"fn": "xiaomi:special", "at": "0xff01", "idx": "0x01", "script": "xiaomi_battery.js"}
        },
```

SwoopX, 2022-07-26:
Please use the suggested DDF as above while applying the necessary fix. There is no need to try to apply any other bindings or attribute reporting, as it will fail or be ignored in 90% of the case.

The reason why you do not see any battery values is that the required script is not in the same location as the DDF. Also, the swversion will not show up unless the new deconz version is out, but preperations are made.

It will be interesting to see, if the device follows zigbee enrollment process or if that fails (can be checked in the IAS zone cluster).

roelvandermade, 2022-08-02:
Hi,

Have been trying the suggested DDF's with the battery-fix and am getting mixed results but not all of them feel like the final and best configuration. 

My results now are;
- the device is showing up in Phoscon and deCONZ with a light-flash trigger identification working when executed in deCONZ's 'Identify cluster'. I have no clue how to have it identify itself (or test itself) with an audible signal from within deCONZ and/or HA
- the battery status SEEMS to be updating (see screenshot) (I have added the xiaomi_battery.js script taken from [https://raw.githubusercontent.com/dresden-elektronik/deconz-rest-plugin/master/devices/xiaomi/xiaomi_battery.js])
- the sensor status also SEEMS to be updating (see screenshot)
- the Smoke Sensor switch-button does not work (it used to work during testing several settings/ddf's but cannot reproduce this anymore somehow, though I was able to toggle the switch but it didn't do anything further).

So my questions now are; 
- how can we be certain the battery and sensor status are indeed accurate and updated? 
- Do we need to flip switches/configurations in the Cluster-zone settings, or is it all just the DDF file for identification and configuration?
- how can we have the smoke sensor identify itself with an audible signal from within deCONZ or HA?

attached the screenshot from HomeAssistant and attached my current DDF.

@SwoopX regarding your IAS Zone enrollment, I have also added a screenshot regarding that, it seems to be enrolled successfully, though I am unsure at this stage if that is due to my fiddling around, or not.

thanks!
Roel.
```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "$MF_LUMI",
  "modelid": "lumi.sensor_smoke.acn03",
  "vendor": "Xiaomi",
  "product": "Aqara smoke sensor JY-GZ-01AQ v2",
  "sleeper": true,
  "status": "Gold",
  "path": "/devices/aqara_smart_smoke_sensor.json",
  "subdevices": [
    {
      "type": "$TYPE_FIRE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0402",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0500"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "parse": {
            "at": "0xff01",
            "fn": "xiaomi:special",
            "idx": "0x01",
            "script": "xiaomi_battery.js"
          },
          "default": 0
        },
        {
          "name": "config/enrolled",
          "public": false
        },
        {
          "name": "config/ledindication",
          "refresh.interval": 300,
          "read": {
            "at": "0x013c",
            "cl": "0xfcc0",
            "ep": 1,
            "fn": "zcl",
            "mf": "0x115f"
          },
          "write": {
            "at": "0x013c",
            "cl": "0xfcc0",
            "dt": "0x20",
            "ep": 1,
            "eval": "Item.val ? false : true",
            "fn": "zcl",
            "mf": "0x115f"
          },
          "parse": {
            "at": "0x013c",
            "cl": "0xfcc0",
            "ep": 1,
            "eval": "Item.val = Attr.val == 1 ? true : false",
            "fn": "zcl",
            "mf": "0x115f"
          },
          "default": true
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/pending"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/fire"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/lowbattery"
        },
        {
          "name": "state/test"
        }
      ]
    },
    {
      "type": "Warning device",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert",
          "default": "none"
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0020",
          "dt": "0x20",
          "min": 300,
          "max": 43200,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 35,
      "cl": "0x0500"
    }
  ]
}
```

[image]

[image]

Smanar, 2022-08-02:
>how can we be certain the battery and sensor status are indeed accurate and updated?

Good question ^^, (you can see request in logs but idk if you can trust them) can try with an end of life battery ? And with the cluster IAS you have too a state/lowbattery

>Do we need to flip switches/configurations in the Cluster-zone settings, or is it all just the DDF file for identification and configuration?

Not sure to understand, but the DDF is here to make ALL the needed stuff (it s trying at least), on a final DDF the user will not need to use the GUI.

>how can we have the smoke sensor identify itself with an audible signal from within deCONZ or HA?

Have you try the siren ? You have some possible values to use here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4432#issuecomment-782841021

GyroGearl00se, 2022-10-28:
Anyone an idea if this device will be officially supported? Currently I'm not able to pair it.

ottelo9, 2022-11-22:
How to pair this device? I pressed the button 3 times, the LED is blinking green but nothing happens.

ottelo9, 2022-11-22:
[image]
[image]
[image]

Smanar, 2022-11-23:
>Ah ok I found out. The device will automatically pair with deconz but we cannot see the success in phoscon, only in the deconz vnc view.

I never use phoscon, as it don't support all device.
All devices can be see on deconz event not supported in the API
The important part is the support in the API, if it have worked

- The device is visible in the API, using third app or phoscon/help/debug view/sensors
- The name in deconz change from 0xXXXX to a real name.

ottelo9, 2022-11-23:
I uploaded the ddf (json file) from @roelvandermade to deconz and open it via the ddf editor for the smoke sensor. But what can I do now? I cannot see the new device in Home Assistant?

Smanar, 2022-11-23:
Strange according to your capture, the DDF is the good one.

After having doing that, if you select the node and make "edit DDF", it s the good that appear ?
You have put the file in a "devices" folder ?

No error when you have loaded the file ?

ottelo9, 2022-11-23:
Ah I had to set the status to gold, save it and do a hot reload. Now its visible in Phoscon and HA. But in HA I got only the Test Mode Entity nothing else? Whats about the battery? Do I have to put the xiaomi_battery.js somwhere? My DDF file is lying in /data/.local/share/dresden-elektronik/deCONZ/devices/

Smanar, 2022-11-24:
>Ah I had to set the status to gold

Wich one DDF have you used ? on the @roelvandermade post the status is already "Gold".

IDK how work HA, can you take a look direclty in the API, suign for exemple phoscon/help/API Information/sensors and search for the ZHAFire sensor ?

ottelo9, 2022-11-24:
I dont know why my status changed back to draft.

> IDK how work HA, can you take a look direclty in the API, suign for exemple phoscon/help/API Information/sensors and search for the ZHAFire sensor ?

```
{
    "config": {
        "on": true,
        "reachable": true
    },
    "etag": "224a4d2989e56be8964cbfb6bc66f988",
    "lastannounced": null,
    "lastseen": "2022-11-24T11:03Z",
    "manufacturername": "LUMI",
    "modelid": "lumi.sensor_smoke.acn03",
    "name": "Rauchmelder",
    "state": {
        "fire": false,
        "lastupdated": "2022-11-24T07:49:42.688"
    },
    "swversion": "2019",
    "type": "ZHAFire",
    "uniqueid": "54:ef:44:10:00:55:ca:4b-01-0500"
}
```

Smanar, 2022-11-24:
There is a problem with the DDF. On this link https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5864#issuecomment-1202987196

The DDF have a config/battery that is not present on your device sjon.

ottelo9, 2022-11-24:
But he wrote that it's working with screenshot of the battery entity in HA. I use his DDF and also put the xiaomi_battery.js in the device folder. See my [comment](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5864#issuecomment-1325531979).

Smanar, 2022-11-24:
Yep, but just mean the DDF have worked for him and not you.

There is somes strange thing during your test, like the status that was not on "Gold" and the missing field.

If you select the node in VNC, and make "edit DDF" on the preview tab, can you compare your DDF with the one on the post ?
Can try too delete the device and re-include it from scratch, no need to make "hot reload" in this situation, as the DDF is in the folder it will be checked by deconz on start and will be used for inclusion.

ottelo9, 2022-11-24:
Now its working. It was my fault, I uploaded two files to the device folder. After a rebind deconz took the one empty draft file :). After delete the draft file everything is working fine!!

roelvandermade, 2022-11-24:
Hey guys :)

Great to see it is working for you now! 

I’m curious if it stays connected for a longer time also, after some time, in Phoscon at least, it becomes unreachable. In HA the status seems to be OK, I haven’t tested it with smoke or something yet :) 

@ottelo9 it’s also unclear to me what or how the ‘fire-test mode’ and siren test entities are used.

@Smanar what would need from me/us to get this DDF included into a next Deconz release?

Cheers
Roel.

roelvandermade, 2022-11-24:
Btw, if you happen to use z2m there seems to be a bit more support for this sensor (https://www.zigbee2mqtt.io/devices/JY-GZ-01AQ.html) I haven't tested this.

ottelo9, 2022-11-25:
Is there a chance to look into the code to figure it out how to implement more functions in deconz ? I will check it as soon as I have more time for that.

roelvandermade, 2022-11-25:
@ottelo9 good suggestion; I'm an infra guy, no coder, so hope you are able to :)

roelvandermade, 2022-11-25:
> > I’m curious if it stays connected for a longer time also, after some time, in Phoscon at least, it becomes unreachable. In HA the status seems to be OK, I haven’t tested it with smoke or something yet :)
> 
> have the issue on bothn entries ?

I only have one device, that one indeed seems to lose connection after 24 hours according to Phoscon. I would need to have to check the coming days after leaving the device alone.

> Can you check the "lastseen", sensor are marked "unavalable after 24h without report, you need to have at least the battery one.

Yes will do in the coming days and report back.

> > @Smanar what would need from me/us to get this DDF included into a next Deconz release?
> 
> Just need someone make a PR, or I can make it if needed.

If you could do that once we've compiled a final version of the DDF, that would be great.

> An important question, you don't have battery drain ?

Indeed, no drain it seems, I have only seen it at 100% while I have the sensor on for at least 6 months now I think, there must be some drain now I'd say.
 
> Have updated the DDF a little.
> 
> * increase some poll value
> * enable the state/test feature

Thanks for checking the options in the z2m version and the update!
There's a small error in your DDF at line 66, the trailing comma needs to go.

Your updated DDF is adding a 'Tampered' entity, as you can see in the following screenshot, was this intentional? 
Also I do remember a switch-option for the test-mode somewhere in the earlier versions of the DDF, now we do see a test-mode but how would we be able to initiate such test? Or maybe this is not relevant at all, don't know.

[image]

roelvandermade, 2022-11-25:
btw @Smanar might make sense to re-open this issue, since there's quite some activity again ;-)

roelvandermade, 2022-11-25:
I was running through some logs and saw these in the syslog, might be relevant:

```
Nov 25 15:13:11 bmax d7d44f20cafc[1849]: 15:13:11:760 [WARNING] - Missing cluster in sensor fingerprint: 0x54EF44100045D934 - 0x2137 (lumi.sensor_smoke.acn03), endpoint: 0x01, cluster: 0xFCC0, payload: 27012000, zclSeq: 229
Nov 25 15:13:11 bmax d7d44f20cafc[1849]: 15:13:11:761 [INFO] - No button map for: lumi.sensor_smoke.acn03, unicast to: 0x0000, endpoint: 0x01, cluster: XIAOMI (0xFCC0), command: ATTRIBUTE_REPORT (0x0A), payload: 27012000, zclSeq: 229
Nov 25 15:13:16 bmax d7d44f20cafc[1849]: 15:13:15:995 [INFO] - No button map for: lumi.sensor_smoke.acn03, unicast to: 0x0000, endpoint: 0x01, cluster: BASIC (0x0000), command: 0x0A, payload: 050042176C756D692E73656E736F725F736D6F6B652E61636E3033, zclSeq: 230
```

Smanar, 2022-11-25:
>I only have one device, that one indeed seems to lose connection after 24 hours according to Phoscon. I would need to have to check the coming days after leaving the device alone

The DDF create 2 entries, a "sensor" a ZHAFire and a "light" a "warning device"

>There's a small error in your DDF at line 66, the trailing comma needs to go.

Corrected, thx.

>Your updated DDF is adding a 'Tampered' entity, as you can see in the following screenshot, was this intentional?

Heeee, no ^^, I don't see this field in the DDF, but yes this device have probably one (alarm), but not normal you have it ....

>Also I do remember a switch-option for the test-mode somewhere in the earlier versions of the DDF, now we do see a test-mode but how would we be able to initiate such test? Or maybe this is not relevant at all, don't know.

IDK how they work in HA.
But you need to have "state/test" as bool (true/false) and "config/ledindication" as bool too (true/false), but this code part is not from me, so idk what it does.

You can trigger them using direclty the api too.
`
curl -H 'Content-Type: application/json' -X PUT -d '{"ledindication": true}' http://IP:PORT/api/API_KEY/sensors/ID/config`

>Nov 25 15:13:11 bmax d7d44f20cafc[1849]: 15:13:11:760 [WARNING] - Missing cluster in sensor fingerprint: 0x54EF44100045D934 - 0x2137 (lumi.sensor_smoke.acn03), endpoint: 0x01, cluster: 0xFCC0, payload: 27012000, zclSeq: 229

This is just a warning, can be solved with adding the cluster in the DDF, will be added in 5 mn on previous post.

>Nov 25 15:13:11 bmax d7d44f20cafc[1849]: 15:13:11:761 [INFO] - No button map for: lumi.sensor_smoke.acn03,

This is used for button switch, no need for this device (at least for now)

ottelo9, 2022-11-25:
@Smanar 
Cool, thanks for the updated DDF. I will wait until you add the missing cluster.

Smanar, 2022-11-25:
Done, I think it's this one

```
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0402",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0500",
          "0xfcc0"
        ]
```

roelvandermade, 2022-11-25:
> Already done, I think it's this one
> 
> ```
>       "fingerprint": {
>         "profile": "0x0104",
>         "device": "0x0402",
>         "endpoint": "0x01",
>         "in": [
>           "0x0000",
>           "0x0001",
>           "0x0500",
>           "0xfcc0"
>         ]
> ```

That indeed removed the warning message from the syslog, though it gave me temperature entities :-)

[image]

The power config cluster shows the battery voltage at 30, it's a 3V battery, so assuming that indeed means 100%..

[image]

@ottelo9 how does this all look for you with the latest DDF loaded?

roelvandermade, 2022-11-25:
hmmm battery now showing 1% ?

[image]

Smanar, 2022-11-25:
For the battery the DDF use the xiaomi cluster

```
        {
          "name": "config/battery",
          "parse": {
            "at": "0xff01",
            "fn": "xiaomi:special",
            "idx": "0x01",
            "script": "xiaomi_battery.js"
          },
          "default": 0,
          "refresh.interval": 86400
        },
```

BTW, where have you put the DDF ? You need to have the xiaomi_battery.js file in the same folder.
As this code is the same used for all other device if it don't work we can try with the cluster 0x0001.

roelvandermade, 2022-11-25:
nvm, it's back to 1%. Maybe this is indeed the case btw, hence my remark about the battery..

Smanar, 2022-11-25:
>I changed the battery cluster as suggested (was thinking about the same) to 0x0001
Something like 
```
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2"
          }
        },

But need too, make the bind/report, try to read the value in deconz, if it update the value in the API, it's a bind/report probleml, else its from the DDF
```

roelvandermade, 2022-11-26:
> > I changed the battery cluster as suggested (was thinking about the same) to 0x0001
> 
> Something like
> 
> ```
>         {
>           "name": "config/battery",
>           "awake": true,
>           "parse": {
>             "at": "0x0021",
>             "cl": "0x0001",
>             "ep": 1,
>             "eval": "Item.val = Attr.val / 2"
>           }
>         },
> ```
> 
> But need too make the bind/report, try to read the value in deconz, if it update the value in the API, it's a bind/report probleml, else its from the DDF
> 
> Edit: Its normal, lol I m seing the capture, the attribute 0x0021 is not working, need to use the 0x0020, wich one have you used ?

Am kind of lost now :) 
The battery part in my DDF is the following, a mix if your remarks which shows me again 1% battery left (as before) and still showing the 25C temperature entities (which make no sense).

```
        {
          "name": "config/battery",
          "awake": "true",
          "parse": {
            "at": "0x0020",
            "cl": "0x0001",
            "ep": "1",
            "eval": "Item.val = Attr.val / 2",
            "fn": "xiaomi:special",
            "idx": "0x01",
            "script": "xiaomi_battery.js"
          },
          "default": 0,
          "refresh.interval": 86400
        },
```

Also the cluster warning is back

```
Nov 26 11:02:59 bmax d7d44f20cafc[1849]: 11:02:59:951 [WARNING] - Missing cluster in sensor fingerprint: 0x54EF44100045D934 - 0x2137 (lumi.sensor_smoke.acn03), endpoint: 0x01, cluster: 0x0019, payload: 000000F00000000000000000020000230F0000000600003000, zclSeq: 190
```

Suggestions?

Smanar, 2022-11-26:
>Also the cluster warning is back

This time it's the cluster 0x0019, the OTAU, the DDF realy don't need it, and this cluster is not used a lot (only for  OTA update), so this message will be not invasive.

And yes, your code is a lot of mix ^^ too much. Try this DDf for battery, I m using the attrbute 0x0020 in the cluster 0x0001 and bind enabled.

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "$MF_LUMI",
  "modelid": "lumi.sensor_smoke.acn03",
  "vendor": "Xiaomi",
  "product": "Aqara smoke sensor JY-GZ-01AQ v2",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_FIRE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0402",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0500",
          "0xfcc0"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x0020",
            "cl": "0x0001",
            "ep": 1,
            "eval": "const vmin = 25; const vmax = 30; let bat = Attr.val; if (bat > vmax) { bat = vmax; } else if (bat < vmin) { bat = vmin; } bat = ((bat - vmin) / (vmax - vmin)) * 100; if (bat > 100) { bat = 100; } else if (bat <= 0)  { bat = 1; } Item.val = bat;",
            "fn": "zcl"
          },
          "read": {
            "fn": "None"
          }
        },
        {
          "name": "config/enrolled",
          "public": false
        },
        {
          "name": "config/ledindication",
          "refresh.interval": 3000,
          "read": {
            "at": "0x013c",
            "cl": "0xfcc0",
            "ep": 1,
            "fn": "zcl",
            "mf": "0x115f"
          },
          "write": {
            "at": "0x013c",
            "cl": "0xfcc0",
            "dt": "0x20",
            "ep": 1,
            "eval": "Item.val ? false : true",
            "fn": "zcl",
            "mf": "0x115f"
          },
          "parse": {
            "at": "0x013c",
            "cl": "0xfcc0",
            "ep": 1,
            "eval": "Item.val = Attr.val == 1 ? true : false",
            "fn": "zcl",
            "mf": "0x115f"
          },
          "default": true
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/pending"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/fire"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/lowbattery"
        },
        {
          "name": "state/test",
          "refresh.interval": 3000,
          "read": {
            "at": "0x0127",
            "cl": "0xfcc0",
            "ep": 1,
            "fn": "zcl",
            "mf": "0x115f"
          },
          "write": {
            "at": "0x0127",
            "cl": "0xfcc0",
            "dt": "0x10",
            "ep": 1,
            "eval": "Item.val ? false : true",
            "fn": "zcl",
            "mf": "0x115f"
          },
          "parse": {
            "at": "0x0127",
            "cl": "0xfcc0",
            "ep": 1,
            "eval": "Item.val = Attr.val == 1 ? true : false",
            "fn": "zcl",
            "mf": "0x115f"
          },
          "default": true
        }
      ]
    },
    {
      "type": "Warning device",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert",
          "default": "none"
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0020",
          "dt": "0x20",
          "min": 300,
          "max": 43200,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0500"
    }
  ]
}
```

If you don't want to wait for the first report, can read the value in deconz, it will update the value in the API.

roelvandermade, 2022-11-26:
Reads and Exec-buttons in deconz are now resulting in a timeout and a red blinking 'light' on the left side of the device in deconz overview. So I assume there's something wrong with the DDF, since this was working before, for example in the Identify cluster I could initiate a blinking led on the smoke sensor itsself (the physical device), this now timeouts in deconz. I do not see any relevant errors in the logfile. 

The battery-temp entities as well the tampering entity are gone and the battery is back at 100% , so this is good :)

ottelo9, 2022-11-26:
I replaced the DDF with the new one, removed the device from deconz and then added the device again (just to be sure). I manually read/update the Basic und Power Cluster to get the actual correct values.

I now have the battery, low battery, test mode under diagnostic in HA and the binary_sensor and the siren entity. So everything exactly as before. Battery = 100%

In deconz I can press the exec button to identify the smoke detector. The LED is blinking green, so its working for me!

Smanar, 2022-11-26:
>In HA the button for the siren dont do anything, as before.

The warning device is not working at all ? I don't know HA, do you know the request made by HA (perhaps visible in HA logs ? )

The green led blinking mean a poll, try adding a `"refresh.interval": 86400,` as it the only new thing in the DDF

```
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x0020",
            "cl": "0x0001",
            "ep": 1,
            "eval": "const vmin = 25; const vmax = 30; let bat = Attr.val; if (bat > vmax) { bat = vmax; } else if (bat < vmin) { bat = vmin; } bat = ((bat - vmin) / (vmax - vmin)) * 100; if (bat > 100) { bat = 100; } else if (bat <= 0)  { bat = 1; } Item.val = bat;",
            "fn": "zcl"
          },
          "refresh.interval": 86400,
          "read": {
            "fn": "None"
          }
        },
```

But as the "read" part in empty, strange deconz trying something here, the poll is not visible in deconz log ? (with the falg "ZCL")

The DDF use a formulea to compute the battery.

```
const vmin = 25;
const vmax = 30;
let bat = Attr.val;
if (bat > vmax) {
    bat = vmax;
} else if (bat < vmin) {
    bat = vmin;
}
bat = ((bat - vmin) / (vmax - vmin)) * 100;
if (bat > 100) {
    bat = 100;
} else if (bat <= 0) {
    bat = 1;
}
Item.val = bat;
```

So 31 give 100%

Smanar, 2022-11-26:
Ok, sorry, bad new, we can give up for the moment.

Not possible to use DDF to bypass the "state/alert" command with the actual code, so not possible to trigger the siren.

ottelo9, 2022-11-26:
> Ok, sorry, bad new, we can give up for the moment.
> 
> Not possible to use DDF to bypass the "state/alert" command with the actual code, so not possible to trigger the siren.

Ok no problem!
The polling has now stopped. I dont know why, maybe the device is sleeping? 

> But as the "read" part in empty, strange deconz trying something here, the poll is not visible in deconz log ? (with the falg > "ZCL")
> This can provoke a battery drain.
You mean I should take a look inside the debug log window in deconz with flag ZCL (disable the INFO flag?)? Actually the smoke sensor is sleeping or deconz not polling it. But I can see these logs (example):

`ZCL got data for node=0x2D8E, cl=0x0006, at=0x0000, status=0x00, type=0x10`

where can I find the node ID for my smoke sensor?

roelvandermade, 2022-11-26:
It's funny to still see differences in results, while we both now have exactly the same config (DDF)

[image]

ottelo9, 2022-11-26:
> **Edit: What I did notice that it took me a complete restart of my homeassistant AND deconz docker instances to get things recognised as they are now. All seems to be working (assuming the battery indeed is at 100%) with the exception of the Sirene test, but that has been explained by @Smanar**

I have restarted my whole system (rpi4), deleted the smoke sensor and rebind it. I have the same results again. No Volate and Temperature in Phoscon.

I use ConBee II with firmware 2.19.01 / 19.9.2022 26720700 and Deconz 2.19.1 (HA Addon).

Smanar, 2022-11-26:
>ZCL got data for node=0x2D8E, cl=0x0006, at=0x0000, status=0x00, type=0x10

It's not the same device ? this one don't have the cluster 0x0006

>where can I find the node ID for my smoke sensor?

In HA IDK, but you can find it in the API, for exemple phoscon/help/API Information/sensors

>Is it normal that the status doesn't update at all, even if I press the button on the smoke sensor to wake up the device?

IDK if it's the "lastseen" field or "lastupdate" from the API, the second one is only updated if the "master" part is updated for exemple
if you have 
```
    "state": {
      "lastupdated": "2022-11-26T13:54:46.786",
      "temperature": 1866
    },
```

"Lastupdated" is only updated if you receive a temperature, not battery.

>Its working if I produce smoke.

Yep, but it's possible to force the siren too, and not possible ATM.

ottelo9, 2022-11-29:
Today I come home from holiday and see that the battery of my smoke sensor drops from 100% to 80%. And I see that the polling from or to the sensor is still there but LED blinking changed little bit.

Battery drop:
[image]

Polling?
[image]

In the debug view in deconz I have LOT OF log messages from lot of devices. So I cannot find the node id from my smoke sensor! 

API deconz:
```
                    {
    "config": {
        "battery": 80,
        "ledindication": false,
        "on": true,
        "pending": [],
        "reachable": true
    },
    "ep": 1,
    "etag": "1563e81ad0f7b254a753979b579263ff",
    "lastannounced": null,
    "lastseen": "2022-11-29T17:24Z",
    "manufacturername": "LUMI",
    "modelid": "lumi.sensor_smoke.acn03",
    "name": "Smoke Sensor",
    "state": {
        "fire": false,
        "lastupdated": "2022-11-26T13:45:04.841",
        "lowbattery": false,
        "test": false
    },
    "swversion": "2019",
    "type": "ZHAFire",
    "uniqueid": "54:ef:44:10:00:55:ca:4b-01-0500"
}
```

[58 more replies]

closed 2023-02-25 by github-actions[bot]