dresden-elektronik/deconz-rest-plugin #5865: Device name

babajun12, 2022-03-14:
## Device

- Product name: Tuya door/window sensor
- Manufacturer: _TZ3000_26fmupbb
- Model identifier: TS0203
- Device type : 
    - Sensor
  

## Screenshots

### Basic

[image]

### Node Info
[image]

### Power Configuration
[image]

### IAS Zone
[image]

[image]

Smanar, 2022-03-14:
Hello, this device relay look standard.
I think it can be added using DDF.
Can start from scratch https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet
Or just editing an existing one, for exemple this one https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/wiser/contact_sensor_cct591011_as.json

babajun12, 2022-03-14:
Nice feature. The more time passed, the more atributes were found. Adding binding for power seems to work. But status in phoscon shows "not reachable" - not greyed out. No status (open/close) displayed. Atribute for open/closed not found.

[image]

[image]

Smanar, 2022-03-15:
You have set the "gold" status for your tests ?

In fact you are not lucky IAS device are harder because they have some more features to know ^^

My personnal advices:
- You can remove State/on, as the device will use instead state/open (just select it and press the delete key)
- This device (like all IAS one) have a tampered and lowbattery return, so you can remove too the state/voltage and add the state/lowbattery
- IAS device, are alarm device, they use a special procedure, the enrollement, so you need to add too on your device config/pending config/enrolled
-  You probably will need to make some change on the state/open to use the IAS cluster, you can show a capture if you have issue but the value you need to set are

```
        {
          "name": "state/open",
          "awake": true,
          "parse": {
            "at": "0x0002",
            "cl": "0x0500",
            "ep": 1,
            "eval": "Item.val = (Attr.val & 0x3) != 0"
          }
```

babajun12, 2022-03-16:
Thanks a lot for you offer. I tried to edit the file but device still not recognized. See config and API info blow:

`{
    "config": {
        "battery": null,
        "on": true,
        "reachable": true
    },
    "ep": 1,
    "etag": "a535fabd578dc9e8b0330a33daf403a2",
    "lastannounced": "2022-03-16T15:42:52Z",
    "lastseen": "2022-03-16T16:10Z",
    "manufacturername": "_TZ3000_26fmupbb",
    "modelid": "TS0203",
    "name": "_TZ3000",
    "state": {
        "battery": null,
        "lastupdated": "none",
        "lowbattery": null,
        "on": null,
        "open": null,
        "reachable": null,
        "tampered": null,
        "voltage": null
    },
    "type": "ZHAOpenClose",
    "uniqueid": "a4:c1:38:cd:ef:7b:9a:8d-01-0006"
}`

babajun12, 2022-03-16:
Thanks a lot for you offer. I tried to edit the file but device still not recognized. See config and API info blow. Status = gold and hot reload done.

```
{
    "config": {
        "battery": null,
        "on": true,
        "reachable": true
    },
    "ep": 1,
    "etag": "a535fabd578dc9e8b0330a33daf403a2",
    "lastannounced": "2022-03-16T15:42:52Z",
    "lastseen": "2022-03-16T16:10Z",
    "manufacturername": "_TZ3000_26fmupbb",
    "modelid": "TS0203",
    "name": "_TZ3000",
    "state": {
        "battery": null,
        "lastupdated": "none",
        "lowbattery": null,
        "on": null,
        "open": null,
        "reachable": null,
        "tampered": null,
        "voltage": null
    },
    "type": "ZHAOpenClose",
    "uniqueid": "a4:c1:38:cd:ef:7b:9a:8d-01-0006"
}
```
..and
```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_26fmupbb",
  "modelid": "TS0203",
  "vendor": "_TZ3000_26fmupbb",
  "product": "TS0203",
  "sleeper": true,
  "status": "Gold",
  "path": "/devices/tuya.sensor_magnet.json",
  "subdevices": [
    {
      "type": "$TYPE_OPEN_CLOSE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0006"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0xFFFF",
        "endpoint": "0x01",
        "in": [
          "0x0006"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "read": {
            "fn": "zcl"
          }
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/lowbattery",
          "description": "True when the device battery runs low."
        },
        {
          "name": "state/open"
          "awake": true,
          "parse": {
            "at": "0x0002",
            "cl": "0x0500",
            "ep": 1,
            "eval": "Item.val = (Attr.val & 0x3) != 0"
        },
        {
          "name": "state/reachable"
        },
        {
          "name": "state/tampered"
        },
        {
          "name": "state/voltage",
          "refresh.interval": 300
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001"
    }
  ]
}
```

[image]

babajun12, 2022-03-16:
Hold on, working on
` DDF: failed to read /root/.local/share/dresden-elektronik/deCONZ/devices/tuya.sensor_magnet.json, err: unterminated object, offset: 1817
`

Smanar, 2022-03-16:
Ha ? strange, the DDF is generated by the editor ? This error message mean a "synthax error" on the json.

To unlock right on a file
`sudo chmod -R 777 Folder_File_Name`

You can too remove the state/on.
Then select the state/open and change values (if needed) on "parse" and "read"

>             "atribute": "0x0002",
>             "cluster": "0x0500",
>             "ep": 1,
>             "expression": "Item.val = (Attr.val & 0x3) != 0"

>/devices/tuya.sensor_magnet.json

Try avoid "." in name to keep it for the extension "tuya.sensor"

But there is some thing I don't understand, you are using the editor or editing direclty the file ? There is lot of difference with the DDF sample, for exemple

```
      "fingerprint": {
        "profile": "0x0104",
        "device": "0xFFFF",
        "endpoint": "0x01",
        "in": [
          "0x0006"
        ]
      },
```

You don't have the cluster 0x0006 on your device, I relay don't understand where it's from, and there is too an issue here  "device": "0xFFFF"

You can remove all the fingerprint part if you have issue with this one, the DDF can work without this part (Not sure at 100 % ^^)

babajun12, 2022-03-16:
I have no idea how to proberly edit the file. Deconz is running as docker container. No "vi" in that image. So I copy "docker cp...." the file and edit it with vi. Then copy back to cantainer. As a result, the file cannot be opened. Tried it with different names. File owner and rights are the same as the working file.

babajun12, 2022-03-16:
...even with this file properties. "tuya.sensor_magnet_org.json" is loaded, "tuya.sensor.json" cannot be opened.
```
root@raspi4:~/.local/share/dresden-elektronik/deCONZ/devices# ls -lisa
total 16
1049546 4 drwxr-xr-x 2 root root 4096 Mar 16 20:53 .
1048585 4 drwxr-xr-x 3 root root 4096 Mar 16 21:05 ..
1050644 4 -rwxrwxrwx 1 root root 2560 Mar 16 20:52 tuya.sensor.json
1056356 4 -rw-r--r-- 1 root root 2526 Mar 16 20:40 tuya.sensor_magnet_org.json
```

Smanar, 2022-03-16:
Something to try.
Right clic on the node, them select "edit DDF", it will open the editor.
Them just save the file, it will choose a working path and display it on the windows title. It will perhaps use a folder with more right for you.

But dockers are definitively hells if you want to custom something. Can make only with the editor from start to the end.
You can copy your DDF from the last tab. I will explain you the values to use.

- delete all previous DDF for this device
- right clic on the node, "edit DDF"
- if not automatically filled, fill manufacture name and model id (using drag and drop)
- set the status to "gold"
- if not made automatically, drag and drop a ZHAopenclose
- add state/lowbattery, state/tampered, config/pending, config/enrolled
- delete the state/on
- select the state/open and set 
```
        "attribute": "0x0002",
        "cluster": "0x0500",
        "ep": 1,
        "expression": "Item.val = (Attr.val & 0x3) != 0"
```
- Save file
- Make hot reload
- Check the result in the API

I think it will be enough, no bind, we can probably add the battery level, it seem working on your capture (cluster 0x0001 attribute 0x0021 (200 for 100 % ? )

babajun12, 2022-03-16:
Thanks for your patience! Followed this procedure but still no luck. This is what it created. open/close on cl 0x0006 which is empty and wrong statements for "fingerprint"...?
I think I have to look for a proper way to edit the file somehow.

```
 {
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_26fmupbb",
  "modelid": "TS0203",
  "product": "TS0203",
  "sleeper": false,
  "status": "Gold",
  "path": "/devices/ts0203.json",
  "subdevices": [
    {
      "type": "$TYPE_OPEN_CLOSE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0006"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0xFFFF",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0006"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "fn": "zcl"
          }
        },
        {
          "name": "config/enrolled",
          "public": false,
          "description": "State of IAS enrollment process."
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/pending",
          "description": "Pending tasks to configure the device."
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/battery",
          "refresh.interval": 3600
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/open",
          "read": {
            "at": "0x0002",
            "cl": "0x0500",
            "ep": 255,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0002",
            "cl": "0x0500",
            "ep": 255,
            "eval": "Item.val = (Attr.val & 0x3) != 0",
            "fn": "zcl"
          }
        },
        {
          "name": "state/tampered"
        }
      ]
    }
  ]
}
```
Zone state not enrolled.

babajun12, 2022-03-16:
Finally, at least installing an editor in the container was easy and file can be edited now.
`apt update`
`apd install `vim`
Edit - save - hot reload
Will proceed tomorrow.

babajun12, 2022-03-17:
### Solved

I changed following line to:
```
  "subdevices": [
    {
      "type": "$TYPE_OPEN_CLOSE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"  <-------  was "0x0006" (on/off)
      ],
```
Phoscon state:

[image]

Cluster 500 - enrolled!

[image]

API response

```
{
    "config": {
        "battery": 100,
        "on": true,
        "pending": [],
        "reachable": true
    },
    "ep": 1,
    "etag": "ff6879f1e2efeae34a65b2e1d1f4e80a",
    "lastannounced": "2022-03-17T08:16:38Z",
    "lastseen": "2022-03-17T08:16Z",
    "manufacturername": "_TZ3000_26fmupbb",
    "modelid": "TS0203",
    "name": "OpenClose 74",
    "state": {
        "battery": 100,
        "lastupdated": "2022-03-17T08:16:39.198",
        "open": true,
        "tampered": false
    },
    "type": "ZHAOpenClose",
    "uniqueid": "a4:c1:38:cd:ef:7b:9a:8d-01-0500"
}
```

closed 2022-03-17 by babajun12

babajun12, 2022-03-18:
You are right, these settings can be done via GUI. I didn´t manage the battery percentage to work.

[image]

[image]

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_26fmupbb",
  "modelid": "TS0203",
  "product": "TS0203",
  "sleeper": true,
  "status": "Gold",
  "path": "/devices/ts0203.json",
  "subdevices": [
    {
      "type": "$TYPE_OPEN_CLOSE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0402",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0500"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "parse": {
            "fn": "None"
          }
        },
        {
          "name": "config/enrolled",
          "public": false,
          "description": "State of IAS enrollment process."
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/pending",
          "description": "Pending tasks to configure the device."
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/lowbattery",
          "description": "True when the device battery runs low."
        },
        {
          "name": "state/open",
          "awake": true,
          "read": {
            "fn": "None"
          },
          "parse": {
            "at": "0x0002",
            "cl": "0x0500",
            "ep": 1,
            "eval": "Item.val = (Attr.val & 0x3) != 0",
            "fn": "zcl"
          }
        },
        {
          "name": "state/tampered"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 300,
          "max": 3600,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0500",
      "report": [
        {
          "at": "0x0002",
          "dt": "0x19",
          "min": 300,
          "max": 3600
        }
      ]
    }
  ]
}
```

Smanar, 2022-03-18:
The battery voltage seem not working = 0 0x0020
The battery level seem * 2 = 200 0x0021, so will use this one.

Binding seem fine, but
```
        {
          "name": "config/battery",
          "parse": {
            "fn": "None"
          }
        },
```

select the config/battery and fill value to have something like
```
        {
          "name": "config/battery",
          "awake": true,
          "refresh.interval": 3700,
          "read": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "fn": "zcl",
            "eval": "Item.val = Attr.val / 2"
          }
        },
```

the "refresh.interval" is a sample, try first with empty value, to let the device make natural report and don't create a poll.

babajun12, 2022-03-18:
Fine, made the changes via GUI. Will observe it for a battery update. I see other sensors updating the battery percentage one a day. I think this can take some time. Tried to read the values via deCONZ GUI, but no update to API yet.

```
{
          "name": "config/battery",
          "awake": true,
          "read": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2",
            "fn": "zcl"
          }
        },
```

Smanar, 2022-03-19:
>Tried to read the values via deCONZ GUI, but no update to API yet.

Can be because the device was sleeping, else we have a problem.

Smanar, 2022-03-21:
>Just "status ok"

You mean if you double clic on the attribute 0x0021 on cluster 0x0001, and try to read it you have a "succed" message ?

Perhaps the screenshoot was taken without having try to read them ? Strange no attributes are disabled.

babajun12, 2022-03-21:
readin failed 

[image]

babajun12, 2022-03-21:
But also if I select another sensor which shows battery level in deCONZ. It reports „reading failed“ as well.
Maye I‘m misunderstanding something…

Smanar, 2022-03-22:
Some IAS device use lowbattery and not battery level, but I don't find information on this device on other zigbee project to know, except on homey where they can have too the battery voltage, but can't be sure.

But thoses device are lazy and can sleep when you make the request, so if you are using the GUI , you need to awake the device in same time you ask the attribute (else the "led" in the node will flash in red = timeout).
Can be done just with using the device or pressing shortly the rest button.

babajun12, 2022-03-22:
Thanks, had the same idea to trigger wakeup - same result.
But battery level is not that important to me. If it reports lowbatt, I´m fine with that.

Smanar, 2022-03-22:
As you want, it's your DDF ^^
If all is working, can you make a PR to add it (in /devices/tuya will be fine) ?

Else I can do it, just copy/paste the last version you are using.

babajun12, 2022-03-22:
to compare; my centralite door sensor, ZHA device as well, reports battery level. DDF shows:

```
{
          "name": "config/battery",
          "description": "The current device battery level in 0&ndash;100&thinsp;%."
        },
```

Smanar, 2022-03-22:
Yep, the DDF is empty at this part, can use just

```
        {
          "name": "config/battery",
        },
```

Deconz will use the defaut code, but I don't think it will be better.

babajun12, 2022-03-23:
My full DDF:

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_26fmupbb",
  "modelid": "TS0203",
  "product": "TS0203",
  "sleeper": true,
  "status": "Gold",
  "path": "/devices/ts0203.json",
  "subdevices": [
    {
      "type": "$TYPE_OPEN_CLOSE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0402",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0500"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true,
          "read": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2",
            "fn": "zcl"
          }
        },
        {
          "name": "config/enrolled",
          "public": false,
          "description": "State of IAS enrollment process."
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/pending",
          "description": "Pending tasks to configure the device."
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/lowbattery",
          "description": "True when the device battery runs low."
        },
        {
          "name": "state/open",
          "awake": true,
          "read": {
            "fn": "None"
          },
          "parse": {
            "at": "0x0002",
            "cl": "0x0500",
            "ep": 1,
            "eval": "Item.val = (Attr.val & 0x3) != 0",
            "fn": "zcl"
          }
        },
        {
          "name": "state/tampered"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0500",
      "report": [
        {
          "at": "0x0002",
          "dt": "0x19",
          "min": 300,
          "max": 3600
        }
      ]
    }
  ]
}
```

Smanar, 2022-03-23:
Hu ?
And it works ? without binding ? and just with empty value ?

Magic ....

But if it work for you, don't hesitate to make a PR to submit it, or ask for someone add it.

babajun12, 2022-03-24:
Shows some values but might not be true:
- not sure why it reports 200%
- seems to report allways same values even with a weak battery
- no battery icon in deCONZ
- API shows "config: battery": 100
   ..but "state: battery": null

[image]

[image]

PS: I´m not aware how to submit a pull request
PPS: Added bindings as well

Smanar, 2022-03-25:
>not sure why it reports 200%

Its the value displayed in deconz, visible on your capture, 
To have correct value
- select config/battery
- set as values for both "parse" and "read" after selected ZCL
cluster 0x0001
attribute 0x0021
endpoint 01
Expression : Item.val = Attr.val / 2

>seems to report allways same values even with a weak battery

You have set the binding for cluster 0x0001 atribute 0x0021 ?

>no battery icon in deCONZ

It mean deconz haven't battery return for the device yet, probably the missing bind

>API shows "config: battery": 100

But it show 100 or 200 ?

>but "state: battery": null

I don't see state/battery on your DDF, It was not removed when making "hot relead" ? This field need to be removed, useless.

babajun12, 2022-03-25:
Anyhow, I am stisfied with the open/close message and the battery status "OK"

Smanar, 2022-03-26:
>state/battery was/is configured:

You need to use config/battery, state/battery is for sensor used only for battery (for exemple for battery covering)

On the reporting you have set 0 for max, try with 2700.

But strange you still have it with the field removed, can be an isue on the code, so not the DDF fault

Do you want I make the PR (need your last version) or do you want I explain how to make a PR ? It will be simple, just try to create a new file on the official github, and it will create a fork, add the file, and help you tu submit the PR
Use the devices/tuya folder

Smanar, 2022-03-28:
>Have to monitor the battery level. Seems to report 100% only - but: deCONZ reads all values.

You mean the voltage ? with the attribute 0x0020 ? It will be same, deconz will use the voltage in a formula and display a percentage too. So not a big difference.

>So I just have to add the xxx.json file into the devices/tuya folder - right?

Yep, you already have inside the _TZE200_e3oitdyu_smart_dimmer_module.json file, so try to use the same typo, for exemple _TZ3000_26fmupbb_door_sensor.json

You will have a message saying you haven't right on this github, and it make a "fork" automatically under you name. The fork is only for you, you can make all you want with it (here you will add a file)
Then it will propose you to make a PR, from your fork to the official github.

Choose a title > Add DDF for ....
Add a link to this issue
A comment, ect, all you want, no danger, you can't break something, and you can delete , edit and restart.

And don't worry if someone ask you to change the status for exemple (need to set "Bronze" for PR) you will just need to edit the file on your fork, and all will be synchronised.
You will see github is realy a nice "machine".

acid115, 2022-06-04:
> Suprise - seems to work now!
> 
> API result:
> 
> ```
> {
>     "config": {
>         "battery": 100,
>         "on": true,
>         "pending": [],
>         "reachable": true
>     },
>     "ep": 1,
>     "etag": "46cd9595a57bd7f1b103c6dc798391a5",
>     "lastannounced": "2022-03-22T17:34:26Z",
>     "lastseen": "2022-03-23T07:32Z",
>     "manufacturername": "_TZ3000_26fmupbb",
>     "modelid": "TS0203",
>     "name": "OpenClose 76",
>     "state": {
>         "battery": null,
>         "lastupdated": "2022-03-23T07:33:12.556",
>         "lowbattery": false,
>         "open": false,
>         "tampered": false
>     },
>     "type": "ZHAOpenClose",
>     "uniqueid": "a4:c1:38:cd:ef:7b:9a:8d-01-0500"
> }
> ```
> 
> Config:
> 
> [image]
> 
> DDF file:
> 
> ```
> {
>   "schema": "devcap1.schema.json",
>   "manufacturername": "_TZ3000_26fmupbb",
>   "modelid": "TS0203",
>   "product": "TS0203",
>   "sleeper": true,
>   "status": "Gold",
>   "path": "/devices/ts0203_test.json",
>   "subdevices": [
>     {
>       "type": "$TYPE_OPEN_CLOSE_SENSOR",
>       "restapi": "/sensors",
>       "uuid": [
>         "$address.ext",
>         "0x01",
>         "0x0500"
>       ],
>       "fingerprint": {
>         "profile": "0x0104",
>         "device": "0x0402",
>         "endpoint": "0x01",
>         "in": [
>           "0x0000",
>           "0x0001",
>           "0x0500"
>         ]
>       },
>       "items": [
>         {
>           "name": "attr/id"
>         },
>         {
>           "name": "attr/lastannounced"
>         },
>         {
>           "name": "attr/lastseen"
>         },
>         {
>           "name": "attr/manufacturername"
>         },
>         {
>           "name": "attr/modelid"
>         },
>         {
>           "name": "attr/name"
>         },
>         {
>           "name": "attr/swversion"
>         },
>         {
>           "name": "attr/type"
>         },
>         {
>           "name": "attr/uniqueid"
>         },
>         {
>           "name": "config/battery",
>           "awake": true,
>           "read": {
>             "fn": "zcl"
>           },
>           "parse": {
>             "fn": "zcl"
>           }
>         },
>         {
>           "name": "config/enrolled",
>           "public": false,
>           "description": "State of IAS enrollment process."
>         },
>         {
>           "name": "config/on"
>         },
>         {
>           "name": "config/pending",
>           "description": "Pending tasks to configure the device."
>         },
>         {
>           "name": "config/reachable"
>         },
>         {
>           "name": "state/lastupdated"
>         },
>         {
>           "name": "state/lowbattery",
>           "description": "True when the device battery runs low."
>         },
>         {
>           "name": "state/open",
>           "awake": true,
>           "read": {
>             "fn": "None"
>           },
>           "parse": {
>             "at": "0x0002",
>             "cl": "0x0500",
>             "ep": 1,
>             "eval": "Item.val = (Attr.val & 0x3) != 0",
>             "fn": "zcl"
>           }
>         },
>         {
>           "name": "state/tampered"
>         }
>       ]
>     }
>   ],
>   "bindings": [
>     {
>       "bind": "unicast",
>       "src.ep": 1,
>       "cl": "0x0500",
>       "report": [
>         {
>           "at": "0x0002",
>           "dt": "0x19",
>           "min": 300,
>           "max": 3600
>         }
>       ]
>     }
>   ]
> }
> ```

Can SOMEONE please commit this DDF File ? Because it WORKS !!! (except the battery level...)
I have bought these sensors too, 5 Euros per Sensor. 
And i needed TWO days, to get them to work. I have read dozen of articles and posts,   created my own ddf with the editor... nothing works. 
But this DDF works well. Please, commit this ddf to help other users with the same problem.
Seems that babajun12 has solved the problem and was never been seen anymore, no one cares.

Smanar, 2022-06-05:
To test the battery, try this DDF (made it with yours)

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_26fmupbb",
  "modelid": "TS0203",
  "product": "TS0203",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_OPEN_CLOSE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0402",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0500"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery"
        },
        {
          "name": "config/enrolled",
          "public": false
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/pending"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/lowbattery"
        },
        {
          "name": "state/open",
          "awake": true,
          "read": {
            "fn": "None"
          },
          "parse": {
            "at": "0x0002",
            "cl": "0x0500",
            "ep": 1,
            "eval": "Item.val = (Attr.val & 0x3) != 0",
            "fn": "zcl"
          }
        },
        {
          "name": "state/tampered"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 60,
          "max": 3600,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0500",
      "report": [
        {
          "at": "0x0002",
          "dt": "0x19",
          "min": 300,
          "max": 3600
        }
      ]
    }
  ]
}
```

acid115, 2022-06-05:
Does not work...

Smanar, 2022-06-06:
Wait a little can take some time to appear.
Else I think you will have only lowbattery instead of the battery level.
We can"t use the attribube 0x0020 as it's 0 on your capture, the attribute 0x0021 seem working, can make a last try with forcing to read the values
```

{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_26fmupbb",
  "modelid": "TS0203",
  "product": "TS0203",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_OPEN_CLOSE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0402",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0500"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "refresh.interval": 18000,
          "awake": true,
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2"
          }
        },
        {
          "name": "config/enrolled",
          "public": false
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/pending"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/lowbattery"
        },
        {
          "name": "state/open",
          "awake": true,
          "read": {
            "fn": "None"
          },
          "parse": {
            "at": "0x0002",
            "cl": "0x0500",
            "ep": 1,
            "eval": "Item.val = (Attr.val & 0x3) != 0",
            "fn": "zcl"
          }
        },
        {
          "name": "state/tampered"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 60,
          "max": 3600,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0500",
      "report": [
        {
          "at": "0x0002",
          "dt": "0x19",
          "min": 300,
          "max": 3600
        }
      ]
    }
  ]
}
```

acid115, 2022-06-06:
Yes, it works :)
It shows 100 % 
Thank you
[image]

Smanar, 2022-06-06:
@acid115 I was doing the PR and I have see a DDF for a similar device recently added
Probably a clone, if you can make a try with this DDF (I m trying to merge the 2 devices)

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": ["_TZ3000_n2egfsli", "_TZ3000_26fmupbb"],
  "modelid": ["TS0203", "TS0203"],
  "vendor": "Tuya",
  "product": "Smart Zigbee Door Window Contact",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_OPEN_CLOSE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0402",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0500"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/checkin"
        },
        {
          "name": "config/enrolled",
          "public": false
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/pending"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2"
          }
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/lowbattery"
        },
        {
          "name": "state/open"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 60,
          "max": 3600,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0500"
    }
  ]
}
```