dresden-elektronik/deconz-rest-plugin #5870: Device name

ZeppDK, 2022-03-15:
## Device

    Product name: EasyAccess EasyFingerTouch
    Manufacturer: Onesti Products AS
    Model identifier: EasyFingerTouch
    Device type :
        Lock

## Screenshots
[image]

### Basic
[image]

### Identify
[image]

### Groups
[image]
[image]

### Door Lock
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

### OTAU
[image]
[image]

ZeppDK, 2022-03-15:
there seems to be an older version of the lock already added, so i hope this one will be straight forward :-)
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4253

Smanar, 2022-03-15:
Hello, but the problem is now we need to use DDF to add device.
Someone have started for a doorlock too, but have give up if you want to test on your side ? https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5775

Smanar, 2022-03-17:
Hello, if you want to start from scratch, I have made a PR for some missing file https://github.com/Smanar/deconz-rest-plugin/commit/770b4e0ea9b65d931591c544aa9778fd11a02e8b
And an error here https://github.com/Smanar/deconz-rest-plugin/commit/db05ea518b2caf4bb1bd783213577e22bd50a517

You need to restart deconz after adding the files

After that

-    delete all previous DDF for this device
-    right clic on the node, "edit DDF"
-    if not automatically filled, fill manufacture name and model id (using drag and drop)
-    set the status to "gold"
-    if not made automatically, drag and drop a Door Lock device
-    Save file
-    Make hot reload,
-    Check the result in the API using third app or phoscon (help / API information)

Smanar, 2022-03-20:
@ZeppDK do you need some help ?
You don't need to compile code, just add/edit file, will be easier on docker install ?

ZeppDK, 2022-03-20:
> @ZeppDK do you need some help ?
> You don't need to compile code, just add/edit file, will be easier on docker install ?

I have added the files, restartet, and gotten a  new DDF file with the new lock type..
But still needs to find out how to get it to use the actual info from the lock shown in the pictures under “door lock”

Smanar, 2022-03-20:
I m not sure to understand ?
Wich one picture ?
The device is reconised ?

ZeppDK, 2022-03-21:
after i added your new files, i made a DDF, and added this from the list manually... 
[image]

ZeppDK, 2022-03-21:
here is the API
[image]

ZeppDK, 2022-03-21:
so like this?

> {
  "schema": "devcap1.schema.json",
  "manufacturername": "Onesti Products AS",
  "modelid": "EasyFingerTouch",
  "vendor": "EasyAccess",
  "product": "EasyFingerTouch",
  "sleeper": false,
  "status": "Gold",
  "path": "/devices/easyfingertouch.json",
  "subdevices": [
    {
      "type": "$TYPE_DOOR_LOCK",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/lock",
          "read": {
            "at": "0x0000",
            "cl": "0x0101",
            "ep": 11,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0000",
            "cl": "0x0101",
            "ep": 11,
            "eval": "Item.val = Attr.val == 1",
            "fn": "zcl"
          }
        },
        {
          "name": "state/lockstate"
        },
        {
          "name": "state/open"
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 11,
      "cl": "0x0101",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x30",
          "min": 1,
          "max": 300
        }
      ]
    }
  ]
}

ZeppDK, 2022-03-22:
[image]
[image]

that works for lock/unlock of the outside of the lock :)

Smanar, 2022-03-22:
Nice, next step the battery
- add a "config/battery" to the DDF
- configure the battery bind, Cluster 0x0001 on right panel, and attribute 0x20 on left panel
- Values for reporting min 300, max 2700, change 1

Make state/lockstate working
- select state/lockstate
- fill the "read" and "parse" part using as value (ZCL)

Cluster 0x0101
Attribute 0x0000
endpoint 0x0B or 11 (I don't remember if it's hexa or not
Expression : if (Attr.val == 1) { Item.val = 'open' } else if (Attr.val == 2) { Item.val = 'closed' } else { Item.val = 'error' }

Smanar, 2022-03-25:
@ZeppDK need more guidance ?

ZeppDK, 2022-03-25:
Maybe.. but have not had time to test yet

Smanar, 2022-03-26:
NP, just ask, I will make a PR to be sure don't forget missing file on future deconz version

ZeppDK, 2022-03-29:
this is after adding the config you provided:
[image]

ZeppDK, 2022-03-29:
the lock state looks weird, as it says open, even though the door is locked...
so maybee i should switch the open/closed like this?
if (Attr.val == 1) { Item.val = 'closed' } else if (Attr.val == 2) { Item.val = 'open' } else { Item.val = 'error' }

i'll double check when i get home.

ZeppDK, 2022-03-29:
[image]

ZeppDK, 2022-03-29:
this is how it looks at the moment

`{
  "schema": "devcap1.schema.json",
  "manufacturername": "Onesti Products AS",
  "modelid": "EasyFingerTouch",
  "vendor": "EasyAccess",
  "product": "EasyFingerTouch",
  "sleeper": false,
  "status": "Gold",
  "path": "/devices/easyfingertouch.json",
  "subdevices": [
    {
      "type": "$TYPE_DOOR_LOCK",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery"
        },
        {
          "name": "config/lock",
          "read": {
            "at": "0x0000",
            "cl": "0x0101",
            "ep": 11,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0000",
            "cl": "0x0101",
            "ep": 11,
            "eval": "Item.val = Attr.val == 1",
            "fn": "zcl"
          }
        },
        {
          "name": "state/lockstate",
          "read": {
            "at": "0x0000",
            "cl": "0x0101",
            "ep": 11,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0000",
            "cl": "0x0101",
            "ep": 11,
            "eval": "if (Attr.val == 1) { Item.val = 'open' } else if (Attr.val == 2) { Item.val = 'closed' } else { Item.val = 'error' }",
            "fn": "zcl"
          }
        },
        {
          "name": "state/open",
          "read": {
            "at": "0x0003",
            "cl": "0x0101",
            "ep": 11,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0003",
            "cl": "0x0101",
            "ep": 11,
            "eval": "if (Attr.val == 0) { Item.val = 'open' } else if (Attr.val == 1) { Item.val = 'closed' } else { Item.val = 'error' }",
            "fn": "zcl"
          }
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 11,
      "cl": "0x0101",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x30",
          "min": 1,
          "max": 300
        },
        {
          "at": "0x0003",
          "dt": "0x30",
          "min": 1,
          "max": 300
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 11,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0020",
          "dt": "0x20",
          "min": 300,
          "max": 2700,
          "change": "0x00000001"
        }
      ]
    }
  ]
}`

Smanar, 2022-03-29:
You can remove state/reachable
For state/open as it's a bool I think it s better to use "Item.val = Attr.val != 1" whith that you will have open= false only if the door is closed, if you have error you will have open = true.
For state/locked I would try "if (Attr.val == 1) { Item.val = 'locked' } else if (Attr.val == 2) { Item.val = 'unlocked' } else { Item.val = 'error' }",

I don't understanbd why battery is not working.
Can you try to read the attribute 0x0020 in the cluster 0x0001 in deconz ?

ZeppDK, 2022-04-04:
[image]

ZeppDK, 2022-04-04:
battery not showing much in deconz either :-)

[image]

ZeppDK, 2022-04-07:
lockstate is showing the correct state now.
thats the most important feature :D

Smanar, 2022-04-07:
Ha ?
State/lockstate  is something new but from my memory it can expose config/lock (so only a bool value) but it s same than State/lockstate

And for state/open ? Need to expose a bool value too (true/false) for the door state.

```

{
  "name": "state/open",
  "read": {
    "at": "0x0003",
    "cl": "0x0101",
    "ep": 11,
    "fn": "zcl"
  },
  "parse": {
    "at": "0x0003",
    "cl": "0x0101",
    "ep": 11,
    "eval": "Item.val = Attr.val != 1",
    "fn": "zcl"
  }
}
```

```
{
  "bind": "unicast",
  "src.ep": 11,
  "cl": "0x0101",
  "report": [
    {
      "at": "0x0003",
      "dt": "0x30",
      "min": 1,
      "max": 300
    }
  ]
}
```

Smanar, 2022-04-29:
@ZeppDK Do you want I remake the full DDF ?

ZeppDK, 2022-04-30:
Not yet, havent had time 😊

oskretc, 2022-09-12:
Anyway we can reopen this and move forward to support the device?

Mimiix, 2022-09-12:
@oskretc  Are you able to test the DDF :)?

oskretc, 2022-09-12:
@Mimiix I can give it a try, 
I'm running docker deconzcommunity/deconz 
I'm not sure where the DDF should go?
Any docs on how to test/debug the DDF?

Smanar, 2022-09-12:
Yes It's easier on docker than on the "HA docker", but you can find usefull information here https://forum.phoscon.de/t/how-to-add-edit-a-ddf-on-home-assistant-using-text-editor/1839

If I m right on docker you have a mounter folder for the /usr/share/deCONZ/ folder, so you will find the "devices" folder in it

The theory : you just need to create a what_you_want.json file with this contain in the deconz "devices" folder, then just re-include the device, if you have the GUI can use the "Hot reload" command in the editor.

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet

oskretc, 2022-09-12:
Yeap, I find all the info, tested the latest DDF and it seems to be working with a few details, 
1. I can see the sensor in the API, but it doesnt show up in the Phoscon app or a third party (HomeAssistant), is something missing from the image below that will prevent it from showing in the UI
2. battery Voltage is readout is 45 (cl 0x0001 at 0x0020) which seems to be the voltage x 10 as in 3 1.5 volts battery
3. battery percentage (cl 0x0001 at 0x0021) readout is 200 maybe we can use that by just dividing by 2, not sure what are the Min interval and Max interval doing.
4.

[image]

oskretc, 2022-09-13:
The image above is using attribute 0x0021 so I have sorted it out. only last thing to verify is when battery percentage decreases, works as expected. I'll keep an eye on it

Yes, I'm sure HA cant see it, I have reloaded both the phosco docker (in a separate rpi) and HA just to  be sure and nothing. 

Any ideas?

Smanar, 2022-09-13:
Heee, I don't use HA, so I can't help you so much, but I m sure it's suported.

I use my joker, @Kane610 have you an idea ^^ ?

Smanar, 2022-09-13:
Har, I think I have see the problem
The type is "Door lock", it s the type for "light", for sensor the type need to be "ZHADoorLock"

oskretc, 2022-09-13:
I guess you meant changing the Doo Lock (1) on the left panel for a ZHADoorLock, but that item is not available in the "Available items" panel. 
If I try to manually change th Type: Door Lock in the middle panel, it turns blue and nothing happens
Also, the file "devices/generic/constants.json" doesnt have an entry for ZHADoorLock not sure if it matters though

Smanar, 2022-09-13:
Yeah, if I m right I will make a PR to correct that, for the moment can you edit direclty the DDF file ?

oskretc, 2022-09-13:
I did, and yes, the sensor is shown in HA, but the lock is unavailable
[image]

Smanar, 2022-09-13:
Does it mean config/recheable  = false ?
Can you check the device json now ?

oskretc, 2022-09-13:
Added reachable and nothing changed

looking at the HA logs it seems that the lock code in deconz plugin for HA is not aligned with this DDF

> `Logger: homeassistant.components.lock
Source: components/deconz/lock.py:61
Integration: Lock (documentation, issues)
First occurred: 22:44:24 (2 occurrences)
Last logged: 22:44:24

Error adding entities for domain lock with platform deconz
Error while setting up deconz platform for lock
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 428, in async_add_entities
    await asyncio.gather(*tasks)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 673, in _async_add_entity
    await entity.add_to_platform_finish()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 777, in add_to_platform_finish
    self.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 533, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 571, in _async_write_ha_state
    state = self._stringify_state(available)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 539, in _stringify_state
    if (state := self.state) is None:
  File "/usr/src/homeassistant/homeassistant/components/lock/__init__.py", line 188, in state
    if (locked := self.is_locked) is None:
  File "/usr/src/homeassistant/homeassistant/components/deconz/lock.py", line 61, in is_locked
    return self._device.is_locked
  File "/usr/local/lib/python3.10/site-packages/pydeconz/models/sensor/door_lock.py", line 45, in is_locked
    return self.lock_state == DoorLockLockState.LOCKED
  File "/usr/local/lib/python3.10/site-packages/pydeconz/models/sensor/door_lock.py", line 50, in lock_state
    return DoorLockLockState(self.raw["state"]["lockstate"])
  File "/usr/local/lib/python3.10/enum.py", line 385, in __call__
    return cls.__new__(cls, value)
  File "/usr/local/lib/python3.10/enum.py", line 710, in __new__
    raise ve_exc
ValueError: 'open' is not a valid DoorLockLockState
`

Kane610, 2022-09-14:
The door lock state are expected to be one of four values and open is not one of them https://github.com/Kane610/deconz/blob/0b5b0b5cfa5d31ac12f121aee2f99c234ad9c426/pydeconz/models/sensor/door_lock.py#L9

Smanar, 2022-09-14:
Arf right.

It's from the DDF

```
          "parse": {
            "at": "0x0000",
            "cl": "0x0101",
            "ep": 1,
            "eval": "if (Attr.val == 1) { Item.val = 'open' } else if (Attr.val == 2) { Item.val = 'closed' } else { Item.val = 'error' }",
            "fn": "zcl"
          }
```

But it can be native, it s in the code, IDK why in the old version it s used direclty in the DDF ....
I have updated the DDF

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "Onesti Products AS",
  "modelid": "EasyFingerTouch",
  "vendor": "EasyAccess",
  "product": "EasyFingerTouch",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_DOOR_LOCK",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0101"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery"
        },
        {
          "name": "config/lock",
          "read": {
            "at": "0x0000",
            "cl": "0x0101",
            "ep": 11,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0000",
            "cl": "0x0101",
            "ep": 11,
            "eval": "Item.val = Attr.val == 1",
            "fn": "zcl"
          }
        },
        {
          "name": "state/lockstate",
          "read": {
            "at": "0x0000",
            "cl": "0x0101",
            "ep": 11,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0000",
            "cl": "0x0101",
            "ep": 11,
            "eval": "if (Attr.val == 0) { Item.val = 'not fully locked' } else if (Attr.val == 1) { Item.val = 'locked' } else if (Attr.val == 2) { Item.val = 'unlocked' } else { Item.val = 'undefined' }",
            "fn": "zcl"
          }
        },
        {
          "name": "state/open",
          "read": {
            "at": "0x0003",
            "cl": "0x0101",
            "ep": 11,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0003",
            "cl": "0x0101",
            "ep": 11,
            "eval": "Item.val = Attr.val != 1",
            "fn": "zcl"
          }
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 11,
      "cl": "0x0101",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x30",
          "min": 1,
          "max": 300
        },
        {
          "at": "0x0003",
          "dt": "0x30",
          "min": 1,
          "max": 300
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 11,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0020",
          "dt": "0x20",
          "min": 300,
          "max": 2700,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 11,
      "cl": "0x0101",
      "report": [
        {
          "at": "0x0003",
          "dt": "0x30",
          "min": 1,
          "max": 300
        }
      ]
    }
  ]
}
```

Take care this DDF will generate a new unique ID, I forget to add the cluster on the previous one.

oskretc, 2022-09-14:
That definitly did the trick, No I can lock and unlock from HA, and see the changes on HA when the door is unlocked.

this is the DDF I end up using, Notice the battery part as well as the type.

`{
  "schema": "devcap1.schema.json",
  "manufacturername": "Onesti Products AS",
  "modelid": "EasyFingerTouch",
  "vendor": "EasyAccess",
  "product": "EasyFingerTouch",
  "sleeper": false,
  "status": "Gold",
  "path": "/devices/easyfingertouch.json",
  "subdevices": [
    {
      "type": "ZHADoorLock",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x0b"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "read": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 11,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 11,
            "eval": "Item.val = Attr.val/2",
            "fn": "zcl"
          },
          "default": 0
        },
        {
          "name": "config/lock",
          "read": {
            "at": "0x0000",
            "cl": "0x0101",
            "ep": 11,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0000",
            "cl": "0x0101",
            "ep": 11,
            "eval": "Item.val = Attr.val ==1",
            "fn": "zcl"
          }
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lockstate",
          "read": {
            "at": "0x0000",
            "cl": "0x0101",
            "ep": 11,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0000",
            "cl": "0x0101",
            "ep": 11,
            "eval": "if (Attr.val == 0) { Item.val = 'not fully locked' } else if (Attr.val == 1) { Item.val = 'locked' } else if (Attr.val == 2) { Item.val = 'unlocked' } else { Item.val = 'undefined' }",
            "fn": "zcl"
          }
        },
        {
          "name": "state/open",
          "read": {
            "at": "0x0003",
            "cl": "0x0101",
            "ep": 11,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0003",
            "cl": "0x0101",
            "ep": 11,
            "eval": "Item.val = Attr.val !=1",
            "fn": "zcl"
          }
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 11,
      "cl": "0x0101",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x30",
          "min": 1,
          "max": 300
        },
        {
          "at": "0x0003",
          "dt": "0x30",
          "min": 1,
          "max": 300
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 11,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 300,
          "max": 2700,
          "change": "0x00000001"
        }
      ]
    }
  ]
}`

Smanar, 2022-09-15:
>Next thing I will like to have is report back the last user operating the lock, any ideas?

You mean according to the pin code used ? Because your doorlock have a keypad ?
There is something started here https://github.com/dresden-elektronik/deconz-rest-plugin/pull/4876 but never valided

closed 2022-10-14 by github-actions[bot]