dresden-elektronik/deconz-rest-plugin #5875: Tuya ZigBee PIR Motion Sensor _TZ3000_bsvqrxru TS0202

BroTo-Typ, 2022-03-16:
Recently bought on Aliexpress (where else? ;-) ) and unfortunally the device is not supported:
https://de.aliexpress.com/item/1005003371197737.html?gatewayAdapt=glo2deu&spm=a2g0o.order_list.0.0.21ef5c5fIZHjNh
Tuya ZigBee PIR Motion Sensor Wireless Infrarot Detektor Sicherheit Einbrecher Alarm Sensor Benötigen Gateway Arbeit für Smart Leben APP

I´ve managed to pair it but I can´t get any of the data. Maybe it should work like most of the other PIR zigbee sensors? 
Hopefully you could help me out - thank you in advance!

## Device

- Product name: The device name as shown on the product or package.
- Manufacturer: As per deCONZ GUI Basic cluster.
- Model identifier: As per deCONZ GUI Basic cluster.
- Device type : 
  - Sensor

## Screenshots

### Basic
[image]

### Identify
[image]

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above
[image]
[image]
[image]
[image]
[image]

Mimiix, 2022-03-16:
@Smanar  IS this doable with DDF?

BroTo-Typ, 2022-03-16:
Unfortunally I don´t have enough knowledge in programming but I´ve seen that there are some _TZ3000_*** with different letters.
Maybe a modified DDF would work? I just need motion/no motion and/or the state of the battery.

By the way - the name "Presence 15" was chosen when I tried to make it work in the phoscon web app.

Thank you in advance and best regards!

Smanar, 2022-03-16:
>Unfortunally I don´t have enough knowledge in programming

You don't need there is an editor to do that (but you need some zigbee knowledge we are happy to give you ^^)

>Maybe a modified DDF would work?

Exactly, you just need to edit 4 lines if you find a compatible one, I will give you a sample after.

>the name "Presence 15" was chosen when I tried to make it work in the phoscon web app

Huuu, It mean the device is reconized, are you sure it's not working ?

BroTo-Typ, 2022-03-17:
Well - it´s being recognized now but I cannot see any change of the state "motion" or "battery"... Will try later again :-)

[image]

(renamed it to Bewegungsmelder 1 instead of Presence)

`{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_bsvqrxru",
  "modelid": "TS0202",
  "product": "TS0202",
  "sleeper": false,
  "status": "Gold",
  "path": "/devices/Bewegungsmelder TS0202.json",
  "subdevices": [
    {
      "type": "$TYPE_PRESENCE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0xff",
        "0x0406"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/alert",
          "description": "The currently active alert."
        },
        {
          "name": "config/battery",
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "cppsrc": "power_configuration.cpp:192",
            "ep": 1,
            "eval": "Item.val = Attr.val"
          }
        },
        {
          "name": "config/duration"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/presence"
        }
      ]
    }
  ]
}`

Smanar, 2022-03-17:
You have not exaclty the same sensor type, yours use IAS cluster 0x0500, and the one you have used use the presence cluster 0x0406

```
"uuid": [
"$address.ext",
"0xff",
"0x0406"
],
```

Need to be 

```
"uuid": [
"$address.ext",
"0x01",
"0x0500"
],
```

and you need to add state/lowbattery, state/tampered, config/pending, config/enrolled

Almost like here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5865#issuecomment-1069647384

BroTo-Typ, 2022-03-21:
Well - the actual JSON-file (with added add state/lowbattery, state/tampered, config/pending, config/enrolled and IAS cluster 0x0500) looks like this:

{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_bsvqrxru",
  "modelid": "TS0202",
  "product": "TS0202",
  "sleeper": false,
  "status": "Gold",
  "path": "/devices/ts0202.json",
  "subdevices": [
    {
      "type": "$TYPE_PRESENCE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/alert"
        },
        {
          "name": "config/battery"
        },
        {
          "name": "config/duration",
          "refresh.interval": 10,
          "read": {
            "fn": "zcl"
          },
          "parse": {
            "fn": "zcl"
          }
        },
        {
          "name": "config/enrolled",
          "public": false,
          "description": "State of IAS enrollment process."
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/pending",
          "description": "Pending tasks to configure the device."
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/battery",
          "refresh.interval": 3600
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/lowbattery",
          "description": "True when the device battery runs low."
        },
        {
          "name": "state/presence",
          "read": {
            "fn": "zcl"
          },
          "parse": {
            "fn": "zcl"
          }
        },
        {
          "name": "state/tampered"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001"
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0000"
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0500"
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0003"
    }
  ]
}

Unfortunately I don´t see any results...
Sadly I am not as good in programming as I would like to be - I´m just an old copy&paste skript"kiddie" :-)

Nevertheless thank you for helping me :-)

Smanar, 2022-03-21:
>But I did not really understand what you´ve meant with "If you don't have typo for manufacture name and model id, when making "hot reload" on the menu, the device need to change it's name "0x0000" for "presence X""

Ha yes sorry, your device was reconized at start, on the capture it was "Presence 15", an unreconized device will be 0xXXXX

>But the "Motion detected" stays and doesn´t change.

Some device don't have 2 state, but send a new notification at every detection.

if not try to complete the state/presence

```
        {
          "name": "state/presence",
          "awake": true,
          "read": {
            "fn": "None"
          },
          "parse": {
            "at": "0x0002",
            "cl": "0x0500",
            "ep": 1,
            "eval": "Item.val = (Attr.val & 0x3) != 0",
            "fn": "zcl"
          }
        },
```

If still not working, try to complete the binding, (you can drag and drop the attribute 0x0002 from the cluster 0x0500 on the right panel, if you want to use the editor)

```
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0500",
      "report": [
        {
          "at": "0x0002",
          "dt": "0x19",
          "min": 300,
          "max": 3600
        }
      ]
    }
```

BroTo-Typ, 2022-04-04:
Sorry for my late reply - work was stressing me a little bit...
Well - the sensor still doesn´t work.
But I´ve tried to pair it with zigbee2mqtt and there it worked like a charm (I´m using Deconz and zigbee2mqtt at the same time with different channels without problems).

So sadly I´m not shure why the sensor doesn´t recognize the motion state in Deconz...

Nevertheless thank you for your effort and best regards!

BroTo-Typ, 2022-04-04:
Update - I don´t know why but after doing some trial and error it´s working now with the configuration above...
Thank you very much for your help :-)

Smanar, 2022-04-04:
On your DDF you can remove the bind to the cluster 0x0003  and the one to the cluster 0x0000
You can too remove the state/tampered part, all you are inside is random ^^

BroTo-Typ, 2022-04-05:
Oookay, I´ve changed the above mentioned things and will tell you if it´s working ^^
But, to be save, I´ve made an extra copy of my configuration - fhem (my homeautomation) already recognized the motion with that so I was able to use the motion sensor :-)

I´ll tell you tomorrow if your changes have been successful.

Thank you very much!

BroTo-Typ, 2022-04-05:
EDIT: The sensor still is recognized - in FHEM (my homeautomation) its shows  up as HUEDevice (like all the bulbs and sensors via Deconz) and it shows alarm:1 and alarm:0 
So everything works like it should.
------------
HUEDevice HUESensor21 alarm: 0
HUEDevice HUESensor21 lastseen: 2022-04-05T16:20Z
HUEDevice HUESensor21 reachable: 1
HUEDevice HUESensor21 alarm: 1
HUEDevice HUESensor21 batteryState: ok
------------

Thank you for your help!

Smanar, 2022-04-05:
If all is working, can you make a PR for other users ?
You can't break something, can make several tries ^^.

BroTo-Typ, 2022-04-05:
> If all is working, can you make a PR for other users ? You can't break something, can make several tries ^^.

Erm - what is a PR?^^
Shure I can try to write a kind of tutorial what I´ve done and upload it here, including the code and screenshots next week... :-)

Smanar, 2022-04-05:
A PR is Pull request, like here https://github.com/dresden-elektronik/deconz-rest-plugin/pulls

The easy way, just try to add the file (in good folder, for exemple here for tuya https://github.com/dresden-elektronik/deconz-rest-plugin/tree/master/devices/tuya) github will say you have not the right on the github, make a fork under you name, create the file, and propose you to make the PR.

I can explain better if needed.

Like that the PR will be merged automatically and under your name.

closed 2022-05-05 by github-actions[bot]

phil-schneider, 2022-10-17:
@BroTo-Typ what about your PR?
You asked for help, you received help.
Please create a pr so that others can profit from your experience. Thank you