dresden-elektronik/deconz-rest-plugin #5876: ERIA Smart Wireless Dimming Switch Remote 81825 doesn't work as advertised

RudolfRendier, 2022-03-16:
## Describe the bug

Device doesn't work. Can be paired. Shows up in deCONZ, with hex-name. No events visible in Home Assistant.

Product name: ERIA Smart Wireless Dimming Switch Remote
Manufacturer: ADUROLIGHT
Model identifier: Adurolight_NCC
Device type: Switch
https://adurosmart.com/product/eria-smart-wireless-dimming-switch-remote/

There are existing issues for this device, and it should work, but it doesn't:
dresden-elektronik/deconz-rest-plugin#4495
dresden-elektronik/deconz-rest-plugin#3383

## Steps to reproduce the behavior

1. Reset the device by holding both arrow-buttons, until it flashes
1. Enable searching for switches in Phoscon
1. Hold the off button until the remote blinks
1. Watch in deCONZ for the device to show up
1. Check HA for deconz_events when pressing a button

## Expected behavior

No events shown.

## Screenshots
https://i.postimg.cc/HV5ChJxS/audrolight.png

## Environment

- Intel NUC i3, running proxmox OS and Home Assistant OS with the core_deconz addon
- Firmware version: (26720700)
- deCONZ version: (2.14.01 / 06/02/2022)
- ConBee II
- extension cable: yes
- other devices: yes, rfxcom

Recently migrated from RPi3 HASSOS with Raspbee, didn't work on that hardware either.

## deCONZ Logs

This line must mean something, it appears once every time I press a button
(every button, twice)
Line 87: 10:58:02:728 No matching sensor found for 0x00158D0001CCB407, endpoint: 0x01
Line 96: 10:58:03:542 No matching sensor found for 0x00158D0001CCB407, endpoint: 0x01
Line 98: 10:58:04:350 No matching sensor found for 0x00158D0001CCB407, endpoint: 0x01
Line 100: 10:58:05:177 No matching sensor found for 0x00158D0001CCB407, endpoint: 0x01
Line 102: 10:58:06:113 No matching sensor found for 0x00158D0001CCB407, endpoint: 0x01
Line 110: 10:58:06:855 No matching sensor found for 0x00158D0001CCB407, endpoint: 0x01
Line 118: 10:58:07:622 No matching sensor found for 0x00158D0001CCB407, endpoint: 0x01
Line 120: 10:58:08:309 No matching sensor found for 0x00158D0001CCB407, endpoint: 0x01

Here's the full log, for context:
10:57:43:502 ZDP Mgmt_Lqi_rsp zdpSeq: 166 from 0x000B57FFFEB16C20 total: 12, startIndex: 3, listCount: 3
10:57:43:503 * neighbor: 0x00158D0006E124F8 (0x4522), LQI: 63, relation: 0x02, depth: 15, rxOnWHenIdle: 1
10:57:43:504 * neighbor: 0x0C4314FFFEFC122E (0x4F9F), LQI: 95, relation: 0x02, depth: 15, rxOnWHenIdle: 1
10:57:43:506 * neighbor: 0x00178801067DF434 (0x66F3), LQI: 123, relation: 0x02, depth: 15, rxOnWHenIdle: 1
10:57:44:041 APS-DATA.request id: 65, addrmode: 0x03, addr: 0x0017880104a0c7cd, profile: 0x0104, cluster: 0x0006, ep: 0x01 -> 0x0B queue: 0 len: 5 tx.options 0x00
10:57:44:066 APS-DATA.confirm id: 65, status: 0x00 SUCCESS
10:57:44:068 APS-DATA.confirm request id: 65 -> erase from queue
10:57:44:123 APS-DATA.indication srcAddr: 0xf794, srcEp: 0x0B dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 255, rssi: -70
10:57:44:125 APS-DATA.request id: 65 erase from queue
10:57:44:127 ZCL got data for node=0xF794, cl=0x0006, at=0x0000, status=0x00, type=0x10
10:57:47:440 Skip idle timer callback, too early: elapsed 923 msec
10:57:48:341 APS-DATA.request id: 86, addrmode: 0x03, addr: 0x588e81fffed3979d, profile: 0x0104, cluster: 0x0B04, ep: 0x01 -> 0x01 queue: 0 len: 9 tx.options 0x00
10:57:48:384 APS-DATA.confirm id: 86, status: 0x00 SUCCESS
10:57:48:386 APS-DATA.confirm request id: 86 -> erase from queue
10:57:48:426 APS-DATA.indication srcAddr: 0xcd84, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0B04, lqi: 255, rssi: -76
10:57:48:427 APS-DATA.request id: 86 erase from queue
[... 143 lines cut ...]
10:58:14:349 APS-DATA.indication srcAddr: 0xf697, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 255, rssi: -65
10:58:14:350 APS-DATA.indication request id: 239 -> finished
10:58:14:352 APS-DATA.request id: 239 erase from queue
10:58:14:353 ZDP status = 0x00 -> SUCCESS
10:58:14:354 ZDP Mgmt_Lqi_rsp zdpSeq: 176 from 0x000B57FFFEB16C20 total: 12, startIndex: 9, listCount: 3
10:58:14:357 * neighbor: 0x588E81FFFED3979D (0xCD84), LQI: 63, relation: 0x02, depth: 15, rxOnWHenIdle: 1
10:58:14:358 * neighbor: 0x588E81FFFED3A0B8 (0xD792), LQI: 42, relation: 0x02, depth: 15, rxOnWHenIdle: 1
10:58:14:359 * neighbor: 0x000B57FFFE9E9305 (0xD93E), LQI: 33, relation: 0x02, depth: 15, rxOnWHenIdle: 1
10:58:15:442 APS-DATA.request id: 245, addrmode: 0x03, addr: 0x0017880104d37752, profile: 0x0104, cluster: 0x0006, ep: 0x01 -> 0x0B queue: 0 len: 5 tx.options 0x00
10:58:15:464 APS-DATA.confirm id: 245, status: 0x00 SUCCESS
10:58:15:465 APS-DATA.confirm request id: 245 -> erase from queue
10:58:15:484 APS-DATA.indication srcAddr: 0x1211, srcEp: 0x0B dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 255, rssi: -54
10:58:15:485 APS-DATA.request id: 245 erase from queue
10:58:15:486 ZCL got data for node=0x1211, cl=0x0006, at=0x0000, status=0x00, type=0x10
10:58:16:842 APS-DATA.request id: 253, addrmode: 0x03, addr: 0x588e81fffed3a0b8, profile: 0x0104, cluster: 0x0B04, ep: 0x01 -> 0x01 queue: 0 len: 9 tx.options 0x00
10:58:16:872 APS-DATA.confirm id: 253, status: 0x00 SUCCESS
10:58:16:874 APS-DATA.confirm request id: 253 -> erase from queue
10:58:16:886 aps request id: 253 finished, erase from queue
10:58:16:895 APS-DATA.indication srcAddr: 0xd792, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0B04, lqi: 255, rssi: -76
10:58:16:896 ZCL got data for node=0xD792, cl=0x0B04, at=0x0505, status=0x00, type=0x21
10:58:16:898 ZCL got data for node=0xD792, cl=0x0B04, at=0x0508, status=0x00, type=0x21
10:58:16:898 ZCL got data for node=0xD792, cl=0x0B04, at=0x050B, status=0x00, type=0x29

## Additional context
Different brands: IKEA, Philips, Xiaomi, Blitzwolf, AduroSmart

manup, 2022-03-16:
>This line must mean something, it appears once every time I press a button
(every button, twice)
>Line 87: 10:58:02:728 No matching sensor found for 0x00158D0001CCB407, endpoint: 0x01

The message comes from this part of the code, I've checked the integration and button_maps.json there the code for integration looks ok on a quick look.

https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/mfspecific_cluster_xiaoyan.cpp#L16-L22

Is the switch showing up in the Phoscon App? It looks like the sensor resource isn't created.

Mimiix, 2022-03-16:
Seems to be a whitelist issue. Hexname

manup, 2022-03-16:
Yeah thought that too, but from the source code the device is already listed in respective places :thinking:

RudolfRendier, 2022-03-16:
> Is the switch showing up in the Phoscon App? It looks like the sensor resource isn't created.

No, device cannot be seen in Phoscon App.
I looked into the result of the REST endpoint for sensors, but I couldn't find anything related, i.e. no matches for its 'uniqueid'.

closed 2022-04-14 by github-actions[bot]

mdolnik, 2023-05-04:
I know this is an old thread, but I have this remote and I can confirm as of Deconz `2.21.2` (and earlier) that it works for me in deConz, Phoscon, and Home Assistant.

Here's what works for me:

1. In Phoscon Add new switch
2. Choose `Other`
3. Pair the switch:
    - Factory reset Eria switch by holding up + down buttons for 3-5 seconds until light flashes green.
    - Try holding Off button for 3-10 seconds:
        - If it flashes green it's successful.
        - If it flashes red, repeat all the steps again.
        - If repeating steps doesn't work, try hold the remote closer to the gateway
        - If this doesn't work try the next step.
    - Try holding On button for 3-10 seconds:
        - If it flashes green it's successful.
        - If it flashes red, repeat all the steps again.
        - If repeating steps doesn't work, try hold the remote closer to the gateway.
        - If this doesn't work... I'm not sure, try removing/re-inserting battery and trying all steps again?
4. If the light flashes green during pairing, then it is successful:
    - Phoscon will take 5-15 seconds to recognize it:
        - When it first recognizes it, the Back button will disappear.
        - After another 5-10 seconds a success message will appear.
    - The switch will have the Vendor name `Trust` and Product `Dimming switch` in Phoscon.
    - Any button press on the switch should be indicated by the left status circle on the deCONZ node.
5. Identify in Home Assistant:
    1. Go to `Developer Tools` > `Events`
    2. Set `Listen to Events` to `deconz_events`
    3. Click `Start Listening`
    4. Click any button on the remote
    5. An event should appear with the corresponding data / ID
6. Note: This remote appears in Home Assistant as a Device, but not an Entity

[image]
[image]
[image]