dresden-elektronik/deconz-rest-plugin #5883: Friends of Hue Scenic switches are not shown up in deconz gui

larslorenz, 2022-03-17:
## Describe the bug
I dont know if its the right place. I have 3 FriendsOfHue Scenic switches. They are successfully paired and configured over the phoscon webapp. But in deconz they are niot listede at all. I have also a several osram 3 button smartswitches they are paired and listed in deconz as aspected. At all the 3 scenic switches are working but I have a lot of problems with them. It seems that they loses the connection after a while and than you have to press them once with no action happen the second press mostly leads to the acpected behavior. I tried a lot with changing wifi channels and stuf like that. So than i want to see in deconz gui if there is a bad connection but I cannot find them there. I tried to reset and reconnect them but no succsess.

## Steps to reproduce the behavior
The switch is listed and useable in phoscon but not in deconz. In deconz it looks like the switch istn there. The switches often work at the second press but not the firtst one.

## Expected behavior
I hope to see the switches in deconz gui to check for bad connections

## Screenshots

## Environment
deconz: 2.14.01 / 6.2.2022
fw: 26660700

 - Raspberry Pi
 - Raspbian Buster
 - Firmware version: 26660700
 - deCONZ version: 2.14.01
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? If so: 433mhz CUL

## deCONZ Logs

## Additional context

Mimiix, 2022-03-17:
All i know is that FOH Switches(ZGP) are not shown in deCONZ. That's why they arent there. I am not sure why(not).

The issue on switches dropping isn't relevant for this report as that is not as per #5113.

Ive asked Manup to reply.

manup, 2022-03-18:
Right, at the moment ZGP switches are only visible via the REST-API and and in the Phoscon App.
(This is since the UI very much is centered on nodes being tratidional Zigbee devices)

I hope to figure out a way to show them then the ZGP DDF integration is done, but it might looks weird as ZGP devices don't have parents ;)

larslorenz, 2022-03-21:
Ok thank you both for the replies. 
I dont have a big problem if theses devices arent shown up in deconz GUI its just good to know that this is a normal behavior. If i understand Mimiix right the problem with that connection losts have different reasons and shouldt be dicussed here rather then in this forum: https://forum.phoscon.de/c/deconz/general-support/12 ?
I m not really sure if that behavior is a bug or not..

Thanks in advance

closed 2022-04-19 by github-actions[bot]