dresden-elektronik/deconz-rest-plugin #5884: Temperature=0 In Phoscon with 2.15.0

LeoeLeoeL, 2022-03-18:
## Describe the bug
Temperature is 0 in Phoscon for Xiaomi Aqara temperature/humidity/pressure sensor

## Steps to reproduce the behavior
Go to Phoscon --> Sensors --> Aqara Sensor
## Expected behavior
 The correct temperature

## Screenshots

[image]
[image]

## Environment

 - Host system: Raspberry Pi
 - Running method: Raspbian
 - Firmware version: 0x26770700
 - deCONZ version: v2.15.0-beta
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? If so: Which? APC UPS

## deCONZ Logs

## Additional context
 Temperature was correct with 2.14.x
Temperature is correct in the summary page of sensors.
Other Aqara sensors (contact and motion) are OK.

Mimiix, 2022-03-18:
What does the RestAPI report?

SwoopX, 2022-03-18:
In my view, the device stopped reporting temperature values for whatever reason. The one I have in my production environment also defaults to 0 right after upgrade, but then moves on

[image]

canedje, 2022-03-19:
I do have the same issue in my sensors used in freezers.
So it is not showing negative values anymore:
[image]

Erikgj65, 2022-03-19:
I have the same problem.
I test my sensor and it  work above 0 and not below 0.

ACrazyConcept, 2022-03-19:
lol I have like 1/3 of my 16 Xiaomi Aqara sensors show any combination of 0 temp, 0 humidity and 0 pressure.

LeoeLeoeL, 2022-03-19:
I have this problem in Phoscon only.
Data in Home Assistant are ok.

Langelus, 2022-03-20:
Can confirm that I have the same problem, both my sensors in the freezer stops reporting at 0 in gui and what's reported to HomeAssistant. Sensordata if I check in Deconz app seems to report correctly at 0402 temperature measurement 0x0000 though

Rolf56, 2022-03-21:
Had the same problem too in my Windows Environment. After reset and new connecting in Phoscon App everithing works fine again.

Langelus, 2022-03-21:
Tried resetting one of my sensors, not any difference unfortunately. Still not reporting sub 0 temperatures

iisworks, 2022-03-25:
Same here on 2 sensors. Both in fridge and reporting around 0 C. Values look fine in deconz on cluster 402, just not in the api.
I just deleted both sensors and paired them again, but the issue persists.

closed 2022-03-27 by manup