dresden-elektronik/deconz-rest-plugin #5887: dresden elektronik "FLS-PP lp" without PowerOn Configuration Capability

marc965, 2022-03-18:
## Describe the bug

When utility power is restored certain lights connected to dresden elektronik "FLS-PP lp" are turned on for default and you can't stop this with configuration. It is very boring.
## Steps to reproduce the behavior

turn off utility power and on again and you see...
## Expected behavior

I expected a configuration possibility like dresden elektronik 'Kobold".
## Screenshots

I expected a configuration possibility like dresden elektronik 'Kobold".
like this:
[image]
For dresden elektronik "FLS-PP lp" it is greyed out:
[image]

## Environment

 - Host system: (Raspberry Pi / PC / NAS)
   NAS: Synology VMM, Home Assistant core-2022.3.5, supervisor-2022.03.5, Home Assistant OS 7.5

 - Running method: Raspbian / Ubuntu / Home Assistent deCONZ Add-on / Marthoc Docker container / Windows / Virtual Machine)
   Home Assistant deCONZ Add-on 6.12.0
 - Firmware version: (26xxyy00) 266B0700
 - deCONZ version: (2.xx.yy) 2.14.01 / 6.2.2022
 - Device: (ConBee I / ConBee II / RaspBee I / RaspBee II) ConBee II
 - Do you use an USB extension cable: (yes / no) -- only relevant for ConBee I/II yes
 - Is there any other USB or serial devices connected to the host system? If so: Which? no

## deCONZ Logs

## Additional context

SwoopX, 2022-03-18:
This is not a deconz bug, but a capability lack of the device.

However, I can share a deconz rule later when I'm at my PC to automatically turn it off one it gets power (requires the coordinator online and deconz running).

You might want to raise this nevertheless as feature request for the device in the forum.

closed 2022-03-18 by SwoopX

Mimiix, 2022-03-18:
@SwoopX the forum and git are based on deconz, not on the flspp.

I think emailing support is appropriate here. However, I did allow flspp bug reports and feature requests in the past.

SwoopX, 2022-03-18:
That's the PUT reqeust body, make sure to exchange the light ID

```
{
    "actions": [
      {
        "address": "/lights/7/state",
        "body": {
          "on": false
        },
        "method": "PUT"
      }
    ],
    "conditions": [
      {
        "address": "/lights/7/attr/lastannounced",
        "operator": "dx"
      }
    ],
    "name": "FLS-PP Off"
}
```

marc965, 2022-03-19:
Thank you very much for your help.
Your code show me that they are much more possibilities to have influence on my Conbee with the API.

Do you have links where i can get more samples and tutorials?

Unfortunately your code works only if the zigbee network is fully operational. But it is a nice temporary solution.

I will try to email dresden elektronik. Perhaps a firmware update can solve this problem at the root...

SwoopX, 2022-03-19:
Check out https://dresden-elektronik.github.io/deconz-rest-doc/ for some inspiration.