dresden-elektronik/deconz-rest-plugin #5889: Device name

panenka, 2022-03-19:
## Device

- Product name: Smart Air House keeper
- Manufacturer: _TZE200_dwcarsat
- Model identifier: TS0601
- Device type : 
  - Sensor
https://a.aliexpress.com/_mthOESy
Tuya zigbee Smart Air Housekeeper PM2.5 Formaldehyde VOC CO2 Temperature Humidity 6 In 1 Smart Air Box Sensor Alarm Detector
Specifications:
Working temperature:-20-85 °C
Working voltage:5 V
Working current:10-496 ma
size:100*100*40 mm
The power supply mode:USB 5V/1A
Communication protocol:ZIGBEE
color:white
Shell material:ABS
The weight of the:95g
Range (Formaldehyde):0-10.00 mg/m3
Range (VOC):0-99.9 ppm
Range (Carbon dioxide):0-1000 ppm
Range (Temperature):0-60 °C
Range (Humidity):0-95 %
Range (PM2.5):0-1000 ug/m3

## Screenshots
[image]
[image]

### Basic
[image]

### Identify

### Alarms

### Device Temperature

### Groups
[image]
[image]
[image]

### Scenes
[image]
[image]
[image]
[image]

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above
Tuya Specific cluster
[image]
OTAU Cluster
[image]
[image]
Time Cluster
[image]
Green Power Cluster
[image]

[image]
[image]
[image]
[image]

panenka, 2022-03-19:
found some info here too https://github.com/zigpy/zha-device-handlers/issues/1313

panenka, 2022-04-10:
Still waiting for support

DieterKoblenz, 2022-04-18:
Is more information required? I have the same device, and missing support.

Mimiix, 2022-04-21:
@cloudbr34k84  this is ZHA, not deCONZ.

cloudbr34k84, 2022-04-21:
I know I use ZHA..

Mimiix, 2022-04-21:
@Smanar Can a DDF be done for htis?

Smanar, 2022-04-21:
>What's that sorry?

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet

Yes probably but will not be easy, can start first for the temperature to test the feasibility.
Start the DDF with filling the "basic cluster identification"
Drag and drop a ZHATemperature.
Select state/temperature and fill the field (in "parse" and "read") to have something like that https://user-images.githubusercontent.com/383386/158289098-e7d947ea-6fa5-47b7-a0df-72c632c8e425.png

The dpid for temperature is 18 (0x12)

Make hot reload, the node name will change from 0xXXXX in deconz

Take a look on the API to see the device, third app or Phosocn / help / API Information / sensor

cs42, 2022-05-14:
Zigpy supports this device. Shouldn't it be easy to create a DDF using the code in [https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/air/__init__.py](https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/air/__init__.py) or isn't this possible with DDFs (yet)?

Smanar, 2022-05-14:
I think it can work, for the moment you can't write tuya request (it's WIP) but you can read them so no problem for temperature and humidity (haven't checked if all gaz sensor are available)

I just need a squeleton, I can make the tuya stuff part.

AssetBurned, 2022-06-11:
Isn't this a pretty similar unit compared to the _TZE200_yvx5lh6k in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4990 ?

Smanar, 2022-06-12:
Hello, can you try this DDF, it will work only for temperature and humidity, but it's to test.

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZE200_dwcarsat",
  "modelid": "TS0601",
  "product": "Smart Air House keeper",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_HUMIDITY_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0405"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/offset",
          "description": "Relative offset to the main measured value.",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/humidity",
          "description": "The current relative humidity in percent.",
          "parse": {"fn": "tuya", "dpid": 19, "eval": "Item.val = 10 * Attr.val;" },
          "read": {"fn": "tuya"},
          "default": 0
        },
        {
          "name": "config/battery",
          "parse": {"fn": "tuya", "dpid": 4, "eval": "Item.val = Attr.val;" },
          "read": {"fn": "none"},
          "default": 0
        },
        {
          "name": "state/lastupdated"
        }
      ]
    },
    {
      "type": "$TYPE_TEMPERATURE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0402"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/offset",
          "description": "Relative offset to the main measured value.",
          "default": 0
        },
        {
          "name": "config/battery",
          "parse": {"fn": "tuya", "dpid": 4, "eval": "Item.val = Attr.val;" },
          "read": {"fn": "none"},
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/temperature",
          "parse": {"fn": "tuya", "dpid": 18, "eval": "Item.val = 10 * Attr.val;" },
          "read": {"fn": "none"},
          "default": 0
        }
      ]
    }
  ]
}
```

muratcorlu, 2022-06-12:
@Smanar I tried to use this DDF but nothing changed. I see this device as light in my Phoscon UI. But that was same even before trying your DDF.

[image]

Smanar, 2022-06-13:
Nothing more in phoscon/help/API Information/sensors ?

What is your OS ? Have you the GUI ?
(To check if the DDF is used)
You have tried to re-include the device after the DDF was in place ?

Mimiix, 2022-09-13:
Re-open by request.

DieterKoblenz, 2022-09-21:
Re-open please.

vgisbert, 2022-09-26:
Hello. I have not been successful.

CONDITIONS:
- My computer has Conbee II mounted on Linux Debian 4.19.0-18-amd64 which communicates with Home Assistant 2022.9.6 via Add-on deConz version 6.16.0

PROCESS:
- Create the file ts0601.json with the content of the Smanar partner.
- Upload the file to the computer.
In my case I left the file with root user in the folder:
/usr/share/hassio/addons/data/core_deconz/.local/share/dresden-elektronik/deCONZ/devices/
- In the control panel change the DDF mode to “Strict”.
- Select deConz select the sensor and with the right mouse button edit its DDF.
- With the editor “File Open” the file ts0601.json and make a “File Hot reload”.
- I deleted the old “Smart plug” in Phoscon application.
- In Phoscon, the sensor detection procedure is repeated (by pressing the button on the back of the device).

As a result of these operations the sensor is correctly detected as “Humiditi xxx” and appears correctly in deConz.
But, it doesn't show up in Phoscon's list of sensors, and of course it doesn't show up in Home Assistant either.

Have I done something wrong?
Thanks for the help.

vgisbert, 2022-09-26:
Yes, in the API information it appears as Sensor.

vgisbert, 2022-09-26:
[image]
[image]

Mimiix, 2022-09-26:
> Yes, in the API information they appear as Sensors, one for humidity and one for temperature.

In that case: The issue is with Home assistant and Phoscon.  Phoscon needs to add support itself. As this is closed source, the community can't do that.

If you restart Home-Assistant it might show up.

vgisbert, 2022-09-26:
I already did, and it doesn't show up.

Mimiix, 2022-09-26:
> I already did, and it doesn't show up.

In that case, the issue is with HA as the API provides it.

vgisbert, 2022-09-26:
What I don't know is if HA communicates with deConz through the API, or through Phoscon. Well, Phoscon does not show the sensor either:
[image]

Mimiix, 2022-09-26:
> What I don't know is if HA communicates with deConz through the API, or through Phoscon. Well, Phoscon does not show the sensor either: [image]

Phoscon and HA talk to the deCONZ Api.

freman, 2022-09-29:
I have just plugged one of these into my network too, same problem. I will try the DDF.

freman, 2022-10-02:
It did detect as a humidity sensor, but after a restart it's back to being a smart plug, and it shows up in Lights
[image]

@vgisbert is your showing up in lights?

Smanar, 2022-10-02:
@freman what is your deconz version ? on all one, you can have a light device created  by defaut, but on last one the DDF prevent that.

But even with the bad light entry, you can have too the 2 sensors, take a  look direclty in the API with phoscon / help / API Information, but strange you can't see them on HA.

leppa, 2022-10-25:
Go away, bot! My device arrived today and it's still detected as smart plug.

Mimiix, 2022-10-25:
How are the Sensors looking in the api?

Smanar, 2022-10-25:
If it's a tuya device, it's a probably a smart plug on zigbee side (so in deconz)
But if the device is reconised in the API, you will have another type visible in the API (and phoscon)

Your device is 

```
  "manufacturername": "_TZE200_dwcarsat",
  "modelid": "TS0601",
```

closed 2022-11-27 by github-actions[bot]

m4k2k, 2023-01-12:
Hello all,
the issue is still existing - please re-open the ticket.
[image]

two questions:
- is there any update to this issue? What is still required to get it fully supported (all sensors) by deconz API and app?
- is there currently a working workaround to get it fixed manually?

Thanks in advance - best regards
Markus

Smanar, 2023-01-12:
Hello, what is the issue, the device reconised as light instead of sensor ?
If yes what is your deconz version ? It' s normally solved since somes versions

And the DDF is not on the official version, no PR made, so you need to add it yourself.

The last one is probably this one https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5889#issuecomment-1153143038

m4k2k, 2023-01-12:
Hi @Smanar,
there are two issues:
- the device is oob detected as light source
- even with the ddf there are only 2 sensor values (at least as I read the ddf), but the device has at least 6 different sensors.

There seems also someone working on this as the DDF is on the deconz api github master (only 2 sensors..):
https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/_TZE200_TS0601_humidity_temp.json

more device info (6 sensors):
https://zigbee.blakadder.com/Tuya_DCR-KQG.html

what is required to get the 6 sensors supported?

Thanks + best regards
Markus

Smanar, 2023-01-12:
Not hard to add them, the problem is finding field already usable in deconz without compilation ^^.

I have :
- state/airquality but need to be a string
- state/airqualityppb this one is a number
- state/carbonmonoxide but this one is a bool not usable for this device.
- state/pm2_5 as number

So can add this one in the DDF

```
    {
      "type": "$TYPE_AIR_QUALITY_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x02",
        "0xfc7d"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0007",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0003",
          "0x0004",
          "0x0005",
          "0xFC7D"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/pm2_5",
          "parse": {"fn": "tuya", "dpid": 4, "eval": "Item.val = Attr.val;" },
          "read": {"fn": "none"},
          "default": 0
        },
        {
          "name": "state/airqualityppb",
          "parse": {"fn": "tuya", "dpid": 2, "eval": "Item.val = Attr.val;" },
          "read": {"fn": "none"},
          "default": 0
        }
      ]
    }
```

state/airqualityppb will be VOC (idk what is it)
state/pm2_5 will be pm2.5 (idk what is it too, but this one can have fake value that need to be ignored, will see if working)

m4k2k, 2023-01-13:
Hello @Smanar,
it was working - partly.
What if found out meanwhile is that the
"_TZE200_dwcarsat" 6 sensor device is completely different than the 5-sensor device in regards of dpid's.

For my (6 sensor device) the dpid's are:

https://github.com/Koenkk/zigbee-herdsman-converters/blob/75db090d32280e21caa8c02957488b16faefda31/lib/tuya.js#L613

mix of smart air box and smart air house keeper
special for the box with 6 sensors "_TZE200_dwcarsat" (complete different than 5 sensors)

> tuyaSabTemp: **18**, /10
> tuyaSabHumidity: **19**, /10
> tuyaSahkMP25: **2**, (fine particles, pm2.5) -> in ug/m3
> tuyaSabVOC: **21**, /10 (volatile organic compounds | aka TVOC | aka airqualityppb) -> in ppm
> 
> tuyaSahkCO2: **22**, (Carbon dioxide) -> in ppm
> tuyaSahkFormaldehyd: **20**, /100 (CH2O) -> in ug/m3

https://github.com/Koenkk/zigbee-herdsman-converters/blob/75db090d32280e21caa8c02957488b16faefda31/converters/fromZigbee.js#L4263

The /10 etc hints that this might need a division before usable. I also added the references where I found the information.

I tested with
```
        {
          "name": "state/airqualityppb",
          "parse": {
            "fn": "tuya",
            "dpid": 21,
            "eval": "Item.val = Attr.val;"
          },
          "read": {
            "fn": "none"
          },
          "default": 0
        },
        {
          "name": "state/pm2_5",
          "parse": {
            "fn": "tuya",
            "dpid": 2,
            "eval": "Item.val = Attr.val;"
          },
          "read": {
            "fn": "none"
          },
          "default": 0
        },
```
and I'm getting promising results:
```
    "manufacturername": "_TZE200_dwcarsat",
    "modelid": "TS0601",
    "name": "AirQuality 26",
    "state": {
        "airqualityppb": 21,
        "lastupdated": "2023-01-13T22:01:00.241",
        "pm2_5": 4
    },
    "type": "ZHAAirQuality",
```
probably require some /10 division als ppb seems too high.

CO2 and Formaldehyd seems not supported yet and require a new types -> core/cpp adaptions.

**Sidenote**: In Homeassistant the values are not usable yet as there seems a bug crashing the deconz integration because of the ppb sensor

* What still puzzles me is why it's not visible in the phoscon app?
* Is there a config required to tell phoscon how to handle it?
* Is there a docker image/helper to build deconz rest / phoscon (maybe I can try to fix and do PR)

Thanks + best regards
Markus

Smanar, 2023-01-14:
>What still puzzles me is why it's not visible in the phoscon app?
Is there a config required to tell phoscon how to handle it?

Phoscon is a big story, we haven't impact on it, You can ask for feature on the forum, but I realy have no information on it, no roadmap, and no it's realy something closed, you can't do something yourself.

>Is there a docker image/helper to build deconz rest / phoscon (maybe I can try to fix and do PR)

Yes, you have a comment here for docker https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Compiling-the-REST-plugin-for-device-specific-testing at the end.
But I will make a PR myself, it will take only 15 mn.

For the HA issue, wana make an issue on the HA forum, or I can ask to plugin dev if he have time to take a look on this side ?

And BTW if you can share your actual DDF ?

m4k2k, 2023-01-15:
Hello @Smanar,

well done! its working - we're near to a solution.

>Phoscon is a big story, we haven't impact on it, You can ask for feature on the forum, but I really have no information on it, no roadmap, and no it's really something closed, you can't do something yourself.

ok, don't really use phoscon currently - just for debug. Nodes are managed in HA

> And BTW if you can share your actual DDF ?

gist: [_tze200_ts0601_dwcarsat-json](https://gist.github.com/m4k2k/f38e765b3800e2dcd1f1d18d165af324#file-_tze200_ts0601_dwcarsat-json)
sure - but disclaimer: its not complete - I removed some features for debugging simplicity (e.g. swversion, offset, etc.).

- Now checking what else information I get from the device (to get best possible configuration).

- Another open point is the correct number format /division if required. If I divide, we lose information as the fields are int16 and not double (or similar)

> For the HA issue, wana make an issue on the HA forum, or I can ask to plugin dev if he have time to take a look on this side ?

one by one - once deconz rest working smooth I'll check HA

> Yes, you have a comment here for docker

Great source! I was able to compile with it.

> But I will make a PR myself, it will take only 15 mn.

used your code 👍 

**Airquality now provides**
```
    "manufacturername": "_TZE200_dwcarsat",
    "modelid": "TS0601",
    "name": "AirQuality 4",
    "state": {
        "airqualityco2": 364,
        "airqualityformaldehyd": 2,
        "airqualityppb": 19,
        "lastupdated": "2023-01-15T15:19:16.891",
        "pm2_5": 7
    },
    "type": "ZHAAirQuality",
```

now all 6 sensors are working.

What I still dont get is whats going on in the deconz log:

```
15:34:12:183 TY_DATA_REPORT: seq 0, dpid: 0x12, type: 0x02, length: 4, val: 232
15:34:12:231 TY_DATA_REPORT: seq 0, dpid: 0x13, type: 0x02, length: 4, val: 524
15:34:13:285 TY_DATA_REPORT: seq 1, dpid: 0x14, type: 0x02, length: 4, val: 3
15:34:13:333 TY_DATA_REPORT: seq 1, dpid: 0x15, type: 0x02, length: 4, val: 21
15:34:13:380 TY_DATA_REPORT: seq 1, dpid: 0x16, type: 0x02, length: 4, val: 366
15:34:13:427 TY_DATA_REPORT: seq 4, dpid: 0x02, type: 0x02, length: 4, val: 7
```

these are completely different dpid's than I'm using..

```
# currently working:
sensor: dpid
Temp: 18
Humidity: 19
MP25: 2
VOC: 21
CO2: 22
Formaldehyd: 20
```
but according to above I also could use

```
dpid = value => sensor
02 = 7 => pm2_5
12 = 232 => temperature
13 = 524 => humidity
14 = 3 => airqualityformaldehyd
15 = 21 => airqualityppb
16 = 366 => airqualityco2
```
1. do you have a clue whats happening here?

for some strange reason it also appears as "light" again:
[image]

2. is there a possibility to "hide" it?

Thank you for your support + effort

best regards

Markus

Smanar, 2023-01-15:
>Another open point is the correct number format /division if required. If I divide, we lose information as the fields are int16 and not double (or similar)

I don't see where you need a division ? Remember for exemple on deconz temperature are * 100, if we have a value too low, perhaps it's a good idea to have it * 1000, like we do for current (in mA instead of A)

>What I still dont get is whats going on in the deconz log:

Values are hexa
0x12 = 18 > you use it
0x13 = 19 > you use it
0x14 = 20 > you use it
0x15 = 21 > you use it
0x02 = 2 > it's Co2 or pm2.5 (this one depend of devices, and 1 devices can use more than one), the code is a little tricky on z2m https://github.com/Koenkk/zigbee-herdsman-converters/blob/75db090d32280e21caa8c02957488b16faefda31/converters/fromZigbee.js#L4275

>for some strange reason it also appears as "light" again:

This is a long story and a bad idea, but If I m right you have it only on phoscon, if you look on HA or direclty on the API, you will not have this result
And no way to hide it, except don't use phoscon.

m4k2k, 2023-02-08:
Hello @Smanar,
thanks for the Feedback. Meanwhile I fixed the homeassistant part:
[image]
I'm checking now if there is some other data to grab on the device, then I prepare the PR for homeassistant.
If there is no new information to grab on the device I'll let you know - maybe you can do the PR for deconz? (with the DDF and your rest-api fix #6673)
best regards
Markus

Smanar, 2023-02-09:
>maybe you can do the PR for deconz?

Sure no problem, but I can explain you how to do if you prefer have them under your name ?

m4k2k, 2023-02-09:
@Smanar I'm ok if you do the PR with your name. I researched but didn't find any new information to get from the device (at least for now). I also double checked the units of the sensor values returned - I still have some doubt if they are correct. I guess we only find out by longterm usage; therefore I suggest to do the PR.
I still have to investigate which tests are required by HA to get the PR merged on the HA side, I'll start working on it this/next week.

Smanar, 2023-02-10:
PR done https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6739

m4k2k, 2023-02-10:
pydeconz PR in preparation: https://github.com/Kane610/deconz/pull/408

m4k2k, 2023-02-10:
homeassistant PR in preparation: https://github.com/home-assistant/core/pull/87866

Smanar, 2023-02-11:
Lol, was fast, all PR was still in waiting list here.

m4k2k, 2023-02-12:
pydeconz PR is now pending for review: https://github.com/Kane610/deconz/pull/408

ghenzler, 2023-03-03:
I'm waiting for this PR to be merged 🤞

rvplan, 2023-03-20:
> I think it can work, for the moment you can't write tuya request (it's WIP) but you can read them so no problem for temperature and humidity (haven't checked if all gaz sensor are available)
> 
> I just need a squeleton, I can make the tuya stuff part.

Hi everybody,
Has there been any progress with the integration of this device since one year?
Original post started with this reference: 
_TZE200_dwcarsat TS0601 Smart Air House keeper
I got the exact same reference, being well recognized in HomeAssistant + Defconz.
**But** as mentioned in this exchange, and [here (french)](https://community.jeedom.com/t/capteur-6-en-1-ts0601-tze200-dwcarsat-donnees-manquantes/85030/2
):
**Sensor for PPM 2.5 is STILL missing...**
Difference between WiFi and Zigbee show that there is indeed a missing parameter in the Zigbee version, but is present on Tuya app + Tuya hub.

Also this sensor is spamming HomeAssistant statistic database by sending values each second!
Is there a way to define inside zigpy some kind of parameter which would define the measurement rate?
So we can set in the UI that we want measure rates every n second, or n minutes, or n hours, etc.

Cheers

Smanar, 2023-03-20:
The device was included before the field was added in deconz, for the moment we need this PR https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6673

rvplan, 2023-03-20:
> The device was included before the field was added in deconz, for the moment we need this PR #6673 And some changes after (on DDF and on plugin side)

Thanks for your reply.
I take a look of your link, but I don't really understand the issues raised in this post, and my issue of lack of specific vocabulary around zigbee devices...

But I keep seeing in many posts, that through a Tuya HUB, the devices behave well and show correctly their parameters and the right scales; while through ZHA and HA there are many things missing, or mis-presented (scales of value).
This is not a criticism per se, it's just that I don't understand why a device definition appears one way in one environment, and differently in another.
Knowing that it is not the manufacturer who hides this or that parameter. It just "seems" to be a problem of interpretation of the reported data.
I tried to understand the logic of a quirks in HA, but with each explanation or tutorial, it seems more nebulous!
While every user would be able to contribute rather easily (I think) with a few well presented notions, and a "where to start for the dummies"! ;-)

As owner of one those devices (_TZE200_dwcarsat TS0601 Smart Air House keeper) and managing this through HomeAssistant + Conbee2, could I do something to help?
I clearly don't understand (yet) all the logic and layers between zigbee, zigpy, zha, quirks, HA UI, etc.
But if there is a quick way to extract some data, please let me know.

Cheers

AnnoyingTechnology, 2024-10-01:
As a disclaimer, I just received this device with a manufacturer id `_TZE204_dwcarsat`, which is not matched by [the latest DDF](https://raw.githubusercontent.com/dresden-elektronik/deconz-rest-plugin/refs/heads/master/devices/tuya/_TZE200_dwcarsat_air_sensor.json). 

Causing it to be miscategorized as a smartplug, just as the original author reported.

You may have to customize the DDF's manufacturer id, as I did, to get it working.

Smanar, 2024-10-02:
Yes, they are probably not managed by the HA plugin.
But IDK if it's from the DDF or the plugin.

Perhaps @Kane610 have an idea ?

Kane610, 2024-10-02:
You're running an out of date version of home assistant. Upgrade to latest and it will work

The log shows that you're running a newer version of deconz as it reports ZHAFormaldehyde `2024-10-01 20:02:13.437 WARNING (MainThread) [pydeconz.models] Unsupported device type ZHAFormaldehyde` which was only added support for a few releases ago in home assistant.

AnnoyingTechnology, 2024-10-02:
Interesting, did think 2024.4.3 was so old. Will update.

Kane610, 2024-10-02:
When failing bug reports its always good to make sure your running the latest release

AnnoyingTechnology, 2024-10-02:
Yeah, I'm quite sorry because I updated the deconz container especially for this issue, checked HA's version and didn't bother updating since it was less than 6 months old.

Kane610, 2024-10-02:
> checked HA's version and didn't bother updating since it was less than 6 months old.

Haha thats an eternity in hass time :)