dresden-elektronik/deconz-rest-plugin #5891: Hue gradient lightstrip

ebaauw, 2022-03-19:
## Device
- Product name: Hue gradient lightstrip
- Manufacturer: Signify Netherlands B.V.
- Model identifier: LCX004
- Device type : Extended Color Light

deCONZ already supports this lightstrip as a regular single-colour lightstrip.  The request is, of course, for support of the gradients.  In the Hue app, the light strip show three linked colour points, allowing for three areas with different colours, smoothly transitioning into each other.  These gradients are supported only through the Hue API v2.

## Screenshots

### Basic
Very similar to the Hue outdoor lightstrip and to the modules in the Hue Ensis, except for the 0xFC03 cluster.

[image]

### Identify
Same as other Hue lights.

### Groups
Same as other Hue lights.

### Scenes
Same as other Hue lights.

### On/Off
Same as other Hue lights.

### Level Control
Same as other Hue lights.

### Color Control
Same as other Hue lights: colour temperature between 153 and 500; colours from gamut-C.

### OTAU
Image type 0x0118; I managed to capture the location of the firmaware file and updated https://github.com/dresden-elektronik/deconz-ota-plugin/blob/master/README.md.

### FC01
This cluster is present on all Hue colour lights.  I don't know what it's for exactly; might be related to Hue Entertainment.

### FC03
The Hue bridge uses this cluster to set the gradient colours.  It polls attribute 0x0002, which is an octec string, and it sends command 0x00 when changing the gradient from the Hue app.  The gradients are supported only by the Hue API v2.

This cluster is also present on the gamut-C E27 bulb, but not on the gamut-A and gamut-B bulbs and spots, and not on the regular lightstrips.

ebaauw, 2022-03-19:
The Hue app shows special animated scenes, _Candle_ and _Fireplace_.  Looks like the animation is handled by the light strip firmware.  When selecting _Candle_, the bridge sends command 0x00 to 0xFC03 with payload 0x200001.  For _Fireplace_, it sends 0x200002.  When stopping the animation from the app, payload 0x200000 is sent.

payload | meaning
-- | --
0x200000 | stop animation
0x200001 | _Candle_ animation
0x200001 | _Fireplace_ animation

Attribute 0x0002 seems to encode the light state, where the first two bytes indicate the mode, the next byte _OnOff_, the next byte _Current Level_, and the following bytes the mode-specific state.  For example:
```
0b:00:01:fe:05:b1:ec:4e
|     |  |  |     + Current Y: 0x4eec = 0.3083
|     |  |  + Current X: 0xb105 = 0.6915
|     |  + Level: 254
|     + OnOff: on
+ mode 0x000b:`xy`
```
This is the red-most colour in gamut-C.  The attribute reflects this after setting `{"xy": [1, 0.0001]}` through the APIv1, as well as after setting `{"hue": 0, "sat": 254}`.

After setting `{"ct": 254}`, 0x0002 reports:
```
0f:00:01:fe:fe:00:4c:62:84:61
|     |  |  |     |     + Current Y: 0x6184 = 0.3809
|     |  |  |     + Current X: 0x624c = 0.384
|     |  |  + CT: 254
|     |  + Level: 254
|     + OnOff: on
+ mode 0x000b = `ct`
```
The _Current X_ and _Current Y_ values match what APIv1 reports for `xy` while in `colormode` `ct`.

After setting _Candle_:
```
ab:00:01:54:1c:7a:e8:69:01:80
|     |  |  |     |     + Scene 0x8001 = Candle
|     |  |  |     + Current y: 0x69e8 = 0.4137
|     |  |  + Current X: 0x7a1c = 0.477
|     |  + Level: 0x54 = 33%
|     + OnOff: on
+ mode 0x00ab = Scene
```
And after setting _Fireplace_:
```
ab:00:01:fe:98:8c:7f:68:02:80
|     |  |  |     |     + Scene 0x8002 = Fireplace
|     |  |  |     + Current y: 0x687f = 0.4082
|     |  |  + Current X: 0x8c98 = 0.5492
|     |  + Level: 0xfe = 100%
|     + OnOff: on
+ mode 0x00ab = Scene
```
After setting `{"effect": "colorloop"}`, 0x0002 reports the current colour in mode 0x000b.

ebaauw, 2022-03-19:
Setting the gradient seems straightforward enough.  It's command 0x00 with a payload:
- `50 01 04 00`
- 1 byte is length of variable part = 4 + 3 * #entries 
- Next byte has #entries << 4
- 3 bytes `00 00 00`;
- Next is 3 bytes per entry with x, y values normalised to 000..ffff.  Bytes need to be read backwards (as u24) and then split in high byte for x and low byte for y.  So [0, 1] is ff0f00 => 0x000fff => [0x000, 0xfff]
- Next byte seems related to #entries shifted `10` = 2; `18` = 3; `20` = 4; `28` = 5
- Next byte `00`.

Attribute 0x0002 seems to follow a similar pattern:
- `4b01` for mode
- 1 byte for _OnOff_
- 1 byte for _Current Level_
- 2 bytes for _Current X_
- 2 bytes for _Current Y_
- 1 byte length
- 1 byte #entries << 4
- 3 bytes `00 00 00`
- 3 bytes per entry
- 1 byte #entries << 3

However, I have no clue yet how the entries are encoded.  `000000` corresponds to `57730e`; `ffffff` to `11ea91`

While the Hue app only visualises 3 points, it actually sends 5.  I suppose the lines connecting the points are significant as well, relating to the intermediary points.

manup, 2022-05-04:
Small update, the Hue gradient lightstrip is ordered so should be ready for integration soon.

Gloomyeye, 2022-05-24:
Do you have something new to the hue gradient integration with more than one light?

nxtshot, 2022-06-14:
Would love to see support for Hue Gradient Signe as well. I guess it's based on the same function. If someone need details from that lamp, please ping me :)

ebaauw, 2022-07-28:
Ping

theCheek, 2022-08-20:
> Would love to see support for Hue Gradient Signe as well. I guess it's based on the same function. If someone need details from that lamp, please ping me :)

I too have the Gradient Signe, so I'm guessing each zone is not available for control via deconz yet correct? If you need anything from me to help get it working happy to help!

ebaauw, 2022-08-20:
It would help to see how the Hue bridge exposes the Signe over API v2.  Also, please attach GUI screenshots of the node with endpoints and clusters, and of the _Basic_ (see my initial comment), and check what the _Image Type_ of its firmware is (in the OTAU panel).

theCheek, 2022-08-24:
Sorry for the delay on this. 

> It would help to see how the Hue bridge exposes the Signe over API v2. 

Sure things, but how can I do this?

> Also, please attach GUI screenshots of the node with endpoints and clusters 
[image]

> and of the _Basic_ (see my initial comment), 
[image]

> and check what the _Image Type_ of its firmware is (in the OTAU panel).
[image]

Hope this helps, please let me know anything else I can provide!

ebaauw, 2022-08-24:
Please read the attributes before taking the screenshot, so the attribute values are populated.  I think there’s a bug in the GUI, that the attribute values of the _OTAU_ cluster aren’t updated, until you select another cluster and then re-select the _OTAU_ cluster.  Alternatively, check the _OTAU_ panel instead.

theCheek, 2022-08-24:
Apologies but I can't seem to get the attributes to populate to anything different than what I posted above. 
I've tried pressing the 'read' button that is above cluster info attributes:
[image]
but as you can see nothing seems to happen even when selecting a different cluster. 

The OTAU panel is showing this:
[image]

ebaauw, 2022-08-24:
> The OTAU panel is showing this:

Cool.

What about the _Basic_ cluster?  We would need at least the _Model Identifier_.

theCheek, 2022-08-24:
I think I managed it:
[image]

Gloomyeye, 2022-08-28:
Maybe this could help to fully integrate the hue gradient lightstrip: https://github.com/Koenkk/zigbee2mqtt/issues/10674

ebaauw, 2022-08-28:
From what I read, they failed to reverse engineer the encoding of the gradient colour points just as badly?

Gloomyeye, 2022-08-29:
Yes, I think so, too. However, they captured few scenes. Maybe this could give a hint in the right direction…

ebaauw, 2022-09-20:
Ping

ebaauw, 2022-10-13:
Ping

theCheek, 2022-10-13:
what can we do to get this fixed? Happy to do anything necessary!

ebaauw, 2022-10-13:
Explain to me how the colour for the gradients is encoded ;-)

ebaauw, 2022-10-16:
So I'm pretty certain that each gradient point is encoded in 3 bytes, or 24 bits.  The Hue API reports each point with `xy` values, but I don't understand how these translate to the 3-byte value.

So far I've tested the following hypothesis, none of which seemed to hold:
- Value holds _Current X_ and _Current Y_ as two 12-bit unsigned integers;
- Value holds fixed point `x` and `y` values as two 12-bit unsigned integers;
- Value holds RGB values in three 8-bit unsigned integers;
- Value holds _Enhanced Hue_ and _Saturation_ as a 16-bit and an 8-bit unsigned integer.

ebaauw, 2022-10-17:
I think I finally cracked it, with help from @manup.

The 3-byte value holds indeed _Current X_ and _Current Y_ as two 12-bit unsigned integers, but the scale differs between _Current X_ and _Current Y_.  X values from 0.7347 onwards and Y values from 0.8431 onwards are encoded as 0xFFF.  So the encoded value for X between 0.0000 and 0.7347 is `X * 0xFFF / 0.7347`.  For Y between 0.0000 and 0.8431 it is `Y * 0xFFF / 0.8431`.

Note that the light strip reports the actual colours displayed, rather than the colours set, so these differ for colours outside gamut C.

ebaauw, 2022-10-20:
Got it working, but still need to do a sanity check on the API input setting the gradient.
```
$ ph put /lights/5/state '{"gradient": {"points": [[0, 0], [1, 0], [0, 1], [1, 0], [0, 0]], "segments": 5, "offset": 0}}'
{
  "gradient": {
    "offset": 0,
    "points": [
      [
        0,
        0
      ],
      [
        1,
        0
      ],
      [
        0,
        1
      ],
      [
        1,
        0
      ],
      [
        0,
        0
      ]
    ],
    "segments": 5
  }
}
```
```
$ ph get /lights/5
{
  "colorcapabilities": 31,
  "colorgamuttype": "C",
  "ctmax": 500,
  "ctmin": 153,
  "etag": "e3eefaa51583ed27c7539173bd3c2d99",
  "hascolor": true,
  "lastannounced": null,
  "lastseen": "2022-10-20T23:05Z",
  "manufacturername": "Signify Netherlands B.V.",
  "modelid": "LCX004",
  "name": "Extended color light 5",
  "startup": {
    "bri": 254,
    "ct": 366,
    "on": true,
    "xy": "previous"
  },
  "state": {
    "alert": "none",
    "bri": 254,
    "colormode": "gr",
    "ct": 450,
    "effect": "none",
    "gradient": {
      "offset": 0,
      "points": [
        [
          0.2963,
          0.0011
        ],
        [
          0.7344,
          0.0466
        ],
        [
          0.2073,
          0.2658
        ],
        [
          0.7344,
          0.6988
        ],
        [
          0.2963,
          0.2646
        ]
      ],
      "segments": 5
    },
    "on": true,
    "reachable": true,
    "xy": [
      0.1532,
      0.0476
    ]
  },
  "swversion": "0x01001204",
  "type": "Extended color light",
  "uniqueid": "00:17:88:01:0b:90:7d:12-0b"
}
```

ebaauw, 2022-10-21:
I'm becoming obsessed by this device.

I noticed the strip includes `uint8` attributes 0x0030, 0x0032, 0x0033, 0x0034, 0x0035, and 0x0036 in the _Report Attributes_ message.   The non-gradient lights don't seem to carry these.

Doing some more experiments:

Attribute | Type | RWE  | Value | Meaning
-- | -- | -- | -- | --
0x0030 | u8 | re | 16 | Seems to match `pixel_count` in the Hue API v2.
0x0031 | u16 | r | 1250 | Assumption: pixel length in mm - my strip is 2m.
0x0032 | u8 | rwe | 0 | ?
0x0033 | u8 | rwe | 0 | ?
0x0034 | u8 | rwe | 1 | ?
0x0035 | u8 | re | 100 | ?
0x0036 | u8 | re | 72 | ?

ebaauw, 2022-10-22:
I think the PR is good to go.  See the PR for an overview of the new attributes.
```
$ ph get /lights/5
{
  "capabilities": {
    "color": {
      "gamut": {
        "blue": [
          0.1532,
          0.0475
        ],
        "green": [
          0.17,
          0.7
        ],
        "red": [
          0.6915,
          0.3083
        ]
      },
      "gamut_type": "C",
      "modes": [
        "ct",
        "effect",
        "gradient",
        "hs",
        "xy"
      ]
    },
    "ct": {
      "max": 500,
      "min": 153
    },
    "effects": [
      "none",
      "colorloop",
      "candle",
      "fireplace",
      "loop"
    ],
    "gradient": {
      "max_segments": 9,
      "pixel_count": 16,
      "pixel_length": 1250
    }
  },
  "colorcapabilities": 31,
  "config": {
    "startup": {
      "bri": 254,
      "ct": 366,
      "on": true,
      "xy": "previous"
    }
  },
  "ctmax": 500,
  "ctmin": 153,
  "etag": "e5c17f94783a9254bfa463b8cbfb8261",
  "hascolor": true,
  "lastannounced": null,
  "lastseen": "2022-10-22T11:54Z",
  "manufacturername": "Signify Netherlands B.V.",
  "modelid": "LCX004",
  "name": "Extended color light 5",
  "state": {
    "alert": "none",
    "bri": 254,
    "colormode": "gradient",
    "ct": 366,
    "effect": "none",
    "gradient": {
      "color_adjustment": 0,
      "offset": 0,
      "offset_adjustment": 0,
      "points": [
        [
          0.6056,
          0.2667
        ],
        [
          0.1533,
          0.0476
        ]
      ],
      "segments": 2
    },
    "hue": 62592,
    "on": true,
    "reachable": true,
    "sat": 254,
    "xy": [
      0.6055,
      0.2667
    ]
  },
  "swversion": "1.97.3",
  "type": "Extended color light",
  "uniqueid": "00:17:88:01:0b:90:7d:12-0b"
}
```

Set set a gradient:
```
$ ph put /lights/5/state '{"gradient": {"points": [[1, 0], [0, 0]]}}'
{
  "gradient": {
    "color_adjustment": 0,
    "offset": 0,
    "offset_adjustment": 0,
    "points": [
      [
        0.7347,
        0
      ],
      [
        0,
        0
      ]
    ],
    "segments": 2
  }
}
```
You see that the API has provided the default values and clipped the `xy` values.

The light will report the actual colours:
```
$ ph get /lights/5/state
{
  "alert": "none",
  "bri": 254,
  "colormode": "gradient",
  "ct": 366,
  "effect": "none",
  "gradient": {
    "color_adjustment": 0,
    "offset": 0,
    "offset_adjustment": 0,
    "points": [
      [
        0.6056,
        0.2667
      ],
      [
        0.1533,
        0.0476
      ]
    ],
    "segments": 2
  },
  "hue": 62592,
  "on": true,
  "reachable": true,
  "sat": 254,
  "xy": [
    0.6055,
    0.2667
  ]
}
```

nxtshot, 2022-10-22:
@ebaauw 
awesome work, thanks a lot! Do you know if this will work for the Hue gradient signe as well?

ebaauw, 2022-10-22:
Yes, it should.  @Thomas-Vos provided me with the model IDs, in https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6316#issuecomment-1286269191.  If you have another, please report it.  It should also work with the gradient tube light, as soon as we'll have identified and whitelisted its model ID.

I would interested to know what (gradient) capabilities the Signe exposes.  I think there might be a difference between the floor and table models.  Thomas mentioned on Discord that his floor model reports `pixel_count` as 10 (on the Hue v2 API) and, experimentally, supports up to 9 segments.  His light strip with three extensions reports a `pixel_count` of 40.  Hoping to receive the extensions I ordered today to do some testing myself.

nxtshot, 2022-10-22:
I have a floor model (915005987201), so I guess there is nothing I can provide which you don't know yet, right? I have a hue bridge somewhere around, wouldn't be much of effort to reactivate it.

ebaauw, 2022-10-22:
@nxtshot I would appreciate the values of the 0x0030 - 0x0036 attributes in the FC03 cluster.  These will also be reported under the light capabilities once my PR will be merged.

macmanroy, 2022-10-26:
> Yes, it should. @Thomas-Vos provided me with the model IDs of the SIgne and PLAY gradient lightstrips, in [#6316 (comment)](https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6316#issuecomment-1286269191). If you have another, please report it. It should also work with the gradient tube light, as soon as we'll have identified and whitelisted its model ID.
> 
> I would interested to know what (gradient) capabilities the Signe exposes. I think there might be a difference between the floor and table models. Thomas mentioned on Discord that his floor model reports `pixel_count` as 10 (on the Hue v2 API) and, experimentally, supports up to 9 segments. His light strip with three extensions reports a `pixel_count` of 40. Hoping to receive the extensions I ordered today to do some testing myself.

I have the signe table lamp, it reports a points_capable value of 5 and a pixel_count of 4

ebaauw, 2022-11-19:
Here's the gradient light strip with the latest commits:
```
{
  "capabilities": {
    "alerts": [
      "none",
      "select",
      "lselect",
      "blink",
      "breathe",
      "okay",
      "channelchange",
      "finish",
      "stop"
    ],
    "bri": {
      "min_dim_level": 0.01
    },
    "color": {
      "ct": {
        "computes_xy": true,
        "max": 500,
        "min": 153
      },
      "effects": [
        "none",
        "colorloop",
        "candle",
        "fireplace",
        "loop",
        "sunrise"
      ],
      "gamut_type": "C",
      "gradient": {
        "max_segments": 17,
        "pixel_count": 32,
        "pixel_length": 1250,
        "styles": [
          "linear"
        ]
      },
      "modes": [
        "ct",
        "effect",
        "gradient",
        "hs",
        "xy"
      ],
      "xy": {
        "blue": [
          0.1532,
          0.0475
        ],
        "green": [
          0.17,
          0.7
        ],
        "red": [
          0.6915,
          0.3083
        ]
      }
    }
  },
  "colorcapabilities": 31,
  "config": {
    "bri": {
      "execute_if_off": true,
      "startup": "previous"
    },
    "color": {
      "ct": {
        "startup": "previous"
      },
      "execute_if_off": true,
      "gradient": {
        "reversed": false
      },
      "xy": {
        "startup": "previous"
      }
    },
    "groups": [
      "0",
      "256"
    ],
    "on": {
      "startup": false
    }
  },
  "ctmax": 500,
  "ctmin": 153,
  "etag": "2aa7dbe83e883ca78718abdf2ee64c92",
  "hascolor": true,
  "lastannounced": null,
  "lastseen": "2022-11-19T21:28Z",
  "manufacturername": "Signify Netherlands B.V.",
  "modelid": "LCX004",
  "name": "Hue Gradient",
  "productid": "Philips-LCX004-1-GALSECLv1",
  "state": {
    "alert": "none",
    "bri": 232,
    "colormode": "gradient",
    "ct": 366,
    "effect": "sunrise",
    "gradient": {
      "color_adjustment": 5,
      "offset": 30,
      "offset_adjustment": 0,
      "points": [
        [
          0.4572,
          0.41
        ],
        [
          0.4531,
          0.4089
        ],
        [
          0.4488,
          0.4079
        ],
        [
          0.4531,
          0.4089
        ],
        [
          0.4572,
          0.41
        ]
      ],
      "segments": 26,
      "style": "linear"
    },
    "hue": 8415,
    "on": false,
    "reachable": true,
    "sat": 140,
    "xy": [
      0.4571,
      0.4099
    ]
  },
  "swversion": "1.97.3",
  "type": "Extended color light",
  "uniqueid": "00:17:88:01:0b:90:7d:12-0b"
}
```

closed 2023-01-07 by manup

Gloomyeye, 2023-02-23:
Hello @ebaauw, I installed the newest stable version of deconz and I thought that I can switch on parts of my lightstrip via Home Assistant (2023.2). However, my installed Hue lightstrip LCX004 is only shown as one light entity in Home Assistant and I have no idea how I can use the gradient points. Is that possible or do I need to change something in deconz for that?