dresden-elektronik/deconz-rest-plugin #5897: Power measure with TH1124ZB

BabaIsYou, 2022-03-22:
## Describe the bug

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

## Environment

 - Host system: Raspberry Pi /
 - Running method: Raspbian
 - Firmware version:26660700
 - deCONZ version: 2.14.1
 - Device: ConBee II 
 - Do you use an USB extension cable:yes 
 - Is there any other USB or serial devices connected to the host system? yes (Dongle Z-Wave and RFXcomm)

## deCONZ Logs

## Additional context

Smanar, 2022-03-22:
But it s you on domoticz forum too ?

I can purpose a PR to correct that, but in the actual situation I think it will be better to try using the DDF ?
Do you know how it work ? https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet
As the device is already working the DDF can be almost complete (I think it will just miss the binding)

Mimiix, 2022-03-22:
Please use the respective device report to report these things. https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2104

Closing this one.

closed 2022-03-22 by Mimiix

BabaIsYou, 2022-03-23:
> But it s you on domoticz forum too ?
> 
> I think it s same on the node itself ? If you check on the cluster 0x0702 the value is 0 for the power ?
> 
> I can purpose a PR to correct that, but in the actual situation I think it will be better to try using the DDF. Do you know how it work ? https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet As the device is already working the DDF can be almost complete (I think it will just miss the binding)
> 
> If you can't finish it, just make a copy/paste here to have a base.

@Smanar As you can see the attribute are already defined the same manner in both cases...

[image]
[image]

BabaIsYou, 2022-03-23:
> Please use the respective device report to report these things. #2104
> 
> Closing this one.

@Mimiix As the other one is also closed, that's looks like a "catch 22" ;-)

Smanar, 2022-03-23:
Don't worry, he is able to re open the other issue. And he is right, better to have one issue by device.
I will answer on the https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2104

BabaIsYou, 2022-04-14:
> Don't worry, he is able to re open the other issue. And he is right, better to have one issue by device. I will answer on the #2104

@Mimiix . Well, both of the issues are still closed but not solved ?!

Mimiix, 2022-04-15:
I can't reopen the other one because it doesn't comply with the template. As that issue contains all relevant information it makes sense to keep all talk there.

Is the "status" of an issue important for you? I mean, Smanar is helping out there afaik.

BabaIsYou, 2022-04-15:
You sent me to the other thread and at this time there was no mention to missing information to be compliant to a template for an issue that was already closed before than the device was fully supported as initially requested.

BTW what is missing ?

Mimiix, 2022-04-15:
We use templates for issues. a bot checks if theres headers in the original post. If they don't exist, the issue gets auto closed. Even on re-opening.

As the issue is quiet old, the template wasnt used. Either way, I am not sure why you are so eager to have it opened... It really doesn't change anything at this moment 😅

BabaIsYou, 2022-04-15:
You’re right it changes nothing at this moment. I mean nothing at all … i plugged the device, got errouneous value, trying to have some help to understand and correct this fact and the only clear answer is « you don’t it, then delete it, not an issue, keep closed ».

My point is : if this sensor is useless, please correct it’s creation process for everybody using it. Whatever the way is to correct it. If it’a  thru DDF then just tell me what to modify then I can do a PR once it’s corrected.

Letting the issue closed send the message that there is no more issue. But there is.

Mimiix, 2022-04-15:
If it's that important to you, let me work around it.

It really makes not difference here at all in the end.

Mimiix, 2022-04-15:
Please continue in the other issue.

BabaIsYou, 2022-10-11:
Many thanks