dresden-elektronik/deconz-rest-plugin #5898: Unable to modifiy attribut Backlight in TH1124ZB

BabaIsYou, 2022-03-23:
## Describe the bug
Despite the fact that Occupancy and Backlight attributes are clearly documented in [general.xml](https://github.com/dresden-elektronik/deconz-rest-plugin/blob/5ce086600955a6bd696a7b74ae1ac1bcb2f0c767/general.xml). I can modifiy Occupancy using deCONZ Attribute Editor, but I can't modify Backlight.
`
		
		<attribute-set id="0x0400" description="Sinope specific" mfcode="0x119c">
			<attribute id="0x0400" name="Occupancy" type="enum8" access="rw" required="m" mfcode="0x119c">
				<value name="Unoccupied" value="0x00"></value>
				<value name="Occupied" value="0x01"></value>
			</attribute>
			<attribute id="0x0402" name="Backlight" type="enum8" access="rw" required="m" mfcode="0x119c">
				<value name="On demand" value="0x00"></value>
				<value name="Sensing" value="0x01"></value>
			</attribute>
		</attribute-set>`

## Steps to reproduce the behavior
When reading the attribute their is no error on ZCL debug
`09:11:39:916 ZCL got data for node=0x3466, cl=0x0201, at=0x0402, status=0x00, type=0x30` but then two options "On Demand" and "Sensing" are not displayed. Then of course a write without it generate an "invalid data" by Attribute Editor (see screenshot) but nothing in ZCL debug.
[image]

## Expected behavior
Expect to have the choice between to options, like it is the case with Occupancy attribute in the same cluster (see other screenshot)
[image]

## Screenshots

## Environment

 - Host system: Raspberry Pi 
 - Running method: Raspbian 
 - Firmware version: (26660700)
 - deCONZ version: (2.14.01)
 - Device: ConBee II 
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? RfxComm and Dongle Z-Wave

## deCONZ Logs

## Additional context

Smanar, 2022-03-23:
So, now I m sure, it s ' you on the DZ forum ^^

I have perhaps found the issue, the attribute 0x0402 is used 2 time on the same cluster, and the second brand don't use manufacture specific flag.
```

		<!-- ELKO specific -->
		<attribute-set id="0x0400" description="ELKO specific" mfcode="0x1002">
			<attribute id="0x0401" name="Unknown" type="u16" access="rw" required="m"></attribute>
			<attribute id="0x0402" name="Unknown name field" type="cstring" access="rw" required="m"></attribute>
```

BabaIsYou, 2022-03-23:
Weird, because in Oct 2020 the "ELKO manufacturer specific"  description seemed to have the mfc [](https://github.com/dresden-elektronik/deconz-rest-plugin/commit/bcad9a936dc35d8f7dc2db2e927902e4d1b30cd9).

closed 2022-04-22 by github-actions[bot]