dresden-elektronik/deconz-rest-plugin #5899: Device name

MiAutomations, 2022-03-24:
The device is now founded as a Smart plug but I don't see the status of the Sensor, can you please add this function also?

https://s.click.aliexpress.com/e/_A7We5d

- Product name: The device name as shown on the product or package.
- Manufacturer: _TZE200_nklqjk62 
- Model identifier: TS0601
- Device type : Remote opening garage door with sensor 
  - Other: 

## Screenshots

  Required screenshots:
  
[image]

 
[image]

Thank you

sebastian-tennant, 2022-04-05:
+1 on this request

Zigbee2MQTT already support it https://www.zigbee2mqtt.io/devices/PJ-ZGD01.html

serrnovik, 2023-02-14:
Have the same device. Same behaviour, button works, state is not working. Woundering if there were any progress on this one?

pdcmoreira, 2023-05-15:
Hey, what do you mean about state not working?
What happens when you trigger it and the sensor/magnet closes?
Does it change state immediately to "on" or "off"? Or it doesn't change state at all?
It would make sense to me that it would change state only when the gate closes, not immediately.

pdcmoreira, 2023-05-23:
So I bought two of these, which arrived today. I now see what you mean.

It is reported as a switch or light, but its state has nothing to do with the open/close sensor. That sensor state is not being displayed.

Additionally, the switch state doesn't represent the dry contact signal state and to send a signal (close the dry contact for 1s) it is needed to switch it on then off and only then the signal will be sent.

So, for now, to make it work in HA to send a signal, I created a script that turns the switch on, waits 500ms then turns it off and it works fine.

Still don't know how to get the state of the open/close sensor though.

[This issue](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5685) apparently has a DDF to load it, but I'm still trying to understand how the whole thing works.

Mimiix, 2023-05-24:
Re-opened on request.  @BabaIsYou  or @Smanar  you got time to check in :)?

Mimiix, 2023-05-24:
@pdcmoreira  or @MiAutomations  I do require to have all related screenshots attached.

pdcmoreira, 2023-05-24:
Screenshots:
[image]
[image]

pdcmoreira, 2023-05-24:
@Smanar 
I left a couple questions in my previous comment, I don't know if it's possible to deal with that through a DDF:
> Now, is it possible to improve the signal trigger so that the script isn't needed?
Does the open/close sensor look OK? Or can it be improved?

Smanar, 2023-05-24:
Yes for me it's fine too, you can send "on" or "off" request, so no problem with trigger.

closed 2023-06-22 by github-actions[bot]

pdcmoreira, 2023-11-04:
I bought another one which came with "_TZE204_nklqjk62" manufacturer code and I want to let you guys know that this exact same DDF still works on it.

lepeno82, 2024-02-05:
Hello
I would like to buy one of this equipment, but can you confirm , with the latest DDF, I can command the switch, and read the state of the sensor?
I would like to use it in jeedom with deconz plugin
Best regards

pdcmoreira, 2024-02-06:
> Hello I would like to buy one of this equipment, but can you confirm , with the latest DDF, I can command the switch, and read the state of the sensor? I would like to use it in jeedom with deconz plugin Best regards

Yes, it works, my explanation above is still true today.

lepeno82, 2024-02-06:
Ok thanks

hem0773, 2025-01-09:
> Ich habe ein anderes gekauft, das mit dem Herstellercode **_TZE204_nklqjk62** geliefert wurde , und ich möchte euch wissen lassen, dass genau dieses DDF darauf immer noch funktioniert. Ich habe im Grunde nur die Datei _TZE200_nklqjk62 geöffnet, den Herstellernamen geändert und dann Datei → Speichern unter mit dem Namen _TZE204_nklqjk62 ausgewählt. Dann Hot Reload (und tatsächlich alle Docker-Instanzen neu gestartet). Das Verhalten ist auch in HA dasselbe, also habe ich das Skript auch dupliziert.

> I bought another one which came with the **_TZE204_nklqjk62** manufacturer code and I want to let you guys know that this exact same DDF still works on it. I basically just opened the _TZE200_nklqjk62 file, changed the manufacturer name and then File → Save as with the _TZE204_nklqjk62 name. Then hot reload (and actually rebooted all docker instances). Behavior is also the same in HA, so I also duplicated the script.

can you upload the _TZE204_nklqjk62.ddf, I'm not sure what to change

Thanks