dresden-elektronik/deconz-rest-plugin #5905: Namron termostat touch ZigBee 16A - Get power consumption?

jarlesb, 2022-03-25:
[image]

## Device

- Product name: Namron termostat touch ZigBee 16A
- Manufacturer: Namron AS
- Model identifier: 4512737
- Device type : 
  - Thermostat

## Screenshots

### Basic
[image]

### Identify
[image]

### Alarms

### Device Temperature

### Groups
[image]

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering
[image]

### Diagnostics

### Electrical measurements
[image]

Is it possible to get kWh consumption on this devices? Is identified in Home Assistant as Thermostat with Climate entity. I am running Home assistant 2022.3.7 with Conbee 2 on Odroid N2+/Home Assistant Blue

closed 2022-04-23 by github-actions[bot]

jreidel, 2022-05-03:
@jarlesb did you solve the power consumption issue?

Smanar, 2022-05-03:
It's a thermostat ? with power comsumption ?

Can take this one as sample to make a DDF https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/sinope/th1124zb.json

jarlesb, 2022-05-04:
I do not get any power or consumption readings...

[image]

Just says unavailable...

Smanar, 2022-05-04:
But you are using a DDF ?
The device don't seem tobe n whitelist, so not logic you already have thoses sensor by defaut.
Can you show the sensors json ? visible in phoscon / help / API Information / sensors

Do you have access to the GUI ? If you read the value in deconz, it update the value in HA ?

jarlesb, 2022-05-04:
I have not worked with DDF... I am using Home Assistant with Conbee2. and just added via Phoscon.

I phoscon I see the Thermostat as a sensor
[image]

    "113": {
        "config": {
            "heatsetpoint": 1950,
            "offset": 0,
            "on": true,
            "reachable": true,
            "schedule": {},
            "schedule_on": false
        },
        "ep": 1,
        "etag": "xxx",
        "lastannounced": null,
        "lastseen": "2022-05-04T14:21Z",
        "manufacturername": "NAMRON AS",
        "modelid": "4512737",
        "name": "Outlet",
        "state": {
            "lastupdated": "2022-05-04T14:17:34.724",
            "on": false,
            "temperature": 2073
        },
        "swversion": "6.9.1.0_r4",
        "type": "ZHAThermostat",
        "uniqueid": "xxx"
    }
}

Smanar, 2022-05-04:
This one is a ZHAThermostat, you haven't a ZHAConsumption or ZHAPower ?

jarlesb, 2022-05-04:
Sorry, I missed those elements. Here they are:

    "111": {
        "config": {
            "on": true,
            "reachable": true
        },
        "ep": 1,
        "etag": "xxx",
        "lastannounced": null,
        "lastseen": "2022-05-04T13:22Z",
        "manufacturername": "NAMRON AS",
        "modelid": "4512737",
        "name": "Consumption 111",
        "state": {
            "consumption": 1189,
            "lastupdated": "2022-05-04T19:25:55.884",
            "power": 0
        },
        "swversion": "6.9.1.0_r4",
        "type": "ZHAConsumption",
        "uniqueid": "xxx"
    },
    "112": {
        "config": {
            "on": true,
            "reachable": true
        },
        "ep": 1,
        "etag": "xxx",
        "lastannounced": null,
        "lastseen": "2022-05-04T13:22Z",
        "manufacturername": "NAMRON AS",
        "modelid": "4512737",
        "name": "Power 112",
        "state": {
            "current": 0,
            "lastupdated": "2022-05-04T19:28:55.355",
            "power": 0,
            "voltage": 0
        },
        "swversion": "6.9.1.0_r4",
        "type": "ZHAPower",
        "uniqueid": "xxx"
    },
    "113": {
        "config": {
            "heatsetpoint": 2150,
            "offset": 0,
            "on": true,
            "reachable": true,
            "schedule": {},
            "schedule_on": false
        },
        "ep": 1,
        "etag": "xxx",
        "lastannounced": null,
        "lastseen": "2022-05-04T19:29Z",
        "manufacturername": "NAMRON AS",
        "modelid": "4512737",
        "name": "Outlet",
        "state": {
            "lastupdated": "2022-05-04T19:29:18.956",
            "on": false,
            "temperature": 2134
        },
        "swversion": "6.9.1.0_r4",
        "type": "ZHAThermostat",
        "uniqueid": "xxx"
    }

Smanar, 2022-05-04:
It seem the ZHAConsumption is woking.
The ZHAPower have 0 values, you have consumption at this moment ? You haven't the GUI to compare ?

aspic, 2022-05-05:
I have very similar setup (Namron 4512737, RaspBee 2 and Home Assistant) and do get power consumption:
[image]

Smanar, 2022-05-05:
And only consumption is working ? You have 0 value like him.

sliann, 2022-05-05:
I'm using this DDF with a Conbee II, Deconz and Home Assistant.

```
,
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        },
        {
          "at": "0x0400",
          "dt": "0x2A",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0B04",
      "report": [
        {
          "at": "0x050B",
          "dt": "0x29",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        },
        {
          "at": "0x0505",
          "dt": "0x21",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        },
```

sliann, 2022-05-05:
I have a DDF for this. Tried to post it but the clip board on my phone cut some of the text. Will try again later tonight.

sliann, 2022-05-05:
> I have very similar setup (Namron 4512737, RaspBee 2 and Home Assistant) and do get power consumption:
> 
> [image]

Do you get temperature readings from the floor sensor? I get it myself, but it's reporting the air sensor at varying points so the graph is very confusing..
[image]

aspic, 2022-05-05:
> And only consumption is working ? You have 0 value like him for others

This was a bad snapshot, it was idling at the time. Yeah, consumption working :-)

@sliann: No, I only get temperature from the room sensor (at least that's what I thought). Would defenitely love floor sensor as well. Here's a more historic graph:

[image]

jreidel, 2022-05-05:
> I have very similar setup (Namron 4512737, RaspBee 2 and Home Assistant) and do get power consumption: 

How did you add it to Deconz? Via Phoscon (sensor)? I just got one Namron unit installed, but I'm not able to pair (RPi, Conbee2, Home Assistant). @aspic

aspic, 2022-05-05:
@jreidel On the namron device → go into menu → network → reset
Afterwards in Home Assistant → Add integration → Add zigbee device (with ZHA) → it should turn up in search

jreidel, 2022-05-05:
Ah, thanks. I'm currently not using ZHA, just Deconz. Will try some more with deconz, might have to create a DDF.

Smanar, 2022-05-06:
You can't include it at all ?
It need to work natively, without DDF you miss only the measurement.

Select "Add new light"on phoscon and make the reset procedure on the device.

jarlesb, 2022-05-06:
I just removed it and added again via Phoscon and now works with power and
consumption 👍

aspic, 2022-05-08:
Nice! Did you get floor temperatur working as well?

marhoy, 2022-09-13:
I'm also interested in learning how to change the temperature reading from the wall sensor to the floor sensor. I've just added it as a sensor in Phoscon, and I'm able to control it from HomeAssistant. But the reported temperature is the wall sensor.

Smanar, 2022-09-13:
This device have 2 temperatures ? can you give more information where to read them on deconz ?
Need the cluster / attribute.

The actual value is
```

          "description": "The current floor temperature",
          "read": {
            "at": "0x0001",
            "cl": "0x0201",
```

marhoy, 2022-09-13:
Not sure what you need, but there's quite a lot on Zigbee in the manual:

https://www.elektroimportoren.no/docs/lib/4512737-Brukerveiledning-5.pdf

sliann, 2022-09-13:
I have the same device. Please see attached screen shots.
Local temperature is the room temperature and outdoor is the floor temperature.

[image]

[image]

Smanar, 2022-09-13:
Ok so it s easy to add it, but there is a problem ^^
On the field list I have only state/floortemperature available. We are making new one https://github.com/dresden-elektronik/deconz-rest-plugin-v2/issues/13 but they are not available yet.

What is your third app ? If you are using nodered, I can use this one as workaround, else better to wait for new fields.

sliann, 2022-09-14:
From what I can see a lot of things with this thermostat will be fix with the new fields 😊

It will probably fix the state/on problem as well. This is not available from the device, just a list of different states:
[image]

Not sure what you mean by third app?
I have fixed the sensor issue by changing my DDF file.

Smanar, 2022-09-14:
>Not sure what you mean by third app?

Phoscon/ OpenHab/ NodeRed, ect ...

>I have fixed the sensor issue by changing my DDF file.

So you have now the value in 0x0001 but not the one in 0x0000 ?

State/on is not by defaut on this device ? You have it in the DDF
```
        {
          "name": "state/on",
          "refresh.interval": 5
        },
```

The legacy code is able to handle it, perhaps you just miss the binding ?

jarlesb, 2022-10-11:
That is strange. I can try to remove and add the thermostat again and see
if it fixes the issue.

Which FW do you have?

jreidel, 2022-10-17:
I'm not sure deConz DDF settings for this termostat is correct out of the box - or mayby I just don't know how this thing is working. I have notised that voltage and power consumption seems to be wrong by default, and I have adjustet how i _think_ it should be.

Voltage - had to devide by 10:
[image]

Power consumption - had to multiply by 100:
[image]

It seems to be correct now, whey I inspect the data. Can anyone confirm their own settings?