dresden-elektronik/deconz-rest-plugin #5909: Blitzwolf SHP-15 Power and Consumption not updated in REST API after update to 2.15

LukasA83, 2022-03-27:
## Describe the bug
After upgrading to 2.15 my Blitzwolf SHP-15 are not reporting updates to the REST API. While, via VNC, I can retrieve the latest values, in REST API calls the sensor indicate lastupdate before upgrading to 2.15, around 10 days ago, see screenshots.

## Steps to reproduce the behavior
1. Call REST API to sensor, e.g. 92
2. Check last update
3. Open VNC
4. Navigate to corresponding cluster 
5. Click "Read" and see updated values
6. Re-run 1. & 2. and see no updates in REST API

## Expected behavior
Devices is reporting updates and manual update should be visible in REST API calls

## Screenshots
[image]
[image]
[image]

## Environment
 - Host system: Raspi 4b - Docker setup
 - Running method: Docker community deconz
 - Firmware version: 26720700
 - deCONZ version: 2.15.00
 - Device: RaspBee II

iisworks, 2022-04-12:
@Smanar I see these new Blitzwolf DDF files reference a Tuya file:
"script": "../tuya/tuya_swversion.js"

What happens if we create a custom DDF from a copy of this json in the Docker Mount (/docker/deconz/data/devices? This folder does not contain that file in the relative patch ? Will it automatically try to find the file in the installed deconz instance as well?

manup, 2022-04-13:
Hi note that the Blitzwolf SHP-15 has now a DDF in v2.15.1-beta https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5945

manup, 2022-04-22:
Closing for now, should be fixed with v2.15.1-beta.

closed 2022-04-22 by manup