dresden-elektronik/deconz-rest-plugin #5911: Water Leak Detecor

stappel, 2022-03-27:
## Device

- Product name: "ZigBee" Water Leak/Food Detector
- Manufacturer: _TZ3000_d16y6col
- Model identifier: TS0207
- Device type : 
  - Sensor
  - Other: Leak Detector

Item: https://www.aliexpress.com/item/4000785145353.html?gatewayAdapt=glo2nld&gatewayAdapt=glo2nld&spm=a2g0o.order_list.0.0.21ef79d2KEv077 

## Screenshots

### Basic
[image]

### Identify

[image]

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above
[image]
[image]

[image]

Smanar, 2022-03-27:
Hello, this device realy look standards.
Can try with DDF editor, or just with editing this one https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/bosch/bwa-1_water_sensor.json

It's almost the same, just need to edit 

```
  "manufacturername": "_TZ3000_d16y6col",
  "modelid": "TS0207",
  "vendor": "Tuya",
  "product": "Tuya water sensor",
```

stappel, 2022-03-29:
Confirmed. Added json file and It shows a water sensor.
[image]

Smanar, 2022-03-29:
If it work, can you make a PR to submit it ?
Or I can add it myself if needed.

Smanar, 2022-03-30:
np, just if you can confirm the DDF ?

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_d16y6col",
  "modelid": "TS0207",
  "vendor": "Tuya",
  "product": "Tuya water sensor",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "ZHAWater",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0402",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0500"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2"
          }
        },
        {
          "name": "config/enrolled"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/pending"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/lowbattery",
          "awake": true
        },
        {
          "name": "state/tampered",
          "awake": true
        },
        {
          "name": "state/water",
          "awake": true
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 60,
          "max": 3600,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0500"
    }
  ]
}
```

closed 2022-04-01 by manup