dresden-elektronik/deconz-rest-plugin #5917: Nous Smart ZigBee Socket A1Z

Brunsben, 2022-03-29:
## Device

- Product name: Smart ZigBee Socket
- Manufacturer: Nous
- Model identifier: A1Z
- Device type : Please remove all unrelated device types. 
  - Socket

[image]

## Screenshots
 
  
[image]

### Basic
[image]

### Identify
[image]

### On/Off
[image]
[image]

### Scenes
[image]

### Groups
[image]

### Electrical Measurement
[image]

### Simple Metering
[image]
[image]

### OTAU
[image]

JanSucco, 2022-04-10:
You can test the two files: https://github.com/JanSucco/various/tree/main/Deconz
Set status in deconz-gui to bronze.

Smanar, 2022-04-10:
Thx, and don't forget to change the status to "Gold" instead of "Bronze" to test files.

Weltraumpenner, 2022-04-11:
Those plugs looks promising, I wiil try them when they are official supported!

Smanar, 2022-04-12:
For that, need someone test the @JanSucco file, confirm they are working for someone make a PR.

Brunsben, 2022-04-12:
I'll try today and report if it works.

Smanar, 2022-04-12:
Ok so bad new, I m asking the same thing https://forum.phoscon.de/t/how-to-add-edit-a-ddf-on-ha-using-text-editor/1839

If you know a little docker https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5459#issuecomment-1018984004 But the last user I have give the link haven't succed.

Brunsben, 2022-04-12:
I have SSH-Access, but all the Folders seems to be ampty, maybe the root is not really a root user.

Brunsben, 2022-04-12:
No, empty folder:
[image]

Smanar, 2022-04-13:
and inside the folder data or addons ?

Brunsben, 2022-04-13:
Data:
[image]

Addons:
[image]

And in /usr/share is no folder hassio:
[image]

Smanar, 2022-04-28:
>I'm running on a RaspberryPI, HomeAssistant in docker, deconz native install via apt or something.

It mean deconz is not on docker ? 

>Saw the plug in the GUI and checked that Bronze is checked (was already enabled?) Had enabled this before on some Aqara sensor, is this setting shared or something?

You are speaking about the setting in the tab "control" ? This is other thing, here you set the minium level autorised by DDF, But you have ronze checked here ?

You don't need to restart deconz, on the DDF editor, on the menu you have "Hot reload"
On the GUI
- select the device
- right clic "edit DDF"

You will see on the editor title the path and the file. Can test changes direclty.

Plug (light devices) don't have consumption, if all is working you will have a new sensor for that.

On the DDF you have downloaded, line 8, edit to put "Gold"
Else the DDF realy look good.

You can see direclty the API in Phoscon / help / API Information.

Brandt1930, 2022-04-28:
Hi,

I have the same situation as @codeaphex above. When I look at the websocket events in phoscon app, I can also see the changing values of "current" etc..
```json
{
    "22:34:26:777": {
        "e": "changed",
        "id": "2",
        "r": "lights",
        "state": {
            "consumption": 142,
            "current": 467,
            "on": true,
            "power": 0,
            "reachable": true,
            "voltage": 236
        },
        "t": "event",
        "uniqueid": "a4:.."
    },
   ```
But I dont have any sensors with this data in the REST-API.

> Plug (light devices) don't have consumption, if all is working you will have a new sensor for that.

@Smanar You mean after adding the light will have a Sensor as well, or should I be able to find the plug in the Phoscon App by searching for Sensors as well?

codeaphex, 2022-04-28:
> > I'm running on a RaspberryPI, HomeAssistant in docker, deconz native install via apt or something.
> 
> It mean deconz is not on docker ?
Yes, deconz is not deployed via docker.

> > Saw the plug in the GUI and checked that Bronze is checked (was already enabled?) Had enabled this before on some Aqara sensor, is this setting shared or something?
> 
> You are speaking about the setting in the tab "control" ? This is other thing, here you set the minium level autorised by DDF, But you have ronze checked here ? By defaut it s "Gold"
> On the DDF you have downloaded, line 8, edit to put "Gold" (even you don't need it if you have checked "Bronze", but like that I m sure it can't be an issue ^^) Else the DDF realy look good.
> 
Yes was bronze, I think it was read directly from the config. I now changed it back to gold and its still running.

> You don't need to restart deconz, on the DDF editor, on the menu you have "Hot reload" On the GUI
> 
> * select the device
> * right clic "edit DDF"
> 
> You will see on the editor title the path and the file, and can test changes direclty before switching to headless mode.
> You can see direclty the API in Phoscon / help / API Information.
Oh thats both pretty nice 😄  

> Plug (light devices) don't have consumption, if all is working you will have a new sensor for that.
Thats the last thing missing, same output as @Brandt1930. What do we need to do, to get the sensor and events recognized by Phoscon and HomeAssistant.

codeaphex, 2022-04-28:
Just compared the config with a similar plug from Innr: <https://github.com/dresden-elektronik/deconz-rest-plugin/blob/1d7f27ea34f3470998998fff8c28a2319817ce1e/devices/innr/sp_120.json>  
Looks like we need to add the configuration for the sensor?
Dont know if I could figure it out with some fiddling around (or if it will blow up that way lol), but it would probably take longer for me, than someone who had worked with it before.
Anyone who is up for that?

eimann, 2022-04-29:
Thanks for providing the json @JanSucco. I successfully imported the DDF and paired the Nous A1Z. Except the metering it works fine. Let me know if you need anything from me.

[image]

Brandt1930, 2022-04-29:
@Smanar Thank you. Now its working and I have sensor data
```json
                    
    "2": {
        "config": {
            "on": true,
            "reachable": true
        },
        "etag": null,
        "lastannounced": null,
        "lastseen": "2022-04-29T20:28Z",
        "manufacturername": "_TZ3000_ksw8qtmt",
        "modelid": "TS011F",
        "name": "Consumption 2",
        "state": {
            "consumption": 0,
            "lastupdated": "2022-04-29T20:28:05.773"
        },
        "type": "ZHAConsumption",
        "uniqueid": "a4:c1:38:7....6-01-0702"
    },
    "3": {
        "config": {
            "on": true,
            "reachable": true
        },
        "etag": null,
        "lastannounced": null,
        "lastseen": "2022-04-29T20:28Z",
        "manufacturername": "_TZ3000_ksw8qtmt",
        "modelid": "TS011F",
        "name": "Power 3",
        "state": {
            "current": 0,
            "lastupdated": "2022-04-29T18:25:20.758",
            "power": 0,
            "voltage": 235
        },
        "type": "ZHAPower",
        "uniqueid": "a4:c1:38:7...:e6-01-0b04"
    },
```
Am I able to set the name or do I have to find the connection to the light(for a user set name in phsocon) via the uniqueid?

Smanar, 2022-04-29:
I m not sure to understand ? What do you want to do ?
You can change the name ofc.

Brandt1930, 2022-04-29:
Lets say I named the plug "TV". Now in my Lights json I can find the name "TV". But in my Sensor JSON I can only find some generic name like Power 3 or Consumption 2. 
The only mapping between lights and sensors I see which should be possible is my cutting of some of the unique id of the sensor and searching for this on the lights json. There I can get my name TV again and can say TV is consuming 50 W or something. Or is there another way? Can I change the name of the sensors?

eimann, 2022-04-29:
@Brandt1930 Just change the name of the entity under Configuration > Devices & Services > Entities

Smanar, 2022-04-29:
Ha yes, all is separed, you can use a different name for the power measurement and the consumption measurement.
But you can rename them to use the same too.

For the procedue it depend of third app, as phosocon dont' support all sensors.

codeaphex, 2022-04-30:
@Smanar Thx for the config, finally  a zigbee smart plug with working measurements that is also available in stores/online.

repiuk, 2022-05-01:
I'm running deconz from homeassistant os in a docker. I can connect with bash from portainer. However nano, vi, ftp etc are not installed. Opening nano and copy paste the json file doesn't work... How do I transfer the .json to the docker container?

Smanar, 2022-05-01:
Ha nice I don't know this method https://forum.phoscon.de/t/how-to-add-edit-a-ddf-on-home-assistant-using-text-editor/1839/2
If you can explain better the command line to use pls ?

repiuk, 2022-05-01:
- Type cat > filename.ext from the console
- The console waits for input from the keyboard. Instead paste the clipboard (the console from portainer supports right click and paste)
- stop the console input with ctrl-c

I managed to place the file (nous_a1z.json)  in the devices folder. I can open the file in DDF editor and hot load it. I don't see how that changes anything in phoscon/home assistant. tried removing the device in deconz(VNC) and adding a new switch / light. It loads the old DDF.
Must have missed something obvious...

Smanar, 2022-05-01:
Perhaps a conflit if there is 2 files.
You can see on logs or on editor title the file and the path of the used DDF.

But to use your commands you need first to travel in docker stuff up to the good directory ?  and some users don't find it.

repiuk, 2022-05-01:
I put the file in ~/.local/share/dresden-elektronik/deCONZ/devices
Or should it go in /usr/share/deCONZ/devices/

I'll try both once I have the other plugs. It's working now.

repiuk, 2022-05-03:
Received new plugs. They pair directly if nous_a1z.json is stored in
/data/.local/share/dresden-elektronik/deCONZ/devices
DeCONZ and home assistant immediately recognize the switch with power and consumption entities. Nice!

shaktimaan, 2022-05-04:
I managed to pair two plugs correctly, but the plugs switch themselves on after a couple of seconds!?
Removed the Integration in FHEM and Homassist - still the same, so it´s not an automation issue.

:(

Smanar, 2022-05-04:
Both of them ?

TheNice4GH, 2022-05-04:
Sorry for reporting this device again, thanks for pointing me here.
Right-clicked on the node, edit DDF and "save-as" to create a json file where appropriate, which I overwrote with the one provided by Smanar.
Rebooted my RPI3 (for some strange reason the webapp was not responding anymore)
Re-added the 2 "lights", it immediately showed the expected devices in Domoticz
[image]

Thanks for the support!

Smanar, 2022-05-06:
And you can see them in phoscon ? the reported state is the real one ?

shaktimaan, 2022-05-08:
> And you can see them in phoscon ? the reported state is the real one ?

yes, state under On/Off changes accordingly
I also returned the switches as defect and got a replacement which behaves exactly the same

TheNice4GH, 2022-05-08:
Don't know if this can be relevant but here is my Phoscon version (installed on RPI3 64bits)
[image]

Weltraumpenner, 2022-05-09:
> Ha yes, lol, you are right, haven't checked the json. There is a problem, it miss the sensor part, all is mixed in the light entry, that's not "deconz rule" (that are from "philips rules" at start)
> 
> Try with this json
> 
> ```
> {
>   "schema": "devcap1.schema.json",
>   "manufacturername": "_TZ3000_ksw8qtmt",
>   "modelid": "TS011F",
>   "vendor": "Nous A1Z",
>   "product": "TS011F",
>   "sleeper": false,
>   "status": "Gold",
>   "subdevices": [
>     {
>       "type": "$TYPE_ON_OFF_PLUGIN_UNIT",
>       "restapi": "/lights",
>       "uuid": [
>         "$address.ext",
>         "0x01"
>       ],
>       "items": [
>         {
>           "name": "attr/id"
>         },
>         {
>           "name": "attr/lastannounced"
>         },
>         {
>           "name": "attr/lastseen"
>         },
>         {
>           "name": "attr/manufacturername"
>         },
>         {
>           "name": "attr/modelid"
>         },
>         {
>           "name": "attr/name"
>         },
>         {
>           "name": "attr/swversion"
>         },
>         {
>           "name": "attr/type"
>         },
>         {
>           "name": "attr/uniqueid"
>         },
>         {
>           "name": "state/alert"
>         },
>         {
>           "name": "state/on",
>           "refresh.interval": 360
>         },
>         {
>           "name": "state/reachable"
>         }
>       ]
>     },
>     {
>       "type": "$TYPE_CONSUMPTION_SENSOR",
>       "restapi": "/sensors",
>       "uuid": [
>         "$address.ext",
>         "0x01",
>         "0x0702"
>       ],
>       "items": [
>         {
>           "name": "attr/id"
>         },
>         {
>           "name": "attr/lastannounced"
>         },
>         {
>           "name": "attr/lastseen"
>         },
>         {
>           "name": "attr/manufacturername"
>         },
>         {
>           "name": "attr/modelid"
>         },
>         {
>           "name": "attr/name"
>         },
>         {
>           "name": "attr/swversion"
>         },
>         {
>           "name": "attr/type"
>         },
>         {
>           "name": "attr/uniqueid"
>         },
>         {
>           "name": "config/on"
>         },
>         {
>           "name": "config/reachable"
>         },
>         {
>           "name": "state/consumption",
>           "refresh.interval": 360,
>           "parse": {
>             "at": "0x0000",
>             "cl": "0x0702",
>             "ep": 1,
>             "eval": "Item.val = Attr.val * 10;"
>           }
>         },
>         {
>           "name": "state/lastupdated"
>         }
>       ]
>     },
>     {
>       "type": "$TYPE_POWER_SENSOR",
>       "restapi": "/sensors",
>       "uuid": [
>         "$address.ext",
>         "0x01",
>         "0x0b04"
>       ],
>       "items": [
>         {
>           "name": "attr/id"
>         },
>         {
>           "name": "attr/lastannounced"
>         },
>         {
>           "name": "attr/lastseen"
>         },
>         {
>           "name": "attr/manufacturername"
>         },
>         {
>           "name": "attr/modelid"
>         },
>         {
>           "name": "attr/name"
>         },
>         {
>           "name": "attr/swversion"
>         },
>         {
>           "name": "attr/type"
>         },
>         {
>           "name": "attr/uniqueid"
>         },
>         {
>           "name": "config/on"
>         },
>         {
>           "name": "config/reachable"
>         },
>         {
>           "name": "state/current",
>           "refresh.interval": 360
>         },
>         {
>           "name": "state/lastupdated"
>         },
>         {
>           "name": "state/power",
>           "refresh.interval": 360
>         },
>         {
>           "name": "state/voltage",
>           "refresh.interval": 360
>         }
>       ]
>     }
>   ],
>   "bindings": [
>     {
>       "bind": "unicast",
>       "src.ep": 1,
>       "cl": "0x0006",
>       "report": [
>         {
>           "at": "0x0000",
>           "dt": "0x10",
>           "min": 1,
>           "max": 300
>         }
>       ]
>     },
>     {
>       "bind": "unicast",
>       "src.ep": 1,
>       "cl": "0x0702",
>       "report": [
>         {
>           "at": "0x0000",
>           "dt": "0x25",
>           "min": 1,
>           "max": 300,
>           "change": "0x0000000A"
>         }
>       ]
>     },
>     {
>       "bind": "unicast",
>       "src.ep": 1,
>       "cl": "0x0B04",
>       "report": [
>         {
>           "at": "0x0505",
>           "dt": "0x21",
>           "min": 1,
>           "max": 300,
>           "change": "0x00000001"
>         },
>         {
>           "at": "0x0508",
>           "dt": "0x21",
>           "min": 1,
>           "max": 300,
>           "change": "0x00000064"
>         },
>         {
>           "at": "0x050B",
>           "dt": "0x29",
>           "min": 1,
>           "max": 300,
>           "change": "0x00000001"
>         }
>       ]
>     }
>   ]
> }
> ```
> 
> This one will create a light entry + 1 ZHAPower + 1 ZHAConsumption

Looks good to me (Docker on an AMD64 machine). How I proceeded: 
1. cd into container (my container is named deconz)
`sudo docker exec -it deconz /bin/bash`
2. copy file into nous.json
`cat > /usr/share/deCONZ/devices/nous.json`
Insert JSON from clipboard with right-click -> paste
finish editing with Ctrl + C
3. Restart Phoscon in Hue Essentials App 
4. Pair plug in phoscon WebApp

[image]

TheNice4GH, 2022-05-09:
No issue with a very low power consumption device (battery charger):
[image]

No issue with a medium power consumption device (warmsteam humidifier):
[image]

No issue with a high power consumption device (steam ironing station):
[image]

TheNice4GH, 2022-05-09:
For the sake of comparison of readings, I plugged this new NOUS device into a ZW099 Smart Dimmer 6+ from AEON labs (ZWave) which I have installed for years now and connected a 400W Halogen light into them:
[image]

Then I reversed the order, NOUS first then AEON, same halogen light:
[image]

Seems pretty well aligned to me :-)

TheNice4GH, 2022-05-10:
While busy in Domoticz I looked into the log and I noticed the following two lines appearing on a very regular basis, always in pair:
[...]
2022-05-10 15:55:34.351 Error: Zigbee: Can't Update Unit > 13 (sensors) Special part
2022-05-10 15:55:34.351 Error: Zigbee: Can't Update Unit > 13 (sensors) Special part 
[...]
2022-05-10 15:54:10.954 Error: Zigbee: Can't Update Unit > 15 (sensors) Special part
2022-05-10 15:54:10.955 Error: Zigbee: Can't Update Unit > 15 (sensors) Special part 
[...]
So far I'm unable to tell what it is about (could it be that little led that can be switched off?) or if it is due to Domoticz, Conbee2 or a combination of both...

TheNice4GH, 2022-05-10:
13 and 15 already appear in my screenshots above.
Additionally, herewith what is logged for the two NOUS plugs I have when I deactivate/reactive the ZigBee plugin:
[...]
2022-05-10 16:17:04.151 Zigbee: ### Device > 12 Name:Consumption 12 Type:ZHAConsumption Details:{'consumption': 490, 'lastupdated': '2022-05-10T14:17:03.386'} and {'on': True, 'reachable': True}
2022-05-10 16:17:04.152 Zigbee: ### Device > 13 Name:Power 13 Type:ZHAPower Details:{'current': 0, 'lastupdated': '2022-05-10T14:15:34.379', 'power': 0, 'voltage': 234} and {'on': True, 'reachable': True}
2022-05-10 16:17:04.153 Zigbee: ### Device > 14 Name:Consumption 14 Type:ZHAConsumption Details:{'consumption': 130, 'lastupdated': '2022-05-10T14:12:20.071'} and {'on': True, 'reachable': True}
2022-05-10 16:17:04.154 Zigbee: ### Device > 15 Name:Power 15 Type:ZHAPower Details:{'current': 0, 'lastupdated': '2022-05-10T14:14:11.019', 'power': 0, 'voltage': 235} and {'on': True, 'reachable': True}
[...]

Smanar, 2022-05-10:
Oups sorry my comment was for @shaktimaan 

have you try with something that consumme enought power ?
Some plugs are realy sensibles https://github.com/Koenkk/zigbee2mqtt/issues/11703#issuecomment-1109036970

And yes it's domoticz related, you can ask on domoticz forum or domoticz plugin github, but to make short.
- There is a new plugin version 2 days ago that concern this feature.
- An user have asked to have more sensor like "tension" and "frequency" but as they are not realy usefull domoticz don't create them by defaut (need to use a special setting), you have this message because domoticz try to update the "tension" and you don't have the widget for it (for exemple)
- on the last version this message is still here, because usefull for debug, but it s no more an error message but a debug one.

TheNice4GH, 2022-05-10:
Many thanks for the update and explanation!

WolfsWelpe, 2022-05-14:
Hello,
is anyone else noticing that you can't rename the devices after using the custom DDF?
If I rename it in Phoscon it changes in deconZ as well. Sadly after I close and reopen deconZ the name changes back to Power52.
In Phoscon it still got the old name I have changed it to.

Anyone knows how to fix that?

Smanar, 2022-05-14:
>If I rename it in Phoscon it changes in deconZ as well. Sadly after I close and reopen deconZ the name changes back to Power52.
In Phoscon it still got the old name I have changed it to.

Wich one deconz are you using ?
When you change the name in phoscon, the name is updated in the API, and saved on the database, it need to be re-used on next start.

You have the issue on light entry and sensors entries ?

WolfsWelpe, 2022-05-14:
Hello,
I'm using deconZ for Windows 10. Version: 2.16.0
Using the ddf you (@Smanar) has posted.

I don't have this issue with any other entry, except the innr sp120 which ddf file looks suprisingly exactly the same as used for the nous a1z except the model names etc.

TheNice4GH, 2022-05-14:
I just tried renaming in Phoscon (RPI3, version in my screenshot) and it appeared in Domoticz as I modified (for the switch), even after a reboot

Smanar, 2022-05-14:
Realy strange, but this device create more than 1 entry in the API.
You have 1 entry in "light" section (for the on/off) and more in the "sensor" (for measurement), it s possible the rename in phoscon, change only the "light" entry ?

WolfsWelpe, 2022-05-14:
Well the name stays the same in HomeAssistant for me as well just in the deconZ Overview it changes back to this weird "Power65" Tag. Also in Phoscon it stays the same.

WolfsWelpe, 2022-05-14:
> change only the "light" entry

Yes this could be it! When I change the name in Phoscon it changes also the name in HomeAssistant but only the switch entity, so the one who is assigned to the tlight entry. The other two entities created by home assistant are not changing at all and are called Power65 and Consumption32

TheNice4GH, 2022-05-14:
Same for me, only the switch name is carried over

Smanar, 2022-05-15:
For me too, only the light, It's normal for me, have already see that.

So the problem can be from HA that use the "sensor" name instead of the "light" name ?
Or the "light" name is realy reset ?

WolfsWelpe, 2022-05-15:
Okay so they changed the handling with devices? But how do I tell deconz then to detect the light entry as the last one so you can see the correct name in deconZ?

WolfsWelpe, 2022-05-15:
> Haaaa, so it only change on the GUI ?
Not really it changes on both ends, in deconZ and in phosocon. Probably because the light entry for the switch is modified and reloaded in deconZ so the name is getting updated as well in the overview. But after a reboot of deconZ it changes back to the Power name, I guess because it is the last device getting loaded. In phosocon it stays the same because it only shows the light entry there.

Smanar, 2022-05-15:
Strange, I have updated my version and I don't have the issue ...

I can make a request for deconz use in priority "light" name before "sensor" name", but IDK why it don't happen on all devices ....

WolfsWelpe, 2022-05-15:
> I can make a request for deconz use in priority "light" name before "sensor" name"

How do you do this? I'm already happy if deconZ uses the name of the light in the overview rather than the sensor name....

Smanar, 2022-05-15:
For the moment have just asked on discord on dev channel, as I can't reproduce the issue on my side.

WolfsWelpe, 2022-05-19:
Heyho @Smanar 
Anyhting new about this?

Smanar, 2022-05-20:
No answer yet, I will ask to manup direcly and make a PR this WE if needed.

WolfsWelpe, 2022-05-21:
> No answer yet, I will ask to manup direcly and make a PR this WE if needed.

Hey would be very nice I now have a few of them and would like to call them accordingly... Still don't know why it resets every time.

WolfsWelpe, 2022-05-23:
@Smanar Hey thank you so much for your help but I'm too stupid to get it to work :(
I added the "order" option to the lights and the sensors but the it still stays the same name "Power65".
Also not showing up in the ddf editor under preview, but I added it to the ddf file.

Smanar, 2022-05-24:
>I also tried it the other way around with the order but didn't work either.

Yes, have think same, order need to be highter for "light", but your DDF look fine for me

If you set "order":10 for the "$TYPE_ON_OFF_PLUGIN_UNIT
And "order" : 1 for all sensors.
No change after a restart ?

WolfsWelpe, 2022-05-24:
Hey,
I tried it now with "order":10 for "$TYPE_ON_OFF_PLUGIN_UNIT and 1 for all sensors. 
Sadly nothing changed at all, but also the preview in the ddf editor doesn't show the order change? Any other change is visible but not order. 
Preview code in ddf editor: https://pastebin.com/MJhtQ0GA
DDF file: https://pastebin.com/xSDRRf1p

[9 more replies]

closed 2022-06-04 by manup