dresden-elektronik/deconz-rest-plugin #5922: Color change not working

OUARZA, 2022-03-30:
## Describe the bug
Hello,
I have a problem with my LEDVANCE - Gardenpole RGBW Z3 lamps. Since Phoscon, Hue Essential, Jeedom, when I put the color red, the lamp displays blue, when I put blue, it displays white...
I tested from Deconz app, it doesn't work if I put values ​​in "X" and "Y". It only works if I put a value in "Hue"

-->

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

## Environment

## deCONZ Logs

## Additional context

Mimiix, 2022-03-30:
What happens if you use the rest api to set the color?

OUARZA, 2022-03-30:
[image]

It gives red

Mimiix, 2022-03-30:
@manup can you check!

OUARZA, 2022-03-30:
Here are 2 sets of identical lamps, but one recognized as LEDVANCE and the other as OSRAM.
Here is the result when I ask for color #0000FF.
The problematic one is LEDVANCE.

[image]

```
{
    "colorcapabilities": 31,
    "ctmax": 526,
    "ctmin": 153,
    "etag": "8e845296eb46d06b17e9a3ec65ea871f",
    "hascolor": true,
    "lastannounced": "2022-03-30T19:11:15Z",
    "lastseen": "2022-03-30T19:33Z",
    "manufacturername": "LEDVANCE",
    "modelid": "Gardenpole RGBW Z3",
    "name": "Extended color light 9",
    "state": {
        "alert": "none",
        "bri": 254,
        "colormode": "xy",
        "ct": 153,
        "effect": "none",
        "hue": 0,
        "on": true,
        "reachable": true,
        "sat": 0,
        "xy": [
            0.14,
            0.4
        ]
    },
    "swversion": "00103103",
    "type": "Extended color light",
    "uniqueid": "f0:d1:b8:00:00:1a:1e:26-01"
}

{
    "colorcapabilities": 31,
    "ctmax": 666,
    "ctmin": 125,
    "etag": "3395c9535983de92fa2011573d0998e5",
    "hascolor": true,
    "lastannounced": null,
    "lastseen": "2022-03-30T19:22Z",
    "manufacturername": "OSRAM",
    "modelid": "Gardenpole RGBW-Lightify",
    "name": "Extended color light 10",
    "state": {
        "alert": "none",
        "bri": 254,
        "colormode": "xy",
        "ct": 526,
        "effect": "none",
        "hue": 21504,
        "on": true,
        "reachable": true,
        "sat": 254,
        "xy": [
            0.1672,
            0.6476
        ]
    },
    "swversion": "V1.04.93",
    "type": "Extended color light",
    "uniqueid": "7c:b0:3e:aa:0a:07:81:ee-03"
}
```

manup, 2022-03-31:
There are several Ledvance lights where we manually had to remove the xy values, but these appear to be tunable white lights. The code for this light should be the generic one assuming the firmware correctly interprets the color mode commands.

Can you try in the Cluster Info Panel for the *Color Control Cluster* &rarr; **Enhanced  move to hue and saturation** command with following values if they set the appropriate colors:

Saturation: 254
Enhanced hue:

  3000 &rarr; red
  14000 &rarr; green
  42500 &rarr; blue

If this works we can make a DDF and strip out xy color mode, so API clients only use hue and saturation.

OUARZA, 2022-03-31:
Hello,
I just did the test on the 2 sets. The colors are identical for the 3 tests.
The red is a little orange (see photo).
The green is a pale green.
Blue is blue.

Mathieu

[image]

OUARZA, 2022-04-01:
Hello @manup ,
I don't understand why when I run the command below
[image]

 I have the following result?
[image]

Mathieu

manup, 2022-04-03:
After commands are send to a device, values are queried again. Here I'd assume that the light converted to a actual used value. The extended hue is a bit complex, it's not a continuous range but has steps which may not be evenly distributed.

closed 2022-05-01 by github-actions[bot]