dresden-elektronik/deconz-rest-plugin #5923: Device name

Je-sse, 2022-03-30:
## Device

- Product name: Philips Hue Enrave (small)
- Manufacturer: Philips
- Model identifier: 915005996501
- Device type : 
  - Light

## Screenshots

### Basic
[image]

### Identify
[image]

### Groups
[image]

### Scenes
[image]

### On/Off
[image]

### Level Control
[image]

### Color Control
[image]

### Other clusters that are not mentioned above
ZLL commissioning
[image]

Mimiix, 2022-03-30:
Can you also provide the cluster list?

Je-sse, 2022-03-31:
Something went wrong, here it is:
[image]

Smanar, 2022-04-01:
Hello, this device don't work natively ?

Je-sse, 2022-04-01:
Hi, I was not able to connect the device via the search light functionality of Phoscon, I had to use deCONZ and enable the Hybrid option in de DDF tab. I have a second light that I didn't connect yet. I will change the setting back to Basic - Gold and I will test it again with the second light (after a reset) tonight or tomorrow to be sure..

Smanar, 2022-04-01:
But there is no DDF for this device, so I don't think changing something in DDF can do something.

This device can have issue because the missing manufacture name.

Je-sse, 2022-04-01:
I wasn't able to connect the new light out of the box also. I did the same with the first one; connect it with the Hue app on my mobile phone via bluetooth, updated the firmware and did a reset. After that I was able to connect to deCONZ with de default DDF mode (basic - gold only).
So of the firmware update or the reset did the trick

My bad! I will close this request.

closed 2022-04-01 by Je-sse

bilak, 2023-05-17:
Sorry to hijack this issue, but is it possible to include any of Hue Enrave through Phoscon?