dresden-elektronik/deconz-rest-plugin #5927: _TZE200_bjawzodf   TS0601

Piccar, 2022-04-01:
## Device

- Product name: The device name as shown on the product or package.
- Manufacturer: _TZE200_bjawzodf
- Model identifier: TS0601
- Device type : Please remove all unrelated device types. 

  - Other: Humidity and Temperature Sensor

## Screenshots
[image]
[image]

### Basic
Hi I picked up a Cheap Humidity and Temperature Sensor from [Amazon](https://www.amazon.de/YRW-Drahtlos-Temperatursensor-Temperatur-Luftfeuchtigkeitssensor/dp/B08NK4SHP9/ref=pd_sbs_sccl_6/258-3611033-5143315?pd_rd_w=vZVnV&pf_rd_p=82d99ec5-b8da-4272-89b1-cfb4cde1f172&pf_rd_r=AMK8FA9CYRCA7TW4ZY3V&pd_rd_r=c1616a37-6867-4cad-b9d6-1aca0079b51f&pd_rd_wg=eg2SJ&pd_rd_i=B08NK4SHP9&psc=1). From the looks of it it seemed to be identical with [this one](https://www.smarthome-tricks.de/software-iobroker/conbee-ii-zigbee-usb-gateway-teil-10-tuya-temperatur-und-luftfeuchtigkeitssensor/) here, but as [Mimiix](https://github.com/Mimiix) mentioned in #5496 it has to be a different one. The reviews on the product also mentioned that it worked for them. I might have a different Version of it. Is there a way to make this one functional as well?

### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

 
  - If you're unsure whether device support was already requested, please ask for advise in our Discord chat: https://discord.gg/QFhTxqN
-->

## Device

- Product name: The device name as shown on the product or package.
- Manufacturer: As per deCONZ GUI Basic cluster.
- Model identifier: As per deCONZ GUI Basic cluster.
- Device type : Please remove all unrelated device types. 
  - Light
  - Lock
  - Remote
  - Sensor
  - Siren
  - Switch
  - Thermostat
  - Other: 

## Screenshots

### Basic

### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

Mimiix, 2022-04-02:
Can you  please add all screenshots?

Piccar, 2022-04-02:
> Can you please add all screenshots?

I udated the request with more Screenshots. Let me know if i missed something.

Piccar, 2022-05-01:
This is sadly still not solved.

Smanar, 2022-05-01:
Hello.
I think it s possible with DDF now https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet

Can you just start the DDF.
- fille the "basic cluster identification"
- Drag and drop a ZHAtemperature
- Drag and drop a ZHAHumidity

And copy/paste the result for I add the tuya stuff. Not sure it will work because I will try to create fake endpoint.

For memory for later.
- Temperature 0x01 0x02
- Humidity 0x02 0x02
- Battery 0x02 0x04

Piccar, 2022-05-03:
Hi, thanks for helping me. Im not quite sure what to do. I added ZHAHumidity and ZHTemperature like this:
[image]
Should I change soe settings? You mentioned something like 0x01. Not quite sure what to do with that...

Smanar, 2022-05-03:
Nope perfect, I just need a skeleton.
You can save the file, remember the path (if needed for later) and paste the DDF here pls

Piccar, 2022-05-03:
Hope this helps
`{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZE200_bjawzodf",
  "modelid": "TS0601",
  "product": "TS0601",
  "sleeper": false,
  "status": "Draft",
  "path": "/devices/ts0601.json",
  "subdevices": [
    {
      "type": "$TYPE_HUMIDITY_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0405"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/offset",
          "description": "Relative offset to the main measured value.",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/humidity",
          "description": "The current relative humidity in percent.",
          "default": 0
        },
        {
          "name": "state/lastupdated"
        }
      ]
    },
    {
      "type": "$TYPE_TEMPERATURE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0402"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/offset",
          "description": "Relative offset to the main measured value.",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/temperature",
          "default": 0
        }
      ]
    }
  ]
}`

Piccar, 2022-05-03:
[image]
This looks like this. I ticked the show Virtual Sensor CheckBox

Smanar, 2022-05-03:
Ha, lol, so it's worse than I imagined.
You have installed the DDF in the good folder ? As you have the gui, when you select "edit DDF" it s the good file that is chosen.
The node name need to mave from 0xXXXX from a better name.

Piccar, 2022-05-03:
I am not quite sure what you want me to do. I have saved the File at /home/pi/.local/share/dresden-elektronik/deCONZ/devices/ts0601.json
I created the File via edit ddf and reloaded deconz.
[image]
[image]
[image]

Smanar, 2022-05-03:
There is a problem, it s not the DDF I have make.

On the editor menu title, you can see the path and the file used.

Piccar, 2022-05-03:
You mean this ?
[image]

I deleted the old one and pasted it with your code

Smanar, 2022-05-03:
Yep exactly, my DDF have

>   "status": "Gold",

And on your capture it still "Draft"

Piccar, 2022-05-03:
Ok I see the problem. Every time i restart or reboot deconz the sensor becomes like this:
[image]

So I have to Open the new File from you and then save then it looks like this:
[image]

Piccar, 2022-05-03:
nvm. 
[image]

I had to reconnect it

Smanar, 2022-05-03:
Humidity 0 !
Same for temperature ?

Piccar, 2022-05-03:
[image]
[image]

Smanar, 2022-05-03:
lol, 200 is better than 0 but why the same value ?

Can you try to catch some logs with "info" "info_l2" and "ddf" (on deconz, help / debug view)
And heating the device ?

Piccar, 2022-05-03:
If you give me time i will put it outside tonight

Smanar, 2022-05-05:
But can you share some logs to check the problem ?

Piccar, 2022-05-05:
19:11:17:183 poll node 58:8e:81:ff:fe:eb:38:95-01
19:11:17:235 Poll APS request to 0x588E81FFFEEB3895 cluster: 0x0006 dropped, values are fresh enough
19:11:17:524 Poll APS request to 0x588E81FFFEEB3895 cluster: 0x0008 dropped, values are fresh enough
19:11:17:622 read attributes of 0x588E81FFFEEB3895 cluster: 0x0300: [ 
19:11:17:624 0x0000 
19:11:17:625 0x0001 
19:11:17:626 0x0003 
19:11:17:626 0x0004 
19:11:17:627 0x0007 
19:11:17:627 0x0008 
19:11:17:628 ]
19:11:17:629 add task 764949 type 19 to 0x588E81FFFEEB3895 cluster 0x0300 req.id 121
19:11:17:630 Poll APS request 121 to 0x588E81FFFEEB3895 cluster: 0x0300
19:11:17:715 Poll APS confirm 121 status: 0x00
19:11:17:716 Erase task req-id: 121, type: 19 zcl seqno: 64 send time 0, profileId: 0x0104, clusterId: 0x0300
19:11:17:727 Node data 0x588e81fffeeb3895 profileId: 0x0104, clusterId: 0x0300
19:11:17:729 0x588E81FFFEEB3895: update ZCL value 0x01/0x0300/0x0000 after 0 s
19:11:17:731 0x588E81FFFEEB3895: update ZCL value 0x01/0x0300/0x0001 after 0 s
19:11:17:732 0x588E81FFFEEB3895: update ZCL value 0x01/0x0300/0x0003 after 0 s
19:11:17:733 0x588E81FFFEEB3895: update ZCL value 0x01/0x0300/0x0004 after 0 s
19:11:17:734 0x588E81FFFEEB3895: update ZCL value 0x01/0x0300/0x0007 after 0 s
19:11:17:736 0x588E81FFFEEB3895: update ZCL value 0x01/0x0300/0x0008 after 0 s
19:11:17:002 poll node 84:2e:14:ff:fe:64:e0:0a-01
19:11:17:004 Poll light node Ventilator
19:11:17:006 Idle timer triggered
19:11:17:008 Force read attributes for ZHAHumidity SensorNode Humidity 10
19:11:17:009 Force binding of attribute reporting for node Humidity 10
19:11:18:026 rule event /config/localtime: 19:11:17.002 -> 19:11:18.001 (999ms)
19:11:18:397 Poll APS request to 0x842E14FFFE64E00A cluster: 0x0006 dropped, values are fresh enough
19:11:18:001 Daylight now: sunsetEnd, status: 200, daylight: 0, dark: 1
19:11:18:007 rule event /config/localtime: 19:11:18.001 -> 19:11:19.003 (1002ms)
19:11:19:058 Node data 0x588e81fffeeb3895 profileId: 0x0104, clusterId: 0x0300
19:11:19:060 0x588E81FFFEEB3895: update ZCL value 0x01/0x0300/0x0007 after 0 s
19:11:19:061 0x588E81FFFEEB3895: update ZCL value 0x01/0x0300/0x0008 after 0 s
19:11:19:062 ZCL attribute report 0x588E81FFFEEB3895 for cluster: 0x0300, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
19:11:19:063 	payload: 070021e50008003002
19:11:19:326 poll node 14:b4:57:ff:fe:33:a4:57-01
19:11:19:378 Poll APS request to 0x14B457FFFE33A457 cluster: 0x0006 dropped, values are fresh enough
19:11:19:002 poll node 84:2e:14:ff:fe:80:6f:ab-01
19:11:19:005 Idle timer triggered
19:11:19:006 Force read attributes for ZHAHumidity SensorNode Humidity 10
19:11:19:008 Force binding of attribute reporting for node Humidity 10
19:11:19:009 read attributes of 0x842E14FFFE806FAB cluster: 0x0006: [ 
19:11:20:011 0x0000 
19:11:20:012 ]
19:11:20:012 add task 764960 type 19 to 0x842E14FFFE806FAB cluster 0x0006 req.id 135
19:11:20:013 Poll APS request 135 to 0x842E14FFFE806FAB cluster: 0x0006
19:11:20:099 rule event /config/localtime: 19:11:19.003 -> 19:11:20.002 (999ms)
19:11:20:138 Poll APS confirm 135 status: 0x00
19:11:20:140 Erase task req-id: 135, type: 19 zcl seqno: 66 send time 0, profileId: 0x0104, clusterId: 0x0006
19:11:20:204 Node data 0x842e14fffe806fab profileId: 0x0104, clusterId: 0x0006
19:11:20:208 0x842E14FFFE806FAB: update ZCL value 0x01/0x0006/0x0000 after 0 s
19:11:20:217 Websocket 192.168.178.27:39424 send message: {"attr":{"id":"10","lastannounced":"2022-03-06T17:41:02Z","lastseen":"2022-05-05T19:11Z","manufacturername":"IKEA of Sweden","modelid":"TRADFRI control outlet","name":"Exit-Lampe","swversion":"2.0.024","type":"On/Off plug-in unit","uniqueid":"84:2e:14:ff:fe:80:6f:ab-01"},"e":"changed","id":"10","r":"lights","t":"event","uniqueid":"84:2e:14:ff:fe:80:6f:ab-01"} (ret = 946657024)
19:11:20:221 Websocket 192.168.178.27:39420 send message: {"attr":{"id":"10","lastannounced":"2022-03-06T17:41:02Z","lastseen":"2022-05-05T19:11Z","manufacturername":"IKEA of Sweden","modelid":"TRADFRI control outlet","name":"Exit-Lampe","swversion":"2.0.024","type":"On/Off plug-in unit","uniqueid":"84:2e:14:ff:fe:80:6f:ab-01"},"e":"changed","id":"10","r":"lights","t":"event","uniqueid":"84:2e:14:ff:fe:80:6f:ab-01"} (ret = 946657024)
19:11:20:226 Websocket 192.168.178.27:39422 send message: {"attr":{"id":"10","lastannounced":"2022-03-06T17:41:02Z","lastseen":"2022-05-05T19:11Z","manufacturername":"IKEA of Sweden","modelid":"TRADFRI control outlet","name":"Exit-Lampe","swversion":"2.0.024","type":"On/Off plug-in unit","uniqueid":"84:2e:14:ff:fe:80:6f:ab-01"},"e":"changed","id":"10","r":"lights","t":"event","uniqueid":"84:2e:14:ff:fe:80:6f:ab-01"} (ret = 946657024)
19:11:20:231 Websocket 192.168.178.22:54170 send message: {"attr":{"id":"10","lastannounced":"2022-03-06T17:41:02Z","lastseen":"2022-05-05T19:11Z","manufacturername":"IKEA of Sweden","modelid":"TRADFRI control outlet","name":"Exit-Lampe","swversion":"2.0.024","type":"On/Off plug-in unit","uniqueid":"84:2e:14:ff:fe:80:6f:ab-01"},"e":"changed","id":"10","r":"lights","t":"event","uniqueid":"84:2e:14:ff:fe:80:6f:ab-01"} (ret = 946657024)
19:11:20:365 APS-DATA.indication from child 0x239B
19:11:20:484 APS-DATA.indication from child 0x239B
19:11:20:676 APS-DATA.indication from child 0x239B
19:11:21:015 rule event /config/localtime: 19:11:20.002 -> 19:11:21.001 (999ms)
19:11:21:059 poll node 84:2e:14:ff:fe:8e:45:c7-01
19:11:21:061 Poll light node DeckenLampe2
[... cut ...]
19:11:47:301 Websocket 192.168.178.22:54170 send message: {"e":"changed","id":"10","r":"sensors","state":{"humidity":200,"lastupdated":"2022-05-05T18:59:39.001"},"t":"event","uniqueid":"a4:c1:38:f7:1a:b3:aa:3c-01-0405"} (ret = 2124272728)
19:11:48:001 poll node 84:2e:14:ff:fe:97:94:bf-01
19:11:48:003 Poll light node DeckenLampe3
19:11:48:005 Idle timer triggered
19:11:48:006 Force read attributes for ZHAHumidity SensorNode Humidity 10
19:11:48:007 Force binding of attribute reporting for node Humidity 10
19:11:48:017 rule event /config/localtime: 19:11:47.002 -> 19:11:48.001 (999ms)

Smanar, 2022-05-06:
Usefull ines are thoses one

>19:11:21:138 TY_DATA_RESPONSE: seq 30, dpid: 0x01, type: 0x02, length: 4, val: 227

0x01 0x02 is the temperature so 22.7

>19:11:21:169 Websocket 192.168.178.27:39424 send message: {"e":"changed","id":"11","r":"sensors","state":{"lastupdated":"2022-05-05T18:59:55.280","temperature":200}

So 200 is not good

>19:11:26:326 TY_DATA_RESPONSE: seq 31, dpid: 0x02, type: 0x02, length: 4, val: 465

0x02 0x02 is the humidity so 46 %

>19:11:26:355 Websocket 192.168.178.27:39424 send message: {"e":"changed","id":"10","r":"sensors","state":{"humidity":200

Same problem

```
19:11:26:322 a4:c1:38:f7:1a:b3:aa:3c-01-0405/state/humidity expression: Item.val = Attr.val; --> 2.1.022
19:11:42:351 a4:c1:38:f7:1a:b3:aa:3c-01-0402/state/temperature expression: Item.val = Attr.val; --> 2.1.022
```

WTF ?
What is this value ?

You are sure your DDF look like that 
```

        {
          "name": "state/humidity",
          "description": "The current relative humidity in percent.",
          "parse": {"fn": "tuya", "dpid": 2, "eval": "Item.val = Attr.val;" },
          "read": {"fn": "tuya"},
          "default": 0
        },
```

I will ask something to manup , had a similar issue recently.

Smanar, 2022-05-08:
Ok, so manup have found the issue and made a PR https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6030
Unfortunately, ATM you need to wait for the next version or recompile the code to test the future code (I can explain how to do if you have a real/complete OS)

Smanar, 2022-05-17:
Hello, can you make a new try with the last version ?

Smanar, 2022-05-18:
And you still have 

```
19:11:26:322 a4:c1:38:f7:1a:b3:aa:3c-01-0405/state/humidity expression: Item.val = Attr.val; --> 2.1.022
19:11:42:351 a4:c1:38:f7:1a:b3:aa:3c-01-0402/state/temperature expression: Item.val = Attr.val; --> 2.1.022
```
You are on 2.16 version ?

Smanar, 2022-05-18:
Ha ok, so it s normal, the manup correction is in the 2.16 version.

Piccar, 2022-05-18:
I updated the packagelists. Deconz is not listed. Also apt list --upgradables is not showing deconz aswell.

Smanar, 2022-05-18:
You have skipped the part to add it ?
https://phoscon.de/en/conbee2/install#raspbian

Piccar, 2022-05-18:
nope i can see the repo listed in my sources

Smanar, 2022-05-18:
Ha ? I will ask on discord, on my side I have find the file here http://deconz.dresden-elektronik.de/raspbian/beta/

Piccar, 2022-05-18:
Maybe 2.16 is the beta? im on stable

Smanar, 2022-05-18:
Harf right, sorry.

Piccar, 2022-05-18:
I switched to beta now the humidity and temperatur is 2°C and 2,0 % better than 0 Haha

Smanar, 2022-05-18:
lol convertion issue, but it mean we still have the same issue, before the corrective was 200 value too.

```
{"lastupdated":"2022-05-05T18:59:55.280","temperature":200}
{"e":"changed","id":"10","r":"sensors","state":{"humidity":200}
```

So I think you still have too

```
19:11:26:322 a4:c1:38:f7:1a:b3:aa:3c-01-0405/state/humidity expression: Item.val = Attr.val; --> 2.1.022
19:11:42:351 a4:c1:38:f7:1a:b3:aa:3c-01-0402/state/temperature expression: Item.val = Attr.val; --> 2.1.022
```

Piccar, 2022-05-19:
[image]
Im sorry, Im a bit confused what i have to do.

Smanar, 2022-05-19:
Ha, nothing special ^^
Just looking in logs, I think the manup patch haven't worked

Here you have the result before the patch https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5927#issuecomment-1119817294

And I think ATM even with the version 2.16 you have same result ?

```
19:11:26:322 a4:c1:38:f7:1a:b3:aa:3c-01-0405/state/humidity expression: Item.val = Attr.val; --> 2.1.022
19:11:42:351 a4:c1:38:f7:1a:b3:aa:3c-01-0402/state/temperature expression: Item.val = Attr.val; --> 2.1.022
```
You had thoses lines in logs too, and they are so important because we can see the code is using the device version as temperature and humidity.

Piccar, 2022-05-23:
Sorry for the late response
currently the log outputs;

`a4:c1:38:f7:1a:b3:aa:3c-01-0405/state/humidity expression: Item.val = Attr.val; --> 556

a4:c1:38:f7:1a:b3:aa:3c-01-0402/state/temperature expression: Item.val = Attr.val; --> 245`

Smanar, 2022-05-23:
>I switched to beta now the humidity and temperatur is 2°C and 2,0 % better than 0 Haha

You have 20 % humidity ?

But need to be corrected in the DDF, you need *10 for both values ?

Piccar, 2022-05-23:
[image]
It Works now! Both values , the ones on the LCD and in HA, are now identical!

Smanar, 2022-05-23:
Hourra ^^.
So you can confirm the first Tuya sensor using tuya cluster is working ? and without issue ?

Piccar, 2022-05-23:
Well I would say yes with a tiny but:
[image]

as you can see the model name and Manufacturer is a bit off. Seems fixable in your JSON. But besides this it works like a charme! Thank you anonymous knight!

Piccar, 2022-05-25:
Ok now I feel stupid as hell. I thought it was a random name because the packaging said something different. Oh well... Sorry for stealing your time. And Thank You for your patience and you help! I think you can close the issue!

Smanar, 2022-05-26:
BTw we have forget the battery, can you try this new DDF ?

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZE200_bjawzodf",
  "modelid": "TS0601",
  "product": "TS0601",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_HUMIDITY_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0405"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/offset",
          "description": "Relative offset to the main measured value.",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/humidity",
          "description": "The current relative humidity in percent.",
          "parse": {"fn": "tuya", "dpid": 2, "eval": "Item.val = 10 * Attr.val;" },
          "read": {"fn": "tuya"},
          "default": 0
        },
        {
          "name": "config/battery",
          "parse": {"fn": "tuya", "dpid": 4, "eval": "Item.val = Attr.val;" },
          "read": {"fn": "none"},
          "default": 0
        },
        {
          "name": "state/lastupdated"
        }
      ]
    },
    {
      "type": "$TYPE_TEMPERATURE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0402"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/offset",
          "description": "Relative offset to the main measured value.",
          "default": 0
        },
        {
          "name": "config/battery",
          "parse": {"fn": "tuya", "dpid": 4, "eval": "Item.val = Attr.val;" },
          "read": {"fn": "none"},
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/temperature",
          "parse": {"fn": "tuya", "dpid": 1, "eval": "Item.val = 10 * Attr.val;" },
          "read": {"fn": "none"},
          "default": 0
        }
      ]
    }
  ]
}
```

Piccar, 2022-05-30:
Works like a charme!

closed 2022-06-04 by manup