dresden-elektronik/deconz-rest-plugin #5928: Aqara Presence Sensor FP1 (RTCZCGQ11LM)

Verteo, 2022-04-02:
## Device

- Product name: Aqara Presence Sensor FP1
- Manufacturer: aqara
- Model identifier: lumi.motion.ac01
- Device type :  
  - Sensor

## Screenshots

[image]

[image]

### Basic
[image]

### Identify
[image]

### Other clusters that are not mentioned above

Mimiix, 2022-04-02:
I am not sure of it is paired correctly. Can you try to re pair it and see if more clusters show up.

Verteo, 2022-04-02:
I'm not sure either. I tried repairing it multiple times and also I tried the method for [missing data on Xiaomi devices](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Xiaomi:-Lost-devices-or-functional-issues) and nothing really changed except a time cluster sometimes appears and disappers on reading simple descriptors
[image]

Is there anything else I can try?

I captured a log of the pairing, maybe it can help.
<details><summary>click to expand log</summary>
```
19:11:52:881 APS-DATA.indication srcAddr: 0xca45, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0xFCC0, lqi: 255, rssi: -64
19:11:52:882 	asdu: 180a0afc001000
19:11:52:883 APS-DATA.indication from unknown node 0x54EF441000471A48
19:11:53:082 poll node 00:17:88:01:09:c7:26:e5-0b
19:11:53:083 Poll light node Lightbar 1
19:11:53:274 Master: read param with arg 0x19
19:11:53:304 Device TTL 2800 s flags: 0x7
19:11:53:764 rule event /config/localtime: 19:11:52.763 -> 19:11:53.763 (1000ms)
19:11:53:782 APS-DATA.request id: 62, addrmode: 0x03, addr: 0x0017880109c72717, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 0 len: 2 tx.options 0x04
19:11:53:784 	asdu (length: 2): 1e06
19:11:53:832 APS-DATA.confirm id: 62, status: 0x00 SUCCESS
19:11:53:834 APS-DATA.confirm request id: 62 -> confirmed, timeout 0
19:11:53:849 APS-DATA.indication srcAddr: 0x4749, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 255, rssi: -63
19:11:53:850 	asdu: 1e00060600
19:11:53:851 APS-DATA.indication request id: 62 -> finished
19:11:53:852 APS-DATA.request id: 62 erase from queue
19:11:53:852 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x4749
19:11:53:921 poll node 00:17:88:01:0b:80:5d:01-0b
19:11:53:923 Poll light node Filament WA
19:11:54:263 Daylight now: solarNoon, status: 170, daylight: 1, dark: 0
19:11:54:763 poll node ec:1b:bd:ff:fe:9e:45:93-01
19:11:54:764 Poll light node On/Off plug-in unit 2
19:11:54:775 rule event /config/localtime: 19:11:53.763 -> 19:11:54.763 (1000ms)
19:11:55:653 poll node 00:17:88:01:09:e8:3c:9c-0b
19:11:55:654 Poll light node Hue Iris
19:11:55:764 rule event /config/localtime: 19:11:54.763 -> 19:11:55.763 (1000ms)
19:11:56:466 poll node 00:17:88:01:09:c7:27:17-0b
19:11:56:467 Poll light node Lightbar 2
19:11:56:659 APS-DATA.request id: 75, addrmode: 0x03, addr: 0x0017880109c726e5, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 0 len: 2 tx.options 0x04
19:11:56:660 	asdu (length: 2): 1f00
19:11:56:712 APS-DATA.confirm id: 75, status: 0x00 SUCCESS
19:11:56:713 APS-DATA.confirm request id: 75 -> confirmed, timeout 0
19:11:56:754 APS-DATA.indication srcAddr: 0x7637, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 255, rssi: -44
19:11:56:755 	asdu: 1f000500038d1f07ffff2e21008d1f07ffff2e21000000240200ff8d1f07ffff2e21009c3ce80901881700442625020fe08d1f07ffff2e21001727c70901881700494725020fdc
19:11:56:756 APS-DATA.indication request id: 75 -> finished
19:11:56:757 APS-DATA.request id: 75 erase from queue
19:11:56:779 rule event /config/localtime: 19:11:55.763 -> 19:11:56.778 (1015ms)
19:11:57:306 poll node 00:17:88:01:09:c7:26:e5-0b
19:11:57:307 Poll light node Lightbar 1
19:11:57:765 rule event /config/localtime: 19:11:56.778 -> 19:11:57.763 (985ms)
19:11:58:188 poll node 00:17:88:01:0b:80:5d:01-0b
19:11:58:189 Poll light node Filament WA
19:11:58:764 rule event /config/localtime: 19:11:57.763 -> 19:11:58.763 (1000ms)
19:11:59:080 poll node ec:1b:bd:ff:fe:9e:45:93-01
19:11:59:081 Poll light node On/Off plug-in unit 2
19:11:59:539 APS-DATA.request id: 89, addrmode: 0x03, addr: 0x0017880109c726e5, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 0 len: 2 tx.options 0x04
19:11:59:540 	asdu (length: 2): 2003
19:11:59:585 APS-DATA.confirm id: 89, status: 0x00 SUCCESS
19:11:59:586 APS-DATA.confirm request id: 89 -> confirmed, timeout 0
19:11:59:602 APS-DATA.indication srcAddr: 0x7637, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 255, rssi: -44
19:11:59:603 	asdu: 20000603038d1f07ffff2e2100015d800b01881700cf7f25020fff8d1f07ffff2e210093459efeffbd1beca3a825020fa98d1f07ffff2e2100481a47001044ef54fc83160002ff
19:11:59:604 APS-DATA.indication request id: 89 -> finished
19:11:59:605 APS-DATA.request id: 89 erase from queue
19:11:59:606 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x7637
19:11:59:606 neigbor 0x54ef441000471a48 is unknown child
19:11:59:737 APS-DATA.indication srcAddr: 0x83fc, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x0013, lqi: 255, rssi: -45
19:11:59:738 	asdu: 01fc83481a47001044ef548c
19:11:59:739 APS-DATA.indication from unknown node 0x54EF441000471A48
19:11:59:743 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 2, node: 0x83FC
19:11:59:744 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 1, node: 0x83FC
19:11:59:745 new node - ext: 0x54ef441000471a48, nwk: 0x83FC
19:11:59:747 device announce 0x54EF441000471A48 (0x83FC) mac capabilities 0x8C
[... 190 lines cut ...]
19:12:01:860 APS-DATA.indication srcAddr: 0x83fc, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0000, lqi: 255, rssi: -44
19:12:01:861 	asdu: 181d0104000042056171617261
19:12:01:862 APS-DATA.request id: 117 erase from queue
19:12:01:863 Node data 0x54ef441000471a48 profileId: 0x0104, clusterId: 0x0000
19:12:01:869 FP indication 0x0104 / 0x0000 (0x54EF441000471A48 / 0x83FC)
[... cut ...]
19:12:01:988 APS-DATA.confirm request id: 124 -> erase from queue
19:12:02:013 APS-DATA.indication srcAddr: 0x83fc, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0000, lqi: 255, rssi: -44
19:12:02:014 	asdu: 181f0105000042106c756d692e6d6f74696f6e2e61633031
19:12:02:015 APS-DATA.request id: 124 erase from queue
19:12:02:016 Node data 0x54ef441000471a48 profileId: 0x0104, clusterId: 0x0000
19:12:02:022 FP indication 0x0104 / 0x0000 (0x54EF441000471A48 / 0x83FC)
19:12:02:023                       ...     (0x54EF441000471A48 / 0x83FC)
```
</details>

Smanar, 2022-04-03:
After some search It perhaps normal.
This device use the cluster 0xFC00
- Attribute 0x0142 > presence
- Attribute 0x0144 > mode
- Attribute 0x0146 > distance

SwoopX, 2022-04-03:
I'd assume we need the linked PR to tickle some more insights out of the device.

ebaauw, 2022-04-03:
Is this the sensor with radar for presence detection and USB power?  Where did you get it, if you don’t mind me asking?

Verteo, 2022-04-03:
> I'd assume we need the linked PR to tickle some more insights out of the device.

Ok, I will look into doing a deconz build with that patch.

> Is this the sensor with radar for presence detection and USB power? Where did you get it, if you don’t mind me asking?

Yes, that's right. It's the radar sensor. I got it off [Aliexpress](https://de.aliexpress.com/item/1005003766378575.html), every now and then it's available to buy. The curiosity won me over so I was okay with a little bit extra

ebaauw, 2022-04-03:
The specs look very promising (apart from requiring permanent power, but I suppose that comes with active sensing).  Unfortunately it' sold out for now...

Verteo, 2022-04-03:
> > I'd assume we need the linked PR to tickle some more insights out of the device.
> 
> Ok, I will look into doing a deconz build with that patch.

I compiled deconz with the PR and nothing changed 

> The specs look very promising (apart from requiring permanent power, but I suppose that comes with active sensing). Unfortunately it' sold out for now...

I agree, that's why it sparked my curiosity. The power draw I can measure at moment is around 0.15 W, maybe it will consume more power once properly configured?

SwoopX, 2022-04-25:
Ping

wvuyk, 2022-04-27:
If anyone is interested, I received my device yesterday. I can send it out to one of the devs as a loan device? I am guessing that with the distance measurement etc it might be efficient if you have the device hands on?

SwoopX, 2022-04-27:
No need, already got one here for testing but thanks.

You might want to try out the current draft DDF. Please note that deconz version 2.15.1 is required for setting the config items and `sensitivity` cannot be changed without any further code changes.

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "aqara",
  "modelid": "lumi.motion.ac01",
  "vendor": "Xiaomi",
  "product": "Aqara FP1 presence sensor RTCZCGQ11LM",
  "sleeper": false,
  "status": "Gold",
  "path": "/devices/fp1_presence_sensor_rtczcgq11lm.json",
  "subdevices": [
    {
      "type": "$TYPE_PRESENCE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0xfcc0"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "refresh.interval": 84000,
          "read": {
            "at": "0x0006",
            "cl": "0x0000",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0006",
            "cl": "0x0000",
            "ep": 1,
            "eval": "Item.val = Attr.val",
            "fn": "zcl"
          }
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/devicemode",
          "refresh.interval": 300,
          "parse": {
            "at": "0x0144",
            "cl": "0xfcc0",
            "ep": 1,
            "eval": "if (Attr.val == 0) { Item.val = 'undirected' } else if (Attr.val == 1) { Item.val = 'leftright' } else { Item.val = 'unknown' }",
            "fn": "zcl"
          },
          "read": {
            "at": "0x0144",
            "cl": "0xfcc0",
            "ep": 1,
            "fn": "zcl"
          },
          "write": {
            "at": "0x0144",
            "cl": "0xfcc0",
            "dt": "0x20",
            "ep": 1,
            "eval": "if (Item.val == 'undirected') { 0 } else if (Item.val == 'leftright') { 1 }",
            "fn": "zcl"
          },
          "values": [
            ["\"undirected\"", "Undirected"],
            ["\"leftright\"", "Left and right"] 
          ],
          "default": "undirected"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "config/sensitivity",
          "refresh.interval": 300,
          "parse": {
            "at": "0x010C",
            "cl": "0xfcc0",
            "ep": 1,
            "eval": "Item.val = Attr.val",
            "fn": "zcl"
          },
          "read": {
            "at": "0x010C",
            "cl": "0xfcc0",
            "ep": 1,
            "fn": "zcl"
          },
          "write": {
            "at": "0x010C",
            "cl": "0xfcc0",
            "dt": "0x20",
            "ep": 1,
            "eval": "Item.val",
            "fn": "zcl"
          },
          "values": [
            [1, "Low"],
            [2, "Medium"],
            [3, "High"]
          ],
          "default": 3
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/presence",
          "refresh.interval": 300,
          "read": {
            "at": "0x0143",
            "cl": "0xfcc0",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0143",
            "cl": "0xfcc0",
            "ep": 1,
            "eval": "Attr.val != 1 ? Item.val = true : Item.val = false",
            "fn": "zcl"
          },
          "default": false
        }
      ]
    }
  ]
}
```

wvuyk, 2022-04-27:
@SwoopX Thanks! 

The DDF helped to get the device going indeed.

I notice a (for me) new attribute devicemode , is that where we will see the extra info like distance and direction in later?
I have played a bit with it and it is indeed a very good occupancy sensor! I sat in sight of the sensor for 15 minutes, not moving (reading) and it reported presence all that time. When I moved away it took maybe 20 seconds before the device reported no presence. Also checked it in the bathroom, while being behind a curtain in the shower... it stil reported occupancy. The price is steep, but it is a very promising device... 

```
{
"config": {
"devicemode": "undirected",
"on": true,
"reachable": true,
"sensitivity": 3
},
"etag": "e460f507dfc0f0457105980852326443",
"lastannounced": null,
"lastseen": "2022-04-27T12:14Z",
"manufacturername": "aqara",
"modelid": "lumi.motion.ac01",
"name": "Presence 18 Motion",
"state": {
"lastupdated": "2022-04-27T11:47:26.872",
"presence": false
},
"swversion": "20210121",
"type": "ZHAPresence",
"uniqueid": "54:ef:44:10:00:47:1c:13-01-fcc0"
}
```

popy2k14, 2022-05-11:
Hey guys,

please can you provide a link where this can be bought?

Dont find it on AliExpress.
Waiting on this sensor since it was announced to prevent the lights go out when a person is still in the room.

ebaauw, 2022-05-26:
Received mine this week!  Got it to pair to my Aqara M2 hub; please let me know if you need sniffer logs how the Aqara hub handles the device.  Got it to pair with deCONZ v1.16.1 as well, thanks for this PR!  Looks like Homebridge Hue already exposes it as (unknown) motion sensor, based on `state.presence`.

Couple of questions:
- What are the valid values for `config.sensitivity`?
- Why not expose `config.sensitivitymax` (and normalise `config.sensitivity` values to 0 (lowest) ... `config.sensitivitymax` (highest))?
- Is the definition for the 0xFCC0 cluster still missing from `general.xml`, or is there something wrong in my setup?

popy2k14, 2022-06-11:
Got also two of them from Ali and two more are on the way. i must say wow, they blow my mind. Perfectly for living room (couch), bathroom (shower), ...
One little thing which could be better is the 5m range. But i can live with that.
Just want to let you know.

radischem, 2022-06-19:
Fortunately I also received a sensor. Unfortunately it shows in Deconz correctly including motion but doesn’t transfer that info to HA. 
Could someone here explain what I need to change and where to get it running properly? Would be much appreciated :)

popy2k14, 2022-06-20:
I had no such issue,
Do you see the device under devices in HA?
If not, restart HA core & run service deconz refresh devices.

radischem, 2022-06-20:
Deleted the device in Phoscon and started over again with the same result. Searching for a sensor in Phoscon, it will populate the list, but Phoscon declares the search as unsuccesful.

The device is then visible in HA (tried both restart and refresh with the same results), but only as binary sensor which has the state "on", no matter if I'm around or 2 rooms away :(

popy2k14, 2022-06-20:
It's normal that the search behaves like that. In my case also deCONZ in the back showed the sensor.
Also HA is just having an binary sensor with my two sensors.

But it should toggle between those states.

Just an few hint's:
If you have something moving in the room it could be stucked at triggered on.
In my case it was a helium-balloon from a party which moved a little bit.
Also dogs or cat's could be an issue.

Be sure the room is completely empty!

With previous radar sensors i also had issues with thin (dry)walls.
Maybe that's the case for you?

You can lower the sensitivity on this sensors.
But i think, currently you have todo this with the rest API.

Good luck

ebaauw, 2022-06-25:
I now have three working and one broken FP1 devices.  Still messaging with the shop on AliExpress to replace the broken one...

There's a new firmware out, version 0.0.0_0054, see https://github.com/Koenkk/zigbee-OTA/pull/127.  Upgrading one device on the Aqara M1 hub, and another one on deCONZ.

I tried one of the sensors in my hallway and it behaves perfectly.  The one in my living room continues to detect presence while my ceiling fans are turning (up to two minutes after turning them off).  Otherwise detection seems fine.  The one in my dining room continues to detect presence regardless, maybe the Hue Ensis hanging above my dining table is moving slightly?

I'm trying out settings the detection region in the Aqara app, but I'm afraid I fail to understand how it's supposed to work.  The app doesn't consistently show the person symbol where it detects presence.  I created a detection region excluding the grid elements where (I think) the ceiling fans are located, and set sensitivity to low, but I don't seem to get consistent results.
Had anyone already figured out the region logic?

SwoopX, 2022-06-25:
@ebaauw Funny that you mention the detection area, as I played around with that just yesterday and I'd say I understand it 90%. Might need to reset the sensor and redo what I did to get the remaining 10%. I got a pretty adventurous setup with an E1 hub here to get the Xiaomi stuff completely isolated and off of my network, but regardless.

So, apparently you're pretty much free to define quite a number of detection areas (presumably limited to 28, but not sure). Each time an area is defined, it is written to 0x0150 as a 7 Byte ostring. The app gives you a 7 x 4 area (top to bottom Y1 - Y7 and left to right X1 - X4), which is composed as follows:

[image]

```
01:05:00:00:00:01:ff
|| || || || || || ||
|| || || || || || ---- reserved (always ff)
|| || || || || ||
|| || || || || |------ Y7 coordinate
|| || || || || ------- reserved (always 0)
|| || || || ||
|| || || || |--------- Y5 coordinate
|| || || || ---------- Y6 coordinate
|| || || ||
|| || || |------------ Y3 coordinate
|| || || ------------- Y4 coordinate
|| || ||
|| || |--------------- Y1 coordinate
|| || ---------------- Y2 coordinate
|| ||
|| ------------------- detection area index
||
---------------------- yet unknown
```

To now mark an area for detection, the respective X-coordinate value needs to represent the Y-coordinate. X1=1, X2=2, X3=4, X4=8.

Example with detection index 1 and detection area x2y4: **01:01:00:20:00:00:ff**

Presence in a detection area, as far as I can tell, is reported via 0x0151 as ostring (I currently observed 2 Bytes here, but could be more), e.g. 07:08. I'm 99% sure the 2nd Byte represents the X-coordinate, but I'm not sure what the 1st Byte is supposed to mean, either the detection area index or the Y-coordinate.

When setting an area of interference (or whatever it is called), I haven't observed any writes. No clue yet if and to what extend that impacts any processing on the device directly. I also haven't seen any way to delete detection areas, but that might have been a screen resolution issue on my end. If there's indeed no button in the app, it might be be reset with zeroing out the detection area for the respective index?

So much for now...

ebaauw, 2022-06-26:
I've seen the 0x0150 and 0x0151 attributes in the sniffer as well, and your analysis makes sense.  Unfortunately, the GUI doesn't display ostring values in any usable way.  @manup could that be changed into showing an array of (hex) byte values?

I've read online that the FP1 does, indeed, detect motion.  I.e. it compares the result of the next scan with that of the previous scan.  It doesn't detect human bodies, it detects subtle changes in the scan results due to breathing (one of the sites suggested it could be used to raise an alarm if someone stops breathing).  That would also explain why it takes some time before it can detect presence: it needs a couple of scan results before it can conclude so.

I also read that the grid as displayed in the Aqara app is misleading.  The detection area is cone shaped, not rectangular.  Consequently, the coordinates are not rectangular (x, y), but polar (angle, distance).  Come to think of it, this makes a lot of sense physically: the FP1 sends out a mmwave pulse and listens for echoes.  The echoes from closer objects arrive earlier than the echoes of more distant objects.  That would imply that 0x0151 would issue a report per increasing distance during each scan.

Funny part: I've seen some repeated reports for 0x0151, but only occasionally (looking at the Aqara M2 hub).  I see a lot of repeated reports for 0xFFF2, which is an ostring of 24 bytes.  I have seen values 01:08 for 0x0151 (and 01:02 and 01:04).  Maybe 0x0151 only reports changes, where 0xFFF2 reports echoes?  There seems to be some on-device logic, to process scan results (0xFFF2) to changes (0x0151) to presence events (0x0143) to presence detected (0x0142); each step taking some time.

I think only the detection area is sent to the FP1; the other areas are just for the benefit of the app.  I can confirm that 0x0150 seems to encode the coordinates as you indicate.  The first two bytes are 03:01 in my case.  And that 01 corresponds to the first byte in the 0x0151 value.

It would seem the FP1 only reports 0xFFF2 (and 0x0151) when the Aqara app is open on the "grid" view.  The M1 writes value 0x01 to 0x0155 while the app is opening that view.  I haven't captured any reset of 0x0155, maybe it just times out?

SwoopX, 2022-06-26:
> @manup could that be changed into showing an array of (hex) byte values?

+1 over here. That would also come in handy e.g. for the Develco/frient firmware version and eliminate the necessity to use a script for displaying it correctly.

Surely, the grid layout presented by the app was chosen to simplify reality. [This here](https://user-images.githubusercontent.com/38351800/169654245-6b6eae8d-5f25-46aa-8632-c5f560b9168b.png) would be indeed more accurate (it's info material from the P1). What now comes into my mind is that setting the detection/approach distance my eventually impact the reported or to be expected values on 0x0151 respectively.

With the E1 hub, I interestingly haven't observed anything on 0xFFF2 but I had the app running all the time doing the sniff. More than happy to doublecheck if that eventually occurs without it. Anyway, I noted them in the capture you shared with me any maybe we can make some sense out of it jointly. There weren't many Bytes inside the (ostring?) attribute changing though.

Also still curious on the very first Byte on my analysis above, but as said, I should reset the sensor first to eventually learn what purpose it might have. Anyway, more than happy to share my capture, just need to carve out all the OTA packages. Somehow, the device tried to update, but the hub hasn't provided any data which flooded the log.

Otnow, 2022-06-26:
@SwoopX, @ebaauw, for a long time [I was going](https://github.com/Koenkk/zigbee2mqtt/issues/11019#issuecomment-1026012894) to analyze the protocol of work with the regions, but there were other priorities.
Thanks for the motivation to finally do it 🙂

Here is what I was able to understand and clarify at the moment in addition to the above:

# Regions

As you rightly noted, the grid of regions looks like this:

```javascript
         ^
    X1 X2 X3 X4
Y1 | 1| 2| 4| 8|
Y2 | 1| 2| 4| 8|
Y3 | 1| 2| 4| 8|
Y4 | 1| 2| 4| 8|
Y5 | 1| 2| 4| 8|
Y6 | 1| 2| 4| 8|
Y7 | 1| 2| 4| 8|
```

## Detection regions

Detection regions are created, modified and deleted using attribute `0x0150` (_cluster_: `0xfcc0`, _type_: `0x41`) in the following format:

```javascript
_ _ : _ _ : _ _ : _ _ : _ _ : _ _ : _ _ 
| |   | |   | |   | |   | |   | |   | |
| |   | |   | |   | |   | |   | |   | - - hex value (0 or F): 0 when deleting and
| |   | |   | |   | |   | |   | |   |     F when creating/changing a region
| |   | |   | |   | |   | |   | |   |
| |   | |   | |   | |   | |   | |   - - - hex value (0 or F): 0 when deleting and
| |   | |   | |   | |   | |   | |         F when creating/changing a region
| |   | |   | |   | |   | |   | |
| |   | |   | |   | |   | |   | - - - - - hex value (0 - F): 0 when deleting and
| |   | |   | |   | |   | |   |           0-F when creating/changing a region 
| |   | |   | |   | |   | |   |           (sum of the selected values X1-X4 for the Y7 row)
| |   | |   | |   | |   | |   |
| |   | |   | |   | |   | |   - - - - - - hex value: always 0
| |   | |   | |   | |   | |
| |   | |   | |   | |   | - - - - - - - - hex value (0 - F): 0 when deleting and
| |   | |   | |   | |   |                 0-F when creating/changing a region
| |   | |   | |   | |   |                 (sum of the selected values X1-X4 for the Y5 row)
| |   | |   | |   | |   |
| |   | |   | |   | |   - - - - - - - - - hex value (0 - F): 0 when deleting and
| |   | |   | |   | |                     0-F when creating/changing a region
| |   | |   | |   | |                     (sum of the selected values X1-X4 for the Y6 row)
| |   | |   | |   | |
| |   | |   | |   | - - - - - - - - - - - hex value (0 - F): 0 when deleting and
| |   | |   | |   |                       0-F when creating/changing a region
| |   | |   | |   |                       (sum of the selected values X1-X4 for the Y3 row)
| |   | |   | |   |
| |   | |   | |   - - - - - - - - - - - - hex value (0 - F): 0 when deleting and
| |   | |   | |                           0-F when creating/changing a region
| |   | |   | |                           (sum of the selected values X1-X4 for the Y4 row)
| |   | |   | |
| |   | |   | - - - - - - - - - - - - - - hex value (0 - F): 0 when deleting and
| |   | |   |                             0-F when creating/changing a region
| |   | |   |                             (sum of the selected values X1-X4 for the Y1 row)
| |   | |   |
| |   | |   - - - - - - - - - - - - - - - hex value (0 - F): 0 when deleting and
| |   | |                                 0-F when creating/changing a region
| |   | |                                 (sum of the selected values X1-X4 for the Y2 row)
| |   | |
| |   | - - - - - - - - - - - - - - - - - hex value (1 - A): region index (application limits 10)
| |   |
| |   - - - - - - - - - - - - - - - - - - hex value: always 0
| |
| - - - - - - - - - - - - - - - - - - - - hex value (1 - 3): command code:
|                                         1 - create a region
|                                         2 - delete region
|                                         3 - change region
|
- - - - - - - - - - - - - - - - - - - - - hex value: always 0
```

Accordingly, after regions are created, the sensor starts reporting events related to them using attribute `0x0151` (cluster: `0xfcc0`, type: `0x41`) in the following format:

```javascript
_ _ : _ _
| |   | |
| |   | - - hex value (1,2,4,8): event code:
| |   |     1 - enter
| |   |     2 - leave
| |   |     4 - presence
| |   |     8 - no presence
| |   |
| |   - - - hex value: always 0
| |
| - - - - - hex value (1 - A): region index (application limits 10)
|
- - - - - - hex value: always 0
```

## Other regions

Other regions are set using attributes:

- `0x0153` (cluster: `0xfcc0`, type: `0x23`) - **exits and entrances**
- `0x0154` (cluster: `0xfcc0`, type: `0x23`) - **interference source**
- `0x0156` (cluster: `0xfcc0`, type: `0x23`) - **edge**

in the following format (_hex value is passed to the sensor as int_):

```javascript
0 x _ _ _ _ _ _ _ _
    | | | | | | | |
    | | | | | | | - - hex value (0 - F): sum of the selected values X1-X4 for the Y1 row
    | | | | | | |
    | | | | | | - - - hex value (0 - F): sum of the selected values X1-X4 for the Y2 row
    | | | | | |
    | | | | | - - - - hex value (0 - F): sum of the selected values X1-X4 for the Y3 row
    | | | | |
    | | | | - - - - - hex value (0 - F): sum of the selected values X1-X4 for the Y4 row
    | | | |
    | | | - - - - - - hex value (0 - F): sum of the selected values X1-X4 for the Y5 row
    | | |
    | | - - - - - - - hex value (0 - F): sum of the selected values X1-X4 for the Y6 row
    | |
    | - - - - - - - - hex value (0 - F): sum of the selected values X1-X4 for the Y7 row
    |
    - - - - - - - - - hex value: always 0
```

***

### Test environment:

- **Aqara Hub M1S CN** (_ZHWG15LM_): 3.4.2_0008.0616
- **Aqara Presence Detector FP1** (_RTCZCGQ11LM_): 0.0.0_0054

SwoopX, 2022-06-26:
@Otnow That's awesome and many thanks for chiming in! Was planning to reach out that evening as well. Haven't had the time yet to really dive deep into that, but that confirms that I seem to have some constraints in my test setup which didn't allow me to see and catch everything. Above that, some observations I made today while sniffing are confirmed 🙂 

Really cool to see how fast things can go if the right people connect. One goodie I just discovered, in case you didn't already know: tag 0x08 of the special reporting is the firmware version as displayed in the app. Seems to be applicable for many devices.

Otnow, 2022-06-26:
@SwoopX, you are welcome 🙂

> One goodie I just discovered, in case you didn't already know: tag 0x08 of the special reporting is the firmware version as displayed in the app. Seems to be applicable for many devices.

I'm also trying to understand the structure of the heartbeat-message and had a hunch about it, so thanks for the confirmation 👍

manup, 2022-06-27:
> I've seen the 0x0150 and 0x0151 attributes in the sniffer as well, and your analysis makes sense. Unfortunately, the GUI doesn't display ostring values in any usable way. @manup could that be changed into showing an array of (hex) byte values?

Yes I'll try to figure something out perhaps simply `QByteArray::toHex()` , I get my FP1 this week for testing sounds like a complex beast :)

ebaauw, 2022-06-27:
So the automation on regions in the Aqara app has four events:
- Region Manned
- Region Unmanned
- Region In
- Region Leave

Attribute 0x0151 corresponds to these events: the first byte indicates the region, the second by the event:

Raw Value | Event
-- | --
01 | In
02 | Leave
04 | Manned
08 | Unmanned

ebaauw, 2022-06-27:
Funny, the device stops reporting 0xFFF2 after a while, and only restart this after the app writes 1 to 0x0155.  Apparently 0xFFF2 encodes the data to display the icon on the grid.  Still don't understand the logic when it will show and when it won't.

I also saw the FP1 requesting and receiving the current time from the coordinator.  And it does poll its parent every 5 minutes or so, even though is has _Receiver On When Idle_.

Here are some successive values for 0xFFF2 (double-checked for successive ZCL sequence numbers):

```
 0                             1                             2
 0  1  2  3  4  5  6  7  8  9  0  1  2  3  4  5  6  7  8  9  0  1  2  3

aa:74:12:84:73:d9:09:41:0f:00:02:62:b9:ba:9f:42:01:01:d8:04:00:0c:6b:00
aa:74:12:84:74:d8:09:41:0f:00:02:62:b9:ba:a4:43:01:01:ef:09:00:0d:d0:00
aa:74:12:84:75:d7:09:41:0f:00:02:62:b9:ba:a7:44:01:01:d0:08:00:09:ff:00
aa:74:12:84:76:d6:09:41:0f:00:02:62:b9:ba:a8:45:01:01:d0:06:00:08:d8:00
aa:74:12:84:77:d5:09:41:0f:00:02:62:b9:ba:b2:46:01:01:d0:06:00:08:d8:00
aa:74:12:84:78:d4:09:41:0f:00:02:62:b9:ba:bc:47:01:01:d0:06:00:08:d8:00
aa:74:12:84:79:d3:09:41:0f:00:02:62:b9:ba:c7:48:01:01:d0:06:00:08:d8:00
```
Bytes 5 and 15 are counting up, and byte 6 is counting down.

After deleting the detection regions, I see no major difference:
```
aa:74:12:84:d8:74:09:41:0f:00:02:62:b9:bf:4e:c6:01:01:fe:00:00:0b:b8:00
aa:74:12:84:d9:73:09:41:0f:00:02:62:b9:bf:58:c7:01:01:fe:00:00:0b:b8:00
```
It would seem this is indeed raw sensor data, before the region logic has been applied.  Still no clue how to interprete it.

SwoopX, 2022-06-28:
I have to say right now, the idea of exposing 10 additional presence sensors (based on the above identified maximum) is not too charming for me. I'd rather vouche for adding something like `areapresence` and `areapresenceevent`. Not too sure about what to do with the other areas yet. Presumably the resource should/must keep track of the max indexes used as well.

Admittedly, the suggested above would not allow to actually see the current state of all areas at a glance, but it would be enough to fire events based on what's been reported. Regarding the "global" presence state, my perception is also that it doesn't change when anything in an area is going on.

Anyway, it's remarkable how feature rich the device is and it amazingly solved the "issues" a regular motion/presence sensor gives you.

ebaauw, 2022-06-28:
Problem with exposing the regions in an array or object is that you’d need to enhance the `rules` syntax to include these in conditions.  That will be C++ coding.  Ideally, we’d only have additional resources (or array entries) for regions that are actually defined.  Not sure if the device actually reports this, or if we need to conclude that from the 0x0151 events.  That would also need C++ code to enhance the DDF syntax for conditional resources and/or resource items.

manup, 2022-06-29:
In a `ResourceItem` we can encode this into a unsigned 32-bit number which can be handled by rules perhaps with some additional operations? As far as I understand we have 10 regions with 4 events / (states?).

```
2-bits for each region:

region 0:    bits 0-1
region 1:    bits 2-3
...
region 10:   bits 17-18
```

A rule condition looking at such a value could have an additional  *mask* like 0x03 to only check against the first two bits.

Quick-n-dirty idea:
```
{
  "address": "/sensors/00:17:88:01:09:a6:06:d1-01-fc00/state/areamap",
  "operator": "mask_eq",
  "value": "3,2"
}
```
This would then trigger if the value in first two bits (mask = 3) is 2.

On the DDF and API side we need a generic approach to describe such binary structure and expose some meaningful attributes.
E.g. this is an array of ten 2-bit enums with following values.

An API client sees:

```
{
  "state": {
    "area0": "in",
    "area1": "leave",
    ...
    "area9": "unmanned"
  }
}
```

(or whatever we come up with what's convenient for API clients to consume)

SwoopX, 2022-06-29:
If currently hacked something together which looks like this, just for the sake being able to play around with it

```
{
    "config": {
        "detectionarea": "010122110000ff",
        "devicemode": "undirected",
        "on": true,
        "reachable": true,
        "sensitivity": 3,
        "triggerdistance": "medium"
    },
    "etag": "13ff209f9401b317987d42506dd4cd79",
    "lastannounced": null,
    "lastseen": "2022-06-28T23:13Z",
    "manufacturername": "aqara",
    "modelid": "lumi.motion.ac01",
    "name": "Presence 165",
    "state": {
        "lastupdated": "2022-06-28T23:13:38.577",
        "presence": true,
        "presenceareaevent": 18,
        "presenceevent": "Leave 1"
    },
    "swversion": "20210121",
    "type": "ZHAPresence",
    "uniqueid": "54:ef:44:10:00:47:1a:48-01-0406"
}
```

In that example, you can configure the areas via `detectionarea` sending the desired hex value. `presenceareaevent` in turn fires the events based on attribute 0x0151 where the last digit is the event and anything before the detection area.

popy2k14, 2022-07-01:
Guys, the cheapest offer on aliexpress since they are available.
Now down under 60 Euro. There are two offers currently.

Here is the link: https://de.aliexpress.com/item/1005004450930450.html?spm=a2g0o.productlist.0.0.1da37b93WF90Tx&algo_pvid=f3acbafa-e171-4c74-ba19-d8e5a4f3b5fc&algo_exp_id=f3acbafa-e171-4c74-ba19-d8e5a4f3b5fc-0&pdp_ext_f=%7B%22sku_id%22%3A%2212000029303525377%22%7D&pdp_npi=2%40dis%21EUR%21%2158.65%21%21%21%21%21%402101e9d016566937349224934ed2d4%2112000029303525377%21sea

arthur-morgan-1, 2022-07-09:
This sensor only shows clear/detected for me, no directions or anything else...

lightheaded, 2022-12-11:
Hi @ebaauw! Are you still working on getting regions to work without the Aqara app?

popy2k14, 2022-12-14:
Hey guys, i have really strange issues with my 8x fp1 sensors.
Sadly, i was so hyped about this type of radar zigbee sensor i have added them to every room.
That was no good idea. I have those issues with them:

- sometimes they are just unreachable and come back again (every few minutes), repairing doesnt solve this.

- two of them are now stucked in "presence" state, re-pair, lower sensitivity doesnt solvethe issue. They are now unusable. Here is a thread on the home assistant forum: https://community.home-assistant.io/t/aqara-fp1-stuck-on-detected-occupancy/438350

Someone flipped the PCB inside the case and this solved the issue for him.
Maybe a custom 3D case could solve the issue?

- The detection range is not 120 degrees as described. You have to get really straight to the sensor to get detected. Also the sensor is nothing for large rooms, because the range is too low.

Anyone of you have similar issues?
Is there a firm,ware newer than 54?

thx

ebaauw, 2022-12-14:
> Hi @ebaauw! Are you still working on getting regions to work without the Aqara app?

It’s still on my todo list, but not in the upper regions.

> sometimes they are just unreachable and come back again (every few minutes)

I saw that as well, until I moved them to a second network, with only end devices.  Same for the Aqara E1 roller blind controller and the IKEA FYRTUR.

> two of them are now stucked in "presence" state

There is a Zigbee call to force-clear the state.  It’s exposed through the API, `config/resetpresence`.

> The detection range is not 120 degrees as described.

As I understand, the detection zone is really a pie shape, rather than a square as suggested by the Aqara app.  Placing the sensor is crucial (and challenging since it’s wired to power).  I’m afraid my apartment isn’t big enough to exceed the range.  Still not sure how the various settings influence the detection range.

I do have an issue with the vertical detection range: the register my ceiling fans, which I want to turn off when there’s no more presence.  I put placed some aluminium foil above the sensor to block their vision, but have been fantasising to place that inside the housing.

popy2k14, 2022-12-14:
> > sometimes they are just unreachable and come back again (every few minutes)
> 
> I saw that as well, until I moved them to a second network, with only end devices. Same for the Aqara E1 roller blind controller and the IKEA FYRTUR.
Dont want to make an new gateway/network just for this. My workaround is to retain the last value, this works reliable. Just wanted to know if i am alone and if there is another workaround.

> > two of them are now stucked in "presence" state
> 
> There is a Zigbee call to force-clear the state. It’s exposed through the API, `config/resetpresence`.
Yeah i know the call, which is also exposed to HA, but sadly half of the time, it just do nothing :-(

> > The detection range is not 120 degrees as described.
> 
> As I understand, the detection zone is really a pie shape, rather than a square as suggested by the Aqara app. Placing the 
>
> I do have an issue with the vertical detection range: the register my ceiling fans, which I want to turn off when there’s no more presence. I put placed some aluminium foil above the sensor to block their vision, but have been fantasising to place that inside the housing.

sensor is crucial (and challenging since it’s wired to power). I’m afraid my apartment isn’t big enough to exceed the range. Still not sure how the various settings influence the detection range.
Yeah, maybe i should playce two of them in my big room, but with all the issues this sensor has. I likley will not buy another one,

PS.: Got answer from aqara with an standard nonsense answer, so i think that's a dead end :-(

Is there a firmware newer than 54?

ebaauw, 2022-12-14:
According to my Aqara M2 hub (set to Mainland China to support the FP1), there's no newer firmware than 54.

popy2k14, 2022-12-14:
ok, thx.

michalmo, 2023-05-16:
@patmalcolm91 thanks for this reverse engineering work on the `0xfff2` data packets, the missing piece is that byte `JJ` is the angle from center in degrees. [Distance, angle and some trigonometry](https://github.com/zigpy/zha-device-handlers/pull/2401/files#diff-3a695c15bf7294c66c68c53407df96ca27267bf36c5fb99b38021d270afbd56dR158-R159) gives you the X and Y coordinates (my testing shows that each square of the grid is 1 meter x 1 meter).

slavendam, 2024-01-09:
Did anyone succeed to get distance and angle from Aqara FP2 sensor?

amaisano, 2024-02-06:
FYI firmware `58` is now available for the FP1.

greyman12, 2024-02-20:
Did the latest firmware break the ability to use regions?

tophee, 2024-02-21:
@greyman12 no, the firmware is not the problem. I upgraded to version 58 several weeks ago and the regions worked fine. It seems more likely that one of the latest releases of Zigbee2MQTT broke something. I managed to get regions to work again after I downgraded to v1.35.1.

greyman12, 2024-02-22:
@tophee thanks for the info I will try downgrading.