dresden-elektronik/deconz-rest-plugin #5929: OWON PIR313-E

sirinveen, 2022-04-02:
## Device

- Product name: ZigBee Multi-Sensor (Motion/Temp/Humi/Light) https://www.owon-smart.com/zigbee-motion-sensor-with-wireless-alarm-for-smart-home-multisensor-pir313-product/
- Manufacturer: OWON
- Model identifier: PIR313
- Device type : 
 
  - Sensor
 

## Screenshots

[image]
[image]

### Basic

[image]

### Identify

[image]

### Power Configuration

[image]

### IAS Zone

[image]

### Illuminance measurement

[image]

### Temperature measurement

[image]

### Relative Humidity measurement

[image]

### Occupancy Sensor is missing!!

Smanar, 2022-04-03:
Hello, so the DDF can work it s not the same device
Your device is a "PIR313" and on your DDF you are using "PIR313-E", you can use drag and drop to fill this field too.

sirinveen, 2022-04-03:
hi smanar,

I forget to precise it's a PIR313-E model (I just update my template) but deCONZ recognise like a PIR313.

Smanar, 2022-04-03:
Hu, It's not work like that ^^, information you see in deconz are from the device firmware, it s the device itself that say it's a "PIR313", you can this value in the capture "Basic"

You need to use this one in the DDF.

sirinveen, 2022-04-03:
Means he sell me the wrong model :/ when I set the name PIR313 I can see now the sensor in Phoscon but the value aren't updated...

[image]

Smanar, 2022-04-03:
Ca you show your DDF ?

sirinveen, 2022-04-03:
I made different change but without success.
Here the preview: 

`{
  "schema": "devcap1.schema.json",
  "manufacturername": "OWON",
  "modelid": "PIR313",
  "vendor": "OWON",
  "product": "PIR313",
  "sleeper": false,
  "status": "Gold",
  "path": "/devices/pir313.json",
  "subdevices": [
    {
      "type": "$TYPE_PRESENCE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0406"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "awake": true,
          "refresh.interval": 84000,
          "parse": {
            "at": "0x0004",
            "cl": "0x0000",
            "ep": 255,
            "eval": "Item.val = Attr.val",
            "fn": "zcl",
            "mf": "0x113c"
          }
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/duration"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
[... 358 lines cut ...]
      "cl": "0x0405",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x21",
          "min": 1,
          "max": 0
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0500",
      "report": [
        {
          "at": "0x0001",
          "dt": "0x31",
          "min": 0,
          "max": 0
        }
      ]
    }
  ]
}`

sirinveen, 2022-04-03:
I change endpoint and now I receive value for temperature and humidity but not light level and motion don't run because I don't have the cluster for that :/

[image]

Smanar, 2022-04-03:
Can you try with this DDF

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "OWON",
  "modelid": "PIR313",
  "vendor": "OWON",
  "product": "PIR313",
  "sleeper": false,
  "status": "Gold",
  "path": "/devices/pir313.json",
  "subdevices": [
    {
      "type": "$TYPE_PRESENCE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "awake": true,
          "refresh.interval": 84000,
          "parse": {
            "at": "0x0004",
            "cl": "0x0000",
            "ep": 255,
            "eval": "Item.val = Attr.val",
            "fn": "zcl",
            "mf": "0x113c"
          }
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/duration"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "config/enrolled",
[... 303 lines cut ...]
          "change": "0x0000000A"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 3,
      "cl": "0x0405",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x21",
          "min": 60,
          "max": 600
        }
      ]
    }
  ]
}
```

For the presence you use the Cluster 0x0500.
I have let the code by defaut, for the state/presence
I have removed the ZHAbattery sensor, as you have config/battery for others sensors
I have removed too the binding for cluster 0x0500 because not all device need it, so if the presence don't work, add the binding (and only it, so only the left panel), and if it's still not working add the report on right panel, attribute 0x0002

sirinveen, 2022-04-03:
I find solution for Light/Level I delete the bindings and let by default (Parse to none) and now I receive the result in Phoscon.

[image]

Only thing don't work is motion.  I try different solution with 0x0500 for motion but no success the sensor stay detected and have not effect. With and without bindings...

We are near of a global solution ... I continue to try different possibility about sensor ...

Smanar, 2022-04-03:
Can you show the device json for the motion sensor ? you can find it in phosocn/ help / api information.
Can set the 

```
        {
          "name": "config/enrolled",
          "public": false
        },
```

To public = true, so we will be able to see if the enrollement have succed.
Have you tried with bindig + reporting ?

```
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0500",
      "report": [
        {
          "at": "0x002",
          "dt": "0xXX",
          "min": 60,
          "max": 600,
          "change": "0x00000001"
        }
```

Can try too, force some value in state/presence

```
          "awake": true,
          "parse": {
            "at": "0x0002",
            "cl": "0x0500",
            "ep": 1,
            "eval": "Item.val = (Attr.val & 0x3) != 0"
          }
```

sirinveen, 2022-04-03:
good news I do a new bindings with reporting and set the new eval (Item.val = (Attr.val & 0x3) != 0) and now it's running. Sensor Mention go back to none!

For your info here below the json of the motion sensor. What do you mean with config/enrolled? Where I find it?

`{
    "config": {
        "battery": 100,
        "duration": 0,
        "on": false,
        "reachable": true
    },
    "etag": "9dee52d0bcb0bb118b274aea3f81adb7",
    "lastannounced": "2022-04-03T15:58:02Z",
    "lastseen": "2022-04-03T18:27Z",
    "manufacturername": "OWON",
    "modelid": "PIR313",
    "name": "Presence 289",
    "state": {
        "lastupdated": "2022-04-03T18:27:28.021",
        "presence": false
    },
    "type": "ZHAPresence",
    "uniqueid": "3c:6a:2c:ff:fe:d1:b1:73-01-0406"
}`

Smanar, 2022-04-04:
>What do you mean with config/enrolled? Where I find it?

No need, if the sensor is working it mean the enrollement is done.
If all is working can you make a PR with the DDF fo other ?

sirinveen, 2022-04-04:
For my knowledge want would be the process if the enrolment wasn't done.
Where you do that?

> When you said: Can set the to public = true `        {
          "name": "config/enrolled",
          "public": false
        },`

I delete the sensor and I have big difficulties again he recognise every sensor. Now it's look like everything is working. I just wait to be sure and I will edit the DDF.

Smanar, 2022-04-04:
IAS device are special, they need 2 "inclusions", 1 you make yourself when including the device in the zigbee network, and after that the gateway make the enrollement (you have nothing to do, deconz do the job)
So the device can be included in the zigbee network but not enrolled, so it can don't work (some device don't use the enrollement), this value is a counter, that increase at every enrollement step, so we can use it to check where we are.
And as this kind of device are lazy, the requests can be delayed.

Smanar, 2022-04-04:
>He refresh only if I manually push the button read on the cluster. Before he refresh also the cluster automatically but now not anymore :-/

It's typically a bind issue.
The DDF is good because it update the value when you read the attribute on deconz.
You just miss the periodic report, take a look on bind and on the attribute 0x0000, defaut value used by deconz are min 5, max 300, change 2000.

sirinveen, 2022-04-04:
When I use DDF bindings for illuminance no chance that the lux is updated in deCONZ and also in Phoscon. 
When I do a bind dropbox (old method) it's now ok :-/
Not easy to switch with new and old method ...

closed 2022-05-05 by github-actions[bot]

ebaauw, 2022-07-24:
@smanar, it looks like the `uniqueid` for the ZHATemperature and ZHAHumidity has the wrong endpoint?

Idaho947, 2022-11-10:
Hello, i've just buy this sensor. The ddf will be include in a stable version ?

Smanar, 2022-11-10:
I don't find it, can you try this new DDF (based on the issue)

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "OWON",
  "modelid": "PIR313",
  "vendor": "OWON",
  "product": "PIR313",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_PRESENCE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0402",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0500"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/duration"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "config/enrolled",
          "public": false
[... 257 lines cut ...]
        {
          "at": "0x0000",
          "dt": "0x29",
          "min": 60,
          "max": 600,
          "change": "0x0000000A"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 3,
      "cl": "0x0405",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x21",
          "min": 60,
          "max": 600
        }
      ]
    }
  ]
}
```

Idaho947, 2022-11-11:
Thanks, everything works fine except the lux.

Smanar, 2022-11-11:
So "state/lightlevel" is working but not "state/lux" ?

For battery can wait somes hours, if it's still not working try to change a 

```
{
  "name": "config/battery"
}
```

by 

```
{
  "name": "config/battery",
  "awake": true,
  "refresh.interval": 3600,
  "parse": {
    "at": "0x0021",
    "cl": "0x0001",
    "ep": 1,
    "eval": "Item.val = Attr.val / 2",
    "fn": "zcl"
  }
}
```

For a sensor, and look it somes hours after (need to read direclty in the API because it will update it only on the sensor you have made the change)

No need to re-include, just make a hot-reload.

Idaho947, 2022-11-11:
Sorry no state lightlevel, daylight, lux and dark don't work.
[image]

Ok for batterie state I'll check

Smanar, 2022-11-11:
Ok, so like the other issue, can you try to read :
- in the endpoint 0x02 (here I have a doubt on z2m they are using 0x03)
- On the cluster 0x400
- Read the attribute 0x0000

It need to update all missing field in the DDF, else I m using perhaps a bad endpoint, or the DDF is broken.

Idaho947, 2022-11-11:
For the 0x02 :
[image]

For the 0x000 :
[image]

Smanar, 2022-11-11:
The endpoint 0x02 is 

[image]

It's just to be sure it's the 0x02 and not the 0x03.

When you have read the value in deconz, it have updated values in jeedom ?

Can be the binding
```

        {
          "at": "0x0000",
          "dt": "0x21",
          "min": 60,
          "max": 600,
          "change": "0x0000FFFF"
        }
```

0xffff is to disable generaly, defaut values are

        {
          "at": "0x0000",
          "dt": "0x21",
          "min": 5,
          "max": 300,
          "change": "0x000007d0"
        }

Idaho947, 2022-11-11:
I'll try this evening I'll tell you

Smanar, 2022-11-11:
Ok so new DDF with all bnattery and new binding for illuminance

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "OWON",
  "modelid": "PIR313",
  "vendor": "OWON",
  "product": "PIR313",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_PRESENCE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0402",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0500"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/duration"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "config/enrolled",
          "public": false
[... 266 lines cut ...]
        {
          "at": "0x0000",
          "dt": "0x29",
          "min": 60,
          "max": 600,
          "change": "0x0000000A"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 3,
      "cl": "0x0405",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x21",
          "min": 60,
          "max": 600
        }
      ]
    }
  ]
}
```

Idaho947, 2022-11-11:
I try you addition but nothing change. And the values don't change in jeedom when I read the cluster in deconz

Smanar, 2022-11-11:
And with a fingerprint for the illuminance cluster ?

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "OWON",
  "modelid": "PIR313",
  "vendor": "OWON",
  "product": "PIR313",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_PRESENCE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0402",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0500"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/duration"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "config/enrolled",
          "public": false
[... 275 lines cut ...]
        {
          "at": "0x0000",
          "dt": "0x29",
          "min": 60,
          "max": 600,
          "change": "0x0000000A"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 3,
      "cl": "0x0405",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x21",
          "min": 60,
          "max": 600
        }
      ]
    }
  ]
}
```

Idaho947, 2022-11-11:
Batterie is good
I test for luminance and wait a little time but now it's still 0 after a hot reload

Smanar, 2022-11-11:
>Batterie is good

Ha it finally work after some time ?

For illuminance, you have same endpoint than him 
[image]

Idaho947, 2022-11-11:
yes batterie is good after four hours.
It's seem to be the same :
[image]

sirinveen, 2022-11-11:
Hi,
 
 Here my DDF see below and work without issue.
 
 ```json
 {
  "schema": "devcap1.schema.json",
  "manufacturername": "OWON",
  "modelid": "PIR313",
  "vendor": "OWON",
  "product": "PIR313",
  "sleeper": true,
  "status": "Gold",
  "path": "/devices/pir313.json",
  "subdevices": [
    {
      "type": "$TYPE_PRESENCE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/duration",
          "awake": true,
          "refresh.interval": 10,
          "read": {
            "at": "0x0002",
            "cl": "0x0500",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0002",
            "cl": "0x0500",
            "ep": 1,
            "eval": "Item.val = 0;",
            "fn": "zcl"
          },
          "default": 0
        },
[... 324 lines cut ...]
          "dt": "0x20",
          "min": 300,
          "max": 3000,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 3,
      "cl": "0x0405"
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0500"
    }
  ]
}
 ```

And here some picture of the Bindings if that can help you. And yes must be patient to have all attribute refresh. A hot reload can help or a reset.

[image]
[image]

Idaho947, 2022-11-11:
I put it three hours ago and restart my Deconz server. Nothing change. Lux stay at 0

Smanar, 2022-11-12:
Ha, strange @sirinveen , you are not using reporting for illuminance cluster.
You haven't 

```
    {
      "bind": "unicast",
      "src.ep": 2,
      "cl": "0x0400",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x21",
          "min": 60,
          "max": 600,
          "change": "0x000007d0"
        }
      ]
    },
```

And the cluster is not visible on your capture on binding tab.

But instead, you are making an agresssive poll

```
       {
         "name": "state/lightlevel",
         "awake": true,
         "refresh.interval": 30,
```

Every 30s .....

Idaho947, 2022-11-12:
Yep for me non lux with this ddf

Smanar, 2022-11-12:
You have your sensor in a dark room ^^ ?

Idaho947, 2022-11-12:
:-p
No, no in my living room near an other luxmeter to be sure 0 is wrong.

Smanar, 2022-11-12:
Yeah and you have the value in the GUI https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5929#issuecomment-1311552738

And when you read the value in deconz It don't update the value in the API, so it's not a binding/report issue.

On the @sirinveen he ise using a more agressive method but 100% with DDF

```
       {
         "name": "state/lightlevel",
         "awake": true,
         "refresh.interval": 30,
         "read": {
           "at": "0x0000",
           "cl": "0x0400",
           "ep": 2,
           "fn": "zcl"
         },
         "parse": {
           "at": "0x0000",
           "cl": "0x0400",
           "ep": 2,
           "eval": "Item.val = Attr.val;",
           "fn": "zcl"
         },
         "default": 0
       },
```

And his DDF haven't worked neither.

Except looking for problem in logs I haven't more ideas

Idaho947, 2022-11-12:
Yep the value is présent and change :
[image]

What logs do you need ?

Smanar, 2022-11-12:
Can try in deconz / help / debug view with flag "info" and "info_l2" and "ddf"', trying to find something about cluster 0x0400 or state/lightlevel.

Can try to make the device react using a light.

Idaho947, 2022-11-12:
In info when I change the lux :
[image]

Smanar, 2022-11-12:
>Edit : OK it seems to be for temperature and humidity.

No, you are right, the first one is the cluster 0x0400 so it s the good one, so the device is making report

>In the api after each time I read the value for the illuminance, this appears in event :

Ha yes, right good idea to simulate a report. And deconz is sending a notification for change, but its just a "attr" one, not a "state", I think it update the "lastseen".

There is nothing in same time if you use the flag "DDF" ?

Idaho947, 2022-11-12:
I'm in the laste stable 2.19.1.

no nothing move in the ddf log.

Smanar, 2022-11-12:
On the ZCL part i'ts on the cluster 0x0000, on the attribute 0x4000.
Its the SW Build ID, it s because it's still null on the DDF, can be solved by this DDF (have take some part from the @sirinveen DDF

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "OWON",
  "modelid": "PIR313",
  "vendor": "OWON",
  "product": "PIR313",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_PRESENCE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0402",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0500"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "refresh.interval": 84000,
          "read": {"cl": "0x0000", "at": "0x0006", "ep": 1, "fn": "zcl"},
          "parse": {"cl": "0x0000", "at": "0x0006", "ep": 1, "eval": "Item.val = Attr.val", "fn": "zcl"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/duration"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
[... 311 lines cut ...]
          "change": "0x0000000A"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 3,
      "cl": "0x0405",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x21",
          "min": 60,
          "max": 600
        }
      ]
    }
  ]
}
```

But it's exactly the issue. You need to see a "atribute report for cluster 0x400" or "read attribute cl:0x0400" and just after the same log than that 
[image]

But for state/lightlevel for the sensor 3c:6a:2c:ff:fe:d1:b1:9f-02-0400

Smanar, 2022-11-12:
BTW on the capture, we can see "Item.val = measuredValue;"

This part is from the illuminance JS script https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/generic/illuminance_cluster/sml_light_level.js

You haven't the same log but with the complete line ?

Idaho947, 2022-11-12:
I'll try your new ddf and see the result tonight.
I'll try to find the line

Idaho947, 2022-11-12:
I see that in error :
[image]

I don't see where you see : "Item.val = measuredValue; --> 4772"

Smanar, 2022-11-13:
Arf, bad luck.
But the error in the SQL base can be a way to explore.

This error mean deconz try to select an entry in the database withut success, I will post the capture in dev channel to have advice.
Can try to remove the device and re-include it ?

Idaho947, 2022-11-13:
Yes I try Yesterday. But without succès.

Idaho947, 2022-11-14:
I deleted the device : no error. Restart and re-inclusion : error again.

Smanar, 2022-11-14:
Yeah you have 2 notifications, for not changed values, notifications need to happen only if you have changes...

Lux are not working (and it's not normal) but lightlevel work again now ?

The first error mean there is a problem when deconz "parse" the request do,e in the IAS cluster (alarm cluster, used for exemple for presence dectection).
Can happen if the device make a request not handled by the code, not sure it's something important.

The first error is more suspicion for me.

Idaho947, 2022-11-14:
Yes at the moment, température, humidity, lux and batterie are ok in the phoscon webapp 
But the sensor spam every second and cpu is very hight... So I disconnect the batterie.

Smanar, 2022-11-14:
>Motion is not working now

So the error message on ias cluster is not negligible ...

I realy have no clue if its from deconz or the sensor, I have send your last capture to dev channel, hoping for a tips.

Idaho947, 2022-11-14:
Thanks for your action

Idaho947, 2022-11-24:
Hello, an other user try your ddf an same result. Every thing works fine excepte the motion and spam every second.

Smanar, 2022-11-24:
I have a doubt, the problem is not since we have added 

```
        {
          "name": "attr/swversion",
          "refresh.interval": 84000,
          "read": {"cl": "0x0000", "at": "0x0006", "ep": 1, "fn": "zcl"},
          "parse": {"cl": "0x0000", "at": "0x0006", "ep": 1, "eval": "Item.val = Attr.val", "fn": "zcl"}
        },
```

Even you have nothing on log about the cluster 0x0000 or the attribute 0x0006.

Perhaps with setting a defaut value for state/duration for the ZHAPresence ?
```
                {
                    "name": "config/duration",
                    "default": 120
                },
```

And if you remove the ZHALightlevel, the motion work again ?

Idaho947, 2022-11-24:
Well done with the configuration/ duration it works.

Smanar, 2022-11-24:
>Well done with the configuration/ duration it works.

Seriously ? Ok, so good to know this magic.

>But always the spam every second.

Spam concern only the light level ?
If you enable the flag "ZCL" you will probably the zcl request that trigger the events.
Have you tried to remove the part with the TYPE_LIGHT_LEVEL_SENSOR ?

Idaho947, 2022-11-24:
Yes realy !

This commande spam :
[image]

[15 more replies]