dresden-elektronik/deconz-rest-plugin #5939: IKEA switches/buttons stopped working

RouNNdeL, 2022-04-04:
## Describe the bug

3 days ago, the Deconz plugin for Home Assistant would not start, crashing with `SIGTERM`. I read that this can happen if the deconz servers are down, and indeed the issue just resolved itself after a while. However, since then all 3 of my IKEA switches (one on/off, two scene/single push buttons) stopped working. Today I attempted to repair one of them, but it does not work. It appears in the deconz panel, but no attributes can be read from it.

## Steps to reproduce the behavior

The issue happened spontaneously without any changes to the configuration.

## Expected behavior

The switches/buttons should pair

## Screenshots

The switch I attempted to reconnect

[image]

Pressing *read* does not do anything

[image]

The other switches cannot be read either

## Environment

 - Host system: Raspberry Pi 4B
 - Running method:  Home Assistent deCONZ Add-on
 - Firmware version: 0x26700700
 - deCONZ version: 2.14.1
 - Device: ConBee II
 - Do you use an USB extension cable: yes - 3m
 - Is there any other USB or serial devices connected to the host system? Yes, an SSD (ConBee is on USB 2.0 port, SSD on 3.0)
 - 
## deCONZ Logs
```
23:44:06:741     * neighbor: 0xB4E3F9FFFEBAAADD (0xA271), LQI: 252, relation: 0x01, depth: 1, rxOnWHenIdle: 0
23:44:06:756 ZDP status = 0x00 -> SUCCESS
23:44:06:756 Mgmt_Lqi_req zdpSeq: 2 to 0x00212EFFFF077861 start index 2
23:44:06:757 ZDP Mgmt_Lqi_rsp zdpSeq: 1 from 0x00212EFFFF077861 total: 19, startIndex: 1, listCount: 1
23:44:06:757     * neighbor: 0xA4C138F53F2DF12B (0xC833), LQI: 252, relation: 0x02, depth: 1, rxOnWHenIdle: 1
23:44:06:776 ZDP status = 0x00 -> SUCCESS
23:44:06:777 Mgmt_Lqi_req zdpSeq: 3 to 0x00212EFFFF077861 start index 3
23:44:06:777 ZDP Mgmt_Lqi_rsp zdpSeq: 2 from 0x00212EFFFF077861 total: 19, startIndex: 2, listCount: 1
23:44:06:777     * neighbor: 0xA4C1381D5052108D (0x373B), LQI: 252, relation: 0x02, depth: 1, rxOnWHenIdle: 1
23:44:06:791 ZDP status = 0x00 -> SUCCESS
23:44:06:791 Mgmt_Lqi_req zdpSeq: 4 to 0x00212EFFFF077861 start index 4
23:44:06:792 ZDP Mgmt_Lqi_rsp zdpSeq: 3 from 0x00212EFFFF077861 total: 19, startIndex: 3, listCount: 1
23:44:06:792     * neighbor: 0xA4C13828101CF0E6 (0xB723), LQI: 252, relation: 0x02, depth: 1, rxOnWHenIdle: 1
23:44:06:811 ZDP status = 0x00 -> SUCCESS
23:44:06:812 Mgmt_Lqi_req zdpSeq: 5 to 0x00212EFFFF077861 start index 5
23:44:06:812 ZDP Mgmt_Lqi_rsp zdpSeq: 4 from 0x00212EFFFF077861 total: 19, startIndex: 4, listCount: 1
23:44:06:812     * neighbor: 0x0017880109165199 (0x27EB), LQI: 252, relation: 0x01, depth: 1, rxOnWHenIdle: 0
23:44:06:828 ZDP status = 0x00 -> SUCCESS
23:44:06:828 Mgmt_Lqi_req zdpSeq: 6 to 0x00212EFFFF077861 start index 6
23:44:06:829 ZDP Mgmt_Lqi_rsp zdpSeq: 5 from 0x00212EFFFF077861 total: 19, startIndex: 5, listCount: 1
23:44:06:829     * neighbor: 0x00158D00073A3146 (0x958B), LQI: 252, relation: 0x01, depth: 1, rxOnWHenIdle: 0
23:44:06:845 ZDP status = 0x00 -> SUCCESS
23:44:06:846 Mgmt_Lqi_req zdpSeq: 7 to 0x00212EFFFF077861 start index 7
23:44:06:846 ZDP Mgmt_Lqi_rsp zdpSeq: 6 from 0x00212EFFFF077861 total: 19, startIndex: 6, listCount: 1
23:44:06:846     * neighbor: 0x001788010917BF80 (0xED94), LQI: 252, relation: 0x01, depth: 1, rxOnWHenIdle: 0
23:44:06:865 ZDP status = 0x00 -> SUCCESS
23:44:06:866 Mgmt_Lqi_req zdpSeq: 8 to 0x00212EFFFF077861 start index 8
23:44:06:866 ZDP Mgmt_Lqi_rsp zdpSeq: 7 from 0x00212EFFFF077861 total: 19, startIndex: 7, listCount: 1
23:44:06:866     * neighbor: 0x00158D0007C5AA2D (0xE072), LQI: 252, relation: 0x01, depth: 1, rxOnWHenIdle: 0
23:44:06:883 ZDP status = 0x00 -> SUCCESS
23:44:06:884 Mgmt_Lqi_req zdpSeq: 9 to 0x00212EFFFF077861 start index 9
23:44:06:884 ZDP Mgmt_Lqi_rsp zdpSeq: 8 from 0x00212EFFFF077861 total: 19, startIndex: 8, listCount: 1
23:44:06:884     * neighbor: 0x00158D0007DF633C (0x5657), LQI: 248, relation: 0x01, depth: 1, rxOnWHenIdle: 0
23:44:06:900 ZDP status = 0x00 -> SUCCESS
[... 46 lines cut ...]
23:44:11:232 MAC Poll 0x02 0x27EB, life time: 60000 sec, dev timeout: 61440 sec
23:44:11:234 ZCL got data for node=0x27EB, cl=0x0406, at=0x0010, status=0x00, type=0x21
23:44:11:337 ZDP send request id: 0x05 to 0x00212effff077861
23:44:11:379 MAC Poll 0x02 0xED94, life time: 60000 sec, dev timeout: 61440 sec
23:44:11:554 ZDP status = 0x00 -> SUCCESS
23:44:11:554 ZDP Simple_Descriptor_rsp 0x00212EFFFF077861 - 0x0000
23:44:11:772 MAC Poll 0x02 0x27EB, life time: 59998 sec, dev timeout: 61440 sec
23:44:11:909 MAC Poll 0x02 0xED94, life time: 59998 sec, dev timeout: 61440 sec
23:44:12:272 MAC Poll 0x02 0x27EB, life time: 60000 sec, dev timeout: 61440 sec
23:44:12:296 Mgmt_Lqi_req zdpSeq: 21 to 0xA4C1381D5052108D start index 0
23:44:12:324 ZDP status = 0x00 -> SUCCESS
23:44:12:324 ZDP Mgmt_Lqi_rsp zdpSeq: 21 from 0xA4C1381D5052108D total: 5, startIndex: 0, listCount: 2
23:44:12:325     * neighbor: 0x2C1165FFFEEEE5D6 (0xEA01), LQI: 148, relation: 0x01, depth: 2, rxOnWHenIdle: 0
23:44:12:326     * neighbor: 0xA4C13828101CF0E6 (0xB723), LQI: 136, relation: 0x02, depth: 1, rxOnWHenIdle: 1
23:44:12:410 MAC Poll 0x02 0xED94, life time: 60000 sec, dev timeout: 61440 sec
23:44:12:803 MAC Poll 0x02 0x27EB, life time: 60000 sec, dev timeout: 61440 sec
23:44:12:910 MAC Poll 0x02 0xED94, life time: 60000 sec, dev timeout: 61440 sec
23:44:13:336 MAC Poll 0x02 0x27EB, life time: 60000 sec, dev timeout: 61440 sec
23:44:13:442 MAC Poll 0x02 0xED94, life time: 60000 sec, dev timeout: 61440 sec
23:44:13:636 MAC Poll 0x02 0xA271, life time: 59994 sec, dev timeout: 61440 sec
```

## Additional context

All devices <10m away from the coordinator, 3 routers, 25 devices in total.

RouNNdeL, 2022-04-05:
Today, the switches suddenly started working again. I even managed to add back the one switch I attempted to repair. It shows up in Home Assistant and triggers the automations just fine. It does not however appear in deconz and phoscon AT ALL.

Mimiix, 2022-04-05:
Hi,

As this is not a bug as described in #5113, i am closing this issue.

Feel free to open an topic on the forums.

Kind regards,

closed 2022-04-05 by Mimiix

RouNNdeL, 2022-04-05:
Seems like the issue resolved itself with Home Assistant Operating System upgrade to 7.6