dresden-elektronik/deconz-rest-plugin #5947: 01MINIZB unrealistic LQI values

mnkbstard, 2022-04-09:
## Describe the bug
**Bug description**
Sonoff relay 01MINIZB (version 2) do not report correct LQI in deconz gui for the device itself and connected end points.
Relays are responding 100% but the reported LQI is way below the expected, considering some of them are located in close proximity of ConbeeII connected with extension cable.
Downgraded Deconz to 2.12.6 because of the major visualization changes occurred in following version, very low LQI values are still reported

```
"2": {
		"etag": "c94bb3703c0e704c9de73e746286b971",
		"hascolor": false,
		"lastannounced": null,
		"lastseen": "2022-04-09T11:23Z",
		"manufacturername": "SONOFF",
		"modelid": "01MINIZB",
		"name": "RELAY Camera Sud",
		"state": {
			"alert": "none",
			"on": false,
			"reachable": true
		},
		"swversion": "2.0.0",
		"type": "On/Off light",
		"uniqueid": "00:12:4b:00:25:8b:b1:d4-01"
	}

```

## Steps to reproduce the behavior
**Steps to reproduce the behavior**
Deconz gui visualization

## Expected behavior
**Expected behavior**
Realistic LQI values reported from 01MINIZB and connected endpoints

## Screenshots
**Screenshot**
[image]

## Environment
**Environment**

 - Host system: Raspberry Pi 4B 4GB - PC
 - Running method: Raspbian - Mint
 - Firmware version: 26660700
 - deCONZ version: 2.12.06 -> 2.14.1
 - Device: ConBeeII
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? no

## deCONZ Logs

## Additional context

manup, 2022-04-10:
The LQI values are taken directly from the neighbor table entries reported by the 01MINIZB there is nothing we can do about them, might be a bug in the device firmware or indeed RF related issues like interference.

The ZDP messages show the entries of the neighbor tables as reported by the device:

```
10:58:44:247     * neighbor: 0x00124B0026B6CE21 (0xC8C2), LQI: 1, relation: 0x02 rxOnWHenIdle: 2
10:58:44:248     * neighbor: 0x00124B00258BB1D4 (0xBEAF), LQI: 1, relation: 0x02 rxOnWHenIdle: 2
```

mnkbstard, 2022-04-10:
thank you.

i'm closing bug report.

closed 2022-04-10 by mnkbstard