dresden-elektronik/deconz-rest-plugin #5950: Danalock V3 (New FW 0.20.0) Losing Connection

Prior99, 2022-04-10:
## Describe the bug

## Steps to reproduce the behavior

## Expected behavior

## Describe the bug

I bought a Danalock V3 and integrated it into my Deconz system.
After a while the lock will always lose connection to the Zigbee Network and does not do anything anymore. I can neither send commands to it, Nor read clusters. This seems to happen after sending the open or close command. If the lock is just left unattended and only used for reading the clusters, the bug is not triggered.
After rebooting the lock it will work again.

## Steps to reproduce the behavior

1.) Connect lock to network using click commands.
2.) Open / Close the lock multiple times

## Expected behavior

The lock will always open / close (perhaps with some delay).

## Screenshots

[image]
[image]

## Environment

 - Host system: x86 PC
 - Running method: Assistent deCONZ Add-on 
 - Firmware version: (26720700)
 - deCONZ version: (2.14.1)
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? A serial adapter.

## deCONZ Logs
```
15:52:57:149 read attributes of 0x00158D0000F2471C cluster: 0x0006: [ 15:52:57:149 0x0000 15:52:57:149 ]
15:52:57:150 add task 3810383 type 19 to 0x00158D0000F2471C cluster 0x0006 req.id 124
15:52:57:150 Poll APS request 124 to 0x00158D0000F2471C cluster: 0x0006
15:52:57:170 Poll APS confirm 124 status: 0x00
15:52:57:170 Erase task req-id: 124, type: 19 zcl seqno: 231 send time 0, profileId: 0x0104, clusterId: 0x0006
15:52:57:261 Node data 0x00158d0000f2471c profileId: 0x0104, clusterId: 0x0006
15:52:57:261 0x00158D0000F2471C: update ZCL value 0x01/0x0006/0x0000 after 0 s
15:52:57:263 Websocket 172.30.32.1:48864 send message: {"attr":{"colorcapabilities":31,"ctmax":1000,"ctmin":100,"id":"50","lastannounced":"2022-04-05T07:16:37Z","lastseen":"2022-04-10T13:52Z","manufacturername":"innr","modelid":"FL 120 C","name":"Living Room LED-Strip Sofa","swversion":"0x28002162","type":"Extended color light","uniqueid":"00:15:8d:00:00:f2:47:1c-01"},"e":"changed","id":"50","r":"lights","t":"event","uniqueid":"00:15:8d:00:00:f2:47:1c-01"} (ret = 405)
15:52:57:752 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x9922
15:52:57:960 Daylight now: solarNoon, status: 170, daylight: 1, dark: 0
15:52:58:208 Door lock debug 0x50325FFFFECFBE4C, data 0x00000001 
15:52:58:213 Door lock debug 0x50325FFFFECFBE4C, data 0x00000001 
15:52:58:262 poll node 0c:43:14:ff:fe:6a:a1:02-01
15:52:58:263 Poll light node Kitchen Window Cover
15:52:58:460 Idle timer triggered
15:52:58:461 binding for attribute reporting SensorNode Motion Sensor Garage of cluster 0x0406 seems to be active
15:52:58:461 Force read attributes for ZHAPresence SensorNode Motion Sensor Garage
15:52:58:461 binding for attribute reporting of ep: 0x02 cluster 0x0000 seems to be active
15:52:58:461 binding for attribute reporting of ep: 0x02 cluster 0x0406 seems to be active
15:52:58:462 Force binding of attribute reporting for node Motion Sensor Garage
15:52:58:575 read attributes of 0x0C4314FFFE6AA102 cluster: 0x0102: [ 15:52:58:575 0x0008 15:52:58:575 ]
15:52:58:575 add task 3810390 type 19 to 0x0C4314FFFE6AA102 cluster 0x0102 req.id 142
15:52:58:576 Poll APS request 142 to 0x0C4314FFFE6AA102 cluster: 0x0102
15:52:58:597 Poll APS confirm 142 status: 0x00
15:52:58:597 Erase task req-id: 142, type: 19 zcl seqno: 232 send time 0, profileId: 0x0104, clusterId: 0x0102
15:52:58:655 Node data 0x0c4314fffe6aa102 profileId: 0x0104, clusterId: 0x0102
15:52:58:655 0x0C4314FFFE6AA102: update ZCL value 0x01/0x0102/0x0008 after 0 s
15:52:58:665 Websocket 172.30.32.1:48864 send message: {"attr":{"id":"31","lastannounced":"2021-11-01T06:21:32Z","lastseen":"2022-04-10T13:52Z","manufacturername":"_TZ3000_vd43bbfq","modelid":"TS130F","name":"Kitchen Window Cover","productid":"QS-Zigbee-C01 Module","swversion":"0x00000040","type":"Window covering device","uniqueid":"0c:43:14:ff:fe:6a:a1:02-01"},"e":"changed","id":"31","r":"lights","t":"event","uniqueid":"0c:43:14:ff:fe:6a:a1:02-01"} (ret = 398)
15:52:59:252 poll node cc:86:ec:ff:fe:ba:aa:d2-01
15:52:59:253 Poll light node Office Window Cover
15:52:59:547 read attributes of 0xCC86ECFFFEBAAAD2 cluster: 0x0102: [ 15:52:59:547 0x0008 15:52:59:548 ]
15:52:59:548 add task 3810395 type 19 to 0xCC86ECFFFEBAAAD2 cluster 0x0102 req.id 154
[... 51 lines cut ...]
15:53:04:457 poll node 80:4b:50:ff:fe:b5:69:b9-01
15:53:04:458 Poll light node Socket Bedroom Cabinet Light
15:53:04:509 Poll APS request to 0x804B50FFFEB569B9 cluster: 0x0006 dropped, values are fresh enough
15:53:05:460 poll node 00:17:88:01:09:ef:cd:c4-0b
15:53:05:460 Poll light node Living Room TV Backlight Left
15:53:05:510 read attributes of 0x0017880109EFCDC4 cluster: 0x0006: [ 15:53:05:510 0x0000 15:53:05:511 ]
15:53:05:511 add task 3810422 type 19 to 0x0017880109EFCDC4 cluster 0x0006 req.id 228
15:53:05:511 Poll APS request 228 to 0x0017880109EFCDC4 cluster: 0x0006
15:53:06:460 sql exec SELECT conf FROM zbconf ORDER BY rowid desc limit 1
15:53:06:461 Idle timer triggered
15:53:06:461 binding for attribute reporting SensorNode Motion Kitchen of cluster 0x0406 seems to be active
15:53:06:461 Force read attributes for ZHAPresence SensorNode Motion Kitchen
15:53:06:462 binding for attribute reporting of ep: 0x02 cluster 0x0000 seems to be active
15:53:06:462 binding for attribute reporting of ep: 0x02 cluster 0x0406 seems to be active
15:53:06:462 Force binding of attribute reporting for node Motion Kitchen
15:53:06:960 GW firmware version: 0x26720700
15:53:06:960 GW firmware version is up to date: 0x26720700
```

## Additional context

Lock is running the new firmware 0.20.0
I've already contacted Danalock, they've sent me a new lock and assured me that they are testing the locks locally with a Conbee II stick and Zigbee2MQTT and do not have any problems there.

I tried different router of different manufacturers (Phillips, LUMI, innr) and the stick is 2m away from the lock. LQI is between 150 and 200.

Mimiix, 2022-04-10:
Hi,

As this is not a bug report as described in #5113, I am closing this issue. Feel free to open an topic on the forum.

Kind regards,
Dennis

closed 2022-04-10 by Mimiix

Prior99, 2022-04-10:
I can understand that you guys are very busy and like to close issues asap, however this is not a question related to an integration, it's supposed to be a bug report. Something is fishy here related to deconz.

The forums look pretty abandoned to me and I can barely see official support there.

I think in order to get the danalock working properly with deconz there are some fixes needed in deconz itself as the lock is supposedly working with Conbee II on zigbee2mqtt.

Prior99, 2022-04-10:
Topic opened at https://forum.phoscon.de/t/danalock-v3-new-fw-0-20-0-losing-connection/1838

Mimiix, 2022-04-11:
@Prior99  I do understand your concerns. However, we keep bugs here that can clearly be reproduced easily. Anything that isn't obvious to fix is pointed to the forums.  

Also: I am able to direct DE Employees on the forum way easier than here which increases the solvability. I will make sure it gets handled properly :)