dresden-elektronik/deconz-rest-plugin #5952: Device name

KrOnAsK, 2022-04-10:
## Device

- Product name: ZigBee MINI motion sensor
- Manufacturer: unkown
- Model identifier: unknown
- Device type : 

  - Sensor

## Screenshots

[image]
### Basic
 
[image]
### Identify
[image]
### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

Mimiix, 2022-04-10:
The model ID and manufacturer name can be found in the basic cluster.

Also, please add screenshots of all clusters.

KrOnAsK, 2022-04-10:
Okay I updated the information

Mimiix, 2022-04-10:
Looks like this one is a duplicate of https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5584

And it seems that the DDF there works. As there's not a pr with a ddf yet, I'll re open the other one. Perhaps you can submit the pr when it works for you.

closed 2022-04-10 by Mimiix

KrOnAsK, 2022-04-11:
Hi I am kinda new to this all could u explain the abbreviations u used ?

KrOnAsK, 2022-04-11:
Thanks but that is above my level of competency, I have no idea how the DDF works or how to create a good pr.

Mimiix, 2022-04-11:
Feel free to ask there :)

KrOnAsK, 2022-04-11:
Will do thanks again for your help