dresden-elektronik/deconz-rest-plugin #5956: Device name

JayJay68, 2022-04-13:
## Device
- Product name: 511.544 Zigbee 3.0 Wand-Dimmer für RGB / RGB+W / RGBW / RGB+Duo LEDs, Touch R, 4 Lichtzonen | weißes Glas
- Manufacturer: iluminize
- Model identifier: ZGRC-TEUR-001
- Device type : Please remove all unrelated device types. 
    - Switch
 - Remote
 - Iluminize has branded this device. About 15 month ago I bought the same one (511.544) from the iluminize-shop. The manufacturer (Basic Cluster) is not iluminize but sunricher.
- not battery powered
- product page: https://www.iluminize.com/de/shop/led-steuerung/wand-dimmer/product/499-511-544-zigbee-wand-dimmer-schalter-rgb-rgbw-rgbws-led-4z.html
- picture:
[image]

If you need further information, please let me know.

## Screenshots
### Node-Info:
[image]

[image]

### Basic
[image]

### Identify
[image]

### Diagnostics
[image]
[image]

### Groups
All Values empty, 0 or 0x0000

### On/Off
Empty

### Level Control
Empty

### Color Control
Empty

### Other clusters that are not mentioned above
OTAU and ZLL comissioning: both empty or all values 0, 0x000,...

Mimiix, 2022-04-13:
Please add all screenshots.

JayJay68, 2022-04-13:
I have edit my first post with all screenshots for 
[image]

Do you need all screnshots for "02 HA", "03 HA" and "04 HA", too (they have all the same values)?

And I found at last this one:
[image]

### Green Power
[image]

JayJay68, 2022-04-23:
Do you have all information to integrate the device, now?

Mimiix, 2022-04-23:
Looks okay now :)

@Smanar is this doable with ddf?

Smanar, 2022-04-23:
IDK, on m y side I have never succed with switch yet.

SwoopX, 2022-04-23:
That one should already do the trick

[4grp_dimmer_switch.zip](https://github.com/dresden-elektronik/deconz-rest-plugin/files/8548621/4grp_dimmer_switch.zip)

JayJay68, 2022-04-27:
Thank you - But what do I have to do with the JSON-file that is in the ZIP file? Copy somewhere?

Smanar, 2022-04-27:
Yep it's a DDF file. https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet

Just need to copy it in a "devices" folder.

JayJay68, 2022-04-28:
OK, I think that this file could work. What version of deCONZ do you recommend for using the DDF-File (above)? I have currently 2.12.06? If I have to update to a newer version, please recommend only a version that is stable enough for production (using at home from the family)?
Thank you very much for your help.

Smanar, 2022-04-29:
This stable https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/v2.14.1 is 2 months old, so I think now all issues on it are corrected.

BTW, don't forget to make a backup of zigbee network using phoscon, will help if you want to rollback.

Its the version used in HA (they are using only stable)

JayJay68, 2022-05-22:
I have to solve other issues first.

2022, 04:00:

JayJay68, 2022-06-23:
Please let it opened. I need much more time to solve other problems with higher priority.

closed 2022-07-23 by github-actions[bot]