dresden-elektronik/deconz-rest-plugin #5959: Wattle D Pro Multipunkt

jokerejoker, 2022-04-14:
## Device

- Product name: Wattle D Pro Multipunkt
- Manufacturer: Home Control AS
- Model identifier: HC-SLM-1
- Device type:
  - Lock

## Screenshots

### Basic

[image]

### Power Configuration

[image]

### Identify

[image]

### Groups

[image]

### Scenes

[image]

### Power Configuration

[image]

### Door Lock

[image]

Smanar, 2022-04-14:
Hello.
The support for doorlock is not yet official, making one here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5870

can try this DDF to make tries.

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "Home Control AS",
  "modelid": "HC-SLM-1",
  "vendor": "Wattle",
  "product": "Wattle D Pro Multipunkt",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_DOOR_LOCK",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery"
        },
        {
          "name": "config/lock",
          "read": {
            "at": "0x0000",
            "cl": "0x0101",
            "ep": 11,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0000",
            "cl": "0x0101",
            "ep": 11,
            "eval": "Item.val = Attr.val == 1",
            "fn": "zcl"
          }
        },
        {
          "name": "state/lockstate",
          "read": {
            "at": "0x0000",
            "cl": "0x0101",
            "ep": 11,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0000",
            "cl": "0x0101",
            "ep": 11,
            "eval": "if (Attr.val == 1) { Item.val = 'open' } else if (Attr.val == 2) { Item.val = 'closed' } else { Item.val = 'error' }",
            "fn": "zcl"
          }
        },
        {
          "name": "state/open",
          "read": {
            "at": "0x0003",
            "cl": "0x0101",
            "ep": 11,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0003",
            "cl": "0x0101",
            "ep": 11,
            "eval": "if (Attr.val == 0) { Item.val = 'open' } else if (Attr.val == 1) { Item.val = 'closed' } else { Item.val = 'error' }",
            "fn": "zcl"
          }
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 11,
      "cl": "0x0101",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x30",
          "min": 1,
          "max": 300
        },
        {
          "at": "0x0003",
          "dt": "0x30",
          "min": 1,
          "max": 300
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 11,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0020",
          "dt": "0x20",
          "min": 300,
          "max": 2700,
          "change": "0x00000001"
        }
      ]
    }
  ]
}

```

jokerejoker, 2022-04-15:
Thanks, will try it out.

My previous danalock was just recognized as a light bulb and that work fine. Any chance that could be a temporary solution, until the door lock support is implemented?

Anything i can do to help on the door lock support?

Smanar, 2022-04-15:
Long time door lock are no more reconized as bulbs (even the code is already here for old device like the danalock)
They are now sensor, ZHADoorlock, what is your third app ? They support probably this sensor.

>Anything i can do to help on the door lock support?

Yep sure, testing the DDF ^^. DDF is the future, I can make change on the core but only to help for DDF support.

jokerejoker, 2022-04-15:
What do you mean by my third app? Is it home assistant you are referring to?

Smanar, 2022-04-15:
Yep, for exemple.
And I can confirm, The HA plugin support ZHADoorlock https://github.com/home-assistant/core/pull/48516

jokerejoker, 2022-04-16:
After adding the DDF to the devices folder, the "Add new sensor" wizard reacted by "New sensor ready", but the lock does not appear afterward in the list of sensors, in Phoscon.

This is the result when pulling the /sensors API:

```
{
    "5": {
        "config": {
            "battery": null,
            "on": true,
            "reachable": true
        },
        "etag": "c8e8698169d827b5044498cb7cc54911",
        "lastannounced": "2022-04-16T16:03:01Z",
        "lastseen": "2022-04-16T16:07Z",
        "manufacturername": "Home Control AS",
        "modelid": "HC-SLM-1",
        "name": "$TYPE_DOOR_LOCK 5",
        "state": {
            "lastupdated": "none",
            "open": null,
            "reachable": null
        },
        "type": "$TYPE_DOOR_LOCK",
        "uniqueid": "68:0a:e2:ff:fe:7b:49:40-01"
    }
}
```

[image]

I think we are close, but I am not too familiar with the DDF declarations yet, so please advice on what is wrong and what should be corrected.

Mimiix, 2022-04-16:
It wont show in phoscon as they need to add support for it.

jokerejoker, 2022-04-16:
Okay, and that is what is going on at issue #5870 ?

Smanar, 2022-04-16:
>"type": "$TYPE_DOOR_LOCK",

Wich one deconz version are you using ? There is so much things missing.

>Okay, and that is what is going on at issue https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5870 ?

Nothing special, just he don't have so much time for test.

jokerejoker, 2022-04-16:
I just quickly tested it with the deconz packaged with the hassio (home assistant) core add-on for deconz. It is version 2.14.1

jokerejoker, 2022-04-16:
@Smanar any thoughts ?

Smanar, 2022-04-16:
Yep you need this PR https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5906, so at least the version v2.15.1

jokerejoker, 2022-04-19:
I will build and test a version, of the HA deCONZ Add-on, with the 2.15.1-beta as soon as possible

Smanar, 2022-04-19:
You don't need to build/recompile the library, the 2.15.1 is available, but not sure how to update it on HA.

jokerejoker, 2022-04-20:
Yes, it was also for the HA plugin that I ment :)

I any ETA on the 2.15.1 stable ?

Smanar, 2022-05-03:
New stable out https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/v2.15.3

Tell me if the DDF is working ?

jokerejoker, 2022-05-08:
Awesome! I will test it with in the next days.

FYI - HA addon has been officially updated with the new version

jokerejoker, 2022-05-15:
This appear to be the result from the API.

[image]

Execution of lock/unlock command does not appear, lock state does not seam to come through and to be working and battery state is not available. This is properbly due to the DDF being configured wrong?

Also the lock is not recognized correctly by the HA plugin, but is might not be you who need to take a look at that :)

[image]

I well get back to about if the lock disconnects it self. How long would you expect for to do so ?

Smanar, 2022-05-16:
Oups sorry m y bad, I m using bad endpoint on the DDF (you need the 0x01 and I m using "11"
Can you try this one

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "Home Control AS",
  "modelid": "HC-SLM-1",
  "vendor": "Wattle",
  "product": "Wattle D Pro Multipunkt",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_DOOR_LOCK",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery"
        },
        {
          "name": "config/lock",
          "read": {
            "at": "0x0000",
            "cl": "0x0101",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0000",
            "cl": "0x0101",
            "ep": 1,
            "eval": "Item.val = Attr.val == 1",
            "fn": "zcl"
          }
        },
        {
          "name": "state/lockstate",
          "read": {
            "at": "0x0000",
            "cl": "0x0101",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0000",
            "cl": "0x0101",
            "ep": 1,
            "eval": "if (Attr.val == 1) { Item.val = 'open' } else if (Attr.val == 2) { Item.val = 'closed' } else { Item.val = 'error' }",
            "fn": "zcl"
          }
        },
        {
          "name": "state/open",
          "read": {
            "at": "0x0003",
            "cl": "0x0101",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0003",
            "cl": "0x0101",
            "ep": 1,
            "eval": "if (Attr.val == 0) { Item.val = 'open' } else if (Attr.val == 1) { Item.val = 'closed' } else { Item.val = 'error' }",
            "fn": "zcl"
          }
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0101",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x30",
          "min": 1,
          "max": 300
        },
        {
          "at": "0x0003",
          "dt": "0x30",
          "min": 1,
          "max": 300
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0020",
          "dt": "0x20",
          "min": 300,
          "max": 2700,
          "change": "0x00000001"
        }
      ]
    }
  ]
}
```

>I will get back to about if the lock disconnects it self. How long would you expect for to do so ?

The previous user that have the issue have a DC after somes days, but there is some improvement ATM for this issue on the Danalock one.

closed 2022-06-14 by github-actions[bot]

micv83, 2022-09-12:
When will this DDF be included in the officiel release ?

Mimiix, 2022-09-12:
@micv83 once we know it works :)

micv83, 2022-09-12:
Would love to test the DDF, but running the official deconz addon there is no easy way, as far as I know ?

Smanar, 2022-09-12:
Hello, what is your OS, I think the " official deconz addon" mean HA ? https://forum.phoscon.de/t/how-to-add-edit-a-ddf-on-home-assistant-using-text-editor/1839

micv83, 2022-09-13:
Sorry, yes it's HA :-) That is actually a pretty good guide. Managed to edit the DDF, so i'll try testing the device later today or tomorrow.

micv83, 2022-09-13:
Uploaded your DDF @Smanar, and the lock is now recognized as a lock, but it is not exposed to HA. 

[image]

Smanar, 2022-09-13:
Ha I have exactly the same problem here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5870#issuecomment-1245631892

micv83, 2022-09-13:
Okay 😝 Hope somebody figure out the problem. It’s supported in zigbee2mqtt, so I might have to give that a try😊

Smanar, 2022-09-13:
Ok, so can you try this new DDF pls ?

{
  "schema": "devcap1.schema.json",
  "manufacturername": "Home Control AS",
  "modelid": "HC-SLM-1",
  "vendor": "Wattle",
  "product": "Wattle D Pro Multipunkt",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "ZHADoorLock",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery"
        },
        {
          "name": "config/lock",
          "read": {
            "at": "0x0000",
            "cl": "0x0101",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0000",
            "cl": "0x0101",
            "ep": 1,
            "eval": "Item.val = Attr.val == 1",
            "fn": "zcl"
          }
        },
        {
          "name": "state/lockstate",
          "read": {
            "at": "0x0000",
            "cl": "0x0101",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0000",
            "cl": "0x0101",
            "ep": 1,
            "eval": "if (Attr.val == 1) { Item.val = 'open' } else if (Attr.val == 2) { Item.val = 'closed' } else { Item.val = 'error' }",
            "fn": "zcl"
          }
        },
        {
          "name": "state/open",
          "read": {
            "at": "0x0003",
            "cl": "0x0101",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0003",
            "cl": "0x0101",
            "ep": 1,
            "eval": "if (Attr.val == 0) { Item.val = 'open' } else if (Attr.val == 1) { Item.val = 'closed' } else { Item.val = 'error' }",
            "fn": "zcl"
          }
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0101",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x30",
          "min": 1,
          "max": 300
        },
        {
          "at": "0x0003",
          "dt": "0x30",
          "min": 1,
          "max": 300
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0020",
          "dt": "0x20",
          "min": 300,
          "max": 2700,
          "change": "0x00000001"
        }
      ]
    }
  ]
}

micv83, 2022-09-14:
A bit better, the lock is now partially presented to HA. But the lock item is unavailable and battery status shows 0%...

[image]

Smanar, 2022-09-14:
I m making the correction on the other issue, but for the battery, can you check in deconz where I can find it ?
On the capture on the first post, on the power configuration cluster, both values are 0.

micv83, 2022-09-15:
After pushing read i deconz Power Configuration Cluster i get this:

[image]

Smanar, 2022-09-15:
Ok so let's try with the attribute 0x0021

Take care this new DDF will generate a new unique ID.

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "Home Control AS",
  "modelid": "HC-SLM-1",
  "vendor": "Wattle",
  "product": "Wattle D Pro Multipunkt",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "ZHADoorLock",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0101"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery"
        },
        {
          "name": "config/lock",
          "read": {
            "at": "0x0000",
            "cl": "0x0101",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0000",
            "cl": "0x0101",
            "ep": 1,
            "eval": "Item.val = Attr.val == 1",
            "fn": "zcl"
          }
        },
        {
          "name": "state/lockstate",
          "read": {
            "at": "0x0000",
            "cl": "0x0101",
            "ep": 11,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0000",
            "cl": "0x0101",
            "ep": 11,
            "eval": "if (Attr.val == 0) { Item.val = 'not fully locked' } else if (Attr.val == 1) { Item.val = 'locked' } else if (Attr.val == 2) { Item.val = 'unlocked' } else { Item.val = 'undefined' }",
            "fn": "zcl"
          }
        },
        {
          "name": "state/open",
          "read": {
            "at": "0x0003",
            "cl": "0x0101",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0003",
            "cl": "0x0101",
            "ep": 1,
            "eval": "if (Attr.val == 0) { Item.val = 'open' } else if (Attr.val == 1) { Item.val = 'closed' } else { Item.val = 'error' }",
            "fn": "zcl"
          }
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0101",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x30",
          "min": 1,
          "max": 300
        },
        {
          "at": "0x0003",
          "dt": "0x30",
          "min": 1,
          "max": 300
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 300,
          "max": 2700,
          "change": "0x00000001"
        }
      ]
    }
  ]
}
```

micv83, 2022-09-15:
Result:

[image]

Smanar, 2022-09-16:
And it works or not ?

micv83, 2022-09-20:
I've tried changing to:

 {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2"
          }
        },

Still shows 0%, but will wait 24h.

micv83, 2022-09-20:
Okay, after pressing read again in Deconz so 0x0021 shows 199 again, battery status updates in HA to 100% :-)

micv83, 2022-09-26:
Any news @Smanar ?

Mimiix, 2022-09-26:
@micv83  Just to let you know: Smanar is on Holiday. I think he will be back in a few days.

micv83, 2022-09-26:
Okay, and in a few days i will be on holiday :-)

micv83, 2022-09-27:
Only the batterystatus worked with the last changes. Did you make any changes to the lock part, in the latest DDF ?

Smanar, 2022-09-27:
Oups my bad, sorry it s the new DDF (working on another device)

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "Home Control AS",
  "modelid": "HC-SLM-1",
  "vendor": "Wattle",
  "product": "Wattle D Pro Multipunkt",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "ZHADoorLock",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0101"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2"
          }
        },
        {
          "name": "config/lock",
          "read": {
            "at": "0x0000",
            "cl": "0x0101",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0000",
            "cl": "0x0101",
            "ep": 1,
            "eval": "Item.val = Attr.val == 1",
            "fn": "zcl"
          }
        },
        {
          "name": "state/lockstate",
          "read": {
            "at": "0x0000",
            "cl": "0x0101",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0000",
            "cl": "0x0101",
            "ep": 1,
            "eval": "if (Attr.val == 0) { Item.val = 'not fully locked' } else if (Attr.val == 1) { Item.val = 'locked' } else if (Attr.val == 2) { Item.val = 'unlocked' } else { Item.val = 'undefined' }",
            "fn": "zcl"
          }
        },
        {
          "name": "state/open",
          "read": {
            "at": "0x0003",
            "cl": "0x0101",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0003",
            "cl": "0x0101",
            "ep": 1,
            "eval": "Item.val = Attr.val !=1",
            "fn": "zcl"
          }
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0101",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x30",
          "min": 1,
          "max": 300
        },
        {
          "at": "0x0003",
          "dt": "0x30",
          "min": 1,
          "max": 300
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 300,
          "max": 2700,
          "change": "0x00000001"
        }
      ]
    }
  ]
}
```

micv83, 2022-09-27:
Still no luck, with the lock part. 

Changed DDF, did hot reload, also complete restart of Deconz and reinitialize Deconz plugin in HA. 

[image]

Smanar, 2022-09-27:
Have you tried to delete the device in HA and resynchronise it ?
Perhaps an error on HA logs ?

Can you share how look the device json ? you can see it in phoscon / help / API Information / sensor

micv83, 2022-09-27:
Tried removing the batteries from the lock an insert again. Now the lock status is shown when unlocking directly on the lock. 😄 But Pushing lock or unlock in HA does nothing.

Smanar, 2022-09-28:
```
"state": {
  "lastupdated": "none",
  "lockstate": null,
  "open": null,
  "reachable": null
},
```

I think here, you have new values now ? reachable is now true ?
It's the only field that can prevent HA to make the request. So if the value is true, you have something in log.

Can try too direclty with the API.

`curl -H 'Content-Type: application/json' -X PUT -d '{"lock": true}' http://IP:PORT/api/KEY/sensors/ID/config`

IP and PORT are the same you can see when using phoscon
KEY is an API key, can use the same used by HA
ID is the device id, probably 65 from your last capture.

micv83, 2022-09-28:
Reachable is still null:

{
    "config": {
        "battery": 100,
        "lock": true,
        "on": true,
        "reachable": true
    },
    "etag": "08487aa99b30bfa47f94a3e59b86717b",
    "lastannounced": "2022-09-27T23:20:29Z",
    "lastseen": "2022-09-28T12:04Z",
    "manufacturername": "Home Control AS",
    "modelid": "HC-SLM-1",
    "name": "DoorLock 65",
    "state": {
        "lastupdated": "2022-09-28T12:04:32.359",
        "lockstate": "locked",
        "open": null,
        "reachable": null
    },
    "type": "ZHADoorLock",
    "uniqueid": "68:0a:e2:ff:fe:7b:60:cc-01-0101"
}

Smanar, 2022-09-28:
```
"state": {
   "lastupdated": "2022-09-28T12:04:32.359",
   "reachable": null
},
```

Ok so new issue ....
But you are on HA ? so you are using the last stable version for deconz available for it ?

Smanar, 2022-09-28:
@micv83 and if you use 

`"sleeper": true,`

micv83, 2022-09-28:
I’m on deconz version 2.17.1 I have 2.18.2 available, but don’t know if an update will wipe the custom ddf, I’m away at the moment so won’t be able to upload the ddf again.

Smanar, 2022-09-28:
ok, I m asking some tips to other devs, I will give you a new DDF "last hope" soon.

Smanar, 2022-09-28:
Ok so ebaauw have found the probleml, the DDF was made with the first one (from may), It was totally out. I m using state/reachable, and for sensor It's config/reachable.

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "Home Control AS",
  "modelid": "HC-SLM-1",
  "vendor": "Wattle",
  "product": "Wattle D Pro Multipunkt",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "ZHADoorLock",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0101"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2"
          }
        },
        {
          "name": "config/lock",
          "read": {
            "at": "0x0000",
            "cl": "0x0101",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0000",
            "cl": "0x0101",
            "ep": 1,
            "eval": "Item.val = Attr.val == 1",
            "fn": "zcl"
          }
        },
        {
          "name": "state/lockstate",
          "read": {
            "at": "0x0000",
            "cl": "0x0101",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0000",
            "cl": "0x0101",
            "ep": 1,
            "eval": "if (Attr.val == 0) { Item.val = 'not fully locked' } else if (Attr.val == 1) { Item.val = 'locked' } else if (Attr.val == 2) { Item.val = 'unlocked' } else { Item.val = 'undefined' }",
            "fn": "zcl"
          }
        },
        {
          "name": "state/open",
          "read": {
            "at": "0x0003",
            "cl": "0x0101",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0003",
            "cl": "0x0101",
            "ep": 1,
            "eval": "Item.val = Attr.val != 1",
            "fn": "zcl"
          }
        },
        {
          "name": "config/reachable"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0101",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x30",
          "min": 1,
          "max": 300
        },
        {
          "at": "0x0003",
          "dt": "0x30",
          "min": 1,
          "max": 300
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 300,
          "max": 2700,
          "change": "0x00000001"
        }
      ]
    }
  ]
}
```

micv83, 2022-09-29:
Super! I’ll test it as soon as I have returned from holiday. 😊

micv83, 2022-10-08:
State in now correcly displayed when locked/unlocked from the lock, but i'm still not able to lock/unlock from HA.

But it shows reachable in API now.
{
    "config": {
        "battery": 100,
        "lock": false,
        "on": true,
        "reachable": true
    },
    "etag": "65fa13a7527403ed7852311da07c1ec0",
    "lastannounced": "2022-10-08T17:22:58Z",
    "lastseen": "2022-10-08T17:23Z",
    "manufacturername": "Home Control AS",
    "modelid": "HC-SLM-1",
    "name": "DoorLock 65",
    "state": {
        "lastupdated": "2022-10-08T17:23:59.541",
        "lockstate": "locked",
        "open": null
    },
    "type": "ZHADoorLock",
    "uniqueid": "68:0a:e2:ff:fe:7b:60:cc-01-0101"
}

Smanar, 2022-10-09:
All seem fine, even the lastupdated have worked, except the state/open=null, perhaps a bind problem, canbe solved later by deconz.
You don't have error message in HA ?

Can you try direclty with API

`curl -H 'Content-Type: application/json' -X PUT -d '{"lock": true}' http://IP:PORT/api/KEY/sensors/ID/config`

IP and PORT are the same used than phoscon
KEY is an apikey, can use the same one than the one used by HA
ID is the device id.

micv83, 2022-10-09:
Tried the 2 changes, but then i was unable to even get the status from the lock, so went back to the DDF from before and got the status working again. Hope I wasn’t to impatient, as the lock sometimes takes a bit to update correctly after a deconz restart. If the changes should work, I can try them again.

Don’t know how to extract the apikey from HA or deconz, to do the api call manually?

Smanar, 2024-12-21:
Next here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/8073