dresden-elektronik/deconz-rest-plugin #5960: BCM500DS-TYZ-22 support

Pherax, 2022-04-14:
## Device

- Product name: BCM500DS-TYZ-22
- Manufacturer: _TZE200_rmymn92d.
- Model identifier: TS0601.
- Device type : 
  - Other:  Cover

## Screenshots

### Basic

### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above
I am about to receive this device (do not have it yet). it is already integrated in zha and zigbee2mqtt

https://github.com/zigpy/zha-device-handlers/issues/1294

https://www.zigbee2mqtt.io/devices/TS0601_cover.html

Mimiix, 2022-04-14:
Hi,

If you aren't able to provide screenshots, it's hard to implement the device.

I'll add a label for now. Please add screenshots when you have it.

Kind regards,

Pherax, 2022-04-30:
I have updated the start post with the requested screenshots.

Pherax, 2022-05-06:
> Hi,
> 
> If you aren't able to provide screenshots, it's hard to implement the device.
> 
> I'll add a label for now. Please add screenshots when you have it.
> 
> Kind regards,

when will you remove the labels?

Mimiix, 2022-05-06:
HI,

I missed your previous post somehow. Sorry!

I've removed them.

Kind regards,

Mimiix, 2022-05-06:
Can you please post a screen of the cluster list on the device?

Mimiix, 2022-05-06:
Yes :)

@smanar can this be done with a DDF?

Smanar, 2022-05-06:
Not yet, sorry, we are finishing the DDF for TRV, if this one work will do same for all other device.
But there is probably a new issue ATM on return from tuya device, even just reading a temperature is broken, I m not sure, I m asking to manup to be sure.

closed 2022-06-04 by github-actions[bot]

Schloerhome, 2022-07-04:
Hi any update on this please ? Got the same device

patrusco, 2023-05-09:
Hi all, I have the same device and would be happy to help with testing @Pherax  can you help ?

Smanar, 2023-05-09:
The support for tuya covering is not possible ATM with DDF.

patrusco, 2023-05-09:
> The support for tuya covering is not possible ATM with DDF.
I do see the device but identified as a switch. I was able to use iobroker to close the curtain but opening would be all I require more to get me going with my planned automation..

below the config I got after discovery 

{
    "1": {
        "etag": "e343140db698b628c616c3d53104e192",
        "hascolor": false,
        "lastannounced": null,
        "lastseen": "2023-05-09T17:41Z",
        "manufacturername": "dresden elektronik",
        "modelid": "RaspBee II",
        "name": "Configuration tool 1",
        "state": {
            "reachable": true
        },
        "swversion": "0x26780700",
        "type": "Configuration tool",
        "uniqueid": "00:21:2e:ff:ff:08:b7:a3-01"
    },
    "2": {
        "capabilities": {
            "alerts": [
                "none",
                "select",
                "lselect"
            ]
        },
        "config": {
            "groups": [
                "0",
                "1"
            ]
        },
        "etag": "3f33b981c18366c84b449f59ea1b3a26",
        "hascolor": false,
        "lastannounced": null,
        "lastseen": "2023-05-09T17:40Z",
        "manufacturername": "_TZE200_rmymn92d",
        "modelid": "TS0601",
        "name": "Smart plug 2",
        "state": {
            "alert": "none",
            "on": false,
            "reachable": true
        },
        "swversion": null,
        "type": "Smart plug",
        "uniqueid": "a4:c1:38:2e:8b:59:5b:30-01"
    }
}

patrusco, 2023-05-28:
> The support for tuya covering is not possible ATM with DDF.

It was possible with the standard discovery above to move the curtain in one direction. So does it mean that this is still not possible? Im not an expert here but one function seams to work so I would assume it is possible to correct the recognition and have the device working? thanks for your help!

Smanar, 2023-05-28:
It's not possible to use custom request for covering and more for Tuya one.
It's perhaps possible with this PR https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6936/files#diff-75257ab83b06c4302e62c9688cd9d6da01b7b773c5a24a436330464d5badc906R1874 but it's something new, never tried yet.

patrusco, 2023-06-11:
Thanks, can you guide me to a "how to" install this? I'm unfortunately not experienced at this point.

patrusco, 2023-06-16:
Thanks, happy to do so but how do I share a DDF? I have a phoscon gateway where I'm able to login with ssh. what would be the steps required to get you the DDF?

Smanar, 2023-06-16:
Ha yep, we can have a problem ^^, you need the GUI to do that ?
What is your OS ?

patrusco, 2023-06-22:
I'm running MAC OS on a Macbook Pro but I got windows VMs and docker containers if that helps..

Smanar, 2023-06-23:
And where is running deconz ?
To know if you have an easy way to access GUI else I can try to make a DDF from scratch, so you will just need to install it.

patrusco, 2023-06-24:
I'm running deconz as part of IOBroker which runs as part of a docker on my synology

Smanar, 2023-06-24:
Ok So I think you are using the official docker image .

Can try this prototype

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZE200_rmymn92d",
  "modelid": "TS0601",
  "product": "BCM500DS-TYZ-22 covering",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_WINDOW_COVERING_DEVICE",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/on",
          "parse": {"fn": "tuya", "dpid": 1, "eval": "Item.val = Attr.val == 0x01;" },
          "write": {"fn": "tuya", "dpid": 1, "dt": "0x30", "eval": "Item.val == 1 ? 0x01 : 0x02;"},
          "read": {"fn": "tuya"}
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ],
  "bindings": [
  ]
}
```

It will support only "state/on" to open/close the covering.
This one need to work, if yes you can replace "state/on" by "state/open", then delete the device and re-include it for it use the new DDF.
But not sure it will work with "state/open".

Command can be reversed, it's just for test.

patrusco, 2023-06-24:
Thanks, great! .. but where do I put this, sorry for the question but I never did edit the .json before..

Smanar, 2023-06-24:
On the official docker image, you just need to put your DDF file in the `/opt/deCONZ` folder https://github.com/deconz-community/deconz-docker/issues/163

Just create a new file called what_you_want.json with the DDF contain.

patrusco, 2023-06-24:
Hey @Smanar, I was able to install the DDF and re-discovered the device. Unfortunately I'm not able to steer anything but  I did get a status back if the device moves (on argument goes from "true" to "false"). any other thing I can try?

Smanar, 2023-06-25:
>(on argument goes from "true" to "false"

It's "state/on" ?
How are you trying to move it ? using direclty the API or using third app ?

With API and a console

`curl -H 'Content-Type: application/json' -X PUT -d '{"on":false}' http://IP:PORT/api/API_KEY/lights/ID/state`

It's hard to get logs on an headless OS, but how many devices have you ? it's your dev environnement ?

patrusco, 2023-06-25:
I move it by using iobrokers objects where I can read and write(screenshot attached). yes it is my environment. 
the curl command I can not execute on the iobroker level I guess or I need to get hold on the API key which must be somewhere..I try to figure that out..

[image]

Smanar, 2023-06-25:
On your picture you are on the "alert" field, and this one is removed on the test DDF, but yes it need to work too.
You are sure you are on the good field ? you haven't log to see the send request by iobrokers  ?

```
          "parse": {"fn": "tuya", "dpid": 1, "eval": "Item.val = Attr.val == 0x01;" },
          "write": {"fn": "tuya", "dpid": 1, "dt": "0x30", "eval": "Item.val == 1 ? 0x01 : 0x02;"},
```

The return is handled by the "parse" part, and if this one is working, I don't see why the "write" part is not working. I m using the same dpid, the dt is good too (enum), and the value need to be good at least for 1 side.

patrusco, 2023-06-26:
iobroker/deconz-module creates the elements automatically based on what's comming from my phoscon gateway. On the phoscon, I was able to build your DDF. I think I can confirm that since I see the correct label "window covering device" within iobroker, but don't ask me where the alert is comming from ;).. I do still have the initial DDF which resides within :/usr/share/deCONZ/devices/tuya,. I have not changed anything there. might this also be used in parallel to your DDF?

I have put the deconz module in iobroker into debug mode. I can see the following message:

`

  | Code  400: Bad request set light state 2:  [{"error":{"address":"/lights/2/state","description":"body contains  invalid JSON","type":2}}]
-- | --

`

Smanar, 2023-06-26:
> body contains invalid JSON

There is something bad in the request, but this part happen before the DDF, Can be  iobroker that make it.

>In addition I was able to execute your command above with the following result

And the covering don't move ? can try with true and false ?

patrusco, 2023-06-26:
> And the covering don't move ? can try with true and false ?

unfortunately not. no movement.. is there any other thing I can try to send?

Smanar, 2023-06-27:
>I see "on: false" which goes back to "on: true" after the movement

Can be my fault, on the code I m using 2 way, the dpid 0x01 (the state, apparently not reliable) or 0x02 (going to position) and 0x03 (position reached).
Other thing your device need to reverse the position return, and it's based on position so you have 1 closed position and 99 open, the value will change only on full closed or full open.
Perhaps better idea to use 

"parse": {"fn": "tuya", "dpid": 1, "eval": "Item.val = Attr.val != 0x01;" },

or

`"parse": {"fn": "tuya", "dpid": 3, "eval": "Item.val = Attr.val < 5;" },`

This is for return, for command send

Tuya covering use thoses dpid for movement
- close: 2
- open: 0
- stop: 1

How many device have you ? There is a way to run deconz on debug mode on the docker, can be usefull to catch the log when deconz make the request to see what is missing.

Edit :

Ha ^^, I have see something strange

>   curl -H 'Content-Type: application/json' -X PUT -d '{"on":false}' http://IP:PORT/api/API_KEY/lights/ID/state

[{"success":{"/lights/2/state/open":true}}]

You are using "state/on" but you have in the response "state/open" and to prevent that, I m using only "state/on" on the DDF to prevent problem with state/open

patrusco, 2023-06-30:
on the curl command above, yes this is what I'm getting back. So what do you suggest I should do? Just to be sure we are on the same page:

- The curl command I'm sending from my mac to my phoscon gateway directly
- The above logs I took from deconz docker/iobroker module which was set to debug in order to get the websocket messages

regarding your parse command changes, what would be the fastest way to try this out? I assume this would be a change on the phoscon gateway / deconz instance? Can I change the DDF without re-learning the device? this took me about 40min before it finally was detected. 
Thanks!

Smanar, 2023-06-30:
40 mn ?
If you have the GUI there is a command "hot reload" on the DDF editor.
Else you can just edit the DDF and restart deconz. But I m not sure the DDF can update the "type" without a re-inclusion.

> the same one but replacing $TYPE_WINDOW_COVERING_DEVICE by $TYPE_ON_OFF_LIGHT

For the "parse" test, I m sure, just a deconz restart and he will use the new command, but the "type" change will have more impact, to test if we can send request to the device using DDF

Smanar, 2023-07-02:
You need a desktop environnement or use VNC "stuff" to have the GUI and yes the web app is another thing.

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Access-deCONZ-GUI-in-headless-setups

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZE200_rmymn92d",
  "modelid": "TS0601",
  "product": "BCM500DS-TYZ-22 covering",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_ON_OFF_LIGHT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/on",
          "parse": {"fn": "tuya", "dpid": 1, "eval": "Item.val = Attr.val != 0x01;" }
          "write": {"fn": "tuya", "dpid": 1, "dt": "0x30", "eval": "Item.val == 1 ? 0x01 : 0x02;"},
          "read": {"fn": "tuya"}
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ],
  "bindings": [
  ]
}
```

On this DDF you have 2 modifcation, the one for the type (will replace the covering by a on/off device) and one to reverse the state/on return.

But not sure the deconz restart will be enought for the type change.

patrusco, 2023-07-02:
**awesome!! success!! finally I got some movement**
If I send this call, the curtain closes. if I send a true, it will stop closing. I can not open the curtain at this point

> curl -H 'Content-Type: application/json' -X PUT -d '{"on":false}' http://192.168..../api/..../lights/2/state

...I can steer the same thing on iobroker now.. I just have changed the code as per your DDF above without any other thing (no additional modification)..

How do I get to open it? I'm excited now ;)

Smanar, 2023-07-02:
Tuya covering use thoses dpid for movement

 -   close: 2
 -   open: 0
 -   stop: 1

`"write": {"fn": "tuya", "dpid": 1, "dt": "0x30", "eval": "Item.val == 1 ? 0x01 : 0x02;"},`

So If I m right if you send a true value (=1) it send the command 0x01 so the stop else it send the command 0x02 the close.
Can try

`"write": {"fn": "tuya", "dpid": 1, "dt": "0x30", "eval": "Item.val == 1 ? 0x02 : 0x00;"},`

For the return

`"parse": {"fn": "tuya", "dpid": 1, "eval": "Item.val = Attr.val != 0x00;" }`

Perhaps need to use 0x02 instead of 0x00, but your device is reversed, and the value need to change only on full closed (even just open as 2 % is open), so can make try to find the good value

patrusco, 2023-07-03:
Great! 

With the code below, I'm able now to close it with "true" and open it with "false". And that works also from iobroker side.
I did change the parse argument from "Attr.val ==" to "Attr.val !=" in addition to make "true" the close command

observation: on iobroker I do see return of "true"  once the curtain has fully opened. So the value to steer is also the value for a status? strange, but again I'm learning ;)

what next ?

>    {
          "name": "state/on",
          "parse": {"fn": "tuya", "dpid": 1, "eval": "Item.val = Attr.val != 0x00;" },
          "write": {"fn": "tuya", "dpid": 1, "dt": "0x30", "eval": "Item.val == 1 ? 0x02 : 0x00;"},
          "read": {"fn": "tuya"}
        },

patrusco, 2023-07-07:
Okey! Now I'm able to move the curtan to a specific position. Below the example where I was able to set "bri" on 175 and tue curtain moves to 68% (also visible in IOBroker). I'm not able to change the % this seems just to be a status but not for control, right?

[image]

regarding linux, I can build one if you like? but the function we got now is already great since I have control in both directions and I'm able to set the position, thats awesome!

Smanar, 2023-07-08:
IDK, this is from your third app, 68% = 175 (max is 255)
Deconz use only the numeric value.

>regarding linux, I can build one if you like? but the function we got now is already great since I have control in both directions and I'm able to set the position, thats awesome!

In fact this DDF is a "hack" we are using bri/on from a "light" device, we need to use lift/open from a "covering" device.
One day you will have issue with this DDF.

I have a code modification that add this feature to DDF, but hard to find testers.
You need a linux OS with complete OS to test code modification. I can explain how to do if you want to make a try ?

patrusco, 2023-07-10:
sure! tell me what linux you prefer and I will get a VM built to test. Please understand that I'm only able to build VM's which have remote-access to my phoscon gateway, so no direct attached zigbee stuff, but if that sufficient I'm happy to help "back"  and give it a try