dresden-elektronik/deconz-rest-plugin #5961: Develco EMIZB-132 is not reporting total enegry consumption with decimal value

baunan, 2022-04-14:
I hace a Develco Norwegian Han type EMIZB-132. I use Home assistant to control my house. I run Home assistant on a NUC with a conbeen II on a separate raspberry pi. I noticed now that when the total enegry consumption reported on my meter went above 100000, then DeConz stoped reporting this value with a decimal figure. It was reported with 1 decimal figure before. Is it possible to get the decimal figure back? 
The Develco unit report the total energy consumptoin value with 3 decimals.

## Steps to reproduce the behavior

n/a
## Expected behavior

Expect that the correct value from the meter is reported.
## Screenshots

Screenshot from Deconz (Simple metering cluster)
[image]
Screenshot from Home Assistant (integration page)
[image]

## Environment

 - Host system: Raspberry Pi
 - Running method: Raspbian
 - Firmware version: 26680700
 - deCONZ version: 2.13.04
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs

n/a
## Additional context

n/a

Smanar, 2022-04-14:
Hello, can you check how is the value direclty on the API ?
Using for exemple phoscon / help  / API Information / sensors

baunan, 2022-04-14:
Hi, this is the info I get for this unit:

"consumption" is the total energy consumption from the meter
   
 "21": {
        "config": {
            "interfacemode": 1,
            "on": true,
            "reachable": true
        },
        "ep": 2,
        "etag": "509bcc8f27ddfa7cdf2aa7b0b62d321d",
        "lastannounced": "2022-04-12T15:33:49Z",
        "lastseen": "2022-04-14T19:10Z",
        "manufacturername": "Develco Products A/S",
        "modelid": "EMIZB-132",
        "name": "Electricity Sensor (2)",
        "state": {
            "consumption": 100811000,
            "lastupdated": "2022-04-14T19:11:12.129",
            "power": 2824
        },
        "swversion": "2019-07-11 23:47",
        "type": "ZHAConsumption",

SwoopX, 2022-04-14:
Deconz had an issue with larger numbers being inadequately rounded, which got resolved with 2.15.1. Therefore, I'm closing this one. Feel free to request reopening if that wasn't resolved nevertheless.

closed 2022-04-14 by SwoopX