dresden-elektronik/deconz-rest-plugin #5966: ubisys J1 sends pct = 0 since update to version v2.15.1-beta

Rolf56, 2022-04-19:
## Describe the bug

ubisys J1 returns pct = 0 since update to version v2.15.1-beta. When sending a pct value to the device, it moves to the right position.

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

[image]

## Environment

 - Host system: PC
 - Running method:  Windows
 - Firmware version: 26400500
 - deCONZ version: 2.15.01
 - Device: ConBee I
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? If so: Which? non

## deCONZ Logs

## Additional context

SwoopX, 2022-05-05:
This is not a bug and has nothing to do with deconz. Rubbing my crystal ball, your device has lost position awareness either by a power loss or by rebooting itself for whatever reason (see ubisys technical reference).

When cover is opened, move it down and then immediately up, that should restore it.

closed 2022-05-05 by SwoopX

Rolf56, 2022-05-06:
Hi SwoopX
Thank you for your answer. I do not agree with you. The value 'Current Position Lift Percentage', what I think is the pct value, corresponds with the cover position in deCONZ UI.

[image]

But the pct value is missing in the REST interface! It is also not defined in the file **j1_5502.json**!
The DDF for J1 was changed in version [v2.15.1-beta](https://github.com/dresden-elektronik/deconz-rest-plugin/tree/v2.15.1-beta) and exact since then it does not work anymore correctly!

[image]

Yes, I moved the cover down and up, also in configuration mode. This changed nothing.
Maybee you can rubbing your crystal ball again, to find out what happen with this release [v2.15.1-beta](https://github.com/dresden-elektronik/deconz-rest-plugin/tree/v2.15.1-beta) https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5894

SwoopX, 2022-05-07:
Well, apparently, rubbing the crystal ball is not such a good fortune teller after all, q.e.d.

Now back to the topic and allow me to set your expectations right.

> But the pct value is missing in the REST interface! It is also not defined in the file j1_5502.json!

Deconz does not expose a `pct` parameter and never has, however, your perception of `Current position lift percentage` is correct, but it maps to `lift` (raw value) and `bri` (with a calculation behind it). Now I know the HUE device perl module never picked up the `lift` values (at least not with the version I checked last time about 9 months ago), but is doing some magic on `bri` to deduce the pct value you mean. Will have a look if the `bri` calculation remained identical while purging it from the code (and moving it to the DDF).

In Phoscon, when you go to _Help -> API Information_, then select lights and choose your cover and finally click on events, you see what deconz sends towards FHEM.

That's what my cover sends over:

```
{
    "10:27:37:779": {
        "e": "changed",
        "id": "12",
        "r": "lights",
        "state": {
            "bri": 13,
            "lift": 5
        },
        "t": "event",
        "uniqueid": "00:1f:ee:00:00:xx:xx:xx-01"
    },
    "10:27:39:311": {
        "e": "changed",
        "id": "12",
        "r": "lights",
        "state": {
            "bri": 20,
            "lift": 8
        },
        "t": "event",
        "uniqueid": "00:1f:ee:00:00:xx:xx:xx-01"
    },
    "10:27:40:307": {
        "e": "changed",
        "id": "12",
        "r": "lights",
        "state": {
            "bri": 0,
            "lift": 0
        },
        "t": "event",
        "uniqueid": "00:1f:ee:00:00:xx:xx:xx-01"
    },
    "10:27:42:990": {
        "e": "changed",
        "id": "12",
        "r": "lights",
        "state": {
            "bri": 13,
            "lift": 5
        },
        "t": "event",
        "uniqueid": "00:1f:ee:00:00:xx:xx:xx-01"
    },
    "10:27:44:43": {
        "e": "changed",
        "id": "12",
        "r": "lights",
        "state": {
            "bri": 28,
            "lift": 11
        },
        "t": "event",
        "uniqueid": "00:1f:ee:00:00:xx:xx:xx-01"
    },
    "10:27:45:197": {
        "e": "changed",
        "id": "12",
        "r": "lights",
        "state": {
            "bri": 43,
            "lift": 17
        },
        "t": "event",
        "uniqueid": "00:1f:ee:00:00:xx:xx:xx-01"
    },
    "10:27:46:66": {
        "e": "changed",
        "id": "12",
        "r": "lights",
        "state": {
            "bri": 58,
            "lift": 23
        },
        "t": "event",
        "uniqueid": "00:1f:ee:00:00:xx:xx:xx-01"
    },
    "10:27:47:596": {
        "e": "changed",
        "id": "12",
        "r": "lights",
        "state": {
            "bri": 71,
            "lift": 28
        },
        "t": "event",
        "uniqueid": "00:1f:ee:00:00:xx:xx:xx-01"
    },
    "10:28:01:839": {
        "attr": {
            "lastseen": "2022-05-07T08:27Z"
        },
        "e": "changed",
        "id": "12",
        "r": "lights",
        "t": "event",
        "uniqueid": "00:1f:ee:00:00:xx:xx:xx-01"
    }
}
```

And FHEM updates fine:

[image]

Now, to be fair, I'm using a completely self written perl module here which is tailored just for deconz' REST API, only processing `lift` for cover position. A bit like comparing apples and pears. Regardless, I'll double check the bri calculation.

SwoopX, 2022-05-07:
I assume the subtype in the attributes is (still) `blind`?

From a code perspective, I cannot see any difference (DDF left, legacy code right)

[image]

Rolf56, 2022-05-09:
Hallo SwoopX

Dank für die Angaben. Schade dass pct nicht mehr geht. Somit werde ich meine Abfragen von pct auf bri umstellen müssen.

Grüsse, Rolf

Von: SwoopX
Gesendet: Samstag, 7. Mai 2022 21:25
An: dresden-elektronik/deconz-rest-plugin
Cc: Rolf56; Author
Betreff: Re: [dresden-elektronik/deconz-rest-plugin] ubisys J1 returns pct = 0since update to version v2.15.1-beta (Issue #5966)

I assume the subtype in the attributes is (still) blind?
From a code perspective, I cannot see any difference (DDF left, legacy code right)

Mimiix, 2022-05-09:
Hi there, Can you please translate to English? Thanks :)!