dresden-elektronik/deconz-rest-plugin #5967: Aqara Motion Sensor P1

spn91, 2022-04-20:
## Device

- Product name: Aqara Motion Sensor P1
- Manufacturer: LUMI
- Model identifier: lumi.motion.ac02
- Device type : Please remove all unrelated device types. 
  - Sensor

## Screenshots
### Basic
[image]

### Identify
[image]
### Other clusters that are not mentioned above
[image]

Mimiix, 2022-04-20:
Please add all screens.

spn91, 2022-04-20:
I'm not sure what you mean with "all" screens, but I have added the clusters which are not mentioned in the manual or template.

Mimiix, 2022-04-20:
> I'm not sure what you mean with "all" screens, but I have added the clusters which are not mentioned in the manual or template.

In the template it mentions the following:

  If available add screenshots of other clusters. Please Remove sections that are not available on your device.

  Relevant clusters are: Simple Metering, Electrical Measurement, Power Configuration, Thermostat, etc. You can typically spare Identify, Alarms, Device Temperature, On/Off. Please ensure data has been read prior to taking any screenshots.

in your case, the power config was missing.

Either way seems good so far.

@Smanar  @SwoopX  Can a DDF be sufficient?

Smanar, 2022-04-20:
Honneslty I can't be sure, I never made DDF for device using the Xiaomi cluster 0xFCC0 and I don't found sample to be sure.
Perhaps @SwoopX know better this cluster.

spn91, 2022-04-21:
@SwoopX regarding the discussion in the forum, I am able to sniff the traffic. But why do we not just use the old sensor as "template". Do you really think they differ that much?

SwoopX, 2022-04-23:
@spn91 I don't think, I see it 🙂 Xiaomi does not follow the zigbee specs and doesn't use the occupancy cluster as far as I can tell. Hence, nothing in common with the old sensor. Same btw for the FP1.

I need to get an impression on what this devices uses when it is reporting motion. Presumably, the respective attribute isn't even part of the already known attributes yet while using the 0xFCC0 cluster for this. I'd also need to know if it reports when no motion is detected anymore. Ideally, that would be configurable on the device itself.

That's why a sniff would be most beneficial in answering all those questions.

techfreakdad, 2022-04-30:
I’m not a Dev,  it an end user with a P1 motion sensor. Let me know if you need any UAT’s.

SwoopX, 2022-05-08:
Ok folks, I'd dare to give it a run with the following DDF but please do take not that I'd require some debug output from deconz.

Create a .json file in `/home/<DECONZUSER>/.local/share/dresden-elektronik/deCONZ/devices` (if you run some Linux), or search for deconz' DB file `zll.db`. In that directory, there should also be a directory `devices` and place it there. Now for the debug logging, I'd require at least DBG_INFO + DBG_INFO_L2, that should be enough. Ideal would be a 10 mins log where you make sure to trigger motion detection 3 - 4 times. I'd expect that the device automatically resets the presence state after 30 seconds, but I hope that's gonna be proven and which attribute is used for that. Consequently, make sure you have at least 1 min between your motion trigger attemps.

Just FYI, I'm mainly interested in those lines for further amendment (please do not strip the log)

```
23:14:23:931 [INFO] - No button map for: lumi.motion.ac01, unicast to: 0x0000, endpoint: 0x01, cluster: XIAOMI (0xFCC0), command: ATTRIBUTE_REPORT (0x0A), payload: 43012004, zclSeq: 18
23:14:23:932 ZCL attribute report 0x54EF441000471A48 for cluster: 0xFCC0, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
23:14:23:933 	payload: 43012004
```

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "aqara",
  "modelid": "lumi.motion.ac02",
  "vendor": "Xiaomi",
  "product": "Aqara P1 motion sensor RTCGQ14LM",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_PRESENCE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0406"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "refresh.interval": 86400
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x0020",
            "cl": "0x0001",
            "ep": 1,
            "eval": "const vmin = 20; const vmax = 30; let bat = Attr.val; if (bat > vmax) { bat = vmax; } else if (bat < vmin) { bat = vmin; } bat = ((bat - vmin) / (vmax - vmin)) * 100; if (bat > 100) { bat = 100; } else if (bat <= 0) { bat = 1; } Item.val = bat;"
          },
          "read": {
            "fn": "None"
[... 145 lines cut ...]
        {
          "name": "state/lux"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0020",
          "dt": "0x20",
          "min": 60,
          "max": 3600,
          "change": "0x00000001"
        }
      ]
    }
  ]
}
```

jan666, 2022-05-09:
@SwoopX I tried it but found nothing about lumi.motion.ac01 in my log. So I guess I did something wrong...?

I put `p1.json` in `/home/pi/.local/share/dresden-elektronik/deCONZ/devices`

```
pi@raspberrypi:~/.local/share/dresden-elektronik/deCONZ $ file /home/pi/.local/share/dresden-elektronik/deCONZ/zll.db
/home/pi/.local/share/dresden-elektronik/deCONZ/zll.db: SQLite 3.x database, user version 9, last written using SQLite version 3027002
```

SwoopX, 2022-05-09:
Perpaps because you searched for the zigbee model ID of the FP1 sensor, and not yours? 😅

jan666, 2022-05-09:
@SwoopX i sent you a DM on the phoscon forum w/ links to 3 logs. the third inlcudes a pairing

SwoopX, 2022-05-09:
@jan666 Hm, apparently, something went wrong here. I do not see the device occurring in any of the files. Have you re-paired the device? In case not, please do so. Also please check if you're on the latest firmware (0x26720700), might be mandatory for the device to behave correctly while pairing. Last thing to check would be if 2 sensor resources have been created.

SwoopX, 2022-05-09:
@jan666 Ah, it was my fault. Classic c&p error (wrong manufacturername). I've updated the sample DDF above, sorry.

jan666, 2022-05-09:
2 Sensors got created using the new DDF file. Log file via. Phoscom Forum DM.

```
{
  "config": {
    "battery": 0,
    "on": true,
    "reachable": true,
    "sensitivity": 3
  },
  "etag": "a8f14b84f4e96f38630063b2282356c3",
  "lastannounced": null,
  "lastseen": "2022-05-09T10:36Z",
  "manufacturername": "LUMI",
  "modelid": "lumi.motion.ac02",
  "name": "Presence 64",
  "state": {
    "lastupdated": "none",
    "presence": false
  },
  "swversion": "0.0.0_0005",
  "type": "ZHAPresence",
  "uniqueid": "54:ef:44:10:00:43:eb:f9-01-0406"
}
```

```
{
  "config": {
    "battery": 0,
    "on": true,
    "reachable": true,
    "tholddark": 12000,
    "tholdoffset": 7000
  },
  "etag": "c77caa11a96307042d9d2b152bd15a0d",
  "lastannounced": null,
  "lastseen": "2022-05-09T10:36Z",
  "manufacturername": "LUMI",
  "modelid": "lumi.motion.ac02",
  "name": "LightLevel 65",
  "state": {
    "dark": null,
    "daylight": null,
    "lastupdated": "none",
    "lightlevel": null,
    "lux": null
  },
  "swversion": "0.0.0_0005",
  "type": "ZHALightLevel",
  "uniqueid": "54:ef:44:10:00:43:eb:f9-01-0400"
}
```

SwoopX, 2022-05-09:
Yes, here we go, thanks. Now, I had a look at the log and I guess I got the illuminance calculation. Surprisingly however, there was nothing in there indicating presence detection (yet). You surely triggered some motion?

I've updated the DDF template above once again. Have some battery reading meanwhile occurred?

jan666, 2022-05-10:
I had the sensor covered on my desk and removed the cover a few times. Should have triggered motion I think.

Battery is till at 0.

I'll update the DDF and add the javascript but can't  test until tomorrow

jan666, 2022-05-11:
@SwoopX see deconz_debug_5.log via DM @ Phoscon Forum. Includes pairing + a few times covering, removing the cover and make some movements. the LED just blinked on the first movement-simulation.

SwoopX, 2022-05-11:
Thanks. Spotted something new, which would potentially make sensse:

```
06:41:40:391 ZCL attribute report 0x54EF441ABCDEF for cluster: 0xFCC0, ep: 0x01, frame control: 0x1C, mfcode: 0x115F 
06:41:40:392 	payload: fc001000
```

The payload contains 10, which is a boolean (hopefully for presence). Also made some other amendments on the DDF and the script above. To get the battery values, you also need to have `xiaomi_battery.js` copied to your folder.

jan666, 2022-05-12:
@SwoopX deconz_debug_6.log

Battery is at 100 % now (so it might work?)

Lux is updating, lightlevel is not.

No Motion

```
> select * from deflux_ZHALightLevel where id = '65' order by time desc limit 30;
name: deflux_ZHALightLevel
time                           age_secs battery dark daylight id lightlevel lux name               source    type
----                           -------- ------- ---- -------- -- ---------- --- ----               ------    ----
2022-05-12T05:38:07.464552576Z 0        100     true false    65 65506      50  Bewegungssensor P1 websocket ZHALightLevel
2022-05-12T05:37:37.740224817Z 0        100     true false    65 65506      123 Bewegungssensor P1 websocket ZHALightLevel
2022-05-12T05:37:08.115709359Z 0        100     true false    65 65506      68  Bewegungssensor P1 websocket ZHALightLevel
2022-05-12T05:35:13.767871454Z 0        100     true false    65 65506      86  Bewegungssensor P1 websocket ZHALightLevel
2022-05-12T05:34:30.177666568Z 0        100     true false    65 65506      40  Bewegungssensor P1 websocket ZHALightLevel
2022-05-12T05:33:03.61126796Z  0        100     true false    65 65506      70  Bewegungssensor P1 websocket ZHALightLevel
2022-05-12T05:32:33.552829992Z 0        100     true false    65 65506      72  Bewegungssensor P1 websocket ZHALightLevel
2022-05-12T05:32:03.863375107Z 0        100     true false    65 65506      64  Bewegungssensor P1 websocket ZHALightLevel
2022-05-12T05:30:25.202388655Z 0        100     true false    65 65506      4   Bewegungssensor P1 websocket ZHALightLevel
2022-05-12T05:30:22.199622913Z 0        100     true false    65 65506      84  Bewegungssensor P1 websocket ZHALightLevel
2022-05-12T05:30:19.206903502Z 0        100     true false    65 65506      248 Bewegungssensor P1 websocket ZHALightLevel
2022-05-12T05:30:16.088652004Z 0        100     true false    65 65506      63  Bewegungssensor P1 websocket ZHALightLevel
2022-05-12T05:30:12.681625529Z 0        100     true false    65 65506      100 Bewegungssensor P1 websocket ZHALightLevel
2022-05-12T05:30:03.955598546Z 0        100     true false    65 65506      96  Bewegungssensor P1 websocket ZHALightLevel
2022-05-12T05:30:00.771976093Z 0        100     true false    65 65506      99  Bewegungssensor P1 websocket ZHALightLevel
2022-05-12T05:29:56.693732692Z 0        100     true false    65 65506      96  Bewegungssensor P1 websocket ZHALightLevel
2022-05-12T05:29:08.21511743Z  0        100     true false    65 65506      100 Bewegungssensor P1 websocket ZHALightLevel
2022-05-12T05:28:37.176518018Z 0        100     true false    65 65506      100 Bewegungssensor P1 websocket ZHALightLevel
2022-05-12T05:28:20.763746217Z 0        100     true false    65 65506      100 Bewegungssensor P1 websocket ZHALightLevel
2022-05-12T05:28:17.772163495Z 0        100     true false    65 65506      101 Bewegungssensor P1 websocket ZHALightLevel
2022-05-12T05:28:13.116965439Z 0        100     true false    65 65506      100 Bewegungssensor P1 websocket ZHALightLevel
2022-05-12T05:28:08.532021433Z 0        100     true false    65 65506      100 Bewegungssensor P1 websocket ZHALightLevel
2022-05-12T05:28:05.5339354Z   0        100     true false    65 65506      81  Bewegungssensor P1 websocket ZHALightLevel
2022-05-12T05:27:01.916076259Z 0        100     true false    65 65506      77  Bewegungssensor P1 websocket ZHALightLevel
2022-05-12T05:26:58.926748769Z 0        100     true false    65 65506      0   Bewegungssensor P1 websocket ZHALightLevel
```

```
> select * from deflux_ZHAPresence where id = '64' order by time desc;
name: deflux_ZHAPresence
time                           age_secs battery id lowbattery name               presence source    tampered type
----                           -------- ------- -- ---------- ----               -------- ------    -------- ----
2022-05-12T05:26:11.203074052Z 0        100     64 false      Bewegungssensor P1 false    websocket false    ZHAPresence
2022-05-11T13:18:22.458505863Z -1       0       64 false      Bewegungssensor P1 false    websocket false    ZHAPresence
2022-05-11T12:28:45.048768703Z -1       0       64 false      Bewegungssensor P1 false    websocket false    ZHAPresence
2022-05-11T11:34:17.45985456Z  0        0       64 false      Bewegungssensor P1 false    websocket false    ZHAPresence
2022-05-11T10:40:26.605122826Z 0        0       64 false      Bewegungssensor P1 false    websocket false    ZHAPresence
2022-05-11T09:48:32.81331667Z  0        0       64 false      Bewegungssensor P1 false    websocket false    ZHAPresence
2022-05-11T08:58:56.793247655Z 0        0       64 false      Bewegungssensor P1 false    websocket false    ZHAPresence
2022-05-11T08:06:16.637863003Z 0        0       64 false      Bewegungssensor P1 false    websocket false    ZHAPresence

```

mac-city, 2022-05-16:
just received one of these today to find it wasn't yet supported by phoscon in home assistant, but deconz seems to have found it.  can it be used in home assistant just using deconz?

jan666, 2022-05-16:
@mac-city it's not (yet?) supported in deCONZ. I sent mine to SwoopX so he can do whatever he does and might add support for it. After it's supported in deCONZ it will be exposed via REST-API and should work in home assistant (assuming web assistant is using the REST-API / Websocket, I never used ha)

closed 2022-05-24 by manup

frest-de, 2022-05-24:
Just updated to 2.16.01. presence and last_updated is working fine, lightlux and battery level does not work yet.

Muto999, 2022-05-26:
Hello. Thanks for all your work. I updated to 2.16.01 and the new P1 motion sensor is recognized by Deconz. I was also able to integrate the sensor to fhem. Any motion is recognized by deconz UI, butthe status is not pushed to fhem. Might this be an issue ob the deconz side, or does fhem not recognized the pushed motion-update-information? Thanks for your help. If you need any logfiles, please let me know.

SwoopX, 2022-05-26:
> Any motion is recognized by deconz UI...

Is that deconz GUI or Phoscon? Which firmware do you run on Raspbee/Conbee? The way presence is exposed doesn't differ from other sensors, so FHEM should recognize it. However, other attributes exposed for the device you eventually do not see as the respective module does not support everything from the API.

DesertNomadBear, 2022-05-26:
Hi everyone, and thanks for all your hard work and quick fix for these new sensors.
I have a small question: I run deconz as an add-on in HA Supervised. I haven’t received any official updates yet to the add-on, so not sure what/how to update so I can use these sensors. Thanks for your help.

SwoopX, 2022-05-26:
You need to ask HA folks on that one. The released version is marked as stable (and those are only considerd by them).

Muto999, 2022-05-26:
> Is that deconz GUI or Phoscon? Which firmware do you run on Raspbee/Conbee? The way presence is exposed doesn't differ from other sensors, so FHEM should recognize it. However, other attributes [...]

Yes, I mean the Phoscon Web-Interface. I have a Rasbee with Firmware 26400500. I have a lot of Aquara door/window sensors and they get updated in nearly realtime. Just the state Motion/nomotion from the Aquara P1 is not updated/pushed. And as this device was just added in the last update, I thought it might be related to the new implemantation in deconz/phoscon. Thanks for your help!

SwoopX, 2022-05-26:
It's hard for me to follow that as the same messages originating from deconz go to all REST API clients. It's more the all or nothing here. Are you sure you added the right sensor in FHEM? There's 2 for the device.

Shamal2008, 2022-05-27:
same problem with luminance on my side. added in phoscon, recognized - but the screen with "finished" (during adding process) "died". Both sensors appear in FHEM (Luminance and Presence), but Luminance does not work. looks like integration seem not to finish in deconz.

Mimiix, 2022-05-27:
@SwoopX @Shamal2008  @frest-de also noticed that lux isn't there yet.

Muto999, 2022-05-28:
> ...Are you sure you added the right sensor in FHEM? There's 2 for the device.
Yes. There is ZHAPresence and ZHANightlevel, and I choosed the Presence-sensor.

SwoopX, 2022-05-28:
Based on @Shamal2008's feedback, it now evident to me that it's the very same behavior as described [here](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5974). It is definitively not related to the DDF, but apparently has its origin somewhere in deconz and is only observed with few Xiaomi devices so far.

This is tidious, as none of the devs was able to reproduce that so far.

manup, 2022-05-28:
Maybe we can do a live session to inspect whats going on, perhaps with some more logging added to pin down where it fails?

Shamal2008, 2022-05-29:
> > ...Are you sure you added the right sensor in FHEM? There's 2 for the device.
> > Yes. There is ZHAPresence and ZHANightlevel, and I choosed the Presence-sensor.

Yes, I'm sure. I already used the "old" ones from aqara for over 2 yrs now. The P1 is imho not recognized correctly. 

It's also reproducable (tested it 5 times), that with the new 2.16.01 deconz Software the pairing has a problem. It will recognize the sensor, "Sensor ready" - and no "finish" button is available. You can only close the Window with upper-right "X", but it appears only one sensor PresenceXXX with very strange data. See enclosed screenshot. 

The fact, that the paring fails shows also, that basic device information - is incomplete/wrong. 

[image]
[image]

Shamal2008, 2022-05-29:
> Maybe we can do a live session to inspect whats going on, perhaps with some more logging added to pin down where it fails?

Yes, we can, just write me, when you're available. On Monday I'm on business trip up to probably 7.00. p.m.

frest-de, 2022-05-29:
> 

got exectly the same situation like described!

SwoopX, 2022-05-29:
@Shamal2008 That's interesting and concerning at the same time. Could you provide a full blown set of debug data? Meaning info, info_l2, aps, aps_l2, ddf, dev, error, error_l2.

This will be extremely noisy, however, it should give us some more insights. Luckily, the sensor's default config is to report presence every 30 secs, so it shouldn't take too long. Please use pastebin then 😉

Shamal2008, 2022-05-29:
> @Shamal2008 That's interesting and concerning at the same time. Could you provide a full blown set of debug data? Meaning info, info_l2, aps, aps_l2, ddf, dev, error, error_l2.
>
> This will be [...]

Do you have any idea, how to avoid that the debug will crash deconz-gui regulary? If I enable the debug, it will crash at least in 2-3 minutes after running. Also its really annoying to have no chance to search within in the debug. I have a prod-system with 60 sensors, it's impossible to "watch" the right lines and copy them. 

ps: I used "add acode", but github doesn't care :)

SwoopX, 2022-05-29:
Now you really caught me. Deconz shouldn't crash at all when debugging is enabled, even with everything turned on. I'm afraid, your wife now needs to take one for the team 🙂 

I'd like to provoke 2 more crashes. Let's make the first one and catch all the debug stuff. On Linux, and I desperately hope you run it, start it headless with `/usr/bin/deCONZ -platform minimal --dbg-info=2 --dbg-error=2 --dbg-aps=2 --dbg-ddf=1 --dbg-dev=1 > debug.txt`. If it should crash, there should be plenty of data to work with.

Now to analyze the crash. Run deconz as follows and do what you did to make it happen: `gdb --args "/usr/bin/deCONZ"`. When it crashed, you return to the command promt of gdb and then type `bt`, followed by hitting enter. That gives us the backtrace and hopefully some more insights of what's going on.

In case you don't run Linux, please state your system setup.

Shamal2008, 2022-05-29:
Hi, it's Linux :) - Raspian, latest patches and updates from friday. I give it a try tommorrow evening. I'll update to 2.16.1 again and try to reproduce.

Muto999, 2022-05-31:
Quick update.... Really don't know why... Just rebooted my Raspberry Pi several times due to other updates, and now presence updates are pushed as from other sensors. So no issues anymore. But I really can't tell what happened, because neither my deconz-config nor my fhem config did change. No "auto"-updates on both either.

SwoopX, 2022-05-31:
@Shamal2008 Many thanks for taking that data and also your feedback. Hope your wife will show some mercy after all.

Tldr, I'd support your conclusion that the DB is the issue or one particular entry, precisely. I recall I had a comparable situation once a while ago, where I also experienced unexplainable crashes in my dev environment. I don't remamber where it came from, but one row had a missing sensor id (it was NULL). Once deleted, everything went back to normal. However, this shouldn't have happened in the first place, as the field is also primary key...

So, looking at your debug output, the first thing I was worried about turned out to be nothing (failed assertion) and hence do not play any role here.
The SQL errors is something which shouldn't happen and rightfully thrown. As I do not know those parts of the code too well, I cannot really anticipate the consequences here. @manup can/does that brick something after all (L994)? This seems to be indeed about the ID.

Bottom line is that I got the feeling some very rare bugger is still dwelling in the depths. However, glad it started working for you in the end.

manup, 2022-05-31:
Might be worth to construct some test scenarios with invalid, null or missing IDs to see if and where the code breaks.
In any case I think there is room to add some more checks for validation and repairing this stuff.

Big challenge is to define the cases which might be seen in the wild and what should happen :) considering also the legacy entries which more than often are a bit wild.

MiAutomations, 2022-10-17:
Hello to all in the deconz where I can find the detection interval configuration for the motion sensor Aqara P1?

Thank you

djsomi, 2022-11-05:
> Hello to all in the deconz where I can find the detection interval configuration for the motion sensor Aqara P1?
> 
> Thank you

Hi All, 

This is what I am looking for also, any hint is highly appreciated!

Thanks
Zoltan

yschen5812, 2022-12-15:
> 

I'm no familiar with anything here but from the screenshots I found the attribute 0x0102 could be it as the default cool down interval is 30 secs.

Smanar, 2023-12-01:
Hello, need to know your deconz version, the device model ID and the manufacture name to compare.

69franky, 2023-12-01:
deconz version: 2.23.02
Firmware 26780700
Manufacturer: Lumi United Technology Co, Ltd
model ID: RTCGQ11LM (I guess its this one)
Model: MS-S02
https://www.amazon.de/dp/B0B9XZ1D51?psc=1&ref=ppx_yo2ov_dt_b_product_details

maybe similar problem as in ioBroker?
https://github.com/ioBroker/ioBroker.zigbee/issues/1851

SwoopX, 2023-12-01:
@69franky as your question is not related to the device integration, please head over to the Phoscon forum. The issue already contains 2/3 unrelated content. Thanks!