dresden-elektronik/deconz-rest-plugin #5974: Aquara Multi Sensors (Temperature/Pressure/Humidity) no longer reporting temperature

UppyJC, 2022-04-22:
## Describe the bug

Since upgrading to latest deCONZ version, Aquara Multi Sensors (Temperature/Pressure/Humidity) no longer reporting temperature. API reports null values, Phoscon reports 0 degrees. Pressure and humidity values fine.

Deleting and re-adding sensors makes no difference.

Working correctly on previous version, but can't remember which version that was!

## Steps to reproduce the behavior

Add any sensor of affected type to deCONZ. Observe null temperature value in API and 0 temperature value in Phoscon. Humidity and pressure values report correctly.

## Expected behavior

Temperature value reports correctly in API and Phoscon.

## Screenshots

[image]

[image]

## Environment

 - Host system: PC
 - Running method: Windows
 - Firmware version: (26580700)
 - deCONZ version: (2.15.02)
 - Device: ConBee II
 - Do you use an USB extension cable: Yes
 - Is there any other USB or serial devices connected to the host system?: No

## deCONZ Logs

## Additional context

Mimiix, 2022-04-22:
What does the respective clusters show in deCONZ? 

Also, did you try this?

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Xiaomi:-Lost-devices-or-functional-issues

UppyJC, 2022-04-22:
Yes, I have re-paired one of the sensors and read node descriptors and simple descriptors.

There is a reading in deCONZ as shown of the screenshot below, but the API and Phoscon both still report null/0 respectively.

[image]

I have 11 of these sensors and they're all exhibiting exactly the same behaviour from the same time (when I upgraded to the latest deCONZ version).

nunoary, 2022-04-24:
I see the same behavior. It was working fine with version 2.14.01 but after installing version 2.15.01 the sensors no longer report temperature.

I did some tests and noticed that if I turn off the DDF of these sensors (put the status in "Draft") the sensors work correctly and report the temperature again. But if you put it back in "Gold" the sensors stop reporting the temperature again.

I saw in another [post](https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5914) that this bug had been fixed in version 2.15.01 but it seems that it remains.

DNC74, 2022-04-24:
please take the similar issue and infos posted on #5978 regarding battery levels only into concern

manup, 2022-04-25:
Can anybody where the problem happens please send me the `zll.db` file (or Phoscon App backup) to mpi@dresden-elektronik.de. I have an assumption that the cause is a leftover in database from v2.15.x – v2.15.1 regression.

UppyJC, 2022-04-25:
Database file sent to you as requested!

manup, 2022-04-25:
Thanks, I had a look the database looks alright :thinking: 
Is Basic/Gold enabled in the control panel?

*Panels &rarr; Control &rarr; Basic &rarr; Gold*

manup, 2022-04-25:
Also if Gold is already enabled, a log with following settings (incl. DDF) would be helpful *Help &rarr; Debug view*:

[image]

UppyJC, 2022-04-25:
Gold is enabled - log is attached, complete with waiting for an update from the affected sensors. Still all reporting null/0 temperatures.

`13:41:09:846 ZCL attribute report 0x0015BC001B02243D for cluster: 0x0702, ep: 0x02, frame control: 0x18, mfcode: 0x0000 
13:41:14:845 ZCL attribute report 0x0015BC001B02243D for cluster: 0x0702, ep: 0x02, frame control: 0x18, mfcode: 0x0000 
13:41:24:848 ZCL attribute report 0x0015BC001B02243D for cluster: 0x0702, ep: 0x02, frame control: 0x18, mfcode: 0x0000 
13:41:30:823 Device TTL 3157 s flags: 0x7
13:41:34:850 ZCL attribute report 0x0015BC001B02243D for cluster: 0x0702, ep: 0x02, frame control: 0x18, mfcode: 0x0000 
13:41:39:852 ZCL attribute report 0x0015BC001B02243D for cluster: 0x0702, ep: 0x02, frame control: 0x18, mfcode: 0x0000 
13:41:49:852 ZCL attribute report 0x0015BC001B02243D for cluster: 0x0702, ep: 0x02, frame control: 0x18, mfcode: 0x0000 
13:41:59:858 ZCL attribute report 0x0015BC001B02243D for cluster: 0x0702, ep: 0x02, frame control: 0x18, mfcode: 0x0000 
13:42:04:860 ZCL attribute report 0x0015BC001B02243D for cluster: 0x0702, ep: 0x02, frame control: 0x18, mfcode: 0x0000 
13:42:14:865 ZCL attribute report 0x0015BC001B02243D for cluster: 0x0702, ep: 0x02, frame control: 0x18, mfcode: 0x0000 
13:42:24:971 ZCL attribute report 0x0015BC001B02243D for cluster: 0x0702, ep: 0x02, frame control: 0x18, mfcode: 0x0000 
13:42:29:869 ZCL attribute report 0x0015BC001B02243D for cluster: 0x0702, ep: 0x02, frame control: 0x18, mfcode: 0x0000 
13:42:30:833 Device TTL 3097 s flags: 0x7
13:42:39:873 ZCL attribute report 0x0015BC001B02243D for cluster: 0x0702, ep: 0x02, frame control: 0x18, mfcode: 0x0000 
13:42:44:876 ZCL attribute report 0x0015BC001B02243D for cluster: 0x0702, ep: 0x02, frame control: 0x18, mfcode: 0x0000 
13:42:54:879 ZCL attribute report 0x0015BC001B02243D for cluster: 0x0702, ep: 0x02, frame control: 0x18, mfcode: 0x0000 
13:42:59:882 ZCL attribute report 0x0015BC001B02243D for cluster: 0x0702, ep: 0x02, frame control: 0x18, mfcode: 0x0000 
13:43:09:885 ZCL attribute report 0x0015BC001B02243D for cluster: 0x0702, ep: 0x02, frame control: 0x18, mfcode: 0x0000 
13:43:14:889 ZCL attribute report 0x0015BC001B02243D for cluster: 0x0702, ep: 0x02, frame control: 0x18, mfcode: 0x0000 
13:43:24:890 ZCL attribute report 0x0015BC001B02243D for cluster: 0x0702, ep: 0x02, frame control: 0x18, mfcode: 0x0000 
13:43:29:894 ZCL attribute report 0x0015BC001B02243D for cluster: 0x0702, ep: 0x02, frame control: 0x18, mfcode: 0x0000 
13:43:30:833 Device TTL 3037 s flags: 0x7
13:43:39:897 ZCL attribute report 0x0015BC001B02243D for cluster: 0x0702, ep: 0x02, frame control: 0x18, mfcode: 0x0000 
13:43:49:900 ZCL attribute report 0x0015BC001B02243D for cluster: 0x0702, ep: 0x02, frame control: 0x18, mfcode: 0x0000 
13:43:54:912 ZCL attribute report 0x0015BC001B02243D for cluster: 0x0702, ep: 0x02, frame control: 0x18, mfcode: 0x0000 
13:44:04:913 ZCL attribute report 0x0015BC001B02243D for cluster: 0x0702, ep: 0x02, frame control: 0x18, mfcode: 0x0000 
13:44:09:912 ZCL attribute report 0x0015BC001B02243D for cluster: 0x0702, ep: 0x02, frame control: 0x18, mfcode: 0x0000 
13:44:19:912 ZCL attribute report 0x0015BC001B02243D for cluster: 0x0702, ep: 0x02, frame control: 0x18, mfcode: 0x0000 
13:44:20:191 ZCL attribute report 0x00158D0007C62C11 for cluster: 0x0000, ep: 0x01, frame control: 0x1C, mfcode: 0x115F 
13:44:20:191 0x00158D0007C62C11 extract Xiaomi special attribute 0xFF01
13:44:20:191 	01 battery 3025 (0x0BD1)
13:44:20:191 	03 Device temperature 27 °C
13:44:20:195 	04 unknown 17320 (0x43A8)
13:44:20:195 	05 RSSI dB (?) 12 (0x000C)
13:44:20:195 	06 LQI (?) 12884901888 (0x0300000000)
13:44:20:195 	0a Parent NWK 42865 (0xA771)
13:44:20:196 	64 on/off 0
13:44:20:196 	0b lightlevel 28 (0x001C)
13:44:24:919 ZCL attribute report 0x0015BC001B02243D for cluster: 0x0702, ep: 0x02, frame control: 0x18, mfcode: 0x0000 
13:44:30:831 Device TTL 2977 s flags: 0x7
13:44:34:921 ZCL attribute report 0x0015BC001B02243D for cluster: 0x0702, ep: 0x02, frame control: 0x18, mfcode: 0x0000 
13:44:39:922 ZCL attribute report 0x0015BC001B02243D for cluster: 0x0702, ep: 0x02, frame control: 0x18, mfcode: 0x0000 
13:44:49:920 ZCL attribute report 0x0015BC001B02243D for cluster: 0x0702, ep: 0x02, frame control: 0x18, mfcode: 0x0000 
13:44:54:976 ZCL attribute report 0x0015BC001B02243D for cluster: 0x0702, ep: 0x02, frame control: 0x18, mfcode: 0x0000 
13:45:04:929 ZCL attribute report 0x0015BC001B02243D for cluster: 0x0702, ep: 0x02, frame control: 0x18, mfcode: 0x0000 
13:45:09:934 ZCL attribute report 0x0015BC001B02243D for cluster: 0x0702, ep: 0x02, frame control: 0x18, mfcode: 0x0000 
13:45:19:932 ZCL attribute report 0x0015BC001B02243D for cluster: 0x0702, ep: 0x02, frame control: 0x18, mfcode: 0x0000 
13:45:25:451 SC tick --> StateRead
13:45:25:451 SC tick --> StateRead
13:45:25:451 SC tick --> StateRead
13:45:25:451 SC tick --> StateRead
13:45:25:451 SC tick --> StateRead
13:45:25:451 SC tick --> StateRead
13:45:29:938 ZCL attribute report 0x0015BC001B02243D for cluster: 0x0702, ep: 0x02, frame control: 0x18, mfcode: 0x0000 
13:45:30:825 Device TTL 2917 s flags: 0x7
13:45:34:943 ZCL attribute report 0x0015BC001B02243D for cluster: 0x0702, ep: 0x02, frame control: 0x18, mfcode: 0x0000 
13:45:44:946 ZCL attribute report 0x0015BC001B02243D for cluster: 0x0702, ep: 0x02, frame control: 0x18, mfcode: 0x0000 
[... cut ...]
13:46:14:950 ZCL attribute report 0x0015BC001B02243D for cluster: 0x0702, ep: 0x02, frame control: 0x18, mfcode: 0x0000 
13:46:24:953 ZCL attribute report 0x0015BC001B02243D for cluster: 0x0702, ep: 0x02, frame control: 0x18, mfcode: 0x0000 
13:46:29:955 ZCL attribute report 0x0015BC001B02243D for cluster: 0x0702, ep: 0x02, frame control: 0x18, mfcode: 0x0000 
13:46:30:826 Device TTL 2857 s flags: 0x7
13:46:39:965 ZCL attribute report 0x0015BC001B02243D for cluster: 0x0702, ep: 0x02, frame control: 0x18, mfcode: 0x0000 `

UppyJC, 2022-04-25:
Apologies!

`15:34:56:941 ZCL attribute report 0x0015BC001B02243D for cluster: 0x0702, ep: 0x02, frame control: 0x18, mfcode: 0x0000 
15:35:01:437 ZCL attribute report 0x00158D0007BC141E for cluster: 0x0402, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
15:35:01:438 00:15:8d:00:07:bc:14:1e-01-0405/state/humidity expression: Item.val = Attr.val + R.item('config/offset').val --> 4205
15:35:01:465 ZCL attribute report 0x00158D0007BC141E for cluster: 0x0405, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
15:35:01:467 00:15:8d:00:07:bc:14:1e-01-0403/state/pressure expression: Item.val = Attr.val + R.item('config/offset').val --> 1009
15:35:01:475 ZCL attribute report 0x00158D0007BC141E for cluster: 0x0403, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
15:35:01:938 ZCL attribute report 0x0015BC001B02243D for cluster: 0x0702, ep: 0x02, frame control: 0x18, mfcode: 0x0000`

Sensor report included above, complete with preceding and following lines. 0x402 attribute report is received, but it doesn't appear to do anything with it.

manup, 2022-04-25:
Thanks, yes exactly this. A classical "impossible"... for humidity and pressure it works but somehow the state/temperature isn't processed although the high level code is correct. I'm debugging with your db in my setup. Something is missing, perhaps during the load of the resources. (tomorrow I get a the sensor myself and can to some live tests)

LeoeLeoeL, 2022-04-26:
I don't know if the problem is the latest version of Deconz.
I stay with 2.15.0.
Yesterday my Aqara Multi Sensor stopped to update Humidity.
I re-paired again and noticed in Phoscon version changed. Before there was a date (2016 etc.) now there is 3000-0001

LeoeLeoeL, 2022-04-26:
I don't know if I can use this thread (device is the same) but even after many re-pairs and change of battery, humidity stops to update after 30 minutes from pairing while temperature continues. For pressure I don't know; It's steady according MET reports.
As written before, version  3000-0001 is strange. Is the new DDF? my other 2 Aqara sensors (motion and contact) continue to have the "date" notation for version.

manup, 2022-04-26:
The 3000-0001 is the attribute *SwBuildId* from the Basic Cluster which is used by most devices, but for Xiaomi the Date Code attribute is a better choice, I'll update the DDF to use that instead.

Got myself the sensor today and start to make some tests to see where it fails, I use @UppyJC database for that, hopefully it can be reproduced.

LeoeLeoeL, 2022-04-26:
It could be hard to find the problem.
Now, after the "nth" re-pairing, it's working
[image]
[image]
The peaks at the end of the graph are because I took sensor in the hand to verify if it was still working :)

LeoeLeoeL, 2022-04-29:
Anyway , it's very strange.
3 days without problems.
Maybe the sensor likes its "new" router.

MrMarkus42, 2022-05-01:
I've got the same problem. I've upgraded today to 2.15.03 and all my Aquara-Temp sensors report "0" Degrees within Phoscon oder Home-Assistant. Hoewever tha values are transmitted:

 "11:03:40:478": {
        "e": "changed",
        "id": "62",
        "r": "sensors",
        "state": {
            "lastupdated": "2022-05-01T09:03:39.798",
            **"temperature": 2545**
        },
        "t": "event",
        "uniqueid": "00:15:8d:00:07:55:b0:09-01-0402"
    }
}

MrMarkus42, 2022-05-01:
> 

Strange: Phoscon is reporting "0" Home Assistant is reporting the correct value.

LeoeLeoeL, 2022-05-01:
I update too but all 3 values are OK in Phoscon too.
I saw the version came back as a date.

Smanar, 2022-05-02:
>Strange: Phoscon is reporting "0" Home Assistant is reporting the correct value.

Not possible, HA and phoscon use the same API and use the sames values, the problem is elsewhere.

MrMarkus42, 2022-05-02:
The problem is solved after rebooting my Raspberry 2 times.

peer69, 2022-05-04:
Last working version for Aquara Temp Sensors was 2.13.1 for me. With 2.14.1 Temps were still reported but battery value wasnt at all. 
With 2.15.3 an OSRAM Smart+ Motion Sensor shows similar behaviour and reports "0" as Battery Value. 

Is there a workaround to set a "dummy" battery value to get rid of low battery warnings?

SVH-Powel, 2022-05-09:
> Well, in my case, it doesnt. For testing purposes I have rolled back my installation to version 2.12.6 which seems to work a lot more stable for me than recent versions. All my sensors report the battery state correctly and some erratic serial connection errors (deCONZ apparently losing connection to the Conbee II Stick and reconnecting) are gone as well. At least for my setup this is definitely caused by something introduced in newer versions of deCONZ.

Yes, I did also rolled back. I have 10 Aqara sensors and they started to fail one by one. I rolled back to 6.12.1 and all sensors started to work again.

wvuyk, 2022-05-09:
@manup I have had users complaining on not updating sensor readings of these Aqara sensors. With the latest release 2.15.3 several users (but also with other issues)  tell me they can resolve it on this level by going to control in deCONZ and enable Gold, Silver and Bronze en subsequently disabling DDF and re-enabling DDF. From that point on the Aqara sensors start reporting again. The same is reported for users that use Blitzwolf plugs.

Not sure, but maybe an area to look into?

manup, 2022-05-09:
The upcoming [v2.16.0-beta milestone](https://github.com/dresden-elektronik/deconz-rest-plugin/milestone/29) should address this and several sensor issues when moving from legacy code paired sensor to DDF.

For the Xiaomi temperature, humidity the config/temperature where exposed and being 0, since these have a proper temperature sensor only state/temperature is exposed now with PR https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6025

UppyJC, 2022-05-10:
Still seeing the same issue with 2.16.0, which I installed this evening. Extract from the debug log shows exactly the same behaviour for each of the climate sensors - I've pasted one below.

22:17:01:049 ZCL attribute report 0x00158D0007B893A5 for cluster: 0x0402, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
22:17:01:051 00:15:8d:00:07:b8:93:a5-01-0405/state/humidity expression: Item.val = Attr.val + R.item('config/offset').val --> 5052
22:17:01:087 ZCL attribute report 0x00158D0007B893A5 for cluster: 0x0405, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
22:17:01:090 00:15:8d:00:07:b8:93:a5-01-0403/state/pressure expression: Item.val = Attr.val + R.item('config/offset').val --> 1007
22:17:01:099 ZCL attribute report 0x00158D0007B893A5 for cluster: 0x0403, ep: 0x01, frame control: 0x18, mfcode: 0x0000

tarasfilonenko, 2022-05-22:
> Still seeing the same issue with 2.16.0, which I installed this evening. Extract from the debug log shows exactly the same behaviour for each of the climate sensors - I've pasted one below. [...]

Can confirm. Same issue on 2.16.0. Removed and re-added sensor multiple times, issue persists.

Smanar, 2022-05-22:
So you have too thoses lines in code

```
22:17:01:051 00:15:8d:00:07:b8:93:a5-01-0405/state/humidity expression: Item.val = Attr.val + R.item('config/offset').val --> 5052
22:17:01:090 00:15:8d:00:07:b8:93:a5-01-0403/state/pressure expression: Item.val = Attr.val + R.item('config/offset').val --> 1007
```

but still 0 in the API ?

manup, 2022-05-23:
Small update, we did quite some tests with this but can't reproduce the issue. It's really weird, I also deleted the ZHATemperature sensor from sensors and sub-devices tables, and it was correctly restored based on the DDF. Another test was to set the reachable to false, but this was also correctly set after receiving a report from the sensor.

The `state/temperature`  "can not" be `null` it should be set to 0 initially if not set. For example when I load the zll.db you've send me in my setup the output looks like this:

```json
{
    "config": {
        "battery": 82,
        "offset": 0,
        "on": true,
        "reachable": false
    },
    "ep": 1,
    "etag": "6a2aa0eda71c5cb8b30afb308254f360",
    "lastannounced": null,
    "lastseen": "2022-05-23T17:25Z",
    "manufacturername": "LUMI",
    "modelid": "lumi.weather",
    "name": "Cloakroom climate",
    "state": {
        "lastupdated": "none",
        "temperature": 0
    },
    "type": "ZHATemperature",
    "uniqueid": "00:15:8d:00:07:b8:93:a5-01-0402"
}
```

Only when I disable DDF support in *config.ini* (or via gui) it's:

```
[ddf-filter]
bronze=0
gold=0
silver=0
```

```json
{
    "config": {
        "battery": 82,
        "offset": 0,
        "on": true,
        "reachable": false
    },
    "ep": 1,
    "etag": "2e9ef39d156fc9f1fab964d16ed461b3",
    "lastannounced": null,
    "lastseen": "2022-05-23T17:25Z",
    "manufacturername": "LUMI",
    "modelid": "lumi.weather",
    "name": "Cloakroom climate",
    "state": {
        "lastupdated": "none",
        "temperature": null
    },
    "type": "ZHATemperature",
    "uniqueid": "00:15:8d:00:07:b8:93:a5-01-0402"
}
```

nunoary, 2022-05-24:
I also see this issue on my system. With the DDF activated the sensors do not update the temperature values, when deactivating the DDF they work fine. The problem is when I restart deCONZ, it activates the DDF and I have to deactivate it again. The version I'm using right now is v2.16.0-beta

kimmeld, 2022-05-28:
I just installed v2.16.1 and have the same issue. Editing the relevant DDF (xiaomi_wsdcgq11lm_temp_hum_press_sensor.json) and changing it to Bronze (I changed the DDF instead of switching DDF off globally since it’s just this one type of device that isn’t working) makes it work correctly.

If another zcl.db would help I can provide it.

manup, 2022-05-31:
Did another test today on a Pi3 to rule out that Qt or OS version may have an impact here, but sensor works normally and reports all values.

One thing which is different is that my sensor shows a firmware version 2016, the problematic ones appear to be newer. It would be really strange if the temperature report is different between versions but who knows..

Can anybody provide a log with APS, APS_L2 and ZCL options enabled where a message as
 `ZCL attribute report 0x00158D0007B893A5 for cluster: 0x0402, ep: 0x01, frame control: 0x18, mfcode: 0x0000 `
pops up? The related APS lines would help to check if the payload format is different compared to the 2016 version.

kimmeld, 2022-06-01:
Hey @manup,

I enabled those debugging options, but didn't get a `ZCL attribute report` line.  I played around with the debugging options and also enabled INFO and INFO_L2 and started getting that line.

I got a report from a sensor with firmware version 20191205.  deCONZ is 2.16.1 on Windows 10.

Here's what the relevant lines (I think) in the log look like:

```
19:01:57:907 APS-DATA.indication srcAddr: 0xcd17, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0402, lqi: 255, rssi: -29
19:01:57:908 	asdu: 18550a000029710a
19:01:57:909 APS-DATA.indication from child 0xCD17
19:01:57:910 Node data 0x00158d00054c562c profileId: 0x0104, clusterId: 0x0402
19:01:57:934 0x00158D00054C562C: update ZCL value 0x01/0x0402/0x0000 after 0 s
19:01:57:934 [INFO] - No button map for: lumi.weather, unicast to: 0x0000, endpoint: 0x01, cluster: 0x0402, command: 0x0A, payload: 000029710A, zclSeq: 85
19:01:57:935 ZCL attribute report 0x00158D00054C562C for cluster: 0x0402, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
19:01:57:935 	payload: 000029710a
19:01:57:936 don't create binding for attribute reporting of sensor Office Sensor
19:01:57:936 don't create binding for attribute reporting of sensor Office Sensor
19:01:57:937 don't create binding for attribute reporting of sensor Office Sensor
19:01:57:938 Websocket 192.168.11.80:59740 send message: {"attr":{"id":"8","lastannounced":null,"lastseen":"2022-06-01T01:01Z","manufacturername":"LUMI","modelid":"lumi.weather","name":"Office Sensor","swversion":"20191205","type":"ZHAHumidity","uniqueid":"00:15:8d:00:05:4c:56:2c-01-0405"},"e":"changed","id":"8","r":"sensors","t":"event","uniqueid":"00:15:8d:00:05:4c:56:2c-01-0405"} (ret = 328)
19:01:57:939 Websocket 192.168.11.80:59738 send message: {"attr":{"id":"8","lastannounced":null,"lastseen":"2022-06-01T01:01Z","manufacturername":"LUMI","modelid":"lumi.weather","name":"Office Sensor","swversion":"20191205","type":"ZHAHumidity","uniqueid":"00:15:8d:00:05:4c:56:2c-01-0405"},"e":"changed","id":"8","r":"sensors","t":"event","uniqueid":"00:15:8d:00:05:4c:56:2c-01-0405"} (ret = 328)
19:01:57:941 Websocket 192.168.11.80:59740 send message: {"attr":{"id":"7","lastannounced":null,"lastseen":"2022-06-01T01:01Z","manufacturername":"LUMI","modelid":"lumi.weather","name":"Office Sensor","swversion":"20191205","type":"ZHATemperature","uniqueid":"00:15:8d:00:05:4c:56:2c-01-0402"},"e":"changed","id":"7","r":"sensors","t":"event","uniqueid":"00:15:8d:00:05:4c:56:2c-01-0402"} (ret = 331)
19:01:57:943 Websocket 192.168.11.80:59738 send message: {"attr":{"id":"7","lastannounced":null,"lastseen":"2022-06-01T01:01Z","manufacturername":"LUMI","modelid":"lumi.weather","name":"Office Sensor","swversion":"20191205","type":"ZHATemperature","uniqueid":"00:15:8d:00:05:4c:56:2c-01-0402"},"e":"changed","id":"7","r":"sensors","t":"event","uniqueid":"00:15:8d:00:05:4c:56:2c-01-0402"} (ret = 331)
19:01:57:944 Websocket 192.168.11.80:59740 send message: {"e":"changed","id":"7","r":"sensors","state":{"lastupdated":"2022-06-01T01:01:57.934","temperature":2673},"t":"event","uniqueid":"00:15:8d:00:05:4c:56:2c-01-0402"} (ret = 164)
19:01:57:945 Websocket 192.168.11.80:59738 send message: {"e":"changed","id":"7","r":"sensors","state":{"lastupdated":"2022-06-01T01:01:57.934","temperature":2673},"t":"event","uniqueid":"00:15:8d:00:05:4c:56:2c-01-0402"} (ret = 164)
19:01:57:946 Websocket 192.168.11.80:59740 send message: {"attr":{"id":"9","lastannounced":null,"lastseen":"2022-06-01T01:01Z","manufacturername":"LUMI","modelid":"lumi.weather","name":"Office Sensor","swversion":"20191205","type":"ZHAPressure","uniqueid":"00:15:8d:00:05:4c:56:2c-01-0403"},"e":"changed","id":"9","r":"sensors","t":"event","uniqueid":"00:15:8d:00:05:4c:56:2c-01-0403"} (ret = 328)
19:01:57:947 Websocket 192.168.11.80:59738 send message: {"attr":{"id":"9","lastannounced":null,"lastseen":"2022-06-01T01:01Z","manufacturername":"LUMI","modelid":"lumi.weather","name":"Office Sensor","swversion":"20191205","type":"ZHAPressure","uniqueid":"00:15:8d:00:05:4c:56:2c-01-0403"},"e":"changed","id":"9","r":"sensors","t":"event","uniqueid":"00:15:8d:00:05:4c:56:2c-01-0403"} (ret = 328)
```

Does this help any?

kimmeld, 2022-06-01:
Hey @MrSolo570 ,

In your deCONZ install there will be a devices directory - mine is `C:\Users\ha_user\AppData\Local\deCONZ\devices`, but yours will likely be different.

From here, go the `xiaomi` directory and open the file for your sensor (`xiaomi_wsdcgq11lm_temp_hum_press_sensor.json` is the one I used), then find the line that reads `"status": "Gold",` and change `Gold` to `Bronze`.  Restart deCONZ and the sensor will now be using the old working code, not the new DDF code.

SwoopX, 2022-06-01:
@kimmeld Yes, thanks a lot. That helps to make one step further.

@manup Apparently, something goes horribly wrong with the marked code

[image]

as this is the only place where message `19:01:57:910 Node data 0x00158d00054c562c profileId: 0x0104, clusterId: 0x0402` can be generated and this part actually shouldn't have been executed. However, a websocket event seems to have been emitted. Looks ok for me, or is is malformed?

SwoopX, 2022-06-01:
Hm, did I misunderstand and the log was running on Bronze?

manup, 2022-06-01:
The ZCL Frame looks ok to me:

```
ASDU: ZCL Frame

18 Frame Control
   0-1 global command
   2   not manufspecific
   3   direction server to client
   4   disable default response

55    Seq. no

0a    CmdID Attribute Report

0000  AttrID
29    Datatype s16

710a  Measured Value (26.73)

```

manup, 2022-06-01:
Crazy idea but could it be that somehow the files like `devices/generic/items/state_temperature_item.json` are messed up in the problematic setups?

UppyJC, 2022-06-01:
Mine looks ok, but not 100% sure if it's valid as I don't know what I'm looking for exactly!

[image]

UppyJC, 2022-06-01:
You're onto something...

[image]

manup, 2022-06-01:
Ok, I've just did a small test on my Win11 box and installed deCONZ also to use home directory, work fine here.

manup, 2022-06-01:
Cool nice findings, there was a bit of struggle in earlier versions to get the files placed, I guess these are the leftovers.
I'll check the installer to auto cleanup such structures, easy solution is to just remove the `Program Files (x86)\deCONZ\devices` before installing the package version to get always the clean copy.

The devices directory in the AppData\Local is created automatically, it's there to be able to create overwrites from the packaged versions.

kimmeld, 2022-06-02:
Wow, great find!

I had the extra devices directory just like @UppyJC - moved it out of the way and the sensors are slowly reporting in.

Thanks for your help everyone!

MrSolo570, 2022-06-02:
Hey thanks for your diagnostics and the detailed how to for a Windows machine. I am running deconz within a docker container / plugin for home assistant. Is this problem going to be fixed with future patches or do I have to remove files and such manually like described above?

manup, 2022-06-03:
In the Docker container this usually shouldn't be a problem, I think they are build from scratch like a first installation?

I've fixed the Windows installer script now to ensure a fresh copy. Also the .deb package preinstall script to do the same (alltough I'm not sure if it is the same problem on Linux..).

thyaoo06, 2022-06-07:
Hi, I have got same trouble, please did you set up a new exe in http://deconz.dresden-elektronik.de/win/.
Last one is from 24May... Thanks a lot for your debug :=)

thyaoo06, 2022-06-09:
Many Thanks for the new file....

xavax59, 2022-06-16:
I have de 2.16.1 on jeedom and i have the same issue. Editing the relevant DDF (xiaomi_wsdcgq11lm_temp_hum_press_sensor.json) and changing it to Bronze and it's ok.

xavax59, 2022-06-16:
> So it work again with disabling the DDF. But on jeedom you are not on windows ?

Yes ! it work with DDF In Bronze. My Jeedom Works on debian 10.10

SwoopX, 2022-06-17:
I'm closing this one, as the solution is to install the latest version.

closed 2022-06-17 by SwoopX

xavax59, 2022-06-17:
I’m in 2.16.1. It’s the latest version ? I have the problème with this version ….

Smanar, 2022-06-17:
The 2.16 is the last stable, the last beta is the 2.17.

xavax59, 2022-07-08:
With the last stable 2.17.1 I have the same problem

MrSolo570, 2022-08-13:
> Hey @MrSolo570 ,
>
> In your deCONZ install there will be a devices directory - mine is `C:\Users\ha_user\AppData\Local\deCONZ\devices`, but yours will likely be different.
>
> From here, go the `xiaomi` [...]

I tried that as you can see but the sensor is still not working...
[image]

Smanar, 2022-08-14:
You have the "Bronze" Status on your capture.