dresden-elektronik/deconz-rest-plugin #5977: Device name

gmitch64, 2022-04-24:
## Device

- Product name: ZM85EL-2Z
- Manufacturer: _TZE200_cf1sl3tj
- Model identifier: TS0601
- Device type : Please remove all unrelated device types. 

  - Other:  Smart Super Curtain robot motor

## Screenshots

[image]

### Basic
[image]

### Identify

### Alarms

### Device Temperature

### Groups
[image]
[image]

### Scenes
[image]
[image]
[image]
[image]

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

### Tuya Specific
[image]

Currently running v2.15.2-beta

If there's any other info that's needed, please let me know.

G

gmitch64, 2022-05-23:
Is there any other info needed from me?

gmitch64, 2022-05-23:
Do you have an idea of a timescale? Just to set expectations for folks finding this thread?
Should we keep this issue open in the meantime?

Smanar, 2022-06-14:
Not realy, manup want to make some tries on his side before validating the PR.

gmitch64, 2022-07-07:
Just wondering how things are progressing?

closed 2022-08-07 by github-actions[bot]

samcourt, 2022-08-24:
Any progress? Would love to integrate my blind with the rest of my Home

Shuttleu, 2023-04-11:
Just got a couple of these, any updates?

gmitch64, 2023-04-11:
I still have a few of these as well.
It looks like they're supported in zigbee2mqtt, so as soon as I get some time, I'm planning on setting up an mqtt broker and checking that out.

Mimiix, 2023-04-11:
@Smanar can this be done ?

Smanar, 2023-04-12:
Nope, nothing have moved for covering.