dresden-elektronik/deconz-rest-plugin #5978: Aquara Multi Sensors (Temperature/Pressure/Humidity) no longer reporting battery level

DNC74, 2022-04-24:
## Describe the bug

Pretty similar to https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5974 : 
My installation was on ``2.14.01`` where anything was running fine. With the update to ``2.15.00`` to ``2.15.02`` versions the reporting of battery levels of all my Aquara WSDCGQ11LM sensors stopped, while all other sensor values from these sensors are working as expected. Any other paired devices to my Conbee I stick are working flawless as well.

## Steps to reproduce the behavior

Add Aquara Multi Sensors (Temperature/Pressure/Humidity) WSDCGQ11LM sensors to deCONZ. Observe null battery values in Phoscon and API. Temperature, Humidity and Pressure values are reported correctly.

## Expected behavior

## Screenshots

Getting Battery values as in versions before ``2.15.x``

## Environment

 - Host system: Synology DS1817+
 - Running method: Virtual Machine Ubuntu 18.04.6 LTS
 - Firmware version: 26390500
 - deCONZ version: 2.15.02 non-gui
 - Device: ConBee I
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? no

## deCONZ Logs

[log.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/8549446/log.txt)

I can see battery values arriving in the logs for Xiaomi Contact sensors ``lumi.sensor_magnet.aq2`` for example
```
12:12:13:233 ZCL attribute report 0x00158D00031B307E for cluster: 0x0000, ep: 0x01, frame control: 0x1C, mfcode: 0x115F
12:12:13:233 0x00158D00031B307E extract Xiaomi special attribute 0xFF01
12:12:13:233    01 battery 3005 (0x0BBD)
12:12:13:233    03 Device temperature 19 °C
12:12:13:233    04 unknown 17320 (0x43A8)
12:12:13:233    05 RSSI dB (?) 39 (0x0027)
12:12:13:233    06 LQI (?) 4294967296 (0x0100000000)
12:12:13:233    0a Parent NWK 21757 (0x54FD)
12:12:13:233    64 on/off 0
```
but similar messages for the Xiaomi Multi sensors ``lumi.weather`` are missing.

## Screenshots

[image]

[image]

All those Sensors are the same and behave the same as descibed above:
[image]

## Additional context

I see differences in regards to battery in ``/usr/share/deCONZ/devices/xiaomi/xiaomi_wsdcgq11lm_temp_hum_press_sensor.json`` (non-working) vs. ``/usr/share/deCONZ/devices/xiaomi/xiaomi_mccgq14lm_openclose_sensor.json`` (working). Maybe there is the culprit?

Mimiix, 2022-04-24:
Hi,

As you mentioned it's similar to #5974, I am closing this one as it's duplicate. 

Please post your report there.

Kind regards,
Dennis

closed 2022-04-24 by Mimiix

DNC74, 2022-04-24:
Similar is not the same. https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5974 has a problem with Temperature/Pressure/Humidity, mine problem is about battery levels. The only match is the device and the appearance with version 2.15.x.

Mimiix, 2022-04-24:
It's the same device. So any new issues with it have to reside in the same issue.

DNC74, 2022-04-24:
ok, good to know. I hope the reference to this here there is sufficient?

Mimiix, 2022-04-24:
I'd rather have you repost in there a bit too

DNC74, 2022-04-24:
@Mimiix 
I've already tried to provide as much information about the bug as I can, and somehow your answers now give me the impression that it's not about the solution and more about unnecessary formalities. Do what you want with the information above.

Mimiix, 2022-04-24:
@DNC74 my job is to make sure the formalities are followed :) I am not doing anything with the bugs themselves, I just make sure it's all in order.