dresden-elektronik/deconz-rest-plugin #5987: Update from 2.15.01 to 2.15.02 breaks some Hue motion sensors, hue dimmer switch and others sensors

minke890, 2022-04-28:
## Describe the bug
Update from 2.15.01 to 2.15.02 breaks two of the three philips hue motion sensors I have, one Philips hue dimmer switch, an aqara vibration sensor, a friend of hue switch, an aqara smoke sensor (3 others not) and one aqara window sensor. The devices or gone in phoscon and in the deCONZ GUI viewer.

## Steps to reproduce the behavior
Switching back to 2.14 or 2.15.01 did not help since the sensors/switches are gone in phoscon. I can add the devices again to phoscon, but after a restart (stop/start of the docker) all these devices are gone again.

## Expected behavior
Devices should still be available after the update to 2.15.02 and also remain available when restarting deconz.

## Screenshots
Screenhots would not help here, the devices are just gone.

## Environment
 - Host system: Qnap 253be NAS
 - Running method: Docker container
 - Firmware version: 26720700
 - deCONZ version: 2.51.01
 - Device: ConBee II 
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? If so: Which?, APC UPS and zwave usb stick

## deCONZ Logs

## Additional context

Mimiix, 2022-04-28:
Hi,

As this is not a bug report as per #5113 i am closing this issue.

Feel free to open an topic on the forums.

Kind regards,
Dennis

closed 2022-04-28 by Mimiix

minke890, 2022-04-28:
I do not undertand, what did I wrong? What is the forums?

Mimiix, 2022-04-28:
Https://forum.phoscon.de