dresden-elektronik/deconz-rest-plugin #5992: SMaBiT AV2010/21C

WhistleMaster, 2022-04-30:
## Device

- Product name: SMaBiT AV2010/21C Flat Magnetic Contact
- Manufacturer: SMaBiT
- Model identifier: AV2010/21C
- Device type : Sensor

https://manuals.smabit.eu/len/av2010_21c.html

## Screenshots

[image]
[image]

### Basic
[image]

### Power Configuration
[image]

### Identify
[image]

### IAS Zone
[image]

### Poll Control
[image]

### Others
[image]
[image]

Mimiix, 2022-04-30:
@Smanar Can this be done with ddf?

Smanar, 2022-04-30:
Sure, can try this DDF

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "SMaBiT",
  "modelid": "AV2010/21C",
  "vendor": "SMaBiT",
  "product": "SMaBiT AV2010/21C Flat Magnetic Contact",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_OPEN_CLOSE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0402",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0500"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/checkin"
        },
        {
          "name": "config/enrolled",
          "public": false
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/pending"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/lowbattery"
        },
        {
          "name": "state/open",
          "awake": true,
          "parse": {
            "at": "0x0002",
            "cl": "0x0500",
            "ep": 1,
            "eval": "Item.val = (Attr.val & 0x3) != 0"
          }
        }
      ]
    }
  ]
}
```

If the value is updated when reading it in deconz but not automatically.
Add a binding from the cluster 0x0500 https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet#bindings-panel
Retry (save the file and make "hot relaod" on the menu)
If It don't work add a report for the attribute 0x0021
To have something like
```
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 3600,
          "max": 43200,
          "change": "0x00000001"
        }
      ]
    }
  ]
```

WhistleMaster, 2022-05-01:
Thanks for the DDF file ! I've tried but unfortunately, I don't get any values, manually or automatically.

In the API, I get:

```
    "196": {
        "config": {
            "on": true,
            "pending": [],
            "reachable": null
        },
        "ep": 1,
        "etag": "<<REDACTED>>",
        "lastannounced": null,
        "lastseen": "2022-05-01T06:53Z",
        "manufacturername": "SMaBiT",
        "modelid": "AV2010/21C",
        "name": "OpenClose 196",
        "state": {
            "lastupdated": "none",
            "lowbattery": null,
            "open": null
        },
        "type": "ZHAOpenClose",
        "uniqueid": "<<REDACTED>>"
```

The 0x0500 cluster is still like that:

[image]

This is the result I got after following your indications:

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "SMaBiT",
  "modelid": "AV2010/21C",
  "vendor": "SMaBiT",
  "product": "SMaBiT AV2010/21C Flat Magnetic Contact",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_OPEN_CLOSE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0402",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0500"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/checkin",
          "public": false,
          "refresh.interval": 3600,
          "default": 14400
        },
        {
          "name": "config/enrolled",
          "public": false
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/pending"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/lowbattery"
        },
        {
          "name": "state/open",
          "awake": true,
          "parse": {
            "at": "0x0002",
            "cl": "0x0500",
            "ep": 1,
            "eval": "Item.val = (Attr.val & 0x3) != 0"
          }
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0500",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 3600,
          "max": 43200,
          "change": "0x00000001"
        }
      ]
    }
  ]
}
```

WhistleMaster, 2022-05-01:
A quick update, after re-pairing the sensor, it worked ! I've updated my previous comment.

Smanar, 2022-05-01:
So you need all the binding part ?

```
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0500",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 3600,
          "max": 43200,
          "change": "0x00000001"
        }
      ]
    }
  ]
```

I m asking because not all device support it, if you can confirm I will make the PR if you don't want to make it yourself ?

SwoopX, 2022-05-01:
I'm afraid the DDF needs to be updated in several ways:

- The reporting of the respective bindings goes against a non-existent attribute. You don't do any reporting configuration on the IAS zone cluster
- There is no `config/battery` resource item
- Please use the default implementation for `state/open`, so remove the `parse` stanza
- There is no binding and reporting against the power configuration cluster
- Please remove "SMaBiT" in the `product` item
- Please read the device request template carefully. It said to **read** all clusters prior to posting screenshots. All screenshots do not have them read and show defaults

Smanar, 2022-05-01:
>The reporting of the respective bindings goes against a non-existent attribute. You don't do any reporting configuration on the IAS zone cluster

Right, so if it work you, mean you can remove the reporting part

>There is no config/battery resource item

There is no vlaue on the capture, so I use state/lowbattery instead

>There is no binding and reporting against the power configuration cluster

Some explanation than before

>Please read the device request template carefully. It said to read all clusters prior to posting screenshots. All screenshots do not have them read and show defaults

Can change the DDF I have value

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "SMaBiT",
  "modelid": "AV2010/21C",
  "vendor": "SMaBiT",
  "product": "AV2010/21C Flat Magnetic Contact",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_OPEN_CLOSE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0402",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0500"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/checkin"
        },
        {
          "name": "config/enrolled",
          "public": false
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/pending"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/lowbattery"
        },
        {
          "name": "state/open"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0500"
    }
  ]
}
```

WhistleMaster, 2022-05-01:
>  Please read the device request template carefully. It said to **read** all clusters prior to posting screenshots. All screenshots do not have them read and show defaults

Thanks for the reply ! I've read all clusters now:

### Basic
[image]

### Power Configuration
[image]

### Identify
[image]

### IAS Zone
[image]

### Poll Control
[image]

Smanar, 2022-05-02:
Ok, so no more changes on my side, It seem your device don't use the cluster power configuration so you will have only the the state/lowbattery.

You can test the last one on my last post.

WhistleMaster, 2022-05-02:
I've updated the DDF+hot-reload and it's still working fine ! Thanks !

closed 2022-05-16 by manup