dresden-elektronik/deconz-rest-plugin #5996: Harmony Hub - Groups dont work / are Empty

almostserious, 2022-04-30:
## Describe the bug

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

## Environment

 - Host system: Proxmox LXC - Debian
 - Running method: (Raspbian / Ubuntu / Home Assistent deCONZ Add-on / Marthoc Docker container / Windows / Virtual Machine)
 - Firmware version: (26390500)
 - deCONZ version: (2.14.01)
 - Device: (ConBee I)
 - Do you use an USB extension cable: (yes )
 - Is there any other USB or serial devices connected to the host system? If so:  ZWAVE
## deCONZ Logs

## Additional context

Mimiix, 2022-04-30:
What if you make the groups in phoscon? Do they work then?

almostserious, 2022-04-30:
Groups in Phoscon are not synced to Harmony.
After first toggle of the Harmony created Group an empty Group popsup.
 This can be edited in Phoscon to make it work.

closed 2022-05-29 by github-actions[bot]