dresden-elektronik/deconz-rest-plugin #5998: Plugin needs to restart

vlumikero, 2022-05-01:
## Describe the bug
I'm using the plugin via the Home Assistant add-on. For me, deCONZ often goes to an endless loop of trying to recover from an error. When it's restarted it goes back online. It would make sense for the plugin to fail so that the watchdog could detect the issue and restart the plugin.

## Steps to reproduce the behavior
The error above happens to me perhaps once a week, sporadically. 

## Expected behavior
The plugin should fail after a number of reconnect attempts, instead of continuing endlessly.

## Screenshots

## Environment
DeCONZ: 2.14.01
FW: 26390500
Conbee I on an RPI 3B+
HomeAssistant 2022.04.7

 - Host system: Raspberry Pi
 - Running method: Home Assistent deCONZ Add-on
 - Firmware version: 26390500
 - deCONZ version: 2.14.01
 - Device: ConBee I
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs

Example of log output, that just goes on forever.

```
20:35:30:447 start reconnect to network
20:35:30:540 dev /dev/ttyAMA0
20:35:30:637 dev /dev/ttyAMA0
20:35:31:526 dev /dev/ttyAMA0
20:35:31:607 dev /dev/ttyAMA0
20:35:35:521 dev /dev/ttyAMA0
20:35:35:599 dev /dev/ttyAMA0
20:35:35:699 failed to reconnect to network try=1
```

## Additional context

Smanar, 2022-05-01:
Hello have you already updated your firmware ?
Power supply issue ?

vlumikero, 2022-05-01:
I’ve had this issue tens of times over the years, with various versions of
the firmware and deCONZ. I’ve got a 3 Amp PSU.

I’m not after troubleshooting the setup, and I’ve seen a number of people
who have similar issues. What I’m after is a more robust behavior of the
plug-in when these failures occur, as the reconnect doesn’t seem to work.

Mimiix, 2022-05-01:
Hi,
As this is not a bug report as described in #5113 I am closing this issue.

Feel free to open an topic on the forum.

Kind regards,
Dennis

closed 2022-05-01 by Mimiix