dresden-elektronik/deconz-rest-plugin #6000: Ikea Blinds (FYRTUR and TREDANSEN) issues

xxLeoxx93, 2022-05-01:
## Describe the bug
Hey all,

I'm struggeling quiet a bit with the Ikea blinds (FYRTUR and TREDANSEN, both have the same issues).
After a dozen resets of the blinds and restarting deconz I finaly managed to get them into deconz.
**BUT:**
- It seems like they are not updating anything back to deconz. The last seen date doesn't change at all.
- Over night none of them responded anymore and I had to press a button on them to make them work again (at least 4/6, the other two are still visible in Phoscon but don't respond at all)
- It seems to be impossible to add them to deconz AND have a button connected directly to them (?).

Is there anyone using the blinds without these issues? As of now there are pretty much unusable for me.
Thanks!

## Steps to reproduce the behavior
Use the ikea blinds with Phoscon I guess :)

## Expected behavior
The need to respond at least on the next day, otherwise the "Smart" in smart blinds doesn't make any sense. 

## Screenshots
Not needed

## Environment
 - Host system: Raspberry Pi
 - Running method:  Docker container
 - Firmware version: 26720700
 - deCONZ version: 2.15.03 / 21.4.2022
 - Device: ConBee II 
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs
No access to GUI since running on docker and VNC seems broken as of now.

## Additional context
None.

redbull2601, 2022-05-01:
Hi @xxLeoxx93 ,

I had exactly the same problem. Also new pairing did not work. 
The problem stays and on new pairing attempt they moved but no actual state of the blinds was delivered.

The solution:

Reset the blind and connect it to another gateway (fortunately my Tradfri Gateway was sold, but not sent away...)
Delete the nodes in deconz if they still appear (was only a problem with one blind)
Do this with all your blinds (sorry :-) )
And then pair the blinds one after another with deconz.

This worked for me and since 3 weeks I´m having no trouble.
All values are transferred and it works perfectly. (now homeassist can stop the blinds if Siri moves them and the window sensor is open)

best regards

xxLeoxx93, 2022-05-01:
@redbull2601 Thanks for your feedback!
I don't have any other gateway unfortunalty and the gui is currently not accessable with docker (https://github.com/deconz-community/deconz-docker/issues/106). But I'll try to reconnect them once the gui is working again and I'm able to delete the nodes.
Do they work for you after a deconz restart directly or do you need to press a button on them at least once?

redbull2601, 2022-05-02:
Hi @xxLeoxx93 ,

also after deconz restart they are directly available and can be moved.

In Phoscon they are not moving!

xxLeoxx93, 2022-05-02:
@redbull2601 Thanks for you feedback. Without a second gateway I'm not sure now what to to... 
Guess I'll repair everything again but this is frustrating as hell and deconz doesn't sho you any issues...

closed 2022-05-05 by xxLeoxx93