dresden-elektronik/deconz-rest-plugin #6001: Issue with Livarno LED lux light

jokerigno, 2022-05-01:
## Describe the bug
Hi,
I have 2 livarno led strip managed with deconz in my Home Assistant istance.
I can easily manage default color light using light_profiles.csv so I'm assuming that the light is correctly integrated by deconz component.
The issue comes when I use scene create and then scene turn on with those specific lights. The issue is absent when using other light already integrated with deconz (hue lights for example).
I already asked support on HA github ( https://github.com/home-assistant/core/issues/69972 ) but they suggested me that can be related to the light instead of the HA instance.

## Steps to reproduce the behavior

- service: scene.create
  data:
  snapshot_entities: light.striscia_led_specchio
  scene_id: prova

and then...

- service: scene.turn_on
  data: {}
  target:
  entity_id: scene.prova

## Expected behavior
light is restored with correct color, brightness etc while instead is restored with default values

## Screenshots

## Environment

 - Host system: Unraid
 - Running method: deconzcommunity/deconz
 - Firmware version: (264A0700)
 - deCONZ version: (2.15.03)
 - Device: ConBee II 
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? If so: Which? Coral USB 

## deCONZ Logs

https://pastebin.com/etbLYJQY

## Additional context

Home Assistant diagnostics

https://pastebin.com/jpMCaTNF

jokerigno, 2022-05-25:
Keep it active please

jokerigno, 2022-06-23:
The issue is still there. Any way to receive support?

Mimiix, 2022-06-23:
Asked manup to check

manup, 2022-06-23:
I have a Livarno lux strip here (not sure if it is the same model) for testing. As I understand the scenes aren't set/restored correctly?
There are a couple of devices which don't have proper support for scenes, I'll schedule a test to check if the device is the problem or something in the plugin code for the beta after the next stable.

closed 2022-08-15 by github-actions[bot]