dresden-elektronik/deconz-rest-plugin #6005: Aqara H1 switch reports 1% battery after 2.15.3 upgrade

lancer73, 2022-05-02:
## Describe the bug
After the update to deCONZ 2.15.3 all Aqara H1 switches (dual rocker; lumi.remote.b28ac1) report 1% battery (after updating the battery percentage) while reporting 90%+ on older versions of deCONZ.

Downgrading to the previous deCONZ version resolves the problem after the device reports the battery percentage.

Using ConbeeII USB stick with firmware 26720700 on Home Assistant Supervised

## Steps to reproduce the behavior
Update to deCONZ 2.15.3

## Expected behavior
Reporting same battery states

## Screenshots
[image]

## Environment
Home Assistant Supervised with Conbee II  (firmware 26720700)

## deCONZ Logs
none

## Additional context
Multiple H1 switches in the network

Mimiix, 2022-05-02:
@SwoopX is there something that changed this?

SwoopX, 2022-05-02:
No, but it demonstrates that nobody is paying attention to the battery levels, including myself 🙂 

It never worked correctly, as the assumed voltage boundaries in the used javascript file do not apply for this device (factor 100 too high) here and the calculation always spits out 1%. Short term workaround would be to find and copy `xiaomi_battery.js`, remove two trailing zeros for `vmin` and `vmax` (virtual division by 100), save it with an altered name and use that JS in the respective DDF.